evcc-io/evcc #17883: Incorrect costs for self-consumption.

StefanA-LU, 2024-12-24:
### Describe the bug

Incorrect costs for self-consumption. This should be the price to be paid when purchasing the grid, as self-consumption saves it. The feed-in price is not saved.

[image]

### Steps to reproduce

1. See GUI and price settings
2.
3.
...

### Configuration details

```yaml
currency: EUR

grid: # price using energy from the grid
  type: fixed
  price: 0.4009 # EUR/kWh

feedin: # price for feeding solar energy to the grid
  type: fixed
  price: 0.0663 # EUR/kWh

#co2: # carbon intensity forecast data
#  type: grünstromindex
#  zip: <zip>
```

### Log details

```text
Not required
```

### What type of operating system are you running?

Windows

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.131.12

naltatis, 2024-12-24:
This is correct and intended behavior. The price when using pure self-produced energy is calculated as the feed-in money you miss out on. More details here: https://docs.evcc.io/docs/faq#ersparnisberechnung

naltatis, 2024-12-24:
BTW: The numbers in the energy flow diagram show current prices, not savings.

closed 2024-12-24 by naltatis