evcc-io/evcc #17884: Funktion "controllable" für "custom" battery

KrawallOPA, 2024-12-24:
Hi, mein Wechselrichter/Batterie sind über die "manuelle/custom" "Vorlage in EVCC per MQTT eingebunden. Die Daten kommen aus meiner Hausautomation. Ich hätte gerne aus EVCC die Rückmeldung, das die Batterie nicht entladen oder zwangsgeladen werden soll. Hierfür gibt es ja schon das MQTT Topic "batteryMode". Nur wird das Topic bei mir nicht von EVCC beschrieben/geändert. Ich hatte gehofft, das ich durch das "True" setzen des Topics "batteryDischargeControl", diese Funktion nutzen kann, aber leider hat ein "True" keine Auswirkung. Ein Neustart von EVCC, setzt zu dem das True auf False zurück. 
Auch ein Versuch, "controllable" für die Batterie in der yaml zu hinterlegen, funktioniert nicht. 

Es wäre somit super, wenn man die Funktion "controllable" für die "custom" Batterie nutzen könnte.

andig, 2024-12-24:
Dann müssen limitsoc und/oder batterymode hinzu gefügt werden

KrawallOPA, 2024-12-24:
Du meinst ich muss diese beiden Parameter in der Yaml bei der Batterie hinzufügen?

VolkerK62, 2024-12-24:
Hilft evtl weiter   https://github.com/evcc-io/evcc/wiki/aaa-Lifehacks#steuerbarer-hausspeicher-als-custom-meter-per-mqtt

andig, 2024-12-24:
It‘s already implemented: https://github.com/evcc-io/evcc/blob/master/meter/meter.go#L33

closed 2024-12-24 by andig

KrawallOPA, 2025-01-28:
> Hilft evtl weiter https://github.com/evcc-io/evcc/wiki/aaa-Lifehacks#steuerbarer-hausspeicher-als-custom-meter-per-mqtt

Leider funktioniert es nicht bei mir. Ich habe folgenden Code in der Yaml, allerdings gibt es das [Topic ](topic: evcc/gobel1/mode) nicht. Die Topics power und soc werden ohne Probleme angelegt und mit Daten gefüttert.

`- type: custom
  name: PV_Speicher
  power:
   source: mqtt
   topic: evcc/gobel1/power
   timeout: 30s
  soc:
   source: mqtt
   topic: evcc/gobel1/soc
   timeout: 30s
  batterymode:
   source: switch
   switch:
   - case: 1 # normal
     set:
      source: sequence
      set:
      - source: const
        value: 1
        set:
         source: mqtt
         topic: evcc/gobel1/mode
   - case: 2 # hold
     set:
      source: sequence
      set:
      - source: const
        value: 2
        set:
         source: mqtt
         topic: evcc/gobel1/mode
   - case: 3 # charge
     set:
      source: sequence
      set:
      - source: const
        value: 3
        set:
         source: mqtt
         topic: evcc/gobel1/mode
`