evcc-io/evcc #17889: Error modbus: exception '3' (illegal data value), function '16

ThomasGoreth, 2024-12-24:
### Describe the bug

Hallo zusammen,

bei unsere Wallbox APL eMH1, die über den originalen Modbus USB Adapter mit einem Raspberry Pi (evcc Installation) kommuniziert, kommt wenn wir unseren Renault angeschlossen haben jede Minute der Log- Eintrag "modbus: exception '3' (illegal data value), function '16" und das rote Stördreieck geht an.
Die Funktion ist gegeben, nur ich möchte gerne ohne Störung evcc betreiben

Gruß

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX

interval: 30s # control cycle interval

sponsortoken: XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: sma-home-manager 
  usage: grid  
  host: 192.168.178.45  
  name: grid1

chargers:
- type: template
  template: abl 
  id: 3  
  device: /dev/ttyUSB0  
  baudrate: 38400  
  comset: 8E1  
  modbus: rs485serial  
  timeout: 10s  
  name: wallbox3

vehicles:
- type: template
  template: renault 
  title: Zoe  
  user: XXXXXXXXXXXXXXXXXXXXXXXXXXXXX 
  password: XXXXXXXXXXXXXXXXXXXXXXXXXXX 
  vin: XXXXXXXXXXXXXXXXXXXXXXXXXXX
  capacity: 52  
  name: ev2

loadpoints:
- title: Garage
  charger: wallbox3
  vehicle: ev2
  mode: pv

site:
  title: Zuhause
  meters:
    grid: grid1

hems:
  type: sma
  AllowControl: false
```

### Log details

```text
[db:3 ] ERROR 2024/12/24 13:06:18 modbus: exception '3' (illegal data value), function '16'
[db:2 ] TRACE 2024/12/24 13:06:18 modbus: send 03 9f 00 00 00 06 03 03 78 35 00 02
[db:2 ] TRACE 2024/12/24 13:06:18 modbus: recv 03 9f 00 00 00 07 03 03 04 80 00 00 00
[db:3 ] TRACE 2024/12/24 13:06:18 modbus: send 03 a0 00 00 00 06 03 04 7a a3 00 02
[db:3 ] TRACE 2024/12/24 13:06:18 modbus: recv 03 a0 00 00 00 07 03 04 04 00 00 00 00
[db:3 ] TRACE 2024/12/24 13:06:18 modbus: send 03 a1 00 00 00 06 03 04 7a a1 00 02
[db:3 ] TRACE 2024/12/24 13:06:18 modbus: recv 03 a1 00 00 00 07 03 04 04 00 00 00 00
[db:2 ] TRACE 2024/12/24 13:06:18 modbus: send 03 a2 00 00 00 06 03 03 78 f1 00 02
[db:2 ] TRACE 2024/12/24 13:06:18 modbus: recv 03 a2 00 00 00 07 03 03 04 80 00 00 00
[db:2 ] TRACE 2024/12/24 13:06:18 modbus: send 03 a3 00 00 00 06 03 03 77 31 00 04
[db:2 ] TRACE 2024/12/24 13:06:18 modbus: recv 03 a3 00 00 00 0b 03 03 08 00 00 00 00 00 43 e2 94
[site ] DEBUG 2024/12/24 13:06:18 pv 2 power: 0W
[site ] DEBUG 2024/12/24 13:06:18 pv power: 180W
[db:3 ] TRACE 2024/12/24 13:06:18 modbus: send 03 a4 00 00 00 06 03 03 7a a9 00 04
[db:3 ] TRACE 2024/12/24 13:06:18 modbus: recv 03 a4 00 00 00 0b 03 03 08 00 00 00 00 00 43 e2 94
[db:3 ] TRACE 2024/12/24 13:06:18 modbus: send 03 a5 00 00 00 06 03 03 78 7d 00 02
[db:3 ] TRACE 2024/12/24 13:06:18 modbus: recv 03 a5 00 00 00 07 03 03 04 00 00 00 1a
[site ] DEBUG 2024/12/24 13:06:18 battery soc: 26%
[site ] DEBUG 2024/12/24 13:06:18 battery power: 0W
[site ] DEBUG 2024/12/24 13:06:18 site power: 301W
[lp-1 ] DEBUG 2024/12/24 13:06:18 charger status: B
[lp-1 ] DEBUG 2024/12/24 13:06:18 pv charge current: 0A = 0A + -0.436A (301W @ 3p)
[site ] DEBUG 2024/12/24 13:06:47 ----
[lp-1 ] DEBUG 2024/12/24 13:06:47 charge power: 0W
[site ] DEBUG 2024/12/24 13:06:47 grid power: 200W
[db:1 ] TRACE 2024/12/24 13:06:47 modbus: send 05 ff 00 00 00 06 03 04 78 37 00 02
[site ] DEBUG 2024/12/24 13:06:47 grid powers: [244 -20 -25]W
[site ] DEBUG 2024/12/24 13:06:47 grid currents: [1.88 -0.33 -1.09]A
[db:2 ] TRACE 2024/12/24 13:06:48 modbus: send 03 a6 00 00 00 06 03 03 78 35 00 02
[db:1 ] TRACE 2024/12/24 13:06:48 modbus: recv 05 ff 00 00 00 07 03 04 04 00 00 00 b5
[db:1 ] TRACE 2024/12/24 13:06:48 modbus: send 06 00 00 00 00 06 03 03 77 31 00 04
[db:1 ] TRACE 2024/12/24 13:06:48 modbus: recv 06 00 00 00 00 0b 03 03 08 00 00 00 00 01 cf 2a 3c
[site ] DEBUG 2024/12/24 13:06:48 pv 1 power: 181W
[db:2 ] TRACE 2024/12/24 13:06:48 modbus: recv 03 a6 00 00 00 07 03 03 04 80 00 00 00
[db:2 ] TRACE 2024/12/24 13:06:48 modbus: send 03 a7 00 00 00 06 03 03 78 f1 00 02
[db:2 ] TRACE 2024/12/24 13:06:48 modbus: recv 03 a7 00 00 00 07 03 03 04 80 00 00 00
[db:3 ] TRACE 2024/12/24 13:06:48 modbus: send 03 a8 00 00 00 06 03 04 7a a3 00 02
[db:3 ] TRACE 2024/12/24 13:06:48 modbus: recv 03 a8 00 00 00 07 03 04 04 00 00 00 00
[db:3 ] TRACE 2024/12/24 13:06:48 modbus: send 03 a9 00 00 00 0b 03 10 9d 2c 00 02 04 00 00 09 78
[db:3 ] TRACE 2024/12/24 13:06:48 modbus: recv 03 a9 00 00 00 06 03 10 9d 2c 00 02
[db:2 ] TRACE 2024/12/24 13:06:48 modbus: send 03 aa 00 00 00 06 03 03 77 31 00 04
[db:2 ] TRACE 2024/12/24 13:06:48 modbus: recv 03 aa 00 00 00 0b 03 03 08 00 00 00 00 00 43 e2 94
[site ] DEBUG 2024/12/24 13:06:48 pv 2 power: 0W
[site ] DEBUG 2024/12/24 13:06:48 pv power: 181W
[db:3 ] TRACE 2024/12/24 13:06:48 modbus: send 03 ab 00 00 00 06 03 04 7a a1 00 02
[db:3 ] TRACE 2024/12/24 13:06:48 modbus: recv 03 ab 00 00 00 07 03 04 04 00 00 00 00
[db:3 ] TRACE 2024/12/24 13:06:48 modbus: send 03 ac 00 00 00 0b 03 10 9f 59 00 02 04 00 00 00 00
[db:3 ] TRACE 2024/12/24 13:06:48 modbus: recv 03 ac 00 00 00 03 03 90 03
[db:3 ] ERROR 2024/12/24 13:06:48 modbus: exception '3' (illegal data value), function '16'
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.131.8

andig, 2024-12-25:
Ich hätte einen Logeintrag für `abl` erwartet, hier gibts aber nur DB-Geräte. Ich vermute, dass das nicht die WB ist. Mach mal bitte ein

    evcc charger

und

    evcc meter

damit wir die db:3 mal zuordnen können (da brauchen wir noch einen besseren Mechanismus).

hyperbart, 2024-12-25:
Also experiencing this all of a sudden. Alfen box here. Let me know if you need additional info.

mfuchs1984, 2024-12-25:
Do you also have sma equipment? There are several older issues with this error message, all that I found are related to sma.

hyperbart, 2024-12-25:
I have SMA material indeed but the error was scoped to the Alfen because it had an issue with the phases which resulted in 11kW charging which is way too much without solar here in Belgium (peak tariff).

hyperbart, 2024-12-25:
`[site ] DEBUG 2024/12/25 17:10:26 site power: 907W
[lp-1 ] DEBUG 2024/12/25 17:10:26 charge total import: 16173.471kWh
[lp-1 ] DEBUG 2024/12/25 17:10:26 charger status: B
[lp-1 ] DEBUG 2024/12/25 17:10:26 charge start energy: 16173.471kWh
[lp-1 ] INFO 2024/12/25 17:10:26 car connected
[lp-1 ] DEBUG 2024/12/25 17:10:26 vehicle api refresh
[lp-1 ] DEBUG 2024/12/25 17:10:26 pv timer elapse
[lp-1 ] DEBUG 2024/12/25 17:10:26 pv timer inactive
[lp-1 ] DEBUG 2024/12/25 17:10:26 charge total import: 16173.471kWh
[main ] DEBUG 2024/12/25 17:10:27 vehicle status: B (i5 Bart)
[lp-1 ] INFO 2024/12/25 17:10:27 vehicle updated: unknown -> i5 Bart
[lp-1 ] DEBUG 2024/12/25 17:10:27 set charge mode: pv
[lp-1 ] DEBUG 2024/12/25 17:10:28 vehicle status: B
[lp-1 ] DEBUG 2024/12/25 17:10:28 vehicle soc: 64%
[lp-1 ] DEBUG 2024/12/25 17:10:28 vehicle soc limit: 100%
[lp-1 ] DEBUG 2024/12/25 17:10:28 vehicle range: 235km
[lp-1 ] ERROR 2024/12/25 17:10:28 switch phases: modbus: exception '3' (illegal data value), function '6'
[site ] DEBUG 2024/12/25 17:10:28 ----
[lp-2 ] DEBUG 2024/12/25 17:10:28 charge power: 0W
[lp-1 ] DEBUG 2024/12/25 17:10:28 charge power: 0W
[lp-1 ] DEBUG 2024/12/25 17:10:28 charge currents: [0 0 0]A
[circuit-AutomaatLaadpaalVoortuin] DEBUG 2024/12/25 17:10:28 power: 0W
[circuit-AutomaatLaadpaalVoortuin] DEBUG 2024/12/25 17:10:28 current: 0A
[circuit-AutomaatNetaansluiting] DEBUG 2024/12/25 17:10:30 power: 913.19W
[circuit-AutomaatNetaansluiting] DEBUG 2024/12/25 17:10:30 current: 2.86A `

andig, 2024-12-25:
Please open new issue if this is a different charger. Thank you.

andig, 2025-01-02:
No reply, closing

closed 2025-01-02 by andig

bluemonk2, 2025-09-22:
Same problem as described above (battery mode: modbus: exception '3' (illegal data value), function '16'
--> doesn't show up after restarting evcc but only after starting to charge with the use of grid (i.e. switching to fast -> error immediately shows up). 
--> I also notice that despite activating  the switch to "prevent battery-discharge in fast mode", the battery is being discharged in fast mode.

Setup is SMA + Compleo eBox

Logs:

```
[site ] DEBUG 2025/09/22 20:05:51 ----
[lp-2 ] DEBUG 2025/09/22 20:05:51 charge power: 0W
[lp-2 ] DEBUG 2025/09/22 20:05:51 charge currents: [0 0 0]A
[lp-3 ] DEBUG 2025/09/22 20:05:51 charge power: 0W
[lp-3 ] DEBUG 2025/09/22 20:05:51 charge currents: [0 0 0]A
[lp-1 ] DEBUG 2025/09/22 20:05:51 charge power: 561W
[lp-1 ] DEBUG 2025/09/22 20:05:51 charge currents: [0.67 0.73 1.03]A
[site ] DEBUG 2025/09/22 20:05:51 grid power: -0W
[site ] DEBUG 2025/09/22 20:05:51 grid powers: [-184 89 94]W
[site ] DEBUG 2025/09/22 20:05:51 pv 2 power: 0W
[site ] DEBUG 2025/09/22 20:05:51 pv 1 power: 0W
[site ] DEBUG 2025/09/22 20:05:51 pv power: 0W
[site ] DEBUG 2025/09/22 20:05:51 grid currents: [-1.17 0.739 0.986]A
[site ] DEBUG 2025/09/22 20:05:51 !! solar production: accumulate set pv3 12883.720kWh meter total (was: Accumulated: 0.000kWh updated: 2025-09-22 20:05:41 +0100 BST meter: 12883.720kWh)
[site ] DEBUG 2025/09/22 20:05:51 !! solar production: accumulate moved pv3 from 0.000 to 0.000
[site ] DEBUG 2025/09/22 20:05:51 !! solar production: accumulate set pv4 14588.118kWh meter total (was: Accumulated: 0.000kWh updated: 2025-09-22 20:05:41 +0100 BST meter: 14588.118kWh)
[site ] DEBUG 2025/09/22 20:05:51 !! solar production: accumulate moved pv4 from 0.000 to 0.000
[site ] DEBUG 2025/09/22 20:05:51 battery 1 power: 264W
[site ] DEBUG 2025/09/22 20:05:51 battery 1 soc: 96%
[site ] DEBUG 2025/09/22 20:05:51 site power: 364W
[lp-2 ] DEBUG 2025/09/22 20:05:51 charge voltages: [0 0 0]V
[lp-2 ] DEBUG 2025/09/22 20:05:51 charge total import: 0.000kWh
[lp-2 ] DEBUG 2025/09/22 20:05:51 charger status: A
[site ] DEBUG 2025/09/22 20:06:01 ----
[lp-2 ] DEBUG 2025/09/22 20:06:01 charge power: 0W
[lp-2 ] DEBUG 2025/09/22 20:06:01 charge currents: [0 0 0]A
[lp-3 ] DEBUG 2025/09/22 20:06:01 charge power: 0W
[lp-3 ] DEBUG 2025/09/22 20:06:01 charge currents: [0 0 0]A
[lp-1 ] DEBUG 2025/09/22 20:06:01 charge power: 11190W
[lp-1 ] DEBUG 2025/09/22 20:06:01 charge currents: [16.2 16.4 16.3]A
[site ] DEBUG 2025/09/22 20:06:01 grid power: 181W
[site ] DEBUG 2025/09/22 20:06:01 pv 2 power: 0W
[site ] DEBUG 2025/09/22 20:06:01 pv 1 power: 0W
[site ] DEBUG 2025/09/22 20:06:01 grid powers: [-133 215 99]W
[site ] DEBUG 2025/09/22 20:06:01 grid currents: [-1.4 1.43 1.38]A
[site ] DEBUG 2025/09/22 20:06:01 pv power: 0W
[site ] DEBUG 2025/09/22 20:06:01 !! solar production: accumulate set pv3 12883.720kWh meter total (was: Accumulated: 0.000kWh updated: 2025-09-22 20:05:51 +0100 BST meter: 12883.720kWh)
[site ] DEBUG 2025/09/22 20:06:01 !! solar production: accumulate moved pv3 from 0.000 to 0.000
[site ] DEBUG 2025/09/22 20:06:01 !! solar production: accumulate set pv4 14588.118kWh meter total (was: Accumulated: 0.000kWh updated: 2025-09-22 20:05:51 +0100 BST meter: 14588.118kWh)
[site ] DEBUG 2025/09/22 20:06:01 !! solar production: accumulate moved pv4 from 0.000 to 0.000
[site ] DEBUG 2025/09/22 20:06:01 battery 1 power: 1661W
[site ] DEBUG 2025/09/22 20:06:01 battery 1 soc: 96%
[site ] DEBUG 2025/09/22 20:06:01 site power: 1942W
[lp-3 ] DEBUG 2025/09/22 20:06:01 charge voltages: [228 0 0]V
[lp-3 ] DEBUG 2025/09/22 20:06:01 detected connected phases: 1p
[lp-3 ] DEBUG 2025/09/22 20:06:01 charge total import: 1.393kWh
[lp-3 ] DEBUG 2025/09/22 20:06:01 charger status: A
[site ] DEBUG 2025/09/22 20:06:11 ----
[lp-2 ] DEBUG 2025/09/22 20:06:11 charge power: 0W
[lp-2 ] DEBUG 2025/09/22 20:06:11 charge currents: [0 0 0]A
[lp-3 ] DEBUG 2025/09/22 20:06:11 charge power: 0W
[lp-3 ] DEBUG 2025/09/22 20:06:11 charge currents: [0 0 0]A
[lp-1 ] DEBUG 2025/09/22 20:06:11 charge power: 11077W
[lp-1 ] DEBUG 2025/09/22 20:06:11 charge currents: [16 16.2 16.2]A
[site ] DEBUG 2025/09/22 20:06:11 grid power: 6452W
[site ] DEBUG 2025/09/22 20:06:11 grid powers: [-912 3680 3684]W
[site ] DEBUG 2025/09/22 20:06:11 pv 1 power: 0W
[site ] DEBUG 2025/09/22 20:06:11 grid currents: [-4.41 16.2 16.3]A
[site ] DEBUG 2025/09/22 20:06:11 pv 2 power: 0W
[site ] DEBUG 2025/09/22 20:06:11 pv power: 0W
[site ] DEBUG 2025/09/22 20:06:11 !! solar production: accumulate set pv3 12883.720kWh meter total (was: Accumulated: 0.000kWh updated: 2025-09-22 20:06:01 +0100 BST meter: 12883.720kWh)
[... 196 lines cut ...]
  phases: 1
  priority: 0
  enable: 
    threshold: -50
    delay: 1m
  disable: 
    threshold: -50
    delay: 30s
  guardduration: 2m
- title: DG_HeizungBad # display name for UI
  charger: my_smartswitch2 # Wallbox Referenz
  #mode: pv # charge mode (off, now, minpv, pv)
  phases: 1
  priority: 0
  enable: 
    threshold: -50
    delay: 1m
  disable: 
    threshold: -50
    delay: 30s
  guardduration: 2m

hems:
  type: sma
  allowControl: false