evcc-io/evcc #17910: Tariffs: publish forecast

andig, 2024-12-27:
I have been looking for a way to get the future Tibber prices as a sensor. There are some ways to get them, but they seem quite cumbersome to configure. I have EVCC configured to send the data to Mosquitto, which works fine. In the MQTT-topic evcc/site/tariffPriceLoadpoints is only the current one The future Tibber prices are already available in the EVCC GUI, so I was wondering how I could query them for further usage in Home Assistant. Does someone have any suggestions?

_Originally posted by @TheFreaker86 in https://github.com/evcc-io/evcc/discussions/17905_

andig, 2025-01-02:
Duplikat: https://github.com/evcc-io/evcc/issues/13725

closed 2025-01-02 by andig

andig, 2025-02-09:
@thefreaker86 worth noting that you can already get the current forecast using REST api, see https://docs.evcc.io/docs/integrations/rest-api.

TheFreaker86, 2025-02-09:
Ja, weiss ich 😄  Habe ich auch bereits als Workaround implementiert. Ich finde MQTT in Home Assistant einfacher zu handhaben, deshalb habe ich ursprünglich [hier](https://github.com/evcc-io/evcc/discussions/17905#discussion-7748656) gefragt. 

Sieht so aus als ob ihr dabei seid die Vorhersagedaten per MQTT zu implementieren. Vielen lieben Dank dafür! 😃  Ihr habt mit EVCC echt was tolles auf die Beine gestellt! Chapeau! 👍