evcc-io/evcc #17923: Einfache Aufgliederung des Verbrauchs

Chris-Acon, 2024-12-28:
Hallo zusammen,

ich würde gerne den Verbrauch meiner Wärmepumpe im EVCC Dashboard mit anzeigen. Aktuell zeige ich den Verbrauch der Wärmepumpe als „Batterie“ an. Wenn man statt Aux-Meter sowas wie ein Ext-Meter anlegen könnte, wäre das klasse. Alle Ext-Meter sind weitere Messpunkte, die den Verbrauch weiter aufschlüsseln. 

Oder habt ihr einen alternativen Ansatz?

Danke für Eure Tipss und Hinweise. 

/Chris

StefanSchoof, 2024-12-28:
Suchst du sowas wie https://github.com/evcc-io/evcc/discussions/15345

VolkerK62, 2024-12-28:
Solche Wünsche kommen in letzter Zeit häufiger.
Das wäre etwas für einen Mini-Loadpoint  https://github.com/evcc-io/evcc/discussions/7235

Chris-Acon, 2024-12-28:
Hi @StefanSchoof, 

ja, genau sowas suche ich. Ich habe einen Fronius WR und auch Fronius Smart Meter. Daher ist es bei mir nicht so einfach einen Custom Charger (wie mit den Shelly’s) zu definieren. 

@VolkerK62: Die Mini-Loadpoints lesen sich vielversprechend. Sind die Mini-Loadpoints schon in Entwicklung? 

Danke für Eure Antworten. 

/Chris

VolkerK62, 2024-12-28:
schau mal hier https://github.com/evcc-io/evcc/discussions/17590#discussioncomment-11585754
musst nur `power` und evtl. `energy` an deine Umstände anpassen.
und mit dem "Charger" dann einen Loadpoint einrichten