evcc-io/evcc #17925: UI dauerhaft nicht mehr erreichbar und Ladesteuerung hängt nach kurzzeitigem Ausfall von Tibber

foto-andreas, 2024-12-28:
### Describe the bug

Es gab letzte Nacht mal wieder einen vermutlich kurzzeitigen Ausfall der Tibber-API, so dass der Zugriff zu den Pulse-Daten für EVCC vorübergehend nicht möglich war. Die gleiche Situation hatte ich neulich schon einmal, jetzt konnte ich aber zumindest ein paar Daten aus dem Log einsammeln.

Es waren zeitgleich beide EVCC-Instanzen betroffen. Das Log ist nicht ganz gleich. Es gibt beim Ausfall einen Unterschied. Siehe unten. 

Auf EVCC-Seite sieht das dann so aus, dass die UI nicht mehr erreichbar ist und "404 page not found" angezeigt wird. Beide Instanzen gleich. Im Log steht weiterhin, dass die GRID-Daten outdated seien. Allerdings sind die Daten in der Tibber-App sichtbar und aktiv. Außerdem führt ein Neustart in 4 von 4 Fällen sofort zu einem funktionierenden EVCC.

Noch blöder an der ganzen Sache ist aber, dass EVCC dann die Ladung nicht freigibt. Geplante Ladung von 4:00 bis 6:00. Es wurde nicht geladen. Die gesamte Ladesteuerung von EVCC ist offenbar bis zum Restart auf Eis gelegt, weil EVCC nicht merkt, dass die Tibber-Pulse-Daten wieder zur Verfügung stünden. 

Bitte baut eine Absicherung ein, dass sich die Steuerung wieder sortiert, so dass auf alle Fälle wie geplant geladen wird und die UI erreichbar ist.

### Steps to reproduce

1. Tibber Pulse nutzen
2.Tibber-Api lahm legen :)
3. Tibber-Api wieder fit machen
4. UI von EVCC ansehen

Ich hab keine Ahnung, ob man wirklich die gleiche Situation wie beschrieben nachstellen kann.

Ich hab zwar unten ankreuzen müssen, dass ich es mit nightly nachstellen kann, da Pflichtfeld, kann ich aber natürlich nicht.

### Configuration details

```yaml
log: debug

interval: 15s

sponsortoken: "{{ evcc.sponsor_token }}"

tariffs:
  currency: EUR
  grid:
    type: tibber
    token: "{{ tibber.api_token }}"
    homeid: "{{ tibber.home_id_wk }}"
  feedin:
    type: fixed
    price: 0.1231

meters:
  - name: tibber
    type: template
    template: tibber-pulse
    usage: grid
    token: "{{ tibber.api_token }}"
    homeid: "{{ tibber.home_id_wk }}"

  - name: pv1
    type: template
    template: solaredge-hybrid
    usage: pv
    modbus: tcpip
    host: "{{ pv.host }}"
    port: 502
    timeout: 5s

vehicles:
  - name: zoe
    type: template
    template: renault
    title: Zoe
    user: "{{ zoe.user }}"
    password: "{{ zoe.password }}"
    vin: "{{zoe.vin}}"
    capacity: 52
    identifiers: {{ zoe.identifiers }}
    mode: pv
[... 151 lines cut ...]
evcc-py  | [lp-1  ] DEBUG 2024/12/28 00:29:47 charger status: A
evcc-py  | [pulse ] ERROR 2024/12/28 00:29:48 Message: unable to start stream api:e90e4c627d25417383ba9b4cb7821497 for device e90e4c62-7d25-4173-83ba-9b4cb7821497;>
evcc-py  | [pulse ] ERROR 2024/12/28 00:29:49 Message: unable to start stream api:e90e4c627d25417383ba9b4cb7821497 for device e90e4c62-7d25-4173-83ba-9b4cb7821497;>
evcc-py  | [site  ] DEBUG 2024/12/28 00:30:02 ----
evcc-py  | [lp-1  ] DEBUG 2024/12/28 00:30:02 charge power: 0W
evcc-py  | [lp-1  ] DEBUG 2024/12/28 00:30:02 charge currents: [0 0 0]A
evcc-py  | [site  ] DEBUG 2024/12/28 00:30:02 grid power: 303W
evcc-py  | [site  ] DEBUG 2024/12/28 00:30:02 grid currents: [0 0 0]A
evcc-py  | [site  ] DEBUG 2024/12/28 00:30:02 pv power: 0W
evcc-py  | [site  ] DEBUG 2024/12/28 00:30:02 site power: 303W
evcc-py  | [lp-1  ] DEBUG 2024/12/28 00:30:02 charge total import: 5104.054kWh
evcc-py  | [lp-1  ] DEBUG 2024/12/28 00:30:02 charger status: A
```

### What type of operating system are you running?

Docker container

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.12

andig, 2024-12-28:
evcc läuft nicht ohne funktionierenden Netzzähler. Wenn Tibber zu unzuverlässig ist muss es aus der Position raus :(

andig, 2024-12-28:
> Auf EVCC-Seite sieht das dann so aus, dass die UI nicht mehr erreichbar ist und "404 page not found" angezeigt wird. 

Bitte evcc mit `--profile` starten und- wenn das passiert- bitte das Go routine Profil von /debug/pprof und Logfile in dem Moment hier ergänzen. 404 sollte tatsächlich nicht sein.

foto-andreas, 2024-12-29:
"--profile" ist eingebaut.

andig, 2025-01-08:
> aber es gibt kein Profile

Wirklich? Das wäre das erste mal.

foto-andreas, 2025-01-08:
Wird das nur beim Crash geschrieben oder auch, wenn evcc weiterläuft? Ich hab den gesamten Container abgesucht nach profile und debug aber nix passendes gefunden. --profile bekommt keinen zusätzlichen Parameter, oder?

foto-andreas, 2025-01-08:
`
evcc-wk  | [site  ] DEBUG 2025/01/07 10:25:50 ----
evcc-wk  | [lp-1  ] DEBUG 2025/01/07 10:25:50 charge power: 0W
evcc-wk  | [lp-1  ] DEBUG 2025/01/07 10:25:50 charge currents: [0 0 0]A
evcc-wk  | [site  ] DEBUG 2025/01/07 10:25:50 grid power: 347W
evcc-wk  | [site  ] DEBUG 2025/01/07 10:25:50 grid currents: [0 0 0]A
evcc-wk  | [site  ] DEBUG 2025/01/07 10:25:51 pv power: 13W
evcc-wk  | [site  ] DEBUG 2025/01/07 10:25:51 site power: 347W
evcc-wk  | [lp-1  ] DEBUG 2025/01/07 10:25:51 charge total import: 11436.146kWh
evcc-wk  | [lp-1  ] DEBUG 2025/01/07 10:25:51 charger status: A
evcc-wk  | [pulse ] ERROR 2025/01/07 10:26:05 Message: unable to start stream api:a7a10770336a4d0198ecea09c28da682 for device a7a10770-336a-4d01-98ec-x; userId 29fbf633-dee7-4959-9d19-x
d5c97b, Locations: [{Line:1 Column:28}], Extensions: map[], Path: [liveMeasurement]
evcc-wk  | [site  ] DEBUG 2025/01/07 10:26:05 ----
evcc-wk  | [lp-1  ] DEBUG 2025/01/07 10:26:05 charge power: 0W
evcc-wk  | [lp-1  ] DEBUG 2025/01/07 10:26:05 charge currents: [0 0 0]A
evcc-wk  | [site  ] DEBUG 2025/01/07 10:26:05 grid power: 347W
evcc-wk  | [site  ] DEBUG 2025/01/07 10:26:05 grid currents: [0 0 0]A
evcc-wk  | [pulse ] ERROR 2025/01/07 10:26:06 Message: unable to start stream api:a7a10770336a4d0198ecea09c28da682 for device a7a10770-336a-4d01-98ec-x; userId 29fbf633-dee7-4959-9d19-x
d5c97b, Locations: [{Line:1 Column:28}], Extensions: map[], Path: [liveMeasurement]
evcc-wk  | [site  ] DEBUG 2025/01/07 10:26:06 pv power: 12W
evcc-wk  | [site  ] DEBUG 2025/01/07 10:26:06 site power: 347W
evcc-wk  | [lp-1  ] DEBUG 2025/01/07 10:26:06 charge total import: 11436.146kWh
evcc-wk  | [lp-1  ] DEBUG 2025/01/07 10:26:06 charger status: A
evcc-wk  | [pulse ] ERROR 2025/01/07 10:26:06 Message: unable to start stream api:a7a10770336a4d0198ecea09c28da682 for device a7a10770-336a-4d01-98ec-x; userId 29fbf633-dee7-4959-9d19-x
d5c97b, Locations: [{Line:1 Column:28}], Extensions: map[], Path: [liveMeasurement]
evcc-wk  | [pulse ] ERROR 2025/01/07 10:26:07 Message: unable to start stream api:a7a10770336a4d0198ecea09c28da682 for device a7a10770-336a-4d01-98ec-x; userId x
d5c97b, Locations: [{Line:1 Column:28}], Extensions: map[], Path: [liveMeasurement]
evcc-wk  | [pulse ] ERROR 2025/01/07 10:26:07 Message: unable to start stream api:a7a10770336a4d0198ecea09c28da682 for device a7a10770-336a-4d01-98ec-x; userId 29fbf633-dee7-4959-9d19-x
d5c97b, Locations: [{Line:1 Column:28}], Extensions: map[], Path: [liveMeasurement]
evcc-wk  | [pulse ] ERROR 2025/01/07 10:26:07 Message: unable to start stream api:x for device a7a10770-336a-4d01-98ec-ea09c28da682; userId 29fbf633-dee7-4959-9d19-x
d5c97b, Locations: [{Line:1 Column:28}], Extensions: map[], Path: [liveMeasurement]
evcc-wk  | [pulse ] ERROR 2025/01/07 10:26:08 Message: unable to start stream api:a7a10770336a4d0198ecea09c28da682 for device a7a10770-336a-4d01-98ec-x; userId 29fbf633-dee7-4959-9d19-x
d5c97b, Locations: [{Line:1 Column:28}], Extensions: map[], Path: [liveMeasurement]
evcc-wk  | [pulse ] ERROR 2025/01/07 10:26:08 Message: unable to start stream api:a7a10770336a4d0198ecea09c28da682 for device a7a10770-336a-4d01-98ec-x; userId 29fbf633-dee7-4959-9d19-x
d5c97b, Locations: [{Line:1 Column:28}], Extensions: map[], Path: [liveMeasurement]
evcc-wk  | [site  ] DEBUG 2025/01/07 10:26:20 ----
evcc-wk  | [lp-1  ] DEBUG 2025/01/07 10:26:20 charge power: 0W
evcc-wk  | [lp-1  ] DEBUG 2025/01/07 10:26:20 charge currents: [0 0 0]A
evcc-wk  | [site  ] DEBUG 2025/01/07 10:26:20 grid power: 347W
evcc-wk  | [site  ] DEBUG 2025/01/07 10:26:20 grid currents: [0 0 0]A
evcc-wk  | [site  ] DEBUG 2025/01/07 10:26:21 pv power: 11W
evcc-wk  | [site  ] DEBUG 2025/01/07 10:26:21 site power: 347W
evcc-wk  | [lp-1  ] DEBUG 2025/01/07 10:26:21 charge total import: 11436.146kWh
evcc-wk  | [lp-1  ] DEBUG 2025/01/07 10:26:21 charger status: A
evcc-wk  | [site  ] DEBUG 2025/01/07 10:26:35 ----
evcc-wk  | [lp-1  ] DEBUG 2025/01/07 10:26:35 charge power: 0W
evcc-wk  | [lp-1  ] DEBUG 2025/01/07 10:26:35 charge currents: [0 0 0]A
evcc-wk  | [site  ] DEBUG 2025/01/07 10:26:35 grid power: 347W
evcc-wk  | [site  ] DEBUG 2025/01/07 10:26:35 grid currents: [0 0 0]A
evcc-wk  | [site  ] DEBUG 2025/01/07 10:26:36 pv power: 10W
evcc-wk  | [site  ] DEBUG 2025/01/07 10:26:36 site power: 347W
evcc-wk  | [lp-1  ] DEBUG 2025/01/07 10:26:36 charge total import: 11436.146kWh
evcc-wk  | [lp-1  ] DEBUG 2025/01/07 10:26:36 charger status: A
evcc-wk  | [site  ] DEBUG 2025/01/07 10:26:50 ----
evcc-wk  | [lp-1  ] DEBUG 2025/01/07 10:26:50 charge power: 0W
evcc-wk  | [lp-1  ] DEBUG 2025/01/07 10:26:50 charge currents: [0 0 0]A
evcc-wk  | [site  ] DEBUG 2025/01/07 10:26:51 pv power: 18W
evcc-wk  | [site  ] ERROR 2025/01/07 10:26:51 grid power: outdated
evcc-wk  | [site  ] DEBUG 2025/01/07 10:27:05 ----
evcc-wk  | [lp-1  ] DEBUG 2025/01/07 10:27:05 charge power: 0W
evcc-wk  | [lp-1  ] DEBUG 2025/01/07 10:27:05 charge currents: [0 0 0]A
evcc-wk  | [site  ] DEBUG 2025/01/07 10:27:06 pv power: 13W
evcc-wk  | [site  ] ERROR 2025/01/07 10:27:06 grid power: outdated
`

foto-andreas, 2025-01-08:
... oder muss das Profile-Verzeichnis vorher angelegt werden?

foto-andreas, 2025-01-08:
Muss jetzt leider erstmal los. Passiert das bei keinem anderen der Tibber-Evcc-Leute? Hätte da jetzt eigentlich ein globaleres Problem erwartet, da bei mir auch immer beide Hütten betrofen sind.

andig, 2025-01-08:
Einfach /debug/pprof im Browser aufrufen.

foto-andreas, 2025-01-08:
Ach so, ich war im Dateisystem unterwegs.... Ergebnis ist leider nicht wie erwartet:
```
/app # curl -vvvv http://localhost:7070/debug/pprof
17:56:55.080976 [0-x] == Info: [READ] client_reset, clear readers
17:56:55.081158 [0-0] == Info: Host localhost:7070 was resolved.
17:56:55.081207 [0-0] == Info: IPv6: ::1
17:56:55.081235 [0-0] == Info: IPv4: 127.0.0.1
17:56:55.081268 [0-0] == Info: [SETUP] added
17:56:55.081303 [0-0] == Info: [HAPPY-EYEBALLS] created ipv6 (timeout 300000ms)
17:56:55.081355 [0-0] == Info: [HAPPY-EYEBALLS] created ipv4 (timeout 300000ms)
17:56:55.081417 [0-0] == Info: [HAPPY-EYEBALLS] ipv6 starting (timeout=300000ms)
17:56:55.081495 [0-0] == Info:   Trying [::1]:7070...
17:56:55.081540 [0-0] == Info: [TCP] cf_socket_open() -> 0, fd=4
17:56:55.082343 [0-0] == Info: [TCP] local address ::1 port 48274...
17:56:55.082760 [0-0] == Info: [HAPPY-EYEBALLS] ipv6 connect -> 0, connected=0
17:56:55.082851 [0-0] == Info: [TCP] adjust_pollset, !connected, POLLOUT fd=4
17:56:55.082927 [0-0] == Info: [HAPPY-EYEBALLS] adjust_pollset -> 1 socks
17:56:55.083082 [0-0] == Info: [TCP] connected
17:56:55.083130 [0-0] == Info: [HAPPY-EYEBALLS] ipv6 connect -> 0, connected=1
17:56:55.083210 [0-0] == Info: [HAPPY-EYEBALLS] Connected to localhost (::1) port 7070
17:56:55.083296 [0-0] == Info: Connected to localhost (::1) port 7070
17:56:55.083365 [0-0] == Info: using HTTP/1.x
17:56:55.083497 [0-0] == Info: [TCP] send(len=89) -> 89, err=0
17:56:55.083560 [0-0] => Send header, 89 bytes (0x59)
0000: GET /debug/pprof HTTP/1.1
001b: Host: localhost:7070
0031: User-Agent: curl/8.11.1
004a: Accept: */*
0057: 
17:56:55.083943 [0-0] == Info: [TCP] recv(len=102400) -> 204, err=0
17:56:55.084079 [0-0] <= Recv header, 32 bytes (0x20)
0000: HTTP/1.1 301 Moved Permanently
17:56:55.084178 [0-0] == Info: [WRITE] cw_out, wrote 32 header bytes -> 32
17:56:55.084252 [0-0] == Info: [WRITE] download_write header(type=c, blen=32) -> 0
17:56:55.084331 [0-0] == Info: [WRITE] client_write(type=c, len=32) -> 0
17:56:55.084415 [0-0] <= Recv header, 40 bytes (0x28)
0000: Content-Type: text/html; charset=utf-8
17:56:55.084526 [0-0] == Info: [WRITE] header_collect pushed(type=1, len=40) -> 0
17:56:55.084614 [0-0] == Info: [WRITE] cw_out, wrote 40 header bytes -> 40
17:56:55.084691 [0-0] == Info: [WRITE] download_write header(type=4, blen=40) -> 0
17:56:55.084771 [0-0] == Info: [WRITE] client_write(type=4, len=40) -> 0
17:56:55.084851 [0-0] <= Recv header, 25 bytes (0x19)
0000: Location: /debug/pprof/
17:56:55.084936 [0-0] == Info: [WRITE] header_collect pushed(type=1, len=25) -> 0
17:56:55.085098 [0-0] == Info: [WRITE] cw_out, wrote 25 header bytes -> 25
17:56:55.085169 [0-0] == Info: [WRITE] download_write header(type=4, blen=25) -> 0
17:56:55.085249 [0-0] == Info: [WRITE] client_write(type=4, len=25) -> 0
17:56:55.085318 [0-0] <= Recv header, 37 bytes (0x25)
0000: Date: Wed, 08 Jan 2025 16:56:55 GMT
17:56:55.085415 [0-0] == Info: [WRITE] header_collect pushed(type=1, len=37) -> 0
17:56:55.085496 [0-0] == Info: [WRITE] cw_out, wrote 37 header bytes -> 37
17:56:55.085569 [0-0] == Info: [WRITE] download_write header(type=4, blen=37) -> 0
17:56:55.085647 [0-0] == Info: [WRITE] client_write(type=4, len=37) -> 0
17:56:55.085719 [0-0] <= Recv header, 20 bytes (0x14)
0000: Content-Length: 48
17:56:55.085799 [0-0] == Info: [WRITE] header_collect pushed(type=1, len=20) -> 0
17:56:55.085878 [0-0] == Info: [WRITE] cw_out, wrote 20 header bytes -> 20
17:56:55.085952 [0-0] == Info: [WRITE] download_write header(type=4, blen=20) -> 0
17:56:55.086050 [0-0] == Info: [WRITE] client_write(type=4, len=20) -> 0
17:56:55.086128 [0-0] <= Recv header, 2 bytes (0x2)
0000: 
17:56:55.086186 [0-0] == Info: [WRITE] header_collect pushed(type=1, len=2) -> 0
17:56:55.086264 [0-0] == Info: [WRITE] cw_out, wrote 2 header bytes -> 2
17:56:55.086335 [0-0] == Info: [WRITE] download_write header(type=4, blen=2) -> 0
17:56:55.086412 [0-0] == Info: [WRITE] client_write(type=4, len=2) -> 0
17:56:55.086481 [0-0] <= Recv data, 48 bytes (0x30)
0000: <a href="/debug/pprof/">Moved Permanently</a>...
<a href="/debug/pprof/">Moved Permanently</a>.

17:56:55.086591 [0-0] == Info: [WRITE] cw_out, wrote 48 body bytes -> 48
17:56:55.086661 [0-0] == Info: [WRITE] download_write body(type=1, blen=48) -> 0
17:56:55.086737 [0-0] == Info: [WRITE] client_write(type=1, len=48) -> 0
17:56:55.086806 [0-0] == Info: [WRITE] xfer_write_resp(len=204, eos=0) -> 0
17:56:55.086886 [0-0] == Info: [WRITE] cw-out is notpaused
17:56:55.086964 [0-0] == Info: [WRITE] cw-out done
17:56:55.087041 [0-0] == Info: [READ] client_reset, clear readers
17:56:55.087119 [0-0] == Info: Connection #0 to host localhost left intact
```

foto-andreas, 2025-01-08:
<html>
<head>
<title>/debug/pprof/</title>
<style>
.profile-name{
	display:inline-block;
	width:6rem;
}
</style>
</head>
<body>
/debug/pprof/
<br>
<p>Set debug=1 as a query parameter to export in legacy text format</p>
<br>
Types of profiles available:
<table>
<thead><td>Count</td><td>Profile</td></thead>
<tr><td>1217</td><td><a href='allocs?debug=1'>allocs</a></td></tr>
<tr><td>0</td><td><a href='block?debug=1'>block</a></td></tr>
<tr><td>0</td><td><a href='cmdline?debug=1'>cmdline</a></td></tr>
<tr><td>49</td><td><a href='goroutine?debug=1'>goroutine</a></td></tr>
<tr><td>1217</td><td><a href='heap?debug=1'>heap</a></td></tr>
<tr><td>0</td><td><a href='mutex?debug=1'>mutex</a></td></tr>
<tr><td>0</td><td><a href='profile?debug=1'>profile</a></td></tr>
<tr><td>13</td><td><a href='threadcreate?debug=1'>threadcreate</a></td></tr>
<tr><td>0</td><td><a href='trace?debug=1'>trace</a></td></tr>
</table>
<a href="goroutine?debug=2">full goroutine stack dump</a>
<br>
<p>
Profile Descriptions:
<ul>
<li><div class=profile-name>allocs: </div> A sampling of all past memory allocations</li>
<li><div class=profile-name>block: </div> Stack traces that led to blocking on synchronization primitives</li>
<li><div class=profile-name>cmdline: </div> The command line invocation of the current program</li>
<li><div class=profile-name>goroutine: </div> Stack traces of all current goroutines. Use debug=2 as a query parameter to export in the same format as an unrecovered panic.</li>
<li><div class=profile-name>heap: </div> A sampling of memory allocations of live objects. You can specify the gc GET parameter to run GC before taking the heap sample.</li>
<li><div class=profile-name>mutex: </div> Stack traces of holders of contended mutexes</li>
<li><div class=profile-name>profile: </div> CPU profile. You can specify the duration in the seconds GET parameter. After you get the profile file, use the go tool pprof command to investigate the profile.</li>
<li><div class=profile-name>threadcreate: </div> Stack traces that led to the creation of new OS threads</li>
<li><div class=profile-name>trace: </div> A trace of execution of the current program. You can specify the duration in the seconds GET parameter. After you get the trace file, use the go tool trace command to investigate the trace.</li>
</ul>
</p>
</body>
</html>

foto-andreas, 2025-01-08:
Stack dump daraus ziehen?

foto-andreas, 2025-01-08:
```
curl http://localhost:7070/debug/pprof/goroutine?debug=2 
goroutine 3348275 [running]:
runtime/pprof.writeGoroutineStacks({0x30f75e0, 0x4000ebc000})
	runtime/pprof/pprof.go:761 +0x6c
runtime/pprof.writeGoroutine({0x30f75e0?, 0x4000ebc000?}, 0x4001d13708?)
	runtime/pprof/pprof.go:750 +0x2c
runtime/pprof.(*Profile).WriteTo(0x54f2130?, {0x30f75e0?, 0x4000ebc000?}, 0xc?)
	runtime/pprof/pprof.go:374 +0x148
net/http/pprof.handler.ServeHTTP({0x40013ae011, 0x9}, {0x3110d88, 0x4000ebc000}, 0x4002222280)
	net/http/pprof/pprof.go:272 +0x444
net/http/pprof.Index({0x3110d88, 0x4000ebc000}, 0x4002222280?)
	net/http/pprof/pprof.go:388 +0xc8
net/http.HandlerFunc.ServeHTTP(0x5549340?, {0x3110d88?, 0x4000ebc000?}, 0x4000fc6be0?)
	net/http/server.go:2220 +0x38
net/http.(*ServeMux).ServeHTTP(0x4002222140?, {0x3110d88, 0x4000ebc000}, 0x4002222280)
	net/http/server.go:2747 +0x1b4
github.com/gorilla/mux.(*Router).ServeHTTP(0x4001d46480, {0x3110d88, 0x4000ebc000}, 0x4002222000)
	github.com/gorilla/mux@v1.8.1/mux.go:212 +0x194
net/http.serverHandler.ServeHTTP({0x4000f173e0?}, {0x3110d88?, 0x4000ebc000?}, 0x6?)
	net/http/server.go:3210 +0xbc
net/http.(*conn).serve(0x400162a000, {0x31192a0, 0x400211ad80})
	net/http/server.go:2092 +0x4fc
created by net/http.(*Server).Serve in goroutine 1
	net/http/server.go:3360 +0x3dc

goroutine 1 [IO wait]:
internal/poll.runtime_pollWait(0xe61f0b1a59d0, 0x72)
	runtime/netpoll.go:351 +0xa0
internal/poll.(*pollDesc).wait(0x400229a900?, 0x87?, 0x0)
	internal/poll/fd_poll_runtime.go:84 +0x28
internal/poll.(*pollDesc).waitRead(...)
	internal/poll/fd_poll_runtime.go:89
internal/poll.(*FD).Accept(0x400229a900)
	internal/poll/fd_unix.go:620 +0x24c
net.(*netFD).accept(0x400229a900)
	net/fd_unix.go:172 +0x28
net.(*TCPListener).accept(0x4002280c80)
	net/tcpsock_posix.go:159 +0x28
net.(*TCPListener).Accept(0x4002280c80)
	net/tcpsock.go:372 +0x2c
net/http.(*Server).Serve(0x4000bde0f0, {0x3110a28, 0x4002280c80})
	net/http/server.go:3330 +0x294
net/http.(*Server).ListenAndServe(0x4000bde0f0)
	net/http/server.go:3259 +0x84
github.com/evcc-io/evcc/cmd.runRoot(0x551cf20, {0x28583d7?, 0x7?, 0x284bf3f?})
	github.com/evcc-io/evcc/cmd/root.go:324 +0x1cd4
github.com/spf13/cobra.(*Command).execute(0x551cf20, {0x400004c090, 0x1, 0x1})
	github.com/spf13/cobra@v1.8.1/command.go:989 +0x81c
github.com/spf13/cobra.(*Command).ExecuteC(0x551cf20)
	github.com/spf13/cobra@v1.8.1/command.go:1117 +0x344
github.com/spf13/cobra.(*Command).Execute(...)
	github.com/spf13/cobra@v1.8.1/command.go:1041
github.com/evcc-io/evcc/cmd.Execute()
	github.com/evcc-io/evcc/cmd/root.go:112 +0x24
main.main()
	github.com/evcc-io/evcc/main.go:44 +0x38

goroutine 7 [chan receive]:
github.com/evcc-io/evcc/cmd.configureDatabase.func2()
	github.com/evcc-io/evcc/cmd/setup.go:541 +0x64
created by github.com/evcc-io/evcc/cmd.configureDatabase in goroutine 1
	github.com/evcc-io/evcc/cmd/setup.go:540 +0x74

goroutine 6 [select, 14673 minutes]:
database/sql.(*DB).connectionOpener(0x400018f040, {0x31192d8, 0x40001bb220})
	database/sql/sql.go:1253 +0x80
created by database/sql.OpenDB in goroutine 1
	database/sql/sql.go:833 +0x128

[... 405 lines cut ...]
nhooyr.io/websocket.(*Conn).timeoutLoop(0x4000fb4000)
	nhooyr.io/websocket@v1.8.17/conn.go:180 +0x130
created by nhooyr.io/websocket.newConn in goroutine 117
	nhooyr.io/websocket@v1.8.17/conn.go:140 +0x484

goroutine 2116404 [select]:
github.com/philippseith/signalr.(*pipe).Read(0x4002346e00, {0x4001238000, 0x8000, 0x54e9320?})
	github.com/philippseith/signalr@v0.6.3/ctxpipe.go:53 +0xa8
github.com/philippseith/signalr.(*PipeReader).Read(0x40022006c0?, {0x4001238000?, 0x4001eb8a50?, 0x0?})
	github.com/philippseith/signalr@v0.6.3/ctxpipe.go:130 +0x24
github.com/philippseith/signalr.readJSONFrames({0x30fed60, 0x4001d0a0f8}, 0x4001eb8a50)
	github.com/philippseith/signalr@v0.6.3/jsonhubprotocol.go:162 +0x78
github.com/philippseith/signalr.(*jsonHubProtocol).ParseMessages(0x40012c24e0, {0x30fed60?, 0x4001d0a0f8?}, 0x0?)
	github.com/philippseith/signalr@v0.6.3/jsonhubprotocol.go:68 +0x38
github.com/philippseith/signalr.(*defaultHubConnection).Receive.func2({0x31192d8, 0x4000b86640}, {0x30fed60, 0x4001d0a0f8}, 0x4002346d90, 0x400209c8c0)
	github.com/philippseith/signalr@v0.6.3/hubconnection.go:138 +0xb4
created by github.com/philippseith/signalr.(*defaultHubConnection).Receive in goroutine 2116402
	github.com/philippseith/signalr@v0.6.3/hubconnection.go:128 +0x2b0

goroutine 3194046 [chan send, 1902 minutes]:
github.com/hasura/go-graphql-client.(*SubscriptionClient).Run.func1()
	github.com/hasura/go-graphql-client@v0.13.1/subscription.go:785 +0x328
created by github.com/hasura/go-graphql-client.(*SubscriptionClient).Run in goroutine 93
	github.com/hasura/go-graphql-client@v0.13.1/subscription.go:775 +0x180
```

foto-andreas, 2025-01-08:
```
curl http://localhost:7070/debug/pprof/goroutine?debug=1
goroutine profile: total 49
3 @ 0x85ee8 0x186a4 0x18274 0x1ff51b8 0x8e704
#	0x1ff51b7	github.com/evcc-io/evcc/util/pipe.(*Dropper).pipe+0x57	github.com/evcc-io/evcc/util/pipe/limiter.go:78

3 @ 0x85ee8 0x6167c 0x68c474 0x8e704
#	0x68c473	google.golang.org/grpc/internal/grpcsync.(*CallbackSerializer).run+0xc3	google.golang.org/grpc@v1.68.0/internal/grpcsync/callback_serializer.go:88

1 @ 0x1e1e4 0x883d0 0x5bc48c 0x8e704
#	0x883cf		os/signal.signal_recv+0x2f	runtime/sigqueue.go:152
#	0x5bc48b	os/signal.loop+0x1b		os/signal/signal_unix.go:23

1 @ 0x42e2c 0x84cf4 0x1fcbbc4 0x1fcb9e0 0x1fc8f78 0x1fddb84 0x1fde498 0x31f608 0x321274 0xf786a4 0x33dfac 0x31e1bc 0x8e704
#	0x1fcbbc3	runtime/pprof.writeRuntimeProfile+0xb3			runtime/pprof/pprof.go:793
#	0x1fcb9df	runtime/pprof.writeGoroutine+0x4f			runtime/pprof/pprof.go:752
#	0x1fc8f77	runtime/pprof.(*Profile).WriteTo+0x147			runtime/pprof/pprof.go:374
#	0x1fddb83	net/http/pprof.handler.ServeHTTP+0x443			net/http/pprof/pprof.go:272
#	0x1fde497	net/http/pprof.Index+0xc7				net/http/pprof/pprof.go:388
#	0x31f607	net/http.HandlerFunc.ServeHTTP+0x37			net/http/server.go:2220
#	0x321273	net/http.(*ServeMux).ServeHTTP+0x1b3			net/http/server.go:2747
#	0xf786a3	github.com/gorilla/mux.(*Router).ServeHTTP+0x193	github.com/gorilla/mux@v1.8.1/mux.go:212
#	0x33dfab	net/http.serverHandler.ServeHTTP+0xbb			net/http/server.go:3210
#	0x31e1bb	net/http.(*conn).serve+0x4fb				net/http/server.go:2092

1 @ 0x85ee8 0x17790 0x173c8 0x1facc78 0x8e704
#	0x1facc77	github.com/evcc-io/evcc/core.(*Site).loopLoadpoints+0x67	github.com/evcc-io/evcc/core/site.go:1068

1 @ 0x85ee8 0x17790 0x173c8 0xe74ff8 0x8e704
#	0xe74ff7	github.com/hasura/go-graphql-client.(*SubscriptionClient).Run.func1+0x327	github.com/hasura/go-graphql-client@v0.13.1/subscription.go:785

1 @ 0x85ee8 0x1842c 0x18254 0xea0cf0 0x8e704
#	0xea0cef	github.com/evcc-io/evcc/meter.NewTibberFromConfig.func3+0x3f	github.com/evcc-io/evcc/meter/tibber-pulse.go:117

1 @ 0x85ee8 0x186a4 0x18254 0x1f80ec8 0x8e704
#	0x1f80ec7	github.com/evcc-io/evcc/tariff.(*Tibber).run+0xd7	github.com/evcc-io/evcc/tariff/tibber.go:84

1 @ 0x85ee8 0x186a4 0x18254 0x1ff44d4 0x8e704
#	0x1ff44d3	github.com/evcc-io/evcc/server/updater.(*watch).watchReleases+0x53	github.com/evcc-io/evcc/server/updater/watch.go:32

1 @ 0x85ee8 0x186a4 0x18254 0x209253c 0x8e704
#	0x209253b	github.com/evcc-io/evcc/cmd.runRoot.func2+0x2b	github.com/evcc-io/evcc/cmd/root.go:262

1 @ 0x85ee8 0x186a4 0x18254 0x2092660 0x8e704
#	0x209265f	github.com/evcc-io/evcc/cmd.runRoot.func1+0x8f	github.com/evcc-io/evcc/cmd/root.go:256

1 @ 0x85ee8 0x186a4 0x18274 0x1016a4c 0x8e704
#	0x1016a4b	github.com/evcc-io/evcc/charger.(*Easee).subscribe.func1+0x4b	github.com/evcc-io/evcc/charger/easee.go:248

1 @ 0x85ee8 0x186a4 0x18274 0x1e244ac 0x8e704
#	0x1e244ab	github.com/evcc-io/evcc/push.(*Hub).Run+0x5b	github.com/evcc-io/evcc/push/hub.go:100

1 @ 0x85ee8 0x186a4 0x18274 0x1effb8c 0x8e704
#	0x1effb8b	github.com/evcc-io/evcc/server.(*MQTT).Run+0x32b	github.com/evcc-io/evcc/server/mqtt.go:286

1 @ 0x85ee8 0x186a4 0x18274 0x1fac8b8 0x8e704
#	0x1fac8b7	github.com/evcc-io/evcc/core.(*Site).Prepare.func1+0x77	github.com/evcc-io/evcc/core/site.go:1033

1 @ 0x85ee8 0x186a4 0x18274 0x1ff43c8 0x8e704
#	0x1ff43c7	github.com/evcc-io/evcc/server/updater.Run+0x197	github.com/evcc-io/evcc/server/updater/run.go:22

1 @ 0x85ee8 0x186a4 0x18274 0x2096c84 0x8e704
#	0x2096c83	github.com/evcc-io/evcc/cmd.configureDatabase.func2+0x63	github.com/evcc-io/evcc/cmd/setup.go:541

1 @ 0x85ee8 0x186a4 0x18274 0x4d99dc 0x8e704
#	0x4d99db	github.com/evcc-io/evcc/util.(*Cache).Run+0x5b	github.com/evcc-io/evcc/util/cache.go:39

1 @ 0x85ee8 0x186a4 0x18274 0x4ddee4 0x8e704
#	0x4ddee3	github.com/evcc-io/evcc/util.(*Tee).Run+0x43	github.com/evcc-io/evcc/util/tee.go:39

[... 139 lines cut ...]
#	0xf2908b	github.com/philippseith/signalr.(*webSocketConnection).Read+0xab		github.com/philippseith/signalr@v0.6.3/websocketconnection.go:47
#	0xf1d4b7	github.com/philippseith/signalr.(*defaultHubConnection).Receive.func1+0x97	github.com/philippseith/signalr@v0.6.3/hubconnection.go:104

1 @ 0x85ee8 0x6167c 0xf1aeb8 0xf1b604 0xf20068 0xf1f588 0xf1d254 0x8e704
#	0xf1aeb7	github.com/philippseith/signalr.(*pipe).Read+0xa7				github.com/philippseith/signalr@v0.6.3/ctxpipe.go:53
#	0xf1b603	github.com/philippseith/signalr.(*PipeReader).Read+0x23				github.com/philippseith/signalr@v0.6.3/ctxpipe.go:130
#	0xf20067	github.com/philippseith/signalr.readJSONFrames+0x77				github.com/philippseith/signalr@v0.6.3/jsonhubprotocol.go:162
#	0xf1f587	github.com/philippseith/signalr.(*jsonHubProtocol).ParseMessages+0x37		github.com/philippseith/signalr@v0.6.3/jsonhubprotocol.go:68
#	0xf1d253	github.com/philippseith/signalr.(*defaultHubConnection).Receive.func2+0xb3	github.com/philippseith/signalr@v0.6.3/hubconnection.go:138

1 @ 0x85ee8 0x6167c 0xf2164c 0xf15c5c 0xf157f8 0x8e704
#	0xf2164b	github.com/philippseith/signalr.(*loop).Run+0xa9b		github.com/philippseith/signalr@v0.6.3/loop.go:77
#	0xf15c5b	github.com/philippseith/signalr.(*client).run+0x10b		github.com/philippseith/signalr@v0.6.3/client.go:245
#	0xf157f7	github.com/philippseith/signalr.(*client).Start.func1+0x107	github.com/philippseith/signalr@v0.6.3/client.go:188

1 @ 0x85ee8 0x6167c 0xf218c4 0x8e704
#	0xf218c3	github.com/philippseith/signalr.(*loop).Run.func1+0xa3	github.com/philippseith/signalr@v0.6.3/loop.go:57

1 @ 0x85ee8 0x62518 0x624f5 0x8781c 0x9d4c4 0x9e0448 0x8e704
#	0x8781b		sync.runtime_Semacquire+0x2b					runtime/sema.go:71
#	0x9d4c3		sync.(*WaitGroup).Wait+0x73					sync/waitgroup.go:118
#	0x9e0447	github.com/eclipse/paho%2emqtt%2egolang.startComms.func3+0x27	github.com/eclipse/paho.mqtt.golang@v1.5.0/net.go:438

1 @ 0x8e704
```

andig, 2025-01-08:
Ich sehe da leider nichts Auffälliges. Offensichtlich laufen Tibber Tarif, Tibber Pulse und eine Easee. Kein Deadlock erkennbar.

/cc @GrimmiMeloni siehst Du evtl. was?

foto-andreas, 2025-01-09:
Es gibt da noch die deadlock-Debug-Ausgabe, die ist binär. Hilft die?

foto-andreas, 2025-01-09:
Ich geh nochmal auf die Suche, ob irgendwas anderes in Container oder drumrum Probleme bereitet. Wenn kein anderer mit diesem Problem konfrontiert ist, sieht das doch irgendwie nach Grund außerhalb von evcc aus.

foto-andreas, 2025-01-09:
* Traefik drumrum neu gestartet, allerdings ohne Erfolg
* Beide EVCC-Container werden in "docker ps" als unhealthy angezeigt
* curl auf die UI von innerhalb des Containers gibt einen 200 und liefert HTML-Code
* curl direkt von außen auf den Container geht nicht, da der Port nicht exportiert wird, das macht traefik
* Port über iptables zu exportieren will auch nicht - bekomme ich weiterhin ein failed to connect

andig, 2025-01-09:
> curl auf die UI von innerhalb des Containers gibt einen 200 und liefert HTML-Code

Also ist das UI doch erreichbar!

foto-andreas, 2025-01-09:
Jepp. Aber nur innerhalb des Containers selbst. Ich habe jetzt einen der Docker-Container gestoppt und wieder gestartet. Das repariert die Situation für diesen Container. Offenbar liegt das Problem also irgendwo in der Docker-Schicht des Containers. Wenn da noch jemand eine Idee zu hat, gerne, ansonsten schau ich mal die Health-Bedingung an und/oder baue was außen drumrum, was den Container alle 15 Minuten neu startet, solange Tibber nicht erreichbar ist. Danke erstmal für das ganze Debug-Kram!

GrimmiMeloni, 2025-01-09:
Ich sehe auf die Schnelle erstmal nichts auffälliges.

Nur zur Sicherheit gefragt: Die Logs von evcc laufen weiter durch, d.h. in jedem Intervall werden Logs geschrieben? Wenn ja, kannst Du mal ein Trace Log erzeugen?

Ansonsten noch als Idee: Vielleicht ist der Tibber auch ein Red Herring und nur ein Seiteneffekt von einer generellen Connectivity Problematik. Wie sieht denn der Netzwerk Stack da ansonsten aus? Traeffik als Reverse Proxy davor hab ich mal aus dem Kontext gelesen. Wie ist der Container denn vernetzt? Bridge, Host?

foto-andreas, 2025-01-09:
Also, Netzwerktechnisch haben die Docker-Compose-Files folgenden Eintrag
```
networks:
  traefik:
    external: true
```

Das Netzwerk dazu:
```
[
    {
        "Name": "traefik",
        "Id": "97197967bfae01a49f4f67f5cbc1099dd92a510d16df45b5013b4a9639a00d9a",
        "Created": "2024-11-14T20:25:28.159153888+01:00",
        "Scope": "local",
        "Driver": "bridge",
        "EnableIPv6": true,
        "IPAM": {
            "Driver": "default",
            "Options": {},
            "Config": [
                {
                    "Subnet": "172.18.0.0/16",
                    "Gateway": "172.18.0.1"
                },
                {
                    "Subnet": "fdd1:c4ab:6c19::/64",
                    "Gateway": "fdd1:c4ab:6c19::1"
                }
            ]
        },
        "Internal": false,
        "Attachable": false,
        "Ingress": false,
        "ConfigFrom": {
            "Network": ""
        },
        "ConfigOnly": false,
        "Containers": {
            "887998dba732b8341c38ea4ecc2fbe37d775f3bd6dc0bee68172bf39114ba3dc": {
                "Name": "evcc-py",
                "EndpointID": "847965ec3cb587cb60db4749cd654bf65db9d699c2260cca5e36185820e719ab",
                "MacAddress": "02:42:ac:12:00:04",
                "IPv4Address": "172.18.0.4/16",
                "IPv6Address": "fdd1:c4ab:6c19::4/64"
            },
            "ce21c690101f134cd2c46e77393a2159d52f6a65365b7e2b7220fdf121fedb0a": {
                "Name": "evcc-wk",
                "EndpointID": "9a03a680e4267ce5cb227819e26ea3489949f02d852fecc53f7d926c95019283",
                "MacAddress": "02:42:ac:12:00:03",
                "IPv4Address": "172.18.0.3/16",
                "IPv6Address": "fdd1:c4ab:6c19::3/64"
            },
            "ecaa1e6e26e73df8295f25f6ef91de7960b2d6f438431da7d3589f209a053c57": {
                "Name": "traefik",
                "EndpointID": "14bd0648f69d5451be50326e10a25f1292d24f7ccacebbb0a81de03b870bcd04",
                "MacAddress": "02:42:ac:12:00:02",
                "IPv4Address": "172.18.0.2/16",
                "IPv6Address": "fdd1:c4ab:6c19::2/64"
            }
        },
        "Options": {},
        "Labels": {}
    }
]
```

Die EVCC-Container exportieren die Ports nicht, da das die einfachste Möglichkeit ist, sie nicht ins Internet zu exportieren. 

Traefik dient als Reverse-Proxy und greift über seine Docker-Integration direkt in die Container, sorgt für eine Authentifizierung, SSL usw. und exportiert die EVCC-UI dann über Hostnamen ins Internet.

Die Logs laufen regelmäßig weiter und liefern konsequent wie oben beschrieben den "grid power: outdated"-Block.

Gerne weiter nachfragen. Ich stelle jetzt mal alles auf trace, muss dann aber warten, bis das Problem wieder auftritt, bis weitere Trace-Ausgaben da sind.

GrimmiMeloni, 2025-01-09:
> Die EVCC-Container exportieren die Ports nicht, da das die einfachste Möglichkeit ist, sie nicht ins Internet zu exportieren.

Ist das so? Muss nicht zumindest ein expose her?

Und steht nicht nur der Traeffik mit einem Bein im Internet?

foto-andreas, 2025-01-09:
Irgendwas muss bei dem Problem hier ja aber auch innerhalb des Containers "sichtbar" sein, denn evcc fühlt sich ja unhealthy.

foto-andreas, 2025-01-10:
Hab nun ein Trace zum Zeitpunkt des Ausfalls:
```
evcc-wk  | [lp-1  ] DEBUG 2025/01/10 15:49:52 !! pvScalePhases EXIT
evcc-wk  | [lp-1  ] DEBUG 2025/01/10 15:49:52 pv charge current: 0A = 0A + -1.86A (427W @ 1p)
evcc-wk  | [cache ] TRACE 2025/01/10 15:49:52 lp-1/planProjectedStart: 0001-01-01 00:00:00 +0000 UTC
evcc-wk  | [cache ] TRACE 2025/01/10 15:49:52 lp-1/planProjectedEnd: 0001-01-01 00:00:00 +0000 UTC
evcc-wk  | [cache ] TRACE 2025/01/10 15:49:52 lp-1/planOverrun: 0s
evcc-wk  | [cache ] TRACE 2025/01/10 15:49:52 greenShareHome: 0.016369273863987135
evcc-wk  | [cache ] TRACE 2025/01/10 15:49:52 greenShareLoadpoints: 0
evcc-wk  | [cache ] TRACE 2025/01/10 15:49:52 tariffPriceHome: 0.35651557131207584
evcc-wk  | [cache ] TRACE 2025/01/10 15:49:52 tariffPriceLoadpoints: 0.3604
evcc-wk  | [pulse ] TRACE 2025/01/10 15:49:52 {"type":"connection_init","payload":{"token":"***"}} client
evcc-wk  | [pulse ] TRACE 2025/01/10 15:49:52 {"type":"connection_init","payload":{"token":"***"}} client
evcc-wk  | [pulse ] TRACE 2025/01/10 15:49:52 {"type":"connection_init","payload":{"token":"***"}} client
evcc-wk  | [pulse ] TRACE 2025/01/10 15:49:52 {"type":"connection_ack"} server
evcc-wk  | [pulse ] TRACE 2025/01/10 15:49:52 {"id":"5e6ad0f4-fc6d-4ee9-8a3e-c9524da99386","type":"subscribe","payload":{"query":"subscription ($homeId:ID!){liveMeasurement(homeId: $homeId){power,powerProduction,lastMeterConsumption,las>
evcc-wk  | [pulse ] TRACE 2025/01/10 15:49:52 {"id":"5e6ad0f4-fc6d-4ee9-8a3e-xxxx","type":"next","payload":{"errors":[{"message":"unable to start stream api:a7a10770336a4d0198ecea09c28da682 for device a7a10770-336a-4d01-98ec-ea0>
evcc-wk  | [pulse ] ERROR 2025/01/10 15:49:52 Message: unable to start stream api:a7a10770336a4d0198ecea09c28da682 for device a7a10770-336a-4d01-98ec-xxxx; userId 29fbf633-dee7-4959-9d19-xxxx, Locations: [{Line:1 Column:>
evcc-wk  | [pulse ] TRACE 2025/01/10 15:49:52 {"id":"5e6ad0f4-fc6d-4ee9-8a3e-xxxx","type":"complete"} server
evcc-wk  | [pulse ] TRACE 2025/01/10 15:49:52 no running subscription. exiting... client
evcc-wk  | [pulse ] TRACE 2025/01/10 15:49:52 {"type":"connection_init","payload":{"token":"***"}} client
evcc-wk  | [pulse ] TRACE 2025/01/10 15:49:52 {"type":"connection_ack"} server
evcc-wk  | [pulse ] TRACE 2025/01/10 15:49:52 no running subscription. exiting... client
evcc-wk  | [site  ] DEBUG 2025/01/10 15:50:07 ----
```

Und dann, wenn "outdated" erkannt wird:

```
evcc-wk  | [site  ] DEBUG 2025/01/10 15:50:22 ----
evcc-wk  | [lp-1  ] DEBUG 2025/01/10 15:50:22 charge power: 0W
evcc-wk  | [lp-1  ] DEBUG 2025/01/10 15:50:22 charge currents: [0.007 0.007 0.009]A
evcc-wk  | [site  ] DEBUG 2025/01/10 15:50:22 grid power: 427W
evcc-wk  | [cache ] TRACE 2025/01/10 15:50:22 lp-1/chargePower: 0
evcc-wk  | [site  ] DEBUG 2025/01/10 15:50:22 grid currents: [0 0 0]A
evcc-wk  | [cache ] TRACE 2025/01/10 15:50:22 lp-1/chargeCurrents: [0.007000000216066837 0.007000000216066837 0.008999999612569809]
evcc-wk  | [cache ] TRACE 2025/01/10 15:50:22 batteryGridChargeActive: false
evcc-wk  | [cache ] TRACE 2025/01/10 15:50:22 gridPower: 427
evcc-wk  | [cache ] TRACE 2025/01/10 15:50:22 gridCurrents: [0 0 0]
evcc-wk  | [sunspec] TRACE 2025/01/10 15:50:22 modbus: send 11 5f 00 00 00 06 01 03 9c 87 00 32
evcc-wk  | [sunspec] TRACE 2025/01/10 15:50:22 modbus: recv 11 5f 00 00 00 67 01 03 64 00 d1 00 38 00 4f 00 49 ff fe 0f b8 0f b8 0f b7 09 1a 09 1b 09 1d ff ff 4e 20 ff fc 13 85 ff fe 13 06 ff ff ec fa ff ff f2 04 ff fc 03 96 02 60 00 00>
evcc-wk  | [sunspec] TRACE 2025/01/10 15:50:22 model 103 block 0 point DCW: 2.0304
evcc-wk  | [pv1   ] TRACE 2025/01/10 15:50:22 modbus: send 11 60 00 00 00 06 01 03 f5 74 00 02
evcc-wk  | [pv1   ] TRACE 2025/01/10 15:50:22 modbus: recv 11 60 00 00 00 07 01 03 04 00 00 00 00
evcc-wk  | [site  ] DEBUG 2025/01/10 15:50:22 pv 1 power: 2W
evcc-wk  | [site  ] DEBUG 2025/01/10 15:50:22 site power: 427W
evcc-wk  | [cache ] TRACE 2025/01/10 15:50:22 pvPower: 2.0304
evcc-wk  | [cache ] TRACE 2025/01/10 15:50:22 pvEnergy: 0
evcc-wk  | [cache ] TRACE 2025/01/10 15:50:22 pv: [{2.0304 0 0 <nil> <nil> <nil>}]
evcc-wk  | [cache ] TRACE 2025/01/10 15:50:22 homePower: 429.0304
evcc-wk  | [lp-1  ] DEBUG 2025/01/10 15:50:22 charge total import: 11476.858kWh
evcc-wk  | [cache ] TRACE 2025/01/10 15:50:22 lp-1/smartCostActive: false
evcc-wk  | [cache ] TRACE 2025/01/10 15:50:22 lp-1/smartCostNextStart: 0001-01-01 00:00:00 +0000 UTC
evcc-wk  | [lp-1  ] DEBUG 2025/01/10 15:50:22 charger status: B
evcc-wk  | [cache ] TRACE 2025/01/10 15:50:22 lp-1/chargeCurrent: 0
evcc-wk  | [lp-1  ] DEBUG 2025/01/10 15:50:22 next soc poll remaining time: 12m15s
evcc-wk  | [cache ] TRACE 2025/01/10 15:50:22 lp-1/sessionEnergy: 4.486388888835791
evcc-wk  | [cache ] TRACE 2025/01/10 15:50:22 lp-1/sessionSolarPercentage: 0
evcc-wk  | [cache ] TRACE 2025/01/10 15:50:22 lp-1/sessionPricePerKWh: 0.3351
evcc-wk  | [cache ] TRACE 2025/01/10 15:50:22 lp-1/sessionPrice: 0.0015033889166488734
evcc-wk  | [cache ] TRACE 2025/01/10 15:50:22 lp-1/sessionCo2PerKWh: <nil>
evcc-wk  | [cache ] TRACE 2025/01/10 15:50:22 lp-1/chargedEnergy: 4.486388888835791
evcc-wk  | [cache ] TRACE 2025/01/10 15:50:22 lp-1/chargeDuration: 15s
evcc-wk  | [cache ] TRACE 2025/01/10 15:50:22 lp-1/chargeTotalImport: 11476.857793333358
evcc-wk  | [lp-1  ] DEBUG 2025/01/10 15:50:22 !! pvScalePhases DOWN activePhases: 1, available power: -427W, scalable: false
[... cut ...]
evcc-wk  | [sunspec] TRACE 2025/01/10 15:50:52 modbus: recv 11 63 00 00 00 67 01 03 64 00 d1 00 39 00 4d 00 4a ff fe 0f d1 0f d2 0f d2 09 1e 09 1a 09 1e ff ff 32 c8 ff fd 13 87 ff fe 13 11 ff ff ec f0 ff ff a2 8b ff fc 03 96 02 60 00 00>
evcc-wk  | [sunspec] TRACE 2025/01/10 15:50:52 model 103 block 0 point DCW: 13.197000000000001
evcc-wk  | [pv1   ] TRACE 2025/01/10 15:50:52 modbus: send 11 64 00 00 00 06 01 03 f5 74 00 02
evcc-wk  | [pv1   ] TRACE 2025/01/10 15:50:52 modbus: recv 11 64 00 00 00 07 01 03 04 00 00 00 00
evcc-wk  | [site  ] DEBUG 2025/01/10 15:50:52 pv 1 power: 13W
```

GrimmiMeloni, 2025-01-10:
> Den zweiten Absatz hab ich nicht verstanden.

Mein Fehler. Ich hatte an einen Forward Proxy gedacht, aber das ist ja im Falle von Traeffik nicht gegeben.

GrimmiMeloni, 2025-01-10:
> Irgendwas muss bei dem Problem hier ja aber auch innerhalb des Containers "sichtbar" sein, denn evcc fühlt sich ja unhealthy.

Der Status unhealthy kommt mit hoher Wahrscheinlichkeit weil vorher der Fehler mit Tibber auftritt. Der health Status vom evcc Core basiert darauf das ein Regelzyklus innerhalb von `1 Minute + Regel Intervall` erfolgreich durchgelaufen ist. Wenn beim Lesen des Grid Meters ein Fehler auftritt kann nicht geregelt werden. Deshalb geht die Instanz dann bei Dir nach 75s in den unhealthy Zustand. Das ist also nur ein Folgefehler.

andig, 2025-01-11:
Trotzdem läuft ja evcc offensichtlich. Da _muss_ auch das UI erreichbar sein.

foto-andreas, 2025-01-11:
Ja, das hatten wir ja oben schon festgestellt, wenn man direkt im Container das UI mit curl aufruft. Ich bin ja jetzt auf der Suche nach dem Problem im Container.

foto-andreas, 2025-01-11:
Ich kanns jetzt zeitweise nachstellen, indem ich EVCC mit Tibber starte, aber eine falsche Home-ID in der Config stehen habe. EVCC fährt hoch, setzt den Container erstmal auf starting nach einer Minute dann aber auf healthy. Und siehe da, mit Debug-Log in Traefik gibts folgende Erklär-Zeile im Log:
```
traefik     | 2025-01-11T18:28:32Z DBG github.com/traefik/traefik/v3/pkg/provider/docker/config.go:200 > Filtering unhealthy or starting container container=evcc-py-evcc-py-95d41e4150b9fed3a1d4928cdc10aa2fcb934be0f3e54c2d52ad2c48b2d6bafa providerName=docker
```
Traefik filtert unhealthy-Container weg. Das erklärt den 404. Frage also erledigt.

Schönen Sonntag dann!

GrimmiMeloni, 2025-01-12:
Erstmal Danke dafür, daß Du die Erklärung für den unresponsive UI geteilt hast.
Was mir aber noch fehllt ist, warum dies _dauerhaft_ der Fall ist.
Nach meinem jetzigen Verständnis würde ich erwarten, daß sobald Tibber wieder online kommt, der Gridmeter check wieder funktioniert, und der Container healthy wird - und somit dann letztlich auch die UI wieder über Traeffik erreichbar ist. Das scheint aber nicht der Fall zu sein?

foto-andreas, 2025-01-12:
Das passiert halt nicht. Es gab auch schon ähnliche Tickets dazu (#17094, #7314). Bisher gabs dazu immer die Antwort: "Ohne grid funktionierts halt nicht, Tibber ist schuld. Die müssen das in den Griff bekommen". Ich sehe das anders, es ist aber wohl müßig, das erneut zu diskutieren.

andig, 2025-01-12:
> Es gab auch schon ähnliche Tickets dazu (https://github.com/evcc-io/evcc/issues/17094, https://github.com/evcc-io/evcc/issues/7314). Bisher gabs dazu immer die Antwort:

V.a. gabs dazu auch schon Fixes die reproduzierbar funktionieren. Lösen die alle Probleme? Anscheinend nicht.

> Tibber ist schuld

Offensichtlich haben Gerät oder API ja eine Macke, also klar. Und natürlich lässt sich das lösen. Dafür brauchts aber einen reproduzierbaren Fall den man sich im Debugger anschauen kann. Bisher gibts den nicht. Wenn Du da bessere Ideen hast: gerne!

foto-andreas, 2025-01-12:
Gerät (Pulse) schließe ich aus, da beider Häuser gleichzeitig betroffen waren. Laut Log kommt offenbar seitens der Tibber-API die Meldung  "unable to start stream api" 8 mal, mit Delay ~0s. 

Im Stack oben steht EVCC in tibber-pulse:117. Das ist das Close auf dem GraphQL-Client. Vielleicht hängt da was. Zumindest wird das in beiden Stacktrace-Varianten ausgegeben. Vielleicht mal eine TRACE-Ausgabe an den Stellen einbauen, wo ein Retry erwartet wird und ggf. nicht kommt.

Die 8 Retrys hab ich im Code nicht nachvollziehen können.

Bombarde76, 2025-01-13:
> Muss jetzt leider erstmal los. Passiert das bei keinem anderen der Tibber-Evcc-Leute? Hätte da jetzt eigentlich ein globaleres Problem erwartet, da bei mir auch immer beide Hütten betrofen sind.

Hallo zusammen,
ich habe das gleiche Problem seit Mitte Dezember, genauso wie du es beschreibst

Bombarde76, 2025-01-13:
ja, ich nutze auch Docker (synology)

Bombarde76, 2025-01-15:
heute wieder ausgefallen:
15:59 Uhr
[evcc-20250115-172345-debug.log](https://github.com/user-attachments/files/18427451/evcc-20250115-172345-debug.log)

andig, 2025-01-15:
> Dafür brauchts aber einen reproduzierbaren Fall den man sich im Debugger anschauen kann. 

Leider ändern Logfiles daran nichts.

bytec77, 2025-01-19:
Ich habe das Problem auch seit einiger Zeit. Es hilft nur ein Neustart von evcc. Die Daten kommen von tibber aber zuverlässig einmal unter developer.tibber.com zu sehen als auch evcc -c evcc.yaml -l debug meter liefert den richtigen wert in der Gui bekomme ich aber nur Grid Power Outdated

`[site ] DEBUG 2025/01/19 12:17:47 ----
[lp-1 ] DEBUG 2025/01/19 12:17:47 charge power: 0W
[lp-1 ] DEBUG 2025/01/19 12:17:47 charge currents: [0 0 0]A
[site ] ERROR 2025/01/19 12:17:48 grid power: outdated`

evcc läuft bei mir in einem LXC unter Proxmox.

GrimmiMeloni, 2025-01-20:
Ich habe mir gerade den Trace von @foto-andreas nochmal angeschaut.

TL;DR - Für mich sieht es nach einem Deadlock im GraphQL Client aus.

Details:
* der GraphQL Client sollte meiner Ansicht nach mit 2 Routinen auf dem Threaddump zu finden sein.
* Ein Thread ist vorhanden - siehe ganz am Ende vom Dump. Das ist ein Thread der versucht auf den Error Kanal zu schreiben. Der Channel ist nicht gebuffert, und der Send hängt. Da fehlt es also seit 1902 Minuten an einem Empfänger.
* Durchsicht des GraphQL Client Codes zeigt, daß es nur einen exakt einen Consumer für diesen Channel gibt, und zwar ist es der der Thread der auch den o.g. Fehlerhandler spawnt. (subscription.go#852) 
* Laut Threaddump hat dieser Thread die Goroutine 93
* Diese Routine 93 ist nicht (mehr) vorhanden

@andig wenn ich mir den GraphQL Client genauer anschaue in Zeile 848ff. (Dieser For Block mit den Selects). Das sieht für mich danach aus, daß der Client selbst retries durchführt (das sind die Fälle wo `return sc.Run()`aufgerufen wird), und dann gibt es aber auch Situation bei denen ausgestiegen (`return nil`, bzw. `return err`). Müssen wir hier also GraphQL Client consumer noch außen herum damit umgehen, daß der Client sich beendet?

andig, 2025-01-20:
@GrimmiMeloni siehst Du eine Möglichkeit, die Situation irgendwie zu provozieren? Und sei es nur durch if/else im GraphQL Client? Dann hätte man was zum testen...

andig, 2025-01-20:
@foto-andreas bitte nightly verwenden- dort gibt es jetzt mehr Logging.

GrimmiMeloni, 2025-01-20:
@andig ja, das ist die Stelle. Das logging in der geforkten Lib sollte zumindest Klarheit darüber schaffen ob dies der Auslöser ist.

Bombarde76, 2025-01-20:
klasse, dass  ihr euch da nochmals dahinterklemmt

pschwed, 2025-01-21:
Bei mir läuft auch der erwähnte Nighly mit --profile. Bisher keine Probleme, aber er verhält sich auch anders, als die Versionen davor. Alle paar Sekunden wird der Pulse Stream neu abboniert...

```
[pulse ] TRACE 2025/01/21 12:03:03 {"id":"24023bc3-781a-4802-a12e-7f1cd5981c41","type":"next","payload":{"data":{"liveMeasurement":{"power":85.31,"powerProduction":0,"lastMeterConsumption":16630.2090932,"lastMeterProduction":527.00758342,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
[pulse ] TRACE 2025/01/21 12:03:06 {"id":"24023bc3-781a-4802-a12e-7f1cd5981c41","type":"next","payload":{"data":{"liveMeasurement":{"power":86.16,"powerProduction":0,"lastMeterConsumption":16630.20914038,"lastMeterProduction":527.00758342,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
[pulse ] TRACE 2025/01/21 12:03:16 {"id":"24023bc3-781a-4802-a12e-7f1cd5981c41","type":"complete"} client
[pulse ] TRACE 2025/01/21 12:03:16 {"type":"connection_init","payload":{"token":"***"}} client
[pulse ] TRACE 2025/01/21 12:03:17 {"type":"connection_ack"} server
[pulse ] TRACE 2025/01/21 12:03:17 {"id":"19702cbc-e226-4945-82a0-976c4334ef28","type":"subscribe","payload":{"query":"subscription ($homeId:ID!){liveMeasurement(homeId: $homeId){power,powerProduction,lastMeterConsumption,lastMeterProduction,currentL1,currentL2,currentL3}}","variables":{"homeId":"***"}}} client
[pulse ] TRACE 2025/01/21 12:03:19 {"id":"19702cbc-e226-4945-82a0-976c4334ef28","type":"next","payload":{"data":{"liveMeasurement":{"power":340.42,"powerProduction":0,"lastMeterConsumption":16630.20991783,"lastMeterProduction":527.00758342,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
[pulse ] TRACE 2025/01/21 12:03:22 {"id":"19702cbc-e226-4945-82a0-976c4334ef28","type":"next","payload":{"data":{"liveMeasurement":{"power":471.77,"powerProduction":0,"lastMeterConsumption":16630.21037559,"lastMeterProduction":527.00758342,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
[pulse ] TRACE 2025/01/21 12:03:25 {"id":"19702cbc-e226-4945-82a0-976c4334ef28","type":"next","payload":{"data":{"liveMeasurement":{"power":519.63,"powerProduction":0,"lastMeterConsumption":16630.21066695,"lastMeterProduction":527.00758342,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
[pulse ] TRACE 2025/01/21 12:03:28 {"id":"19702cbc-e226-4945-82a0-976c4334ef28","type":"next","payload":{"data":{"liveMeasurement":{"power":645.83,"powerProduction":0,"lastMeterConsumption":16630.21133298,"lastMeterProduction":527.00758342,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
[pulse ] TRACE 2025/01/21 12:03:32 {"id":"19702cbc-e226-4945-82a0-976c4334ef28","type":"next","payload":{"data":{"liveMeasurement":{"power":699.6,"powerProduction":0,"lastMeterConsumption":16630.21171062,"lastMeterProduction":527.00758342,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
[pulse ] TRACE 2025/01/21 12:03:34 {"id":"19702cbc-e226-4945-82a0-976c4334ef28","type":"next","payload":{"data":{"liveMeasurement":{"power":817.31,"powerProduction":0,"lastMeterConsumption":16630.21256707,"lastMeterProduction":527.00758342,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
[pulse ] TRACE 2025/01/21 12:03:44 {"id":"19702cbc-e226-4945-82a0-976c4334ef28","type":"complete"} client
[pulse ] TRACE 2025/01/21 12:03:44 {"type":"connection_init","payload":{"token":"***"}} client
[pulse ] TRACE 2025/01/21 12:03:44 {"type":"connection_ack"} server
[pulse ] TRACE 2025/01/21 12:03:44 {"id":"204d4ce5-b03c-4fc8-9660-ab2550de0356","type":"subscribe","payload":{"query":"subscription ($homeId:ID!){liveMeasurement(homeId: $homeId){power,powerProduction,lastMeterConsumption,lastMeterProduction,currentL1,currentL2,currentL3}}","variables":{"homeId":"***"}}} client
[pulse ] TRACE 2025/01/21 12:03:49 {"id":"204d4ce5-b03c-4fc8-9660-ab2550de0356","type":"next","payload":{"data":{"liveMeasurement":{"power":1412.2,"powerProduction":0,"lastMeterConsumption":16630.21769725,"lastMeterProduction":527.00758342,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
[pulse ] TRACE 2025/01/21 12:03:52 {"id":"204d4ce5-b03c-4fc8-9660-ab2550de0356","type":"next","payload":{"data":{"liveMeasurement":{"power":1442.84,"powerProduction":0,"lastMeterConsumption":16630.21849545,"lastMeterProduction":527.00758342,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
[pulse ] TRACE 2025/01/21 12:03:55 {"id":"204d4ce5-b03c-4fc8-9660-ab2550de0356","type":"next","payload":{"data":{"liveMeasurement":{"power":1403.66,"powerProduction":0,"lastMeterConsumption":16630.22007545,"lastMeterProduction":527.00758342,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
[pulse ] TRACE 2025/01/21 12:03:59 {"id":"204d4ce5-b03c-4fc8-9660-ab2550de0356","type":"next","payload":{"data":{"liveMeasurement":{"power":1356.9,"powerProduction":0,"lastMeterConsumption":16630.22083847,"lastMeterProduction":527.00758342,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
[pulse ] TRACE 2025/01/21 12:04:02 {"id":"204d4ce5-b03c-4fc8-9660-ab2550de0356","type":"next","payload":{"data":{"liveMeasurement":{"power":1285.31,"powerProduction":0,"lastMeterConsumption":16630.22229392,"lastMeterProduction":527.00758342,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
[pulse ] TRACE 2025/01/21 12:04:05 {"id":"204d4ce5-b03c-4fc8-9660-ab2550de0356","type":"next","payload":{"data":{"liveMeasurement":{"power":1267.05,"powerProduction":0,"lastMeterConsumption":16630.22300145,"lastM [... cut ...]
[pulse ] TRACE 2025/01/21 12:05:32 {"id":"83c8d890-ee65-448a-bf09-46255ca70244","type":"subscribe","payload":{"query":"subscription ($homeId:ID!){liveMeasurement(homeId: $homeId){power,powerProduction,lastMeterConsumption,lastMeterProduction,currentL1,currentL2,currentL3}}","variables":{"homeId":"***"}}} client
[pulse ] TRACE 2025/01/21 12:05:33 {"id":"83c8d890-ee65-448a-bf09-46255ca70244","type":"next","payload":{"data":{"liveMeasurement":{"power":882.48,"powerProduction":0,"lastMeterConsumption":16630.2469786,"lastMeterProduction":527.00758342,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
```

machristoph1, 2025-01-21:
So, jetzt ist es wieder passiert. Hier das Log-File rund um den entsprechenden Zeitpunkt
```
[lp-1  ] DEBUG 2025/01/21 16:59:41 plan: charge 4h15m32s between 2025-01-21 23:44:28 +0100 CET until 2025-01-22 07:00:00 +0100 CET (power: 22080W, avg cost: 0.358)
[pulse ] TRACE 2025/01/21 16:59:43 {"id":"5e9c53b3-9b53-41fb-a5c4-4489f7f64a03","type":"next","payload":{"data":{"liveMeasurement":{"power":0,"powerProduction":0,"lastMeterConsumption":26231.9406,"lastMeterProduction":0,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
[pulse ] TRACE 2025/01/21 16:59:46 {"id":"5e9c53b3-9b53-41fb-a5c4-4489f7f64a03","type":"next","payload":{"data":{"liveMeasurement":{"power":0,"powerProduction":0,"lastMeterConsumption":26231.9406,"lastMeterProduction":0,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
[pulse ] TRACE 2025/01/21 16:59:49 {"id":"5e9c53b3-9b53-41fb-a5c4-4489f7f64a03","type":"next","payload":{"data":{"liveMeasurement":{"power":0,"powerProduction":0,"lastMeterConsumption":26231.9406,"lastMeterProduction":0,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
!! ReadJSON: failed to read JSON message: failed to get reader: use of closed network connection
!! ReadJSON: EOF
!! Run: <-sc.errorChan retry subscription client
!! Run: <-sc.errorChan: errRetry
[pulse ] TRACE 2025/01/21 16:59:59 {"id":"5e9c53b3-9b53-41fb-a5c4-4489f7f64a03","type":"complete"} client
[pulse ] TRACE 2025/01/21 16:59:59 {"type":"connection_init","payload":{"token":"***"}} client
[pulse ] TRACE 2025/01/21 16:59:59 {"type":"connection_ack"} server
[pulse ] TRACE 2025/01/21 16:59:59 {"id":"0f63cb94-7358-4a02-8137-11f6a1e3001d","type":"subscribe","payload":{"query":"subscription ($homeId:ID!){liveMeasurement(homeId: $homeId){power,powerProduction,lastMeterConsumption,lastMeterProduction,currentL1,currentL2,currentL3}}","variables":{"homeId":"***"}}} client
!! ReadJSON: failed to read JSON message: failed to get reader: use of closed network connection
!! ReadJSON: EOF
!! Run: <-sc.errorChan retry subscription client
!! Run: <-sc.errorChan: errRetry
[pulse ] TRACE 2025/01/21 17:00:09 {"id":"0f63cb94-7358-4a02-8137-11f6a1e3001d","type":"complete"} client
[pulse ] TRACE 2025/01/21 17:00:09 {"type":"connection_init","payload":{"token":"***"}} client
[pulse ] TRACE 2025/01/21 17:00:09 {"type":"connection_ack"} server
[pulse ] TRACE 2025/01/21 17:00:09 {"id":"3217583b-6397-4fb0-a58a-d45b32cd1d1f","type":"subscribe","payload":{"query":"subscription ($homeId:ID!){liveMeasurement(homeId: $homeId){power,powerProduction,lastMeterConsumption,lastMeterProduction,currentL1,currentL2,currentL3}}","variables":{"homeId":"***"}}} client
[pulse ] TRACE 2025/01/21 17:00:09 {"id":"3217583b-6397-4fb0-a58a-d45b32cd1d1f","type":"next","payload":{"errors":[{"message":"unable to start stream api:afeb520a2ce24c9ab87154ebcc9a6e5b for device afeb520a-2ce2-4c9a-b871-54ebcc9a6e5b; userId 73f50685-248e-419b-bfca-9163490dbe58","locations":[{"line":1,"column":28}],"path":["liveMeasurement"]}]}} server
!! Run: <-sc.errorChan Message: unable to start stream api:afeb520a2ce24c9ab87154ebcc9a6e5b for device afeb520a-2ce2-4c9a-b871-54ebcc9a6e5b; userId 73f50685-248e-419b-bfca-9163490dbe58, Locations: [{Line:1 Column:28}], Extensions: map[], Path: [liveMeasurement]
[pulse ] TRACE 2025/01/21 17:00:09 {"id":"3217583b-6397-4fb0-a58a-d45b32cd1d1f","type":"complete"} server
[pulse ] ERROR 2025/01/21 17:00:09 Message: unable to start stream api:afeb520a2ce24c9ab87154ebcc9a6e5b for device afeb520a-2ce2-4c9a-b871-54ebcc9a6e5b; userId 73f50685-248e-419b-bfca-9163490dbe58, Locations: [{Line:1 Column:28}], Extensions: map[], Path: [liveMeasurement]
[pulse ] TRACE 2025/01/21 17:00:09 no running subscription. exiting... client
!! ReadJSON: failed to read JSON message: failed to get reader: use of closed network connection
!! ReadJSON: EOF
[pulse ] TRACE 2025/01/21 17:00:09 {"type":"connection_init","payload":{"token":"***"}} client
!! Run: <-sc.errorChan retry subscription client
!! Run: <-sc.errorChan: errRetry
!! ReadJSON: failed to read JSON message: failed to read: use of closed network connection
!! ReadJSON: EOF
[pulse ] TRACE 2025/01/21 17:00:09 {"type":"connection_init","payload":{"token":"***"}} client
!! Run: <-sc.errorChan retry subscription client
!! Run: <-sc.errorChan: errRetry
!! ReadJSON: failed to read JSON message: failed to get reader: received close frame: status = StatusNormalClosure and reason = "close websocket"
[pulse ] TRACE 2025/01/21 17:00:09 {"type":"connection_init","payload":{"token":"***"}} client
[pulse ] TRACE 2025/01/21 17:00:10 {"type":"connection_ack"} server
[pulse ] TRACE 2025/01/21 17:00:10 no running subscription. exiting... client
!! ReadJSON: failed to read JSON message: failed to get reader: use of closed network connection
!! ReadJSON: EOF
[site  ] DEBUG 2025/01/21 17:00:11 ----
[lp-1  ] DEBUG 2025/01/21 17:00:11 charge power: 0W
[lp-1  ] DEBUG 2025/01/21 17:00:11 charge currents: [0 0 0]A
[site  ] DEBUG 2025/01/21 17:00:11 grid power: 0W
[site  ] DEBUG 2025/01/21 17:00:11 grid currents: [0 0 0]A
[site  ] DEBUG 2025/01/21 17:00:11 pv power: 0W
[site  ] DEBUG 2025/01/21 17:00:11 site power: 0W
[lp-1  ] DEBUG 2025/01/21 17:00:11 charge total import: 13919.720kWh
[lp-1  ] DEBUG 2025/01/21 17:00:11 charger status: A
[lp-1  ] DEBUG 2025/01/21 17:00:11 plan: charge 4h15m32s between 2025-01-21 23:44:28 +0100 CET until 2025-01-22 07:00:00 +0100 CET (power: 22080W, avg cost: 0.358)
[site  ] DEBUG 2025/01/21 17:00:41 ----
[... cut ...]
[lp-1  ] DEBUG 2025/01/21 17:00:41 plan: charge 4h15m32s between 2025-01-21 23:44:28 +0100 CET until 2025-01-22 07:00:00 +0100 CET (power: 22080W, avg cost: 0.358)
[site  ] DEBUG 2025/01/21 17:01:11 ----
[lp-1  ] DEBUG 2025/01/21 17:01:11 charge power: 0W
[lp-1  ] DEBUG 2025/01/21 17:01:11 charge currents: [0 0 0]A
[site  ] ERROR 2025/01/21 17:01:11 grid power: outdated
[site  ] DEBUG 2025/01/21 17:01:41 ----

```

Pprof-Sachen habe ich im Anhang als tar.gz zusammengepackt. Ich hoffe, damit könnt Ihr was anfangen...

[pprof.tar.gz](https://github.com/user-attachments/files/18494163/pprof.tar.gz)

pschwed, 2025-01-21:
.. und vor einer Stunde ist er dann abgeschmiert. Per http normal erreichbar, aber evcc natürlich ohne Funktion:

```
[site ] DEBUG 2025/01/21 16:59:31 ----
[lp-2 ] DEBUG 2025/01/21 16:59:31 charge power: 0W
[lp-1 ] DEBUG 2025/01/21 16:59:31 charge power: 0W
[lp-1 ] DEBUG 2025/01/21 16:59:31 charge currents: [0.872 0.424 0.622]A
[site ] DEBUG 2025/01/21 16:59:31 grid power: 1307W
[site ] DEBUG 2025/01/21 16:59:31 grid currents: [0 0 0]A
[site ] DEBUG 2025/01/21 16:59:31 pv 1 power: 0W
[pulse ] TRACE 2025/01/21 16:59:31 {"id":"cc7b1cc4-56a1-4f56-85bb-6a7a5babe9af","type":"next","payload":{"data":{"liveMeasurement":{"power":1308.21,"powerProduction":0,"lastMeterConsumption":16634.55085575,"lastMeterProduction":527.00758342,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
[site ] DEBUG 2025/01/21 16:59:32 site power: 1307W
[lp-2 ] DEBUG 2025/01/21 16:59:32 charger status: B
[lp-2 ] DEBUG 2025/01/21 16:59:32 pv charge current: 0A = 0A + -5.68A (1307W @ 1p)
[pulse ] TRACE 2025/01/21 16:59:41 {"id":"cc7b1cc4-56a1-4f56-85bb-6a7a5babe9af","type":"complete"} client
[pulse ] TRACE 2025/01/21 16:59:41 {"type":"connection_init","payload":{"token":"***"}} client
[pulse ] TRACE 2025/01/21 16:59:41 {"type":"connection_ack"} server
[pulse ] TRACE 2025/01/21 16:59:41 {"id":"2d8a8df1-bbf1-42f4-8ed3-08d452148776","type":"subscribe","payload":{"query":"subscription ($homeId:ID!){liveMeasurement(homeId: $homeId){power,powerProduction,lastMeterConsumption,lastMeterProduction,currentL1,currentL2,currentL3}}","variables":{"homeId":"***"}}} client
[pulse ] TRACE 2025/01/21 16:59:47 {"id":"2d8a8df1-bbf1-42f4-8ed3-08d452148776","type":"next","payload":{"data":{"liveMeasurement":{"power":1307.53,"powerProduction":0,"lastMeterConsumption":16634.55666693,"lastMeterProduction":527.00758342,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
[pulse ] TRACE 2025/01/21 16:59:50 {"id":"2d8a8df1-bbf1-42f4-8ed3-08d452148776","type":"next","payload":{"data":{"liveMeasurement":{"power":1308.83,"powerProduction":0,"lastMeterConsumption":16634.55739379,"lastMeterProduction":527.00758342,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
[pulse ] TRACE 2025/01/21 17:00:00 {"id":"2d8a8df1-bbf1-42f4-8ed3-08d452148776","type":"complete"} client
[pulse ] TRACE 2025/01/21 17:00:00 {"type":"connection_init","payload":{"token":"***"}} client
[pulse ] TRACE 2025/01/21 17:00:01 {"type":"connection_ack"} server
[pulse ] TRACE 2025/01/21 17:00:01 {"id":"93547553-f2f6-4a6c-a8b3-b52471b842a8","type":"subscribe","payload":{"query":"subscription ($homeId:ID!){liveMeasurement(homeId: $homeId){power,powerProduction,lastMeterConsumption,lastMeterProduction,currentL1,currentL2,currentL3}}","variables":{"homeId":"***"}}} client
[site ] DEBUG 2025/01/21 17:00:01 ----
[lp-2 ] DEBUG 2025/01/21 17:00:01 charge power: 0W
[lp-1 ] DEBUG 2025/01/21 17:00:01 charge power: 0W
[lp-1 ] DEBUG 2025/01/21 17:00:01 charge currents: [0.872 0.424 0.622]A
[site ] DEBUG 2025/01/21 17:00:01 grid power: 1309W
[site ] DEBUG 2025/01/21 17:00:01 grid currents: [0 0 0]A
[site ] DEBUG 2025/01/21 17:00:01 pv 1 power: 0W
[site ] DEBUG 2025/01/21 17:00:01 site power: 1309W
[lp-1 ] DEBUG 2025/01/21 17:00:01 charger status: A
[pulse ] TRACE 2025/01/21 17:00:11 {"id":"93547553-f2f6-4a6c-a8b3-b52471b842a8","type":"complete"} client
[pulse ] TRACE 2025/01/21 17:00:11 {"type":"connection_init","payload":{"token":"***"}} client
[pulse ] TRACE 2025/01/21 17:00:11 {"type":"connection_ack"} server
[pulse ] TRACE 2025/01/21 17:00:11 {"id":"f0e1f917-3db9-454d-974f-faf8ff8af836","type":"subscribe","payload":{"query":"subscription ($homeId:ID!){liveMeasurement(homeId: $homeId){power,powerProduction,lastMeterConsumption,lastMeterProduction,currentL1,currentL2,currentL3}}","variables":{"homeId":"***"}}} client
[pulse ] TRACE 2025/01/21 17:00:11 {"id":"f0e1f917-3db9-454d-974f-faf8ff8af836","type":"next","payload":{"errors":[{"message":"unable to start stream api:90c423074bbc4ff58fe47ddb8fce7020 for device 90c42307-4bbc-4ff5-8fe4-7ddb8fce7020; userId 6d634dc6-cb9b-48fd-9ff2-1a665ed80b9b","locations":[{"line":1,"column":28}],"path":["liveMeasurement"]}]}} server
[pulse ] TRACE 2025/01/21 17:00:11 {"id":"f0e1f917-3db9-454d-974f-faf8ff8af836","type":"complete"} server
[pulse ] TRACE 2025/01/21 17:00:11 no running subscription. exiting... client
[pulse ] ERROR 2025/01/21 17:00:11 Message: unable to start stream api:90c423074bbc4ff58fe47ddb8fce7020 for device 90c42307-4bbc-4ff5-8fe4-7ddb8fce7020; userId 6d634dc6-cb9b-48fd-9ff2-1a665ed80b9b, Locations: [{Line:1 Column:28}], Extensions: map[], Path: [liveMeasurement]
[pulse ] TRACE 2025/01/21 17:00:11 {"type":"connection_init","payload":{"token":"***"}} client
[pulse ] TRACE 2025/01/21 17:00:11 {"type":"connection_ack"} server
[pulse ] TRACE 2025/01/21 17:00:11 no running subscription. exiting... client
[site ] DEBUG 2025/01/21 17:00:31 ----
[lp-2 ] DEBUG 2025/01/21 17:00:31 charge power: 0W
[lp-1 ] DEBUG 2025/01/21 17:00:31 charge power: 0W
[lp-1 ] DEBUG 2025/01/21 17:00:31 charge currents: [0.872 0.424 0.622]A
[site ] DEBUG 2025/01/21 17:00:31 grid power: 1309W
[site ] DEBUG 2025/01/21 17:00:31 grid currents: [0 0 0]A
[site ] DEBUG 2025/01/21 17:00:31 pv 1 power: 0W
[site ] DEBUG 2025/01/21 17:00:33 site power: 1309W
[lp-2 ] DEBUG 2025/01/21 17:00:33 charger status: B
[lp-2 ] DEBUG 2025/01/21 17:00:33 pv charge current: 0A = 0A + -5.69A (1309W @ 1p)
[site ] DEBUG 2025/01/21 17:01:01 ----
[lp-2 ] DEBUG 2025/01/21 17:01:01 charge power: 0W
[lp-1 ] DEBUG 2025/01/21 17:01:01 charge power: 0W
[lp-1 ] DEBUG 2025/01/21 17:01:01 charge currents: [0.872 0.424 0.622]A
[site ] DEBUG 2025/01/21 17:01:01 pv 1 power: 0W
[site ] ERROR 2025/01/21 17:01:02 grid power: outdated
[site ] DEBUG 2025/01/21 17:01:31 ----
```

leider unter /debug/pprof nur ein 404 Fehler.. Vor dem Absturz hat es aber funktioniert..

bytec77, 2025-01-21:
Bei mir auch seit 17:00 Uhr. Nur noch Grid power outdated im log. 

GUI geht noch aber bis zu einem restart wird nix gehen.

andig, 2025-01-21:
@pschwed nicht das nightly? Solche Logs kennen wir schon, wo ist die neue Information?

andig, 2025-01-21:
@machristoph1 dein evcc läuft weiter, ui erreichbar?

machristoph1, 2025-01-22:
Habe inzwischen neu gestartet, damit ich laden konnte. Aber ja. UI erreichbar. Zeigt das rote Warn- Dreieck für Fehler. Laden geht aber nicht.

SmartVolker, 2025-01-22:
Ich bin auf der Version 0.133.0 und sehe das gleiche Problem, dass kurzzeitige Tibber Pulse Ausfälle zum Ausfall / Störung bei EVCC führt und kein Ladevorgang gestartet werden kann. Das Problem hat auch in der Version davor bestanden. Erst ein Restart/Reboot von evcc führt dazu das die Meter Werte von Tibber Pulse wieder gelesen werden.  Einige Screenshot dazu habe ich angefügt.

[image]
[image]
[image]
[image]
[image]
[image]

andig, 2025-01-22:
@SmartVolker bitte keine epischen Screenshots die einfach nur "ich auch" sagen. Das hilft Niemandem. Danke.

andig, 2025-01-22:
@machristoph1 bitte in 30min nochmal neues Log mit neuem Nightly. Vermutung: beim retry wird `Run()` schon frühzeitig beendet.

machristoph1, 2025-01-22:
Ist ok, starte ich nachher. Kann halt etwas dauern, bis der Fehler auftritt…

GrimmiMeloni, 2025-01-22:
@andig ich schaue gerne nochmal mit drauf - würde jetzt aber erst mal abwarten was der neue Nightly bringt.

machristoph1, 2025-01-23:
Von heute Nacht gibt es wieder ein paar Log-Meldungen, die so aussehen, als ob tibberseitig das Problem aufgetreten wäre. EVCC ist aber diesmal nicht in den gridpower: outdated Status gefallen und hat weiterhin funktioniert. 
```

Jan 23 02:42:38 raspicm evcc[29161]: [site  ] DEBUG 2025/01/23 02:42:38 ----
Jan 23 02:42:38 raspicm evcc[29161]: [lp-1  ] DEBUG 2025/01/23 02:42:38 charge power: 19310W
Jan 23 02:42:38 raspicm evcc[29161]: [lp-1  ] DEBUG 2025/01/23 02:42:38 charge currents: [29 28.9 29]A
Jan 23 02:42:38 raspicm evcc[29161]: [site  ] DEBUG 2025/01/23 02:42:38 grid power: 20599W
Jan 23 02:42:38 raspicm evcc[29161]: [site  ] DEBUG 2025/01/23 02:42:38 grid currents: [0 0 0]A
Jan 23 02:42:38 raspicm evcc[29161]: [site  ] DEBUG 2025/01/23 02:42:38 pv power: 0W
Jan 23 02:42:38 raspicm evcc[29161]: [site  ] DEBUG 2025/01/23 02:42:38 site power: 20599W
Jan 23 02:42:38 raspicm evcc[29161]: [lp-1  ] DEBUG 2025/01/23 02:42:38 detected active phases: 3p
Jan 23 02:42:38 raspicm evcc[29161]: [lp-1  ] DEBUG 2025/01/23 02:42:38 charge total import: 13956.346kWh
Jan 23 02:42:38 raspicm evcc[29161]: [lp-1  ] DEBUG 2025/01/23 02:42:38 charger status: C
Jan 23 02:42:38 raspicm evcc[29161]: [lp-1  ] DEBUG 2025/01/23 02:42:38 soc estimated: 83.41% (vehicle: 83.00%)
Jan 23 02:42:38 raspicm evcc[29161]: [lp-1  ] DEBUG 2025/01/23 02:42:38 vehicle soc: 83%
Jan 23 02:42:38 raspicm evcc[29161]: [lp-1  ] DEBUG 2025/01/23 02:42:38 vehicle soc limit: 100%
Jan 23 02:42:38 raspicm evcc[29161]: [lp-1  ] DEBUG 2025/01/23 02:42:38 vehicle range: 423km
Jan 23 02:42:38 raspicm evcc[29161]: [lp-1  ] DEBUG 2025/01/23 02:42:38 plan: charge 1h40m15s between 2025-01-23 02:49:45 +0100 CET until 2025-01-23 04:30:00 +0100 CET (power: 22080W, avg cost: 0.331)
Jan 23 02:42:38 raspicm evcc[29161]: [lp-1  ] DEBUG 2025/01/23 02:42:38 plan: continuing until end of slot
Jan 23 02:42:39 raspicm evcc[29161]: [pulse ] TRACE 2025/01/23 02:42:39 {"id":"656f7a3e-d1de-454b-9702-c7f8ea8cdc1b","type":"next","payload":{"data":{"liveMeasurement":{"power":20599,"powerProduction":0,"lastMeterConsumption":26271.1956,"lastMeterProduction":0,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
Jan 23 02:42:42 raspicm evcc[29161]: [pulse ] TRACE 2025/01/23 02:42:42 {"id":"656f7a3e-d1de-454b-9702-c7f8ea8cdc1b","type":"next","payload":{"data":{"liveMeasurement":{"power":20602,"powerProduction":0,"lastMeterConsumption":26271.2187,"lastMeterProduction":0,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
Jan 23 02:42:46 raspicm evcc[29161]: [pulse ] TRACE 2025/01/23 02:42:46 {"id":"656f7a3e-d1de-454b-9702-c7f8ea8cdc1b","type":"next","payload":{"data":{"liveMeasurement":{"power":20610,"powerProduction":0,"lastMeterConsumption":26271.2303,"lastMeterProduction":0,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
Jan 23 02:42:49 raspicm evcc[29161]: [pulse ] TRACE 2025/01/23 02:42:49 {"id":"656f7a3e-d1de-454b-9702-c7f8ea8cdc1b","type":"next","payload":{"data":{"liveMeasurement":{"power":20602,"powerProduction":0,"lastMeterConsumption":26271.2535,"lastMeterProduction":0,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
Jan 23 02:42:52 raspicm evcc[29161]: [pulse ] TRACE 2025/01/23 02:42:52 {"id":"656f7a3e-d1de-454b-9702-c7f8ea8cdc1b","type":"next","payload":{"data":{"liveMeasurement":{"power":20613,"powerProduction":0,"lastMeterConsumption":26271.2651,"lastMeterProduction":0,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
Jan 23 02:42:55 raspicm evcc[29161]: [pulse ] TRACE 2025/01/23 02:42:55 {"id":"656f7a3e-d1de-454b-9702-c7f8ea8cdc1b","type":"next","payload":{"data":{"liveMeasurement":{"power":20608,"powerProduction":0,"lastMeterConsumption":26271.2883,"lastMeterProduction":0,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
Jan 23 02:43:05 raspicm evcc[29161]: !! ReadJSON: failed to read JSON message: failed to get reader: context deadline exceeded
Jan 23 02:43:05 raspicm evcc[29161]: !! Run: <-sc.errorChan retry subscription client
Jan 23 02:43:05 raspicm evcc[29161]: !! Run: <-sc.errorChan: errRetry
Jan 23 02:43:05 raspicm evcc[29161]: [pulse ] TRACE 2025/01/23 02:43:05 {"id":"656f7a3e-d1de-454b-9702-c7f8ea8cdc1b","type":"complete"} client
Jan 23 02:43:05 raspicm evcc[29161]: [pulse ] TRACE 2025/01/23 02:43:05 {"type":"connection_init","payload":{"token":"***"}} client
Jan 23 02:43:05 raspicm evcc[29161]: [pulse ] TRACE 2025/01/23 02:43:05 {"type":"connection_ack"} server
Jan 23 02:43:05 raspicm evcc[29161]: [pulse ] TRACE 2025/01/23 02:43:05 {"id":"739a79e0-d56d-453f-a52a-bb4689642ba2","type":"subscribe","payload":{"query":"subscription ($homeId:ID!){liveMeasurement(homeId: $homeId){power,powerProduction,lastMeterConsumption,lastMeterProduction,currentL1,currentL2,currentL3}}","variables":{"homeId":"***"}}} client
Jan 23 02:43:08 raspicm evcc[29161]: [site  ] DEBUG 2025/01/23 02:43:08 ----
Jan 23 02:43:08 raspicm evcc[29161]: [lp-1  ] DEBUG 2025/01/23 02:43:08 charge power: 19310W
Jan 23 02:43:08 raspicm evcc[29161]: [lp-1  ] DEBUG 2025/01/23 02:43:08 charge currents: [29 28.8 29.2]A
Jan 23 02:43:08 raspicm evcc[29161]: [site  ] DEBUG 2025/01/23 02:43:08 grid power: 20608W
Jan 23 02:43:08 raspicm evcc[29161]: [site  ] DEBUG 2025/01/23 02:43:08 grid currents: [0 0 0]A
Jan 23 02:43:08 raspicm evcc[29161]: [site  ] DEBUG 2025/01/23 02:43:08 pv power: 0W
Jan 23 02:43:08 raspicm evcc[29161]: [site  ] DEBUG 2025/01/23 02:43:08 site power: 20608W
[... cut ...]
Jan 23 02:43:55 raspicm evcc[29161]: [pulse ] TRACE 2025/01/23 02:43:55 {"id":"8120c6a9-1be7-42b9-a482-d7f8834a7f55","type":"next","payload":{"data":{"liveMeasurement":{"power":20582,"powerProduction":0,"lastMeterConsumption":26271.6358,"lastMeterProduction":0,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
Jan 23 02:43:58 raspicm evcc[29161]: [pulse ] TRACE 2025/01/23 02:43:58 {"id":"8120c6a9-1be7-42b9-a482-d7f8834a7f55","type":"next","payload":{"data":{"liveMeasurement":{"power":20632,"powerProduction":0,"lastMeterConsumption":26271.6474,"lastMeterProduction":0,"currentL1":null,"currentL2":null,"currentL3":null}}}} server

```

pschwed, 2025-01-23:
> Von heute Nacht gibt es wieder ein paar Log-Meldungen, die so aussehen, als ob tibberseitig das Problem aufgetreten wäre. EVCC ist aber diesmal nicht in den gridpower: outdated Status gefallen und hat weiterhin funktioniert.

Gab es kurz nach 01:09:53 heute Nacht irgendeine Besonderheit in Deinen Logs, @machristoph1 ?

Bei mir gab es danach keine weiteren Einträge im Log. UI war heute morgen unresponsive. EVCC lief manuell gestartet mit --profile im Testmodus - aktuelles Nightly (evcc:arm64 (0.133.0+1737367712, 0.133.0+1737568052))

```
pschwed@evcc:~ $ tail -n 300 evcc.log | grep -B 2 -A 4 '!!'
[lp-1  ] DEBUG 2025/01/23 01:04:30 charger status: A
[pulse ] TRACE 2025/01/23 01:04:32 {"id":"e9817e00-8075-4129-b632-16191561898b","type":"next","payload":{"data":{"liveMeasurement":{"power":1353.22,"powerProduction":0,"lastMeterConsumption":16669.13348237,"lastMeterProduction":527.31779904,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
!! ReadJSON: failed to read JSON message: failed to get reader: use of closed network connection
!! ReadJSON: EOF
!! Run: <-sc.errorChan retry subscription client
!! Run: <-sc.errorChan: errRetry
[pulse ] TRACE 2025/01/23 01:04:42 {"id":"e9817e00-8075-4129-b632-16191561898b","type":"complete"} client
[pulse ] TRACE 2025/01/23 01:04:42 {"type":"connection_init","payload":{"token":"***"}} client
[pulse ] TRACE 2025/01/23 01:04:42 {"type":"connection_ack"} server
[pulse ] TRACE 2025/01/23 01:04:42 {"id":"f16b31f9-4a22-4fc6-a80e-84b52e424e7b","type":"subscribe","payload":{"query":"subscription ($homeId:ID!){liveMeasurement(homeId: $homeId){power,powerProduction,lastMeterConsumption,lastMeterProduction,currentL1,currentL2,currentL3}}","variables":{"homeId":"***"}}} client
--
[pulse ] TRACE 2025/01/23 01:06:13 {"id":"f16b31f9-4a22-4fc6-a80e-84b52e424e7b","type":"next","payload":{"data":{"liveMeasurement":{"power":1382.67,"powerProduction":0,"lastMeterConsumption":16669.17155641,"lastMeterProduction":527.31779904,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
[pulse ] TRACE 2025/01/23 01:06:16 {"id":"f16b31f9-4a22-4fc6-a80e-84b52e424e7b","type":"next","payload":{"data":{"liveMeasurement":{"power":1383.69,"powerProduction":0,"lastMeterConsumption":16669.17309289,"lastMeterProduction":527.31779904,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
!! ReadJSON: failed to read JSON message: failed to get reader: use of closed network connection
!! ReadJSON: EOF
!! Run: <-sc.errorChan retry subscription client
!! Run: <-sc.errorChan: errRetry
[pulse ] TRACE 2025/01/23 01:06:26 {"id":"f16b31f9-4a22-4fc6-a80e-84b52e424e7b","type":"complete"} client
[pulse ] TRACE 2025/01/23 01:06:26 {"type":"connection_init","payload":{"token":"***"}} client
[pulse ] TRACE 2025/01/23 01:06:26 {"type":"connection_ack"} server
[pulse ] TRACE 2025/01/23 01:06:26 {"id":"4d796b07-a848-4805-8119-38a1ab11999a","type":"subscribe","payload":{"query":"subscription ($homeId:ID!){liveMeasurement(homeId: $homeId){power,powerProduction,lastMeterConsumption,lastMeterProduction,currentL1,currentL2,currentL3}}","variables":{"homeId":"***"}}} client
pschwed@evcc:~ $ tail -n 5 evcc.log
[pulse ] TRACE 2025/01/23 01:09:41 {"id":"4d796b07-a848-4805-8119-38a1ab11999a","type":"next","payload":{"data":{"liveMeasurement":{"power":1388.65,"powerProduction":0,"lastMeterConsumption":16669.25067771,"lastMeterProduction":527.31779904,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
[pulse ] TRACE 2025/01/23 01:09:44 {"id":"4d796b07-a848-4805-8119-38a1ab11999a","type":"next","payload":{"data":{"liveMeasurement":{"power":1390.58,"powerProduction":0,"lastMeterConsumption":16669.25222038,"lastMeterProduction":527.31779904,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
[pulse ] TRACE 2025/01/23 01:09:47 {"id":"4d796b07-a848-4805-8119-38a1ab11999a","type":"next","payload":{"data":{"liveMeasurement":{"power":1390.53,"powerProduction":0,"lastMeterConsumption":16669.2529922,"lastMeterProduction":527.31779904,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
[pulse ] TRACE 2025/01/23 01:09:51 {"id":"4d796b07-a848-4805-8119-38a1ab11999a","type":"next","payload":{"data":{"liveMeasurement":{"power":1362.56,"powerProduction":0,"lastMeterConsumption":16669.25451971,"lastMeterProduction":527.31779904,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
[pulse ] TRACE 2025/01/23 01:09:53 {"id":"4d796b07-a848-4805-8119-38a1ab11999a","type":"next","payload":{"data":{"liveMeasurement":{"power":1356.03,"powerProduction":0,"lastMeterConsumption":16669.25527258,"lastMeterProduction":527.31779904,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
pschwed@evcc:~ $ 
```

andig, 2025-01-23:
Dann haben wir immer noch nicht die Stelle gefunden wo das passiert :(

machristoph1, 2025-01-23:
> Gab es kurz nach 01:09:53 heute Nacht irgendeine Besonderheit in Deinen Logs, [@machristoph1]

Nein, der Zeitpunkt ist bei mir völlig unauffällig. 

> Dann haben wir immer noch nicht die Stelle gefunden wo das passiert :(

Kann es bei @pschwed was anderes gewesen sein? Stehendes UI ist eigentlich auch nicht das Kennzeichen des Fehlers (wobei es evtl. in Docker-Umgebungen anders ist.). Bei mir war immer das UI ansprechbar, hat ERROR-Hinweis gezeigt, hat aber nicht mehr registriert, ob ich das Auto angesteckt habe
Immerhin sieht das 02:43 Event ziemlich genau so aus, wie das, was zuvor zur Dysfunktionalität geführt hat, diesmal war das aber nicht so. Er war allerdings zu der Zeit am Laden, das hatte ich sonst nicht. Insofern muss man vielleicht noch ein bisschen abwarten. 

08:52 ist übrigens gerade genau das gleiche passiert wie heute Nacht 02:43. Hat er aber überlebt, diesmal ohne angeschlossenes Auto. 
```Jan 23 08:52:38 raspicm evcc[29161]: [lp-1  ] DEBUG 2025/01/23 08:52:38 plan: charge 0s between 0001-01-01 00:53:28 +0053 LMT until 2025-01-24 07:00:00 +01>
Jan 23 08:52:38 raspicm evcc[29161]: [lp-1  ] DEBUG 2025/01/23 08:52:38 plan: avoid re-start within 1h0m0s, continuing for remaining -2562047h47m16.8547758>
Jan 23 08:52:41 raspicm evcc[29161]: !! ReadJSON: failed to read JSON message: failed to get reader: context deadline exceeded
Jan 23 08:52:41 raspicm evcc[29161]: !! Run: <-sc.errorChan retry subscription client
Jan 23 08:52:41 raspicm evcc[29161]: !! Run: <-sc.errorChan: errRetry
Jan 23 08:52:41 raspicm evcc[29161]: [pulse ] TRACE 2025/01/23 08:52:41 {"id":"13c8ef8a-63d7-4ae0-b623-51f5a1aba9a9","type":"complete"} client
Jan 23 08:52:42 raspicm evcc[29161]: [pulse ] TRACE 2025/01/23 08:52:42 {"type":"connection_init","payload":{"token":"***"}} client
Jan 23 08:52:42 raspicm evcc[29161]: [pulse ] TRACE 2025/01/23 08:52:42 {"type":"connection_ack"} server
Jan 23 08:52:42 raspicm evcc[29161]: [pulse ] TRACE 2025/01/23 08:52:42 {"id":"ac6fe6ae-8b3a-40b4-b7e9-3888d19bc79f","type":"subscribe","payload":{"query":>
Jan 23 08:52:52 raspicm evcc[29161]: !! ReadJSON: failed to read JSON message: failed to get reader: use of closed network connection
Jan 23 08:52:52 raspicm evcc[29161]: !! ReadJSON: EOF
Jan 23 08:52:52 raspicm evcc[29161]: !! Run: <-sc.errorChan retry subscription client
Jan 23 08:52:52 raspicm evcc[29161]: !! Run: <-sc.errorChan: errRetry
Jan 23 08:52:52 raspicm evcc[29161]: [pulse ] TRACE 2025/01/23 08:52:52 {"id":"ac6fe6ae-8b3a-40b4-b7e9-3888d19bc79f","type":"complete"} client
Jan 23 08:52:52 raspicm evcc[29161]: [pulse ] TRACE 2025/01/23 08:52:52 {"type":"connection_init","payload":{"token":"***"}} client
Jan 23 08:52:52 raspicm evcc[29161]: [pulse ] TRACE 2025/01/23 08:52:52 {"type":"connection_ack"} server
Jan 23 08:52:52 raspicm evcc[29161]: [pulse ] TRACE 2025/01/23 08:52:52 {"id":"5505812d-60a1-4603-85f8-e75bd5e6693a","type":"subscribe","payload":{"query":>
Jan 23 08:53:02 raspicm evcc[29161]: !! ReadJSON: failed to read JSON message: failed to get reader: context deadline exceeded
Jan 23 08:53:02 raspicm evcc[29161]: !! Run: <-sc.errorChan retry subscription client
Jan 23 08:53:02 raspicm evcc[29161]: !! Run: <-sc.errorChan: errRetry
Jan 23 08:53:02 raspicm evcc[29161]: [pulse ] TRACE 2025/01/23 08:53:02 {"id":"5505812d-60a1-4603-85f8-e75bd5e6693a","type":"complete"} client
Jan 23 08:53:02 raspicm evcc[29161]: [pulse ] TRACE 2025/01/23 08:53:02 {"type":"connection_init","payload":{"token":"***"}} client
Jan 23 08:53:02 raspicm evcc[29161]: [pulse ] TRACE 2025/01/23 08:53:02 {"type":"connection_ack"} server
Jan 23 08:53:02 raspicm evcc[29161]: [pulse ] TRACE 2025/01/23 08:53:02 {"id":"449b78ba-38c8-4d36-992b-c468112635df","type":"subscribe","payload":{"query":>
Jan 23 08:53:05 raspicm evcc[29161]: [pulse ] TRACE 2025/01/23 08:53:05 {"id":"449b78ba-38c8-4d36-992b-c468112635df","type":"next","payload":{"data":{"live>
Jan 23 08:53:08 raspicm evcc[29161]: [site  ] DEBUG 2025/01/23 08:53:08 ----
Jan 23 08:53:08 raspicm evcc[29161]: [lp-1  ] DEBUG 2025/01/23 08:53:08 charge power: 0W
Jan 23 08:53:08 raspicm evcc[29161]: [lp-1  ] DEBUG 2025/01/23 08:53:08 charge currents: [0 0 0]A
Jan 23 08:53:08 raspicm evcc[29161]: [site  ] DEBUG 2025/01/23 08:53:08 grid power: 0W
```

Wie sieht das bei Dir aus @pschwed?

[78 more replies]

PR #18504 (2025-02-01): Tibber: recover from disconnect
fix #17925 

@andig I have no tibber to actually test this, but the logic is simple enough...
PR #18643 (2025-02-14): Tibber: resubscribe on clean disconnect
fix #17925

Resubscribes in case of a clean exit.
closed 2025-02-08 by GrimmiMeloni