evcc-io/evcc #17926: Desynchronization between main window and config part

bigbig6, 2024-12-28:
### Describe the bug

Since I upgraded from 0.131.8 to 0.131.12 I have different information between the main window (in the charger information) and de config window about my car.
The main window or charger window is no more updated the information displayed (SoC and estimated kilometers) are always the same as the first one.

In the config window we can see that information are not the same.

As a result, the charge is not stopped when the target SoC is reached.

### Steps to reproduce

1. plug the car
2. verify information in the two windows
3.
...

### Configuration details

```yaml
interval: 30s # control cycle interval. Interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/co>

log: trace
levels:
  site: trace
  lp-1: trace
  lp-2: trace
  cache: trace
  e-208: trace
  peugeot: trace
  ocpp: trace

site:
  title: Maison # display name for UI
  meters:
    grid: reseau # grid meter
  residualPower: 120 # additional household usage margin

meters:
  - name: reseau
    type: template
    template: demo-meter
    usage: grid
    power: 1243

loadpoints:
  - title: chargeur_gauche
    charger: chargeur_gauche
#    soc:
#      poll:
#        mode: charging
#       interval: 60m
#     estimate: false

  - title: chargeur_droit
    charger: chargeur_droit
    soc:
      poll:
        mode: charging
        interval: 60m
      estimate: false

chargers:
  - name: chargeur_gauche
    type: template
    template: demo-charger

  - name: chargeur_droit
    type: template
    template: elvi

The car is defined in the DB
```

### Log details

```text
kkk
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

bigbig6, 2024-12-28:
Here is the log file:

[evcc-20241227-135654-trace.log](https://github.com/user-attachments/files/18265923/evcc-20241227-135654-trace.log)

bigbig6, 2024-12-28:
Here is the log file:

[evcc-20241227-135654-trace.log](https://github.com/user-attachments/files/18265941/evcc-20241227-135654-trace.log)

StefanSchoof, 2024-12-28:
The config dialog will make a request to the vehicle API.
In the main window evcc will make a request and calculate with the loading power a estimation. With your dummy charger that report no charging power this will be not accurate.
So the behavior is expected and when you use a real charger the estimate should be better.

closed 2024-12-28 by andig

bigbig6, 2024-12-28:
No, Charger 1 is dummy, but not the second one where the car is plugged.
Althrought there were a problem with the car that does not update it information, when I force it everythings was ok.

But since the last upgrade, I have this problem.

Regards,

bigbig6, 2024-12-28:
Hello, sorry, but the answer done is not correct.
Please do not close this bug now.

Regards,

-- HENNEQUIN Olivier

Closed #17926<https://github.com/evcc-io/evcc/issues/17926> as completed.

StefanSchoof, 2024-12-28:
If you use a real charger,  you need to find out why it does report 0,0 kWh charge. If you need help open a discussion where the community will help.

At https://github.com/evcc-io/evcc/discussions/17208 is a German Thread about problems with the meter values from the elvi.

bigbig6, 2024-12-28:
It does report nothing because it have no meter included.

In version 0.131.8 everything were ok, the charger had correcte values and was able to stop the charge.
Since I upgraded in version 0.131.12 (not tested all intermediate versions), I have this problem.

The window with the charger does not display the correct information, but the config part does, as shown by the screenshots.