evcc-io/evcc #17927: Unexpected end of JSON input error when no data available from Nordpool

Millox, 2024-12-28:
### Describe the bug

Every day before the new forecasted prices are available from Nordpool the download fails with error "unexpected end of JSON input". Everything works as expected after about 13-14 o'clock when the data for the next day is available.

### Steps to reproduce

1. Configure Nordpool tariffs
2. Request data before 13 every day

### Configuration details

```yaml
tariffs:
  currency: SEK
  grid:
    type: template
    template: nordpool
    region: SE4 
    currency: SEK
    charges: 0.73744
    tax: 0.25
  feedin:
    type: template
    template: nordpool
    region: SE4
    currency: SEK
    charges: 0.6039
    tax: 0
```

### Log details

```text
[main  ] INFO 2024/12/28 11:24:48 evcc 0.131.12
[main  ] INFO 2024/12/28 11:24:48 using config file: /etc/evcc.yaml
[db    ] INFO 2024/12/28 11:24:48 using sqlite database: /var/lib/evcc/evcc.db
[db    ] TRACE 2024/12/28 11:24:48 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/12/28 11:24:48 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/12/28 11:24:48 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/12/28 11:24:48 SELECT * FROM `settings` 3 <nil>
[mqtt  ] INFO 2024/12/28 11:24:49 connecting evcc-1381454247 at tcp://localhost:1883
[mqtt  ] DEBUG 2024/12/28 11:24:49 tcp://localhost:1883 connected
[db    ] TRACE 2024/12/28 11:24:49 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2024/12/28 11:24:49 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2024/12/28 11:24:49 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2024/12/28 11:24:49 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/12/28 11:24:49 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/12/28 11:24:49 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/12/28 11:24:49 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[grid  ] TRACE 2024/12/28 11:24:49 GET https://dataportal-api.nordpoolgroup.com/api/DayAheadPrices?market=DayAhead&date=2024-12-28&deliveryArea=SE4&currency=SEK
[feedin] TRACE 2024/12/28 11:24:49 GET https://dataportal-api.nordpoolgroup.com/api/DayAheadPrices?market=DayAhead&date=2024-12-28&deliveryArea=SE4&currency=SEK
[grid  ] TRACE 2024/12/28 11:24:49 {"deliveryDateCET":"2024-12-28","version":3,"updatedAt":"2024-12-27T12:11:02.8881218Z","deliveryAreas":["SE4"],"market":"DayAhead","multiAreaEntries":[{"deliveryStart":"2024-12-27T23:00:00Z","deliveryEnd":"2024-12-28T00:00:00Z","entryPerArea":{"SE4":325.04}},{"deliveryStart":"2024-12-28T00:00:00Z","deliveryEnd":"2024-12-28T01:00:00Z","entryPerArea":{"SE4":288.00}},{"deliveryStart":"2024-12-28T01:00:00Z","deliveryEnd":"2024-12-28T02:00:00Z","entryPerArea":{"SE4":254.51}},{"deliveryStart":"2024-12-28T02:00:00Z","deliveryEnd":"2024-12-28T03:00:00Z","entryPerArea":{"SE4":239.71}},{"deliveryStart":"2024-12-28T03:00:00Z","deliveryEnd":"2024-12-28T04:00:00Z","entryPerArea":{"SE4":241.54}},{"deliveryStart":"2024-12-28T04:00:00Z","deliveryEnd":"2024-12-28T05:00:00Z","entryPerArea":{"SE4":237.53}},{"deliveryStart":"2024-12-28T05:00:00Z","deliveryEnd":"2024-12-28T06:00:00Z","entryPerArea":{"SE4":250.49}},{"deliveryStart":"2024-12-28T06:00:00Z","deliveryEnd":"2024-12-28T07:00:00Z","entryPerArea":{"SE4":280.43}},{"deliveryStart":"2024-12-28T07:00:00Z","deliveryEnd":"2024-12-28T08:00:00Z","entryPerArea":{"SE4":323.21}},{"deliveryStart":"2024-12-28T08:00:00Z","deliveryEnd":"2024-12-28T09:00:00Z","entryPerArea":{"SE4":349.36}},{"deliveryStart":"2024-12-28T09:00:00Z","deliveryEnd":"2024-12-28T10:00:00Z","entryPerArea":{"SE4":346.26}},{"deliveryStart":"2024-12-28T10:00:00Z","deliveryEnd":"2024-12-28T11:00:00Z","entryPerArea":{"SE4":350.73}},{"deliveryStart":"2024-12-28T11:00:00Z","deliveryEnd":"2024-12-28T12:00:00Z","entryPerArea":{"SE4":349.70}},{"deliveryStart":"2024-12-28T12:00:00Z","deliveryEnd":"2024-12-28T13:00:00Z","entryPerArea":{"SE4":351.88}},{"deliveryStart":"2024-12-28T13:00:00Z","deliveryEnd":"2024-12-28T14:00:00Z","entryPerArea":{"SE4":342.02}},{"deliveryStart":"2024-12-28T14:00:00Z","deliveryEnd":"2024-12-28T15:00:00Z","entryPerArea":{"SE4":344.42}},{"deliveryStart":"2024-12-28T15:00:00Z","deliveryEnd":"2024-12-28T16:00:00Z","entryPerArea":{"SE4":350.27}},{"deliveryStart":"2024-12-28T16:00:00Z","deliveryEnd":"2024-12-28T17:00:00Z","entryPerArea":{"SE4":335.48}},{"deliveryStart":"2024-12-28T17:00:00Z","deliveryEnd":"2024-12-28T18:00:00Z","entryPerArea":{"SE4":339.38}},{"deliveryStart":"2024-12-28T18:00:00Z","deliveryEnd":"2024-12-28T19:00:00Z","entryPerArea":{"SE4":291.32}},{"deliveryStart":"2024-12-28T19:00:00Z","deliveryEnd":"2024-12-28T20:00:00Z","entryPerArea":{"SE4":59.64}},{"deliveryStart":"2024-12-28T20:00:00Z","deliveryEnd":"2024-12-28T21:00:00Z","entryPerArea":{"SE4":52.87}},{"deliveryStart":"2024-12-28T21:00:00Z","deliveryEnd":"2024-12-28T22:00:00Z","entryPerArea":{"SE4":41.98}},{"deliveryStart":"2024-12-28T22:00:00Z","deliveryEnd":"2024-12-28T23:00:00Z","entryPerArea":{"SE4":31.66}}],"blockPriceAggregates":[{"blockName":"Off-peak 1","deliveryStart":"2024-12-27T23:00:00Z","deliveryEnd":"2024-12-28T07:00:00Z","averagePricePerArea":{"SE4":{"average":264.66,"min":237.53,"max":325.04}}},{"blockName":"Peak","deliveryStart":"2024-12-28T07:00:00Z","deliveryEnd":"2024-12-28T19:00:00Z","averagePricePerArea":{"SE4":{"average":339.50,"min":291.32,"max":351.88}}},{"blockName":"Off-peak 2","deliveryStart":"2024-12-28T19:00:00Z","deliveryEnd":"2024-12-28T23:00:00Z","averagePricePerArea":{"SE4":{"average":46.54,"min":31.66,"max":59.64}}}],"currency":"SEK","exchangeRate":11.46936,"areaStates":[{"state":"Final","areas":["SE4"]}],"areaAverages":[{"areaCode":"SE4","price":265.73}]}
[grid  ] TRACE 2024/12/28 11:24:49 GET https://dataportal-api.nordpoolgroup.com/api/DayAheadPrices?market=DayAhead&date=2024-12-29&deliveryArea=SE4&currency=SEK
[feedin] TRACE 2024/12/28 11:24:49 {"deliveryDateCET":"2024-12-28","version":3,"updatedAt":"2024-12-27T12:11:02.8881218Z","deliveryAreas":["SE4"],"market":"DayAhead","multiAreaEntries":[{"deliveryStart":"2024-12-27T23:00:00Z","deliveryEnd":"2024-12-28T00:00:00Z","entryPerArea":{"SE4":325.04}},{"deliveryStart":"2024-12-28T00:00:00Z","deliveryEnd":"2024-12-28T01:00:00Z","entryPerArea":{"SE4":288.00}},{"deliveryStart":"2024-12-28T01:00:00Z","deliveryEnd":"2024-12-28T02:00:00Z","entryPerArea":{"SE4":254.51}},{"deliveryStart":"2024-12-28T02:00:00Z","deliveryEnd":"2024-12-28T03:00:00Z","entryPerArea":{"SE4":239.71}},{"deliveryStart":"2024-12-28T03:00:00Z","deliveryEnd":"2024-12-28T04:00:00Z","entryPerArea":{"SE4":241.54}},{"deliveryStart":"2024-12-28T04:00:00Z","deliveryEnd":"2024-12-28T05:00:00Z","entryPerArea":{"SE4":237.53}},{"deliveryStart":"2024-12-28T05:00:00Z","deliveryEnd":"2024-12-28T06:00:00Z","entryPerArea":{"SE4":250.49}},{"deliveryStart":"2024-12-28T06:00:00Z","deliveryEnd":"2024-12-28T07:00:00Z","entryPerArea":{"SE4":280.43}},{"deliveryStart":"2024-12-28T07:00:00Z","deliveryEnd":"2024-12-28T08:00:00Z","entryPerArea":{"SE4":323.21}},{"deliveryStart":"2024-12-28T08:00:00Z","deliveryEnd":"2024-12-28T09:00:00Z","entryPerArea":{"SE4":349.36}},{"deliveryStart":"2024-12-28T09:00:00Z","deliveryEnd":"2024-12-28T10:00:00Z","entryPerArea":{"SE4":346.26}},{"deliveryStart":"2024-12-28T10:00:00Z","deliveryEnd":"2024-12-28T11:00:00Z","entryPerArea":{"SE4":350.73}},{"deliveryStart":"2024-12-28T11:00:00Z","deliveryEnd":"2024-12-28T12:00:00Z","entryPerArea":{"SE4":349.70}},{"deliveryStart":"2024-12-28T12:00:00Z","deliveryEnd":"2024-12-28T13:00:00Z","entryPerArea":{"SE4":351.88}},{"deliveryStart":"2024-12-28T13:00:00Z","deliveryEnd":"2024-12-28T14:00:00Z","entryPerArea":{"SE4":342.02}},{"deliveryStart":"2024-12-28T14:00:00Z","deliveryEnd":"2024-12-28T15:00:00Z","entryPerArea":{"SE4":344.42}},{"deliveryStart":"2024-12-28T15:00:00Z","deliveryEnd":"2024-12-28T16:00:00Z","entryPerArea":{"SE4":350.27}},{"deliveryStart":"2024-12-28T16:00:00Z","deliveryEnd":"2024-12-28T17:00:00Z","entryPerArea":{"SE4":335.48}},{"deliveryStart":"2024-12-28T17:00:00Z","deliveryEnd":"2024-12-28T18:00:00Z","entryPerArea":{"SE [... cut ...]
[main  ] ERROR 2024/12/28 11:24:49 creating tariff feedin failed: cannot create tariff type 'template': cannot create tariff type 'custom': unexpected end of JSON input
feedin:
[main  ] FATAL 2024/12/28 11:24:49 tariff not available: cannot create tariff type 'template': cannot create tariff type 'custom': unexpected end of JSON input
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.131.12

StefanSchoof, 2024-12-28:
Did you try the nightly? 

There was https://github.com/evcc-io/evcc/commit/6d783e15650a44d02653377f04d4c916187b3912 that I think would fix this.

andig, 2024-12-28:
>  I have verified that the issue is reproducible with the latest nightly build

It would be really nice to actually to so...

closed 2024-12-28 by andig

Millox, 2025-01-01:
Yes, it works now with 0.132.0.

I did try the nightly before registering this bug report, but for some reason it apparently did not download the correct nightly and I have no idea what mistake I did with the git repo. Sorry for the redundant report and thanks for the good work and swift fixes.