evcc-io/evcc #17940: ABL eMH1 - invalid status: Outlet disabled

dpeger, 2024-12-28:
### Describe the bug

Hi, seit einiger Zeit habe ich öfters das Problem, dass meine ABL eMH1 in den Status "deaktiviert" wechselt. Es leuchten beide LEDs und EVCC meldet `invalid status: Outlet disabled`. Das passiert vor allem, wenn der Ladevorgang über einen längeren Zeitraum pausiert wurde (z.B. zu wenig Sonne), das Fahrzeug (Enyaq) den Ladevorgang länger nicht startet) oder auch nachdem der Ladevorgang durch EVCC abgeschlossen wurde. Ich behelfe mir aktuell damit die Sicherung in der Wallbox ein und wieder auszuschalten. Danach tut die Wallbox erst mal wieder "ganz normal".

Tatsächlich tritt das Problem erst auf seit wir beim Enyaq das ME3.7 Update in der Werkstatt haben aufspielen lassen. Vorher lief die Wallbox mit EVCC 2 Jahre komplett problemlos. Ich hatte seit längerem auch kein EVCC Update mehr gemacht - tat alles zu gut, aber auch mit der aktuellen Version kommt es immer wieder zu dem Problem.

Nicht sicher ob das hier mit reinspielt aber ich habe auch beobachtet, dass der Wagen seit dem Update das Laden nicht immer direkt startet. Manchmal (insb. wenn der Wagen länger stand) muss man nochmal die Verriegelung betätigen bevor der Ladevorgang gestartet werden kann. Das war vor dem ME3.7 Update mit ME3.0 nicht so.

### Steps to reproduce

1. Lasse Wagen einige Zeit stehen
2. Schließe Wallbox an
3. Warte Ladegrenze von EVCC erreicht ist
4. Mit Pech geht die Wallbox danach in den "Deaktiviert" status

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc01.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

interval: 10s # control cycle interval

sponsortoken: ****

meters:
- type: template
  template: fronius-solarapi-v1
  usage: grid
  host: symo
  name: grid1
- type: template
  template: fronius-solarapi-v1
  usage: pv
  host: symo
  name: pv2

chargers:
- type: template
  template: abl
  id: 1
  device: /dev/ttyUSB0
  baudrate: 38400
  comset: 8E1
  modbus: rs485serial
  name: wallbox5
- type: template
  template: abl
  id: 2
  device: /dev/ttyUSB0
  baudrate: 38400
  comset: 8E1
  modbus: rs485serial
  name: wallbox6

vehicles:
- type: template
  template: enyaq
  title: Enyaq
  user: ****
  password: ****
  vin: ****
  capacity: 62
[... 111 lines cut ...]
[abl   ] TRACE 2024/12/28 23:30:09 modbus: send 3a 30 32 30 33 30 30 30 46 30 30 30 35 45 37 0d 0a
[abl   ] TRACE 2024/12/28 23:30:09 modbus: recv 3e 30 32 30 33 30 41 30 46 39 34 30 32 31 35 30 31 30 42 30 30 36 34 30 30 36 34 36 33 0d 0a
Power:          0W
Current L1..L3: 0A 0A 0A
Charge status:  invalid status: Outlet disabled
Enabled:        true
Diagnostic dump:
[abl   ] TRACE 2024/12/28 23:30:09 modbus: send 3a 30 32 30 33 30 30 30 31 30 30 30 32 46 38 0d 0a
[abl   ] TRACE 2024/12/28 23:30:09 modbus: recv 3e 30 32 30 33 30 34 30 31 34 32 34 31 42 45 42 35 0d 0a
[abl   ] TRACE 2024/12/28 23:30:09 modbus: send 3a 30 32 30 33 30 30 30 31 30 30 30 32 46 38 0d 0a
[abl   ] TRACE 2024/12/28 23:30:09 modbus: recv 3e 30 32 30 33 30 34 30 31 34 32 34 31 42 45 42 35 0d 0a
Firmware: 01 42 41 be
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.131.12

andig, 2024-12-29:
Es fehlt eine gute Idee: „Steckdose abgeschaltet“- was sollte evcc mit diesem machen? Und warum tritt der auf?

dpeger, 2024-12-29:
Ja, ich vermute auch, dass irgendwas im der Kommunikation mit dem Auto dazu führt, dass die Box sich abschaltet. Ich hatte die Hoffnung, dass man evtl aus der Kommunikation zwischen Box und EVCC erkennen könnte was da schief läuft. Vielleicht könnte EVCC die Box ja auch ohne Stromfreischaltung wieder in einen normalen Status versetzen??

Ticket bei Skoda halte ich leider für wenig aussichtsreich, da wir doch alle wissen, dass die Leute, die diese Tickets bearbeiten, keine Ahnung haben, vor allem darauf bedacht sind Kosten und Arbeit für sich und die Hersteller zu vermeiden und wenig Interesse an einer wirklichen Analyse oder Lösung des Problems haben. Spätestens wenn man dann sagt, dass die Wallbox über eine drittanbieter Software (so gut sie auch sein mag - ich liebe dieses Projekt hier!!!) geregelt wird, steht die Antwort zu 100% fest...

Im Enyaq-Forum habe ich noch folgendes gefunden:
 
> Bekannter Workaround nach den Werkstatt Updates >= 3.2
> 
> • PV-Überschussladen mit Phasenumschaltung, Auto schläft nach einigen Umschaltungen (z.B. bei Wolkenbildung) ein.
> 
> Lösung: Einmalig die Sicherung SB3 im Sicherungskasten im Motorraum ziehen und wieder einsetzen (das ist die Sicherung für den Onboard-Lader). Im Sicherungskasten ist eine weiße Kunststoffpinzette für das Entnehmen der Sicherung. Dadurch startet der Lader einmalig neu.

Das werde ich jetzt mal ausprobieren und berichten...

andig, 2024-12-29:
> Dadurch startet der Lader einmalig neu.

Das klingt ja richtig prima... Skoda zur Nachbesserung auffordern und dann das Fahrzeug wandeln?

> Vielleicht könnte EVCC die Box ja auch ohne Stromfreischaltung wieder in einen normalen Status versetzen??

Solange niemand eine Idee hat, _wie_ das bei der ABL gehen soll wird das nix. Leider...

premultiply, 2024-12-29:
Das kann in Zusammenhang mit der Wakeup()-Funktion stehen, mit der evcc offensichtlich erfolglos versucht das Fahrzeug mit der neuen Software aus dem Tiefschlaf zu wecken. Dazu wird der Ausgang softwareseitig für 3 Sekunden deaktiviert und anschließend wieder reaktiviert.

https://github.com/evcc-io/evcc/blob/bd0e9708d9598b253afcc72f2683f9642bb1223c/charger/abl.go#L258-L271

Allerdings ist verwunderlich weshalb die Box in diesem Zustand dann offensichtlich gelegentlich hängen bleibt.
Gibt es da vielleicht Kommunikationsstörungen auf dem Bus?
Wir schicken schon jeden Befehl zweimal an die Box...

dpeger, 2024-12-30:
Ich hatte gestern mal die besagte Sicherung im Wagen gezogen und wieder eingesetzt. Danach mit "Min+PV" geladen, über Nacht pausiert und heute wieder mit "Min+PW" bis auf 80% geladen. Tat soweit alles wie erwartet. Allerdings hing der Wagen danach noch ein paar Stunden an der Box und nachher war die Box doch wieder im "outlet disabled" status. 

Ich hab das log mal angehängt: [evcc-outlet-disabled.log](https://github.com/user-attachments/files/18277858/evcc-outlet-disabled.log)

Da sind zwar auch ein paar vereinzelte "serial: timeouts" zu sehen. Allerdings auch sowas hier

```
Dec 30 13:12:47 visions-evcc01 evcc[62159]: [lp-1  ] ERROR 2024/12/30 13:12:47 charger status: modbus: response length '45' is not an even number
...
Dec 30 13:14:35 visions-evcc01 evcc[62159]: [lp-1  ] ERROR 2024/12/30 13:14:35 charge power: modbus: response length '61' is not an even number
...
Dec 30 13:15:36 visions-evcc01 evcc[62159]: [lp-1  ] ERROR 2024/12/30 13:15:36 charger status: modbus: response length '45' is not an even number
...
Dec 30 13:17:23 visions-evcc01 evcc[62159]: [lp-1  ] ERROR 2024/12/30 13:17:23 wake-up charger: modbus: response length '47' is not an even number
...
```

PR #17950 (2025-01-07): ABL eMH: force re-enable outlet
Fixes #17940
closed 2025-01-07 by andig