evcc-io/evcc #17945: Kostal Enector: heartbeat: modbus: exception '2' (illegal data address), function '6'

Exezz78, 2024-12-29:
### Describe the bug

My setup: 
Kostal Plenticore 8.5 with BYD battery
- Artikelnummer 10335958
- UI-Version 01.30.12092
- MC-Version 01.80
- IOC-Version 01.80
- HW-Version 0200

Kostal KSEM (connected to plenticore and in LAN)
- Gerätetyp hw0100
- Software-Version 2.5.0

Kostal Enector (connected via RS485 to KSEM)
- Hardware 0003
- VersionFW: 2023.21.11024-20; COM: 1.03

Settings of Enector DIP are:
4,5 and 6 = ON
7 = OFF (same for all others incl. Bank 2)

evcc version 0.131.12

I tried running evcc via Home Assistant AddOn (my preferred option) as well as via proxmox. Same issue in both cases.

Issue: 
Wallbox does not respond; log shows "**heartbeat: modbus: exception '2' (illegal data address), function '6'**"

Related to cvcc.yaml: 
The evcc website states that for Enector configuration protocol "_modbus:_ **rs485**_tcpip_" shall be used. This did not work for me. evcc does not start an reports issue: 
"cannot create charger type 'mennekes-compact': connection already registered with different protocol"

When changed to "_modbus:_ **tcpip**", evcc is starting but reports the mentioned issue in the log files: 

**heartbeat: modbus: exception '2' (illegal data address), function '6'**

I tried several different configurations and IDs without success. 

Please have a look and let me know what might solve the issue.

[image]

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
meters:
- type: template
  template: kostal-plenticore #-hw0200 
  id: 71  
  host: 192.168.1.79  
  port: 1502  
  endianness: little # optional - big or little
  usage: pv  
  modbus: tcpip  
  name: pv

- type: template
  template: kostal-plenticore #-hw0200 
  id: 71  
  host: 192.168.1.79  
  port: 1502  
  endianness: little # optional - big or little
  usage: battery
  capacity: 10 # in kWh  
  modbus: tcpip  
  name: battery

- type: template
  template: kostal-ksem 
  id: 71  
  host: 192.168.1.233
  port: 502  
  usage: grid  
  modbus: tcpip  
  name: grid
```

### Log details

```text
root@evcc:~# evcc -l debug charger
[main  ] INFO 2024/12/29 14:21:31 evcc 0.131.12
[main  ] INFO 2024/12/29 14:21:31 using config file: /etc/evcc.yaml
[db    ] INFO 2024/12/29 14:21:31 using sqlite database: /root/.evcc/evcc.db
Power:          modbus: exception '2' (illegal data address), function '3'
Energy:         modbus: exception '2' (illegal data address), function '3'
Current L1..L3: modbus: exception '2' (illegal data address), function '3'
Voltage L1..L3: modbus: exception '2' (illegal data address), function '3'
Charge status:  modbus: exception '2' (illegal data address), function '3'
Enabled:        modbus: exception '2' (illegal data address), function '3'

root@evcc:~# evcc -l debug meter
[main  ] INFO 2024/12/29 14:21:57 evcc 0.131.12
[main  ] INFO 2024/12/29 14:21:57 using config file: /etc/evcc.yaml
[db    ] INFO 2024/12/29 14:21:57 using sqlite database: /root/.evcc/evcc.db
battery
-------
Power:        5W
Soc:          10%
Capacity:     10.0kWh
Controllable: true

pv
--
Power:  173W
Energy: 3863.9kWh

grid
----
Power:          3070W
Energy:         24857.6kWh
Current L1..L3: 4.45A 9.35A 1.29A
Voltage L1..L3: 232V 231V 232V
Power L1..L3:   760W 2140W 130W
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

andig, 2024-12-29:
Missing config

Exezz78, 2024-12-29:
@andig Please find the full evcc.yaml attached.

[evcc.yaml.txt](https://github.com/user-attachments/files/18269940/evcc.yaml.txt)

Exezz78, 2024-12-29:
```
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: xxx

interval: 30s # control cycle interval

sponsortoken: xxx

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

meters:
- type: template
  template: kostal-plenticore #-hw0200 
  id: 71  
  host: 192.168.1.79  
  port: 1502  
  endianness: little # optional - big or little
  usage: pv  
  modbus: tcpip  
  name: pv

- type: template
  template: kostal-plenticore #-hw0200 
  id: 71  
  host: 192.168.1.79  
  port: 1502  
  endianness: little # optional - big or little
  usage: battery
  capacity: 10 # in kWh  
  modbus: tcpip  
  name: battery

- type: template
  template: kostal-ksem 
  id: 71  
  host: 192.168.1.233
  port: 502  
  usage: grid  
  modbus: tcpip  
  name: grid

vehicles:
- type: template
  template: tesla-command
  title: xxx
  accessToken: xxx
  refreshToken: xxx
  vin: xxx
  capacity: 76
  name: modely
  icon: car
#  cache: 15m

chargers:
- type: template
  template: mennekes-compact 
  id: 50
  host: 192.168.1.233
  port: 502  
  modbus: tcpip  
  name: enector

loadpoints:
- title: Carport
  charger: enector
  mode: now
  vehicle: modely
  
site:
  title: Power
  meters:
    grid: grid
    pv:
    - pv
    battery:
    - battery
    
tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price: 0.26 # EUR/kWh
  feedin:
    type: fixed
    price: 0.0832 # EUR/kWh
```

andig, 2024-12-29:
Was für Protokolle? Das ist doch alles ModbusTCP?

premultiply, 2024-12-29:
Die Mennekes-Wallbox hat nur eine serielle RS485-Schnittstelle und diese scheint statt an evcc an das KSEM angebunden zu sein.

Hier wird versucht irgendwie über das KSEM auf diese zuzugreifen was aber wohl nicht funktionieren wird, da das KSEM kein Schnittstellenkonverter, Proxy oder sonst wie gearteter Protokollumsetzer ist.

andig, 2024-12-29:
Bleibt die Frage: was soll die Box am Ksem wenn evcc sie steuert?

D4UG, 2025-01-01:
Vielleicht kann ich die (durchaus berechtigte) Frage beantworten:
Die Enector kann eigentlich nur über das KSEM mit Firmwareupdates versorgt werden. Als "Normalo" ist das die einzige Möglichkeit dies zu tun. Es gibt zwar eine Lösung mit einem speziellen Kabel und der Hersteller-Inbetriebnahmesoftware, aber offiziell ist das nicht für den Normalverbraucher zugänglich oder so gedacht. Daher verstehe ich schon, wenn jemand auf diese Möglichkeit nicht unbedingt verzichten möchte und auch nicht unbedingt immer wieder den RS485 umstecken will. KSEM, Wechselrichter und Energiespeicher nach einem Umstecken wieder sauber miteinander hochzufahren, sodass auch alles miteinander kommuniziert ist ein Drahtseilakt - ich habe ca. 20 KSEM und Wechselrichter Neustarts in unterschiedlichen Abfolgen gebraucht - macht also eher keinen Spaß. Ein weiterer Grund könnte sein, dass jemand die Werte, die die Enector über das KSEM an das Kostal Solar Portal überträgt, weiterhin tracken möchte (auch wenn EVCC das wahrscheinlich besser kann und besser aufbereitet).

Ich stehe tatsächlich vor dem gleichen Problem; die Funktionalität der Enector Wallbox samt Komfortfunktionen reicht mir nicht mehr aus und ich habe daher versucht sie in EVCC zu integrieren. Das Verhalten ist bei mir identisch zum Ticket oben. Einbinden per rs485tcpip scheitert mit "cannot create charger type 'mennekes-compact': connection already registered with different protocol", einbinden mit tcpip lässt EVCC zumindest starten, aber führt dann zu heartbeat: modbus: exception '2 "illegal data address" function '6'.

Ist "Box vom KSEM abkabeln und direkt per USB-RS485-Interface oder sonst einem RS485/Modbus-IP-Konverter anbinden." wirklich der einzige Lösungsweg?

andig, 2025-01-02:
Hersteller fragen, wie das Ding durch den Ksem ansprechbar wäre? Wir können leider nicht hellsehen. Doku gibts nur für direkte Ansprache.

premultiply, 2025-01-02:
Man könnte auch erstmal andersrum fragen:
Wie kommst du denn überhaupt zu der Annahme dass evcc über das KSEM auf die WB zugreifen könnte?
Das steht nirgends und würde wenn es denn theoretisch funktionieren würde direkt zu dem nächsten Problemen führen, dass dann zwei Steuerungen gegeneinander arbeiten würden.
Keine gute Idee.

Andersrum würde ein Schuh draus werden: Wallbox an evcc anbinden und KSEM notfalls über den Modbus-Proxy im Nur-Lesen-Modus mit Daten versorgen.
Aber ist das wirklich so relevant?

Die Wallbox ist jedenfalls eine ganz normale Mennekes-Wallbox und enthält auch deren Standardfirmware. Die bekommt man vom echten Hersteller. Genau wie die dafür erforderlichen Tools.
Daher kann man die Box auch ganz normal wie im Originalhandbuch(!) beschrieben verwenden.
Kostal hat zur allgemeinen Verwirrung nur das eigene Handbuch etwas modifiziert.
Leg die mal nebeneinander 😉

Es gibt keine Notwendigkeit die Box in irgendeiner Weise an das KSEM anzubinden.
MWn wäre für die ungewollte einfache Ladesteuerung durch das KSEM sogar nochmal eine kostenpflichtige Zusatzlizenz von Kostal notwendig.

D4UG, 2025-01-04:
> Man könnte auch erstmal andersrum fragen: Wie kommst du denn überhaupt zu der Annahme dass evcc über das KSEM auf die WB zugreifen könnte? Das steht mWn nirgends und selbst wenn es denn theoretisch funktionieren würde, würde es direkt zu dem nächsten Problemen führen, dass dann zwei Steuerungen gegeneinander arbeiten würden. Keine gute Idee.

Unwissenheit meinerseits. Bin noch nicht so lange mit EVCC unterwegs (und da auch nur mit EEBus) und hatte mit Modbus bisher nicht viel am Hut und lerne noch - bitte verzeiht mir das. Ich gebe mein bestes, aber mir war nicht klar, dass die Kommunikation durch das KSEM eher nicht durchgeschleust werden kann (selbst, wenn ich die aktive Steuerung der Enector im KSEM logischerweise deaktiviere). Ich bin ein großer Fan von EVCC und der ganzen Möglichkeiten, die es bietet, daher wollte ich versuchen die Enector auch zu integrieren (zunächst mit Trial Token) und wenn es mir glückt natürlich ein Sponsor Token erwerben.

> 
> Andersrum würde ein Schuh draus werden: Wallbox an evcc anbinden und KSEM notfalls über den Modbus-Proxy im Nur-Lesen-Modus mit Daten versorgen. Aber ist das wirklich so relevant?

Guter Einwand, darüber habe ich auch schon gehirnt, ob ich das KSEM irgendwie zumindest mitlesen lassen könnte. Wäre schon irgendwie cool, in der Kostal App die Wallbox-Statistik auch zu sehen.

> Die Wallbox ist jedenfalls eine ganz normale Mennekes-Wallbox und enthält auch deren Standardfirmware. Die bekommt man vom echten Hersteller. Genau wie die dafür erforderlichen Tools. Daher kann man [...]

Ich habe die Wallbox erst 2 Wochen - davon habe ich die Tage gelesen, dass sie am Ende eine Mennekes ist.

> 
> Es gibt keine Notwendigkeit die Box in irgendeiner Weise an das KSEM anzubinden. MWn wäre für die ungewollte einfache Ladesteuerung durch das KSEM sogar nochmal eine kostenpflichtige Zusatzlizenz von Kostal notwendig.

Ich habe die Enector via Freischaltcode im KSEM aktiviert. Funktioniert ja auch alles ganz gut mit den PV-Überschussmodi - aber ich war leider nach wenigen Tagen schon angefressen von fehlenden Einstellmöglichkeiten, die ich mir gewünscht hätte (Batteriespeicher für Ladung sperren geht nur im Solar-Pure-Modus und sonst in keinem anderen und ab April würde ich zudem gern einen dynamischen Stromtarif einbinden). Daher glaube ich, dass mich diese Kostal Lösung auf Dauer nicht zufriedenstellen wird - leider. Ich versuche mein Glück mit einem RS485-USB Adapter und freue mich auch schon darauf, dann eine gemeinsame Statistik mit meiner EEBus-Wallbox innerhalb von EVCC zu führen - weil das ist in EVCC wirklich richtig schick! Danke vielmals für euren Input!

PS: kann jemand einen RS485-USB Adapter empfehlen, mit dem gute Erfahrungen gemacht wurden - bevor ich da irgendwie direkt wieder ins Klo greife?

Exezz78, 2025-01-04:
Hi @D4UG,

mir geht/ging es ganz genauso. Ich war davon ausgegangen des evcc die Daten vom Enector via KSEM bezieht. Daher meine dumme Frage bzw. Konfiguration.

Gestern habe ich einen RS485-USB Adapter angeklemmt (bei amazon gekauft; einfach 788046388466 in die Suche eingeben; ist von Waveshare; kostet ca. 20 EUR; Anbindung in Home Assistant Add-On via proxmox; Hardware wird in die VM durchgereicht) und es funktionierte auf Anhieb und so wie in der Dokumentation von evcc beschrieben. 

Konnte wetterbedingt noch nicht alles testen, aber die Werte sind korrekt und es gibt keinen Fehler in den Logs.
Den relevanten Teil der Config siehe unten. Der USB-Port muss ggf. angepasst werden. In Home Assistant is er unter _Settings - System - Hardware - All Hardware_ ersichtlich.

```
chargers:
  - name: enector
    type: template
    template: mennekes-compact

    modbus: rs485serial
    id: 50
    device: /dev/ttyUSB2 # USB-RS485 Adapter Adresse
    baudrate: 57600 # Prüfe die Geräteeinstellungen, typische Werte sind 9600, 19200, 38400, 57600, 115200
    comset: "8N2" # Kommunikationsparameter für den Adapter
```

D4UG, 2025-01-14:
> Hi @D4UG,
> 
> Gestern habe ich einen RS485-USB Adapter angeklemmt (bei amazon gekauft; einfach 788046388466 in die Suche eingeben; ist von Waveshare; kostet ca. 20 EUR; Anbindung in Home Assistant Add-On via proxmox; Hardware wird in die VM durchgereicht) und es funktionierte auf Anhieb und so wie in der Dokumentation von evcc beschrieben.
> 

Ich habe mir den o.g. USB Adapter auch mal besorgt und versucht die Enector darüber innerhalb meiner Konstellation (Home Assistant via VM auf der Synology NAS) einzubinden. Das Durchreichen der Hardware in die VM scheint zwar geklappt zu haben, aber ich hatte so meine Schwierigkeiten den Stick richtig zu konfigurieren bzw. überhaupt das Konfigurationsmenü zu finden - auch mangels Doku. Da ich das befürchtet hatte, hatte ich mir parallel auch noch ein Modbus RTU <-> Modbus TCP Ethernet Gateway besorgt (USR TCP232-410s) und es damit nach längerem Rumspielen in der Konfiguration nun endlich ohne Fehler im Log zum Laufen bekommen. Geladen habe ich bisher noch nicht, aber sieht mal nicht schlecht aus. Für den USB Adapter finde ich evtl. irgendwann anders noch eine Verwendung.

Danke für den Support

lbriest, 2025-04-23:
> Hi [@D4UG](https://github.com/D4UG),
>
> mir geht/ging es ganz genauso. Ich war davon ausgegangen des evcc die Daten vom Enector via KSEM bezieht. Daher meine dumme Frage bzw. Konfiguration.
>
> Gestern [...]

Danke für deine Anleitung!

Ich habe den gleichen Set-Up wie du:
Enector wallbox
Cat7 Kabel
USB-to_RS485 Adapter (B0B87YJLJQ amazon Artikel)
RPI 4 mit HA, auf dem EVCC als ADD On läuft.

Nun bekomme ich die Kommunikation zwischen EVCC und Wallbox nicht hin. Ich verwende deine Einstellung (außer bei mir geht nur comset: 8N1).
Muss ich noch irgendwas am Adapter einstellen?

Ich habe schon über einen anderen PC per mbpoll daten zu pingen. Bei deiner Variante kam dann auch immer:
Polling slave 50... Ctrl-C to stop) Read output (holding) register failed: Illegal data address

Eventuell kann mir jemand weiterhelfen?