evcc-io/evcc #17947: Since 0.132.0 evcc can‘t start with Solax components

SpiritofGS, 2024-12-29:
### Describe the bug

Just updated to 0.132.0 on HA
Since that, evcc didn‘t start anymore. 
Saw in the Log „apacity“ instead of „capacity“ - eveventually that‘s the Problem. But I don’t know how to fix it…

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
PV/Battery is Solax/Qcells
```

### Log details

```text
[main  ] INFO 2024/12/29 16:45:52 listening at :7070
[main  ] FATAL 2024/12/29 16:45:52 cannot create meter 'my_battery': cannot create meter type 'template': invalid key: apacity
[main  ] FATAL 2024/12/29 16:45:52 will attempt restart in: 15m0s
```

### What type of operating system are you running?

HomeAssistant Add-on

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.132.0

andig, 2024-12-29:
Missing config

closed 2024-12-29 by andig

SpiritofGS, 2024-12-29:
> Missing config

@andig Sorry, just typed that quickly on my smartphone earlier. Config has been added. No changes have been made to it for weeks. And until the update to 0.132.0 everything worked fine. I suspect a typing error in the template for the Solax inverters, as the log says that it cannot create the meter type because of “invalid key: apacity”

0x3d13f, 2024-12-29:
`apacity: 12 # Akkukapazität in kWh (optional)`
In this line there is obviously missing a "c" at the beginning.

andig, 2024-12-29:
…and that‘s exactly why we‘re closing issues without config…

SpiritofGS, 2024-12-29:
Oh dear. Sorry about that. I had recently added the battery capacity. However, everything went well until the update today. The restart may have caused the yaml to be reloaded and my typo meant that evcc could no longer be started. I was just surprised because it fitted in with the update. Sorry again to everyone, and many thanks for the quick help! 
Have a good new year and thank you for your excellent work!