evcc-io/evcc #17969: Einschalten/ Ausschalten der Ladeplanung von Außen

alexonbandit, 2024-12-30:
### Feature
Es wäre super wenn man den "Aktiv" Schalter der Ladenplanung von extern bedienen könnte.

### Hintergrund:
Ich nutze die sich wiederholenden Ladepläne um sicher zustellen, dass das Auto an meinen Arbeitstagen den mindest benötigten SoC hat um zur Arbeit und zurück zu kommen. Wenn ich nun länger weg bin (zum Beispiel Urlaub, Dienstreise) möchte ich die einzelnen Ladepläne gerne (individuell) deaktivieren können und wenn ich dann wieder da bin wieder aktivieren können.
Hierzu haben ich einen Kalender (in meinem Fall in Home Assistant) in welchem die Tage gespeichert sind an denen ich auswärts bin. Sobald einer dieser Tage ist, soll die Ladeplanung deaktiviert werden damit das Auto nicht unnötigerweise über Nacht mit Netzstrom geladen wird.

**Describe the solution you'd like**
Es wäre super wenn ich diese Funktion via MQTT an bzw. ausschalten könnte

**Describe alternatives you've considered**
Bisher haben ich diesen Anwendungsfall wie folgt abgedeckt:

```
rest_command: 
  # set
  evcc_set_targettime:
    url: >-
      {% set lt = strptime(states('input_datetime.evcc_targettime'),'%H:%M:%S').replace(tzinfo=now().astimezone().tzinfo) -%}
      {% set vehicle = state_attr('sensor.evcc_plugged_status','vehicleName') -%}
      {% set soc = states('input_number.evcc_targetsoc') | float | round(0) -%}
      {% set utct = lt | as_timestamp | timestamp_custom('T%H:%M:%SZ',false) -%}
      {% set utcts = (as_timestamp(now()) + 86400) | timestamp_custom('%Y-%m-%d') ~utct -%}
      http://192.168.178.xx:xxxx/api/vehicles/{{vehicle}}/plan/soc/{{soc}}/{{utcts}}
    method: POST
    #delete
  evcc_del_targettime:
    url: >-
      {% set vehicle = state_attr('sensor.evcc_plugged_status','vehicleName') -%}
      http://192.168.178.xx:xxxx/api/vehicles/{{vehicle}}/plan/soc
    method: DELETE

```

Aber wäre doch klasse wenn man das neue und wirklich gelungene Feature der "Wiederholden Ladeplanung" in der GUI bedienen kann aber trotzdem von extern Einfluss darauf nehmen kann. 

[image]

andig, 2024-12-30:
Die Sst sind dokumentiert.

SolarPower2024, 2024-12-31:
Hi @andig
Könntest du vielleicht einen Hinweis geben wo es dokumentiert ist?
In der Doku auf der Homepage steht bei der API Doku nur für einen Ladeplan, aber nicht für wiederholende bzw Ladeplan 1, 2, 3,etc..

Vielen Dank.

andig, 2024-12-31:
Ahh, verstehe- dann sollten wir die ergänzen!

naltatis, 2024-12-31:
Ja, @Maschga kümmert sich aktuell um eine bessere/strukturiertere REST API Doku. Siehe https://github.com/evcc-io/docs/pull/698

Kurzfristige Lösung: Schau dir am besten mal den Netzwerkverkehr in den Browser-Devtools an, wenn du die Pläne bearbeitest. Das hier ist die Struktur mit der die Pläne gesetzt werden:

```
curl 'https://demo.evcc.io/api/vehicles/vehicle_1/plan/repeating' \
-X 'POST' \
-H 'Content-Type: application/json' \
--data-binary '{"plans":[{"weekdays":[1,2,4,5],"time":"07:00","soc":80,"tz":"Europe/Berlin","active":true}]}'
```

Die aktuellen Pläne für ein Fahrzeug bekommst du über diesen Endpunkt

```
curl https://demo.evcc.io/api/state\?jq\=.vehicles.vehicle_1.repeatingPlans
[{"active":true,"soc":80,"time":"07:00","tz":"Europe/Berlin","weekdays":[1,2,4,5]}]
```

Die wiederholenden Pläne werden als ein Block gespeichert. Heißt du musst ggf. den aktuellen Stand einmal vorher lesen und dann die Abwandlung per POST speichern.

closed 2024-12-31 by naltatis