evcc-io/evcc #17970: Sigenergy Wechselrichter hinzufügen

Hornet100182, 2024-12-30:
Ich würde mir gerne einen neuen Wechselrichter mit Speicher (Sigenstor Fa. Sigenergy) in EVCC wünschen.

Über den Händler kann im Sigenergy Wechselrichter mit Speicher (Sigenstor) die Modbus Kommunikation aktiviert werden.

Der Wechselrichter kommuniziert bei mir schon erfolgreich mit Home Assistant.

Anbei die Registerbeschreibung von Sigenergy.
[Modbus.pdf](https://github.com/user-attachments/files/18277239/Modbus.pdf)

0x3d13f, 2024-12-30:
Da gab es schon mal was dazu: https://github.com/evcc-io/evcc/discussions/16246

andig, 2024-12-30:
Got stuck on https://github.com/evcc-io/evcc/discussions/16246#discussioncomment-10711985. Feel free to open PR.

Hornet100182, 2024-12-31:
Vielen Dank, bin an der Suche gescheitert.
Bin neu bei Github und habe die geschlossenen Posts nicht mitgesucht 🙈

closed 2024-12-31 by Hornet100182

andig, 2024-12-31:
Seems grid was still missing to get a complete PR ready.