evcc-io/evcc #17975: UI : kleiner Bug : Hausbatterie : Netzladen : Preisgrenze

vkr2791, 2024-12-31:
### Describe the bug

Bei einem Strompreis von z.B. 20 ct greift nicht die Einstellung "<= 20ct" 
hier fehlt wohl das "<=" im Code, siehe Screenshots:

[image]

Mit "<=20,5 ct" greift es dann:

[image]

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
-
```

### Log details

```text
-
```

### What type of operating system or environment does evcc run on?

Docker container

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.132.0

VolkerK62, 2024-12-31:
Da wird der Preis vermutlich 20,04 ct betragen

andig, 2024-12-31:
Sollten wir im Code auf 1 Nachkommastelle runden? Oder anders visualisieren?

StefanSchoof, 2024-12-31:
Ist runden hier überhaupt sinnvoll? Warum nicht für eine Stunde genau den Preis den der Anbieter haben will angezeigen?

naltatis, 2024-12-31:
> Ist runden hier überhaupt sinnvoll? Warum nicht für eine Stunde genau den Preis den der Anbieter haben will angezeigen?

Weil wir sonst je nach Schnittstelle viele Nachkommastellen anzeigen.

naltatis, 2024-12-31:
> Sollten wir im Code auf 1 Nachkommastelle runden? Oder anders visualisieren?

Die Visualisierung und die Logik/Regelung sind in sync. Heißt er würde auch in dieser Stunde nicht laden. Wir könnten hier aber in der Tat direkt beim einlesen der Daten schon Runden/Abschneiden. Gerade wenn wir es nicht genauer im UI anzeigen wollen führt das dann zu weniger Verwirrung.

Ist allerdings nicht total trivial, weil die Präzision in der wir das Anzeigen abhängig ist von der Währung (ct wenn EUR). Die gesamte Liste gibts hier: https://github.com/evcc-io/evcc/blob/master/assets/js/mixins/formatter.js#L2-L15

StefanSchoof, 2024-12-31:
> Weil wir sonst je nach Schnittstelle viele Nachkommastellen anzeigen.

Ja, aber über verschiedene Währungen ist es unmöglich die Anzahl der für den Nutzer signifikanten Nachkommastellen zu bestimmen. Z.B. hat der Kuwait Dinar hat eine ganz andere Auflösung als der Russische Rubel.

vkr2791, 2024-12-31:
Ach so, das Problem besteht auch bei der Preisgrenze für billige Netzenergie im Modus PV:

[image]

Wisley7, 2025-01-07:
Wenn man aber direkt auf den Balken drückt, dann funktioniert es bei mir…

closed 2025-01-19 by github-actions[bot]

SebStange, 2025-11-17:
Spricht etwas dagegen die Zahl der Nachkommastellen an der Preisquelle festzumachen? EPEX Spot liefert wie Tibber meines Wissens ebenso die Preise in ct/kWh mit 2 Nachkommastellen. Das wird in anderen Währungen sicherlich anders sein, dürfte jedoch dann auch über die Daten-/Preisquelle definiert werden.