evcc-io/evcc #17978: Batterieladung mit Sungrow SH6 wird im GUI in Version 0.132.0 nicht mehr ausgegeben

JuMa1967, 2024-12-31:
### Describe the bug

Seit dem Update auf v0.132.0 funktioniert die Auswertung der Ladung zur Batterie nicht mehr. 
Diese wurde bei v0.131.11 noch korrekt dargestellt. Der Ladestand in % wird korrekt ausgewertet.
Ein Fehler wird nicht ausgegeben.

### Steps to reproduce

Fehler ist direkt sichtbar wenn Solarstrom vorhanden ist und die Batterie geladen wird. Das Laden der Hausbatterie wird mit als Verbrauch dargestellt und nicht gesondert als Batterieladung.

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: error
levels:
  cache: error

# Sponsortoken GitHub
sponsortoken: xxx
# unique installation id
plant: 6d27670fdeef1eee39aafd29b222b48ddaae727c810b8f6c40c6a81b2c0ac8c8

interval: 30s # control cycle interval

meters:
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.133.27  
  port: 502  
  usage: grid  
  modbus: tcpip  
  timeout: 10s  
  name: grid1
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.133.27  
  port: 502  
  usage: pv  
  modbus: tcpip  
  timeout: 10s  
  name: pv2
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.133.27  
  port: 502  
  usage: battery  
  modbus: tcpip  
  timeout: 10s  
  name: battery3

vehicles:
 - name: ev4
   type: template
   template: tesla 
   title: Tesla 
   accessToken:
   refreshToken: xx
   vin: xx  
   capacity: 60
 - name: MyPV_ELWA2
   type: custom
[... 277 lines cut ...]
[lp-1  ] DEBUG 2024/12/31 11:38:57 charge power: 0W
[lp-1  ] DEBUG 2024/12/31 11:38:57 charge currents: [0 0 0]A
[site  ] DEBUG 2024/12/31 11:38:57 grid power: -0W
[site  ] DEBUG 2024/12/31 11:38:57 pv power: 2945W
[site  ] DEBUG 2024/12/31 11:38:57 grid currents: [-0.0173 0.254 0.0387]A
[site  ] DEBUG 2024/12/31 11:38:57 battery soc: 25%
[site  ] DEBUG 2024/12/31 11:38:57 battery power: 0W
[site  ] DEBUG 2024/12/31 11:38:57 site power: 100W
[lp-1  ] DEBUG 2024/12/31 11:38:57 charge voltages: [231 229 230]V
[lp-1  ] DEBUG 2024/12/31 11:38:57 detected connected phases: 3p
[lp-1  ] DEBUG 2024/12/31 11:38:57 charge total import: 1179.900kWh
[lp-1  ] DEBUG 2024/12/31 11:38:57 charger status: A
```

### What type of operating system or environment does evcc run on?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.132.0

JuMa1967, 2024-12-31:
Hier noch die Darstellung des Fehlers im Vergleich evcc vs. Sungrow

[image]
[image]

andig, 2024-12-31:
Was sagt

    evcc meter

stimmen die Werte? Welcher stimmt nicht? In welcher Version sind sie korrekt? Wie sieht evcc meter dazu aus?

JuMa1967, 2024-12-31:
Die Erzeugung stimmt.
Bei "Out" Batterie Laden müsste ein Wert von 2,2kW angegeben sein. (es wird der komplette Out als verbrauch angesehen, inkl. Batterieladung
Ich hatte das Problem mit dem Sperren der Batterie deshalb der Uptdate. Davor hatte ich wie geschrieben v0.131.11 hier hatte es noch funktioniert.
Die Frage zu evcc meter versthe ich nicht da musst du mir genauer sagen was du meinst.

JuMa1967, 2024-12-31:
So müsste es aussehen:
[image]

VolkerK62, 2024-12-31:
auf der Kommandozeile `evcc meter` aufrufen und die Ausgabe hier posten.
Zusammen mit dem Hinweis, welche Werte richtig und welche falsch sind.

JuMa1967, 2024-12-31:
pi@EVCC1:~ $ pi@EVCC1:~ $ evcc meter
[main  ] INFO 2024/12/31 13:56:51 evcc 0.132.0
[main  ] INFO 2024/12/31 13:56:51 using config file: /home/pi/evcc.yaml
[main  ] FATAL 2024/12/31 13:56:51

Found systemd service database at "/var/lib/evcc/evcc.db", evcc has been invoked with database "~/.evcc/evcc.db".
Running evcc with vehicles configured in evcc.yaml may lead to expiring the yaml configuration's vehicle tokens.
This is due to the fact, that the token refresh will be saved to the local instead of the service's database.
If you have vehicles with touchy tokens like PSA or Tesla, make sure to remove vehicle configuration from the yaml file.

If you know what you're doing, you can run evcc ignoring the service database with the --ignore-db flag.

JuMa1967, 2025-01-01:
Das Problem kommt nicht vom Update auf v0.132.0 sondern vom Netzwerkanschlulss. Wenn ich am Wechselrichter vom WINET-Dongle auf den Netzwerkport des Wechselrichters wechsle (hinterer Anschluss) funktioniert die Auswertung korrekt. Auf dem WINET ist der Fehler. Ich habe auf den WINET die Version WINET-SV200.001.00.P026 installiert und lasse mir kommende Woche auf die Version WINET-SV200.001.00.P030 aktualisieren.

zachelnet, 2025-01-01:
Ja, das Fehlverhalten mit der Version "WINET-SV200.001.00.P026" kann ich bestätigen, da die Auswertung vom Modbus Register 13000 falsche Werte am WiNet Port liefert, mit der Version "WINET-SV200.001.00.P030" werden wieder die richtigen Werte geliefert und damit sollte die Berechnung wieder stimmen.

naltatis, 2025-01-03:
Ok, dann mach ich den hier zu.

closed 2025-01-03 by naltatis

JuMa1967, 2025-01-07:
Kurzes Feedback:
Nach dem Update des Sungrow WiNET auf Version WINET-SV200.001.00.P030 in Verbindung mit evcc v0.132.0 funktioniert alles einwandfrei. Danke für die Unterstützung.