evcc-io/evcc #17979: ABB Terra AC: OCPP

lupinve, 2024-12-31:
### Describe the bug

I am facing an issue when trying to connect my ABB Terra AC-W11-G5-R-0 charger to EVCC, which is running in a Docker container. Despite following the instructions and attempting to reset the ABB Terra to factory defaults, I am still unable to establish a connection.

### Steps to reproduce

1. Set up EVCC in Docker according to the installation guide.
2. Attempt to connect the ABB Terra AC-W11-G5-R-0 charger to EVCC.
3. I have tried resetting the charger to factory defaults, but the issue persists.
...

### Configuration details

```yaml
chargers:
  - name: my_charger
    type: template
    template: ocpp-abb
    connecttimeout: 10m
    metervalues: Energy.Active.Import.Register
```

### Log details

```text
[main ] INFO 2024/12/31 11:42:51 evcc 0.132.0
[main ] INFO 2024/12/31 11:42:51 using config file: /etc/evcc.yaml
[db ] INFO 2024/12/31 11:42:51 using sqlite database: /var/lib/evcc/evcc.db
[db ] TRACE 2024/12/31 11:42:51 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db ] TRACE 2024/12/31 11:42:51 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db ] TRACE 2024/12/31 11:42:51 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db ] TRACE 2024/12/31 11:42:51 SELECT * FROM `settings` 3 <nil>
[db ] TRACE 2024/12/31 11:42:51 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db ] TRACE 2024/12/31 11:42:51 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db ] TRACE 2024/12/31 11:42:51 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db ] TRACE 2024/12/31 11:42:51 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db ] TRACE 2024/12/31 11:42:51 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db ] TRACE 2024/12/31 11:42:51 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db ] TRACE 2024/12/31 11:42:51 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[main ] INFO 2024/12/31 11:42:51 listening at :7070
[db ] TRACE 2024/12/31 11:42:51 SELECT * FROM `configs` WHERE `configs`.`class` = 2 0 <nil>
[db ] TRACE 2024/12/31 11:42:51 SELECT * FROM `configs` WHERE `configs`.`class` = 1 0 <nil>
[ocpp-1] DEBUG 2024/12/31 11:42:51 waiting for chargepoint: 10m0s
[ocpp ] INFO 2024/12/31 11:49:21 charge point connected, registering: TACW1140422G0297
[ocpp ] TRACE 2024/12/31 11:49:21 send TACW1140422G0297: [2,"2205367450","ChangeAvailability",{"connectorId":0,"type":"Operative"}]
[ocpp ] TRACE 2024/12/31 11:49:21 recv TACW1140422G0297: [2, "0809972", "StatusNotification", {"connectorId": 0, "errorCode": "NoError", "info": "null", "status": "Available", "vendorErrorCode": "0x0000"}]
[ocpp ] TRACE 2024/12/31 11:49:21 send TACW1140422G0297: [3,"0809972",{}]
[ocpp ] TRACE 2024/12/31 11:49:22 recv TACW1140422G0297: [2, "0285008", "StatusNotification", {"connectorId": 1, "errorCode": "NoError", "info": "null", "status": "Available", "vendorErrorCode": "0x0000"}]
[ocpp ] TRACE 2024/12/31 11:49:22 send TACW1140422G0297: [3,"0285008",{}]
[ocpp-1] DEBUG 2024/12/31 11:49:51 failed configuring availability: timeout
[ocpp ] TRACE 2024/12/31 11:49:51 send TACW1140422G0297: [2,"3797512919","GetConfiguration",{}]
[ocpp ] TRACE 2024/12/31 11:49:53 recv TACW1140422G0297: [3, "3797512919", {"configurationKey": [{"key": "GetConfigurationMaxKeys", "readonly": true, "value": "20"}, {"key": "HeartbeatInterval", "readonly": false, "value": "60"}, {"key": "MeterValueSampleInterval", "readonly": false, "value": "10"}, {"key": "AuthorizationKey", "readonly": false}, {"key": "LocalAuthorizeOffline", "readonly": false, "value": "true"}, {"key": "LocalPreAuthorize", "readonly": false, "value": "false"}, {"key": "LocalAuthListEnabled", "readonly": false, "value": "true"}, {"key": "LocalAuthListMaxLength", "readonly": true, "value": "16"}, {"key": "SendLocalListMaxLength", "readonly": true, "value": "8"}, {"key": "SupportedFeatureProfiles", "readonly": true, "value": "Core,FirmwareManagement,LocalAuthListManagement,SmartCharging,RemoteTrigger"}, {"key": "ChargeProfileMaxStackLevel", "readonly": true, "value": "3"}, {"key": "ChargingScheduleAllowedChargingRateUnit", "readonly": true, "value": "Current,Power"}, {"key": "ChargingScheduleMaxPeriods", "readonly": true, "value": "25"}, {"key": "MaxChargingProfilesInstalled", "readonly": true, "value": "16"}, {"key": "NumberOfConnectors", "readonly": true, "value": "1"}, {"key": "AllowOfflineTxForUnknownId", "readonly": false, "value": "false"}, {"key": "AuthorizeRemoteTxRequests", "readonly": false, "value": "false"}, {"key": "ConnectionTimeOut", "readonly": false, "value": "120"}, {"key": "WebSocketPingInterval", "readonly": false, "value": "30"}, {"key": "StopTransactionOnInvalidId", "readonly": false, "value": "true"}, {"key": "MaxEnergyOnInvalidId", "readonly": false, "value": "1000"}, {"key": "MeterValuesAlignedData", "readonly": false, "value": "Energy.Active.Import.Register"}, {"key": "MeterValuesAlignedDataMaxLength", "readonly": true, "value": "4"}, {"key": "ClockAlignedDataInterval", "readonly": false, "value": "0"}, {"key": "MeterValuesSampledData", "readonly": false, "value": "Current.Import,Current.Offered,Energy.Active.Import.Register,Power.Active.Import,Power.Offered,SoC,Voltage"}, {"key": "MeterValuesSampledDataMaxLength", "readonly": true, "value": "22"}, {"key": "StopTxnSampledData", "readonly": false, "value": ""}, {"key": "StopTxnSampledDataMaxLength", "readonly": true, "value": "5"}, {"key": "StopTxnAlignedData", "readonly": false, "value": ""}, {"key": "StopTxnAlignedDataMaxLength", "readonly": true, "value": "7"}, {"key": "ResetRetries", "readonly": false, "value": "0"}, {"key": "TransactionMessageAttempts", "readonly": false, "value": "10"}, {"key": "TransactionMessageRetryInterval", "readonly": false, "value": "10"}, {"key": "ConnectorPhaseRotation", "readonly": false, "value": "Unknown"}, {"key": "UnlockConnectorOnEVSideDisconnect", "readonly": false, "value": "true"}, {"key": "StopTransactionOnEVSideDisconnect", "readonly": false, "value": "true"}]}]
[ocpp ] TRACE 2024/12/31 11:49:53 send TACW1140422G0297: [2,"3145653213","TriggerMessage",{"requestedMessage":"BootNotification"}]
[ocpp ] TRACE 2024/12/31 11:49:53 recv TACW1140422G0297: [3, "3145653213", {"status": "Accepted"}]
[ocpp ] TRACE 2024/12/31 11:49:54 recv TACW1140422G0297: [2, "5827758", "BootNotification", {"chargePointModel": "CDT_TACW11::NET_WIFI", "chargePointVendor": "ABB", "chargeBoxSerialNumber": "TACW1140422G0297", "chargePointSerialNumber": "TACW1140422G0297", "firmwareVersion": "TAC3Z9119006710273::V1.8.21", "meterType": "V1"}]
[ocpp ] TRACE 2024/12/31 11:49:54 send TACW1140422G0297: [3,"5827758",{"currentTime":"2024-12-31T10:49:54Z","interval":60,"status":"Accepted"}]
[ocpp ] TRACE 2024/12/31 11:49:54 send TACW1140422G0297: [2,"1303722021","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Energy.Active.Import.Register"}]
[ocpp ] TRACE 2024/12/31 11:49:54 recv TACW1140422G0297: [2, "3406334", "SecurityEventNotification", {"type": "StartupOfTheDevice", "timestamp": "2024-12-29T21:25:48.000Z"}]
[ocpp ] ERROR 2024/12/31 11:49:54 ocpp message (3406334): NotSupported - Unsupported feature SecurityEventNotification ([2, "3406334", "SecurityEventNotification", {"type": "StartupOfTheDevice", "timestamp": "2024-12-29T21:25:48.000Z"}])
[ocpp ] TRACE 2024/12/31 11:49:54 recv TACW1140422G0297: [2, "4817968", "SecurityEventNotification", {"type": "SettingSystemTime", "techInfo": "ocppBoot", "timestamp": "2024-12-31T10:49:54.000Z"}]
[ocpp ] ERROR 2024/12/31 11:49:54 ocpp message (4817968): NotSupported - Unsupported feature SecurityEventNotification ([2, "4817968", "SecurityEventNotification", {"type": "SettingSystemTime", "techInfo": "ocppBoot", "timestamp": "2024-12-31T10:49:54.000Z"}])
[ocpp ] TRACE 2024/12/31 11:49:55 recv TACW1140422G0297: [3, "1303722021", {"status": "Accepted"}]
[ocpp ] TRACE 2024/12/31 11:49:55 send TACW1140422G0297: [2,"3437939851","TriggerMessage",{"requestedMessage":"MeterValues"}]
[ocpp ] TRACE 2024/12/31 11:49:55 recv TACW1140422G0297: [3, "3437939851", {"status": "Accepted"}]
[... cut ...]  {"value": "0.00", "context": "Trigger", "format": "Raw", "measurand": "Current.Import", "unit": "A"}, {"value": "0", "context": "Trigger", "format": "Raw", "measurand": "Power.Active.Import", "unit": "W"}]}]}]
[ocpp ] TRACE 2024/12/31 11:52:57 send TACW1140422G0297: [3,"5445730",{}]
```

### What type of operating system or environment does evcc run on?

Docker container

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

 0.132.0

mfuchs1984, 2024-12-31:
`[main ] FATAL 2024/12/31 11:49:57 cannot create charger 'my_charger': cannot create charger type 'template': cannot create charger type 'ocpp': sponsorship required, see https://github.com/evcc-io/evcc#sponsorship`

mfuchs1984, 2024-12-31:
Abgesehen vom fehlenden sponsortoken, warum hast du das drin?
`metervalues: Energy.Active.Import.Register`

closed 2024-12-31 by andig