evcc-io/evcc #17981: Tesla: remove client ID

andig, 2024-12-31:
With https://github.com/evcc-io/evcc/discussions/17501 we're relying on users to supply their own Tesla Client ID. They can use https://www.myteslamate.com/tesla-api-application-registration/ which provides this free of charge. As a benefit, they get 10$ free requests per month for personal use.

To complete this transition by end of January, we should remove the Tesla Client ID from the build process and hint users accordingly.

This prepares for evcc Tesla client _only_ providing command proxy. We can then better understand, what cost and efforts are involved with running the command proxy in future.

rklueber, 2025-01-09:
On the nighly build my tesla connects fine. 

I run a own build docker, bc of some changes für my PV system which are not yet in the nightly. I am not able to get tesla connected in my self build. I found out the hard way that you can need the comfy way of access/refresh token generator from evcc. 

I registered a 3rd party app at tesla developer account. I have clientId and clientSecret. But how can I connect now. I am missing to create access and resfresh token.

I tried ENV variables TESLA_CLIENT_ID and TESLA_CLIENT_SECRET with no success.

naltatis, 2025-01-10:
@rklueber See the explanation in https://github.com/evcc-io/evcc/discussions/17501#discussioncomment-11620698. Our plan is to deprecate tesla.evcc.io and let Tesla owners generate the credentials via the free tier of myteslamate.com.

closed 2025-01-11 by andig