evcc-io/evcc #17983: Unterstützung des 'smart' (Klever) Lademodus in EVCC-Konfiguration und MQTT hinzufügen

Fl0x23, 2024-12-31:
### Describe the bug

Bei EVCC fehlt die Konfigurationseinstellung Smart (Klever). Der Modus kann über die Konfigurationsdatei evcc.yaml nicht gesetzt werden. Des Weiteren ist das Abfragen oder Setzen per MQTT aktuell nicht möglich.

[image]

**Aktuelle Dokumentation:** [configuration/loadpints](https://docs.evcc.io/docs/reference/configuration/loadpoints)
[image]

Derzeit wird das Setzen verschiedener Ladepunktmodi unterstützt. Die Modi sind off, now, minpv und pv. Der Modus 'smart' (Klever) ist aktuell nicht enthalten.

Ich schlage vor, die Unterstützung für das Setzen und Abfragen des Modus 'smart' über die Konfigurationsdatei und MQTT zu implementieren und anschließend die Dokumentation zu aktualisieren.

Die Dokumentation könnte wie folgt erweitert werden:

- `smart`: Lade zu Zeiten mit den günstigsten Strompreisen, basierend auf einem hinterlegten dynamischen Stromtarif.

INFO: Beachten Sie, dass für die Nutzung des Smart-Modus ein entsprechender dynamischer Stromtarif hinterlegt sein muss.

### Steps to reproduce

Füge den Modus 'smart' unter Ladepunkte in der Konfigurationsdatei evcc.yaml hinzu.
```
loadpoints:
   mode: smart # Lade zu Zeiten mit den günstigsten Strompreisen, basierend auf einem hinterlegten dynamischen Stromtarif.
```

### Configuration details

```yaml
loadpoints:
   mode: smart
```

### Log details

```text
not necessary
```

### What type of operating system or environment does evcc run on?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

StefanSchoof, 2024-12-31:
Smart ist nur der Anzeigename für PV wenn man kein PV hat. Siehe https://github.com/evcc-io/evcc/pull/14341

closed 2024-12-31 by andig

Fl0x23, 2024-12-31:
@StefanSchoof, @andig
Danke für die schnelle Rückmeldungen. Leider besteht das Problem weiterhin. Ich denke, es wäre hilfreich, das Issue offen zu lassen, da die Funktionalität aktuell nicht wie erwartet gegeben ist.

Aktuell ist es nach meinem Verständnis weiterhin nicht möglich, den Modus über die Konfigurationsdatei evcc.yaml oder MQTT dauerhaft zu setzen. Zwar wird der Modus "pv" erkannt, und es treten dabei keine Fehler in den Logs auf (siehe Info), allerdings bleibt der zuletzt verwendete Lademodus aktiv – in meinem Fall z. B. "Off".

Hier einige ergänzende Informationen, die das Verhalten verdeutlichen:
1. **Auszug MQTT Explorer mit über UI gesetzten Modus Smart (Klever).**
[image]

2. **Auszug evcc.yaml**
[image]

3. **Auszug Logfile**
`[lp-1 ] INFO 2024/12/31 20:51:15 loadpoint 1: `
`[lp-1 ] INFO 2024/12/31 20:51:15 mode: pv`

4. **Abbildung UI**
[image]

Ich hoffe, diese zusätzlichen Details helfen, das Verhalten besser nachzuvollziehen. Sollte ich irgendwo einen Fehler in der Konfiguration haben, wäre ich für Hinweise natürlich sehr dankbar. Falls das Problem jedoch reproduzierbar ist, würde ich mich freuen, wenn wir das Thema nochmal öffnen könnten, um gemeinsam eine Lösung zu finden.

Vielen Dank

naltatis, 2025-01-01:
> Aktuell ist es nach meinem Verständnis weiterhin nicht möglich, den Modus über die Konfigurationsdatei evcc.yaml oder MQTT dauerhaft zu setzen.

Dauerhaft setzten, also dass man den nicht mehr im UI ändern kann, geht in der Tat nicht. Du kannst aber bestimmen, dass der Modus beim Trennen/Verbinden eines Fahrzeuges auf einen bestimmten Wert gesetzt wird.

In deinem Screenshot sieht man, dass du das für den Ladepunkt ja gemacht hast `mode: pv`. Hast du am Fahrzeug auch was konfiguriert? Die Logik ist "1. Fahrzeug, 2. Ladepunkt". Ist 1 und 2 nicht gegeben, wird der Modus nicht geändert.