evcc-io/evcc #17991: Vaillant-SGready integration "crasht" EVCC nach ca 5 bis 6 Minuten

flushbug, 2025-01-01:
### Describe the bug

Mit aktiver Vaillant SGready integration hängt EVCC nach ca. 5 bis 6 Minuten reproduzierbar bei der Aktualisierung der Loadpoints. Das log bleibt beim update des lp-2 (Vaillant) stehen und es passiert nichts mehr. 
Sämtliche MQTT Topics werden ebenfalls nicht mehr beschrieben. 
Der EVCC modbus Proxy läuft allerdings weiterhin.

(Nightly build habe ich noch nicht getestet)

### Steps to reproduce

1. Hinzufügen einer Vaillant Wärmepumpe über SGready SensorNet template
2. Starten von EVCC
3. ca. 5 bis 6 Minuten warten

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  site: debug
  cache: error
  lp-1: trace
  lp-2: trace

# mqtt message broker
mqtt:
  broker: XXX:1883
  topic: evcc # root topic for publishing, set empty to disable publishing
  user: XXX
  password: XXX

interval: 15s # control cycle interval

meters:
- type: template
  template: kostal-plenticore 
  id: 71  
  host: XXX 
  port: 1502  
  usage: pv  
  modbus: tcpip  
  name: pv1
- type: template
  template: kostal-plenticore 
  id: 71  
  host: XXX
  port: 1502  
  usage: pv  
  modbus: tcpip  
  name: pv2
- type: template
  template: kostal-plenticore 
  id: 71  
  host: XXX
  port: 1502  
  usage: battery  
  modbus: tcpip  
  name: battery2
- type: template
  template: kostal-ksem 
  id: 71  
  host: XXX
  port: 502  
  usage: grid  
[... 542 lines cut ...]
Jan 01 11:58:26 unifi evcc[9757]: [site  ] DEBUG 2025/01/01 11:58:26 grid power: -10W
Jan 01 11:58:26 unifi evcc[9757]: [site  ] DEBUG 2025/01/01 11:58:26 grid powers: [790 -410 -390]W
Jan 01 11:58:26 unifi evcc[9757]: [site  ] DEBUG 2025/01/01 11:58:26 grid currents: [3.68 -2.02 -1.79]A
Jan 01 11:58:26 unifi evcc[9757]: [site  ] DEBUG 2025/01/01 11:58:26 pv 2 power: 650W
Jan 01 11:58:27 unifi evcc[9757]: [site  ] DEBUG 2025/01/01 11:58:27 battery soc: 36%
Jan 01 11:58:27 unifi evcc[9757]: [site  ] DEBUG 2025/01/01 11:58:27 battery power: -2781W
Jan 01 11:58:27 unifi evcc[9757]: [site  ] DEBUG 2025/01/01 11:58:27 pv 1 power: 3815W
Jan 01 11:58:27 unifi evcc[9757]: [site  ] DEBUG 2025/01/01 11:58:27 pv power: 4465W
Jan 01 11:58:27 unifi evcc[9757]: [site  ] DEBUG 2025/01/01 11:58:27 battery has priority at soc 36% (< 40%)
Jan 01 11:58:27 unifi evcc[9757]: [site  ] DEBUG 2025/01/01 11:58:27 site power: 90W
Jan 01 11:58:27 unifi evcc[9757]: [lp-2  ] DEBUG 2025/01/01 11:58:27 charge total import: 9336.936kWh
Jan 01 11:58:27 unifi evcc[9757]: [lp-2  ] DEBUG 2025/01/01 11:58:27 charger status: B
```

### What type of operating system or environment does evcc run on?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

v0.132.0

andig, 2025-01-01:
Offensichtlich crasht da nichts. Bitte 1) trace log und 2) mit --profile starten und Goroutine Profile von /debug/pprof. (edited)

andig, 2025-01-01:
Closed in https://github.com/evcc-io/evcc/commit/2739488a91cc76a9e4e5daa30715c20cd0595910

closed 2025-01-01 by andig