evcc-io/evcc #18000: Fahrzeugverbindung wird nach Softwarestart nicht erneut geprüft, Fahrzeug lädt unerwünscht

abuesen, 2025-01-01:
### Describe the bug

Wir haben hier leider derzeit kurze Stromausfälle im Minutenbereich. Nach einem Stromausfall sind von den beteiligten Geräten zuerst die Wallboxen, dann HomeAssitant mit evcc und danach erst der Internetrouter mit Internetverbindung verfügbar.

Evcc erkennt beim Start das angschlossene Fahrzeug, kann aber den SoC nicht abrufen. Es fällt in den Zustand "SoC unter minimale Ladung" und startet den Ladevorgang mit 11kW (Maximum). Danach erfolgt kein weiterer Versuch, den SoC abzurufen. Selbst Trennen und erneutes Verbinden des Fahrzeugs mit der Wallbox löst keine erneute Abfrage des SoC aus. 

Abhilfe schafft nur eine Neustart von evcc. In der Oberfläche wird das auch entsprechend symbolisiert und ein manueller Neustart angeraten.

Meine Erwartung: Evcc sollte nach dem Start mit erfolgloser Verbindung nach 15 minuten und 30 minuten weiter Verbindungsversuche unternehmen. Ebenso bei erneuter Verbindung des Fahrzeuges. 

Evcc sollte auch nicht unkontrolliert laden, wenn der SoC sich nicht ermitteln lässt. Falls doch sollte maximal die gewünschte Mindestladung ab theorteischen 10% geladen werden und nicht bis zum Abschalten des Ladevorgangs durch andere Ereignisse. 

Ich habe NICHT den letzten nightly probiert, da ich nicht weiß, wie das geht.

### Steps to reproduce

1. Fahrzeug verbinden
2. Wallbox, evcc und router abschalten
3. Wallbox einschalten
4. evcc einschalten
5. router aktivieren

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: ID#

interval: 15s # control cycle interval

# mqtt message broker
mqtt:
  topic: evcc # root topic for publishing, set empty to disable publishing
  broker: localhost:1883
  user: mqtt
  password: EinPasswort

meters:
  - name: grid-fox
    type: custom
    power:
      source: mqtt
      topic: sensor/ct_power_w
      timeout: 30s
    energy:
      source: mqtt
      topic: sensor/fox_feed_in_energy_today

  - name: pv
    type: custom
    power:
      source: mqtt
      topic: ha/sensor/pv_power_helper/state
      scale: 1000  # kW to W
      timeout: 30s
    energy:
      source: mqtt
      topic: sensor/fox_pv1_energy_total
      timeout: 30s

[... 679 lines cut ...]
[lp-1  ] DEBUG 2025/01/01 14:45:31 charge total import: 10647.576kWh
[lp-1  ] DEBUG 2025/01/01 14:45:31 charger status: B
[site  ] DEBUG 2025/01/01 14:45:46 ----
[lp-2  ] DEBUG 2025/01/01 14:45:46 charge power: 0W
[lp-1  ] DEBUG 2025/01/01 14:45:46 charge power: 0W
[lp-2  ] DEBUG 2025/01/01 14:45:46 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2025/01/01 14:45:46 charge currents: [0 0 0]A
[site  ] DEBUG 2025/01/01 14:45:46 grid power: 60W
[site  ] DEBUG 2025/01/01 14:45:46 pv power: 232W
[site  ] DEBUG 2025/01/01 14:45:46 site power: 160W
[lp-2  ] DEBUG 2025/01/01 14:45:46 charge total import: 0.000kWh
[lp-2  ] DEBUG 2025/01/01 14:45:46 charger status: A
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.132.0

andig, 2025-01-01:
Einfach minsoc auf 0 setzen. Dann wird ohne soc auch nicht geladen. Oder HA so konfigurieren, dass Anwendungen erst mit Internet gestartet werden.

andig, 2025-01-01:
Und: ohne Fahrzeugzuordnung würde hier auch nichts passieren! Es gibt also eine grosse Bandbreite von möglichen workarounds.

naltatis, 2025-01-02:
> Den minsoc anhand bekannter Energiemenge plus Verlusten ab 0% zu laden ist eine interessante Idee.

Das könnten wir ja sogar transparent (ohne neues Setting) machen. Also zumindest wenn wir eine Fahrzeugzuordnung und und Kapazität haben. Spricht aus meiner Sicht nichts gegeben.

abuesen, 2025-01-02:
Äh, sorry Leute: Ich wünsche euch ein gutes neues Jahr! 
Hab ich wegen der Stromausfälle total vergessen :)

Wie ist die Meinung zum erneuten Verbindungsversuch (hier Bluelink) nach Zeit oder beim erneuten Verbinden des Fz?

abuesen, 2025-01-02:
> Oder HA so konfigurieren, dass Anwendungen erst mit Internet gestartet werden. Noch besser: root cause lösen

Startgeschwindigkeit der Systeme kann ich schlecht "lösen". Hab jetzt eine HA-Automatisuerung erstellt:
- Bei Wechsel der Internetverbindung zu "Connected" wird evcc neu gestartet.

Theoretische kann ja auch durch andere Ursachen evcc starten, wenn gerade kein Internet verfügbar ist. Dann hilft das auch.

andig, 2025-01-02:
> Wie ist die Meinung zum erneuten Verbindungsversuch (hier Bluelink) nach Zeit oder beim erneuten Verbinden des Fz?

Steht im Backlog...

andig, 2025-01-02:
> Falls doch sollte maximal die gewünschte Mindestladung ab theorteischen 10% geladen werden und nicht bis zum Abschalten des Ladevorgangs durch andere Ereignisse.

Lustig: genau das ist schon umgesetzt:

```
	if lp.vehicleSoc != 0 {
		active := lp.vehicleSoc < float64(minSoc)
		if active {
			lp.log.DEBUG.Printf("forced charging at vehicle soc %.0f%% (< %.0f%% min soc)", lp.vehicleSoc, float64(minSoc))
		}
		return active
	}

	minEnergy := v.Capacity() * float64(minSoc) / 100 / soc.ChargeEfficiency
	return minEnergy > 0 && lp.getChargedEnergy() < minEnergy
```

closed 2025-01-02 by andig

abuesen, 2025-01-03:
> Steht im Backlog...

Da: https://github.com/evcc-io/evcc/issues/14496