evcc-io/evcc #18002: Batterie läd immer nur 50%

Basti77, 2025-01-01:
### Describe the bug

die Haus Batterie läd immer nur +50% obwohl ein viel größerer Zeitpunkt angegeben ist. 

E3DC S10E 18Kwh
[evcc-20250101-190939-debug.zip](https://github.com/user-attachments/files/18286551/evcc-20250101-190939-debug.zip)
[image]
[image]

### Steps to reproduce

das Verhalten hat sich bei keiner Version verbessert und kann jederzeit wenn der Preis stimmt reproduziert werden. 
nach ca 2h ist Schluss. 

### Configuration details

```yaml
interval: 20s # control cycle interval
#mqtt:
#  broker: localhost:1883
#  topic: evcc # root topic for publishing, set empty to disable publishing
#  clientid: 666

 
meters:  
  - name: grid  
    type: template  
    template: e3dc-rscp  
    usage: grid  
    host: 192.168.200.243 # IP-Adresse oder Hostname  
    port: 5033 # Port (optional)  
    user: 
    password: a
    key: 123456 # optional   
  
  - name: pv  
    type: template  
    template: e3dc-rscp  
    usage: pv  
    host: 192.168.200.243 # IP-Adresse oder Hostname  
    port: 5033 # Port (optional)  
    user:  
    password:   
    key: 123456 # optional  
  
  - name: Battery  
    type: template  
    template: e3dc-rscp  
    usage: battery  
    host: 192.168.200.243 # IP-Adresse oder Hostname  
    port: 5033 # Port (optional)  
    user: 
    password: 
    key: 123456 # optional  

chargers:
- name: charger
  type: go-e
  uri: http://192.168.200.249

vehicles:

site:
  title: Skynet Überschussladen # display name for UI
  meters:
    pv: pv
    grid: grid # grid meter
    battery: Battery # battery meter

loadpoints:
- title: Garage # display name for UI
  charger: charger
  mode: pv
  soc:
   poll:
    mode: charging
    interval: 10m
   estimate: false # set true to unterpolate between api updates
  guardduration: 5m
  mincurrent: 6
  maxcurrent: 16

sponsortoken: 
tariffs:
  currency: EUR # (default EUR)
  grid:
    type: tibber
    token: ""
  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.08 # [currency]/kWh
```

### Log details

```text
das komplette Logfile hängt als Zip an.
```

### What type of operating system or environment does evcc run on?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.12

andig, 2025-01-01:
Ohne ins Log zu schauen: schaltet die e3dc die Ladung ab? Gibts da eine maximale Energiemenge?

Basti77, 2025-01-02:
Das ist eine gute Frage: Wo ist eine Energiemenge vorgegeben? Eventuell irgendwo hardcodiert?

Zwei Sachen irritieren mich:

Heute Nacht hat das Gerät, weil es zufällig mit dem Plan übereingestimmt hat, um 0 Uhr wieder angefangen zu laden, was ja auch dem Plan entsprach. Nicht, dass ich darüber böse bin, aber es ist ein Widerspruch zur angeblichen 24-Stunden-Sperre. Also, dass man immer nur alle 24 Stunden über die Webseite laden kann. Es sieht so aus, als würde ein Datumswechsel ausreichen.
[image]

Über das Portal kann man nur 4500 Wh laden. Aber diese Grenze kann es nicht sein, weil 50 % bei mir 8 kWh sind. Eine harte Grenze gibt es im Gerät aber nicht. Über ein CLI-Tool kann ich den Speicher auch zu 100 % beladen (e3dcset). Dort kann man als Parameter die Ladeleistung und die Lademenge angeben.

marleyan333, 2025-01-02:
Ich habe bei meiner Sonnen eco 8.0 mit 12 kWh Kapazität das gleiche Problem, lädt immer nur bis 80%, obwohl die Kapazität in der yaml hinterlegt ist. Mit dem Tool sonnenbuddy kann ich die Batterie problemlos auf 100% laden. 
 @andig kannst Du genauer erklären, was Du im Letzten Post als Problem identifiziert hast? Wie kann man es ändern?

Basti77, 2025-01-02:
> Grund gefunden: https://github.com/evcc-io/evcc/blob/master/meter/e3dc.go#L179

Wäre es möglich den Wert via GUI setzen zu können? Zur Not über die yaml überschreiben...

andig, 2025-01-02:
Was spricht dagegen einfach einen anderen zu setzen? 50kWh sollten für jeden Bedarf reichen.

closed 2025-01-02 by andig

Basti77, 2025-01-02:
> > Was spricht dagegen einfach einen anderen zu setzen? 50kWh sollten für jeden Bedarf reichen.
> 
> Gerne, aber wo genau in der yaml muss ich das eingeben?

Der Parameter wird im Quellcode gesetzt, also müsste man das Projekt selber kompilieren. Die Frage ist nur ob das jeder kann, oder 50kwh ist ab jetzt der neue Standard. Die zitierten Codezeilen waren aber auch nur für E3DC und nicht für Sonnen. 
Die Frage ist nun, ist das ein allgemeines "Problem"!?

marleyan333, 2025-01-02:
@andig wie basti77 schon schrieb: kann man 50.000 dann als allgemeines Limit setzen auch für andere Batterien?

marleyan333, 2025-01-02:
Denke nicht, dass das Problem damit behoben ist, denn bei mir ist die Kapazität mit 12 kWh ja richtig hinterlegt und dennoch lädt es nur bis 75%… aber der admin liest hier offenbar schon nicht mehr mit…

Basti77, 2025-01-02:
... oder ist genervt weilt das vermutlich ein separater issue wäre.
Andere Hardware, vielleicht auch nee andere Ursache... 🤷