evcc-io/evcc #18006: Wallbox aus - EVCC Fehler beim Starten. option für Wallbox hinzufügen damit EVCC starten kann

mxcxpx, 2025-01-02:
Fehler beim Starten. Überprüfe deine Konfiguration und starte neu.
cannot create charger 'wallbox4': cannot create charger type 'template': cannot create charger type 'wattpilot': failed to WebSocket dial: failed to send handshake request: Get "http://192.168.0.22/ws": dial tcp 192.168.0.22:80: connect: no route to host

ich schalte meine Wallbox nur bei Bedarf ein (Schaltaktor - KNX) 
Wenn die Wallbox aus ist meckert EVCC

kann man für die Wallbox ein optional an/aus button einrichten.
danke

StefanSchoof, 2025-01-02:
Klingt nach https://github.com/evcc-io/evcc/issues/14496

andig, 2025-01-02:
> ich schalte meine Wallbox nur bei Bedarf ein 

Entweder evcc neu starten (geht auch über Api) oder WB an lassen.

closed 2025-01-02 by andig

premultiply, 2025-01-02:
Lösung: WB immer anlassen und in evcc den Lademodus auf aus stellen.