evcc-io/evcc #18007: Since 0.132.0 evcc can‘t start with SMA

ionni73, 2025-01-02:
### Describe the bug

Hi,

After the update 0.132.0, EVCC does not start anymore.
I am using evcc as Home Assistant plug in, with an SMA Solar system.
No configuration changes has been made. 
I wonder if I missed some tasks/actions to be taken into consideration with the latest update.

I was not able to find, if this issue has been handled in another thread. If so, please kindly reference it to me.
If there are information missing in the description of the bug, I apologize in advance (first time I created a bug report) and let me know, what else information I should add.

Many thanks in advance for any support.

### Steps to reproduce

1. Start Home Assistant
2. click on EVCC icon to access EVCC
3. "Turning wheel" appears for several minutes trying to start EVCC

### Configuration details

```yaml
meters:
- name: grid1
  type: template
  template: sma-home-manager 
  usage: grid  
  host: 192.xxx.xxx.xxx  

- name: pv2
  type: template
  template: sma-inverter 
  usage: pv  
  host: 192.xxx.xxx.xxx  
  password: xxxxx 

- name: battery3
  type: template
  template: sma-inverter 
  usage: battery  
  host: 192.xxx.xxx.xxx   
  password: xxxx
  capacity: 8.8  

chargers:
- name: wallbox5
  type: template
  template: go-e-v3 
  host: 192.xxx.xxx.xxx   

vehicles:
- name: byd_atto3
  title: BYD Atto3
  type: template
  template: iso15118
  icon: car  
  capacity: 60  
  phases: 3  
  mode: pv  
  minCurrent: 6  
  maxCurrent: 16  

loadpoints:
- title: Carport
  charger: wallbox5
  vehicle: byd_atto3
  mode: pv
  phases: 0 
  mincurrent: 6
  maxcurrent: 16
  enable:  
    threshold: 0  
    delay: 1m  
  disable:  
    delay: 3m 
    threshold: 0  

site:
  residualPower: 250  
  title: Home
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3

hems:
  type: sma

telemetry: true
```

### Log details

```text
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2025/01/02 11:08:02 evcc 0.132.0
[main  ] INFO 2025/01/02 11:08:02 using config file: /config/evcc.yaml
[db    ] INFO 2025/01/02 11:08:02 using sqlite database: /data/evcc.db
[main  ] INFO 2025/01/02 11:08:02 listening at :7070
[main  ] FATAL 2025/01/02 11:08:06 cannot create meter 'battery3': cannot create meter type 'template': cannot create meter type 'sma': failed to get device: no Speedwire ping response for 192.168.178.64
[main  ] FATAL 2025/01/02 11:08:06 will attempt restart in: 15m0s
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.132.0

andig, 2025-01-02:
> no Speedwire ping response for 192.168.178.64

closed 2025-01-02 by andig

ionni73, 2025-01-02:
Thanks Andig.
Your (brief) comment brought me to look into "Speedwire" (I was not aware about what Speedwire is till today, i.e. an evcc or SMA component or else). 
It turns out that the problem was very trivial: the router decided "on its own" to change of IP address of the inverter (although is not allow to. Why this happened is unclear and not for this thread).

This happened concurrently to the evcc update (unsure, if there is a correlation).

Soultion: IP update in the evcc.yaml. It works as before.

Just documenting for all those, who might have a similar problem in the future.

Cheers

andig, 2025-01-02:
Thank you for updating the root cause. I‘m always cautious (and experience supports this) with reports of „since this version that isn‘t working“. Most of the time there are other changes beside the version upgrade that cause this.