evcc-io/evcc #18008: Auswertung Ladeenergie: Bug „dieses Jahr“ (this year)

An-Wi, 2025-01-02:
### Describe the bug

reference: #15578

Mein Ladevorgang am 31.12.2024 (8:40-13:59) wird bei mir „sporadisch“ dazugerechnet:
[image]
[image]
[image]

Sporadisch deshalb, weil es gestern (01.01.) am Nachmittag/Abend nach meinem Ladevorgang (10:09-15:41) richtig gerechnet wurde. 
Sobald ich das genauer spezifizieren kann, ergänze ich das.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
-
```

### Log details

```text
-
```

### What type of operating system or environment does evcc run on?

Windows

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.132.0

An-Wi, 2025-01-02:
bereits behoben?

An-Wi, 2025-01-08:
Fehler tritt in Version 0.132.1 nach wie vor auf:
[image]
[image]

andig, 2025-01-08:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

An-Wi, 2025-01-08:
Helfen diese Log-Zeilen? Ansonsten bitte präzisieren, welches Log zur Analyse weiterhelfen kann.

### Logzeiten, an denen die Auswertung inkorrekt ist:
<pre>
[db    ] TRACE 2025/01/07 07:04:55 
		SELECT COALESCE(SUM(charged_kwh * solar_percentage) / SUM(charged_kwh), 0)
		FROM sessions
		WHERE finished >= "2024-12-08 07:04:55.562" 
		AND charged_kwh > 0 
		AND solar_percentage IS NOT NULL 1 <nil>
[db    ] TRACE 2025/01/07 07:04:55 
		SELECT COALESCE(SUM(charged_kwh), 0)
		FROM sessions
		WHERE finished >= "2024-12-08 07:04:55.562" 
		AND charged_kwh > 0 
		AND solar_percentage IS NOT NULL 1 <nil>
[db    ] TRACE 2025/01/07 07:04:55 
		SELECT COALESCE(SUM(charged_kwh * price_per_kwh) / SUM(charged_kwh), 0)
		FROM sessions
		WHERE finished >= "2024-12-08 07:04:55.562" 
		AND charged_kwh > 0 
		AND price_per_kwh IS NOT NULL 1 <nil>
[db    ] TRACE 2025/01/07 07:04:55 
		SELECT COALESCE(SUM(charged_kwh * co2_per_kwh) / SUM(charged_kwh), 0)
		FROM sessions
		WHERE finished >= "2024-12-08 07:04:55.562" 
		AND charged_kwh > 0 
		AND co2_per_kwh IS NOT NULL 1 <nil>
[db    ] TRACE 2025/01/07 07:04:55 
		SELECT COALESCE(SUM(charged_kwh * solar_percentage) / SUM(charged_kwh), 0)
		FROM sessions
		WHERE finished >= "2024-01-08 07:04:55.573" 
		AND charged_kwh > 0 
		AND solar_percentage IS NOT NULL 1 <nil>
[db    ] TRACE 2025/01/07 07:04:55 
		SELECT COALESCE(SUM(charged_kwh), 0)
		FROM sessions
		WHERE finished >= "2024-01-08 07:04:55.573" 
		AND charged_kwh > 0 
		AND solar_percentage IS NOT NULL 1 <nil>
[db    ] TRACE 2025/01/07 07:04:55 
		SELECT COALESCE(SUM(charged_kwh * price_per_kwh) / SUM(charged_kwh), 0)
		FROM sessions
		WHERE finished >= "2024-01-08 07:04:55.573" 
		AND charged_kwh > 0 
		AND price_per_kwh IS NOT NULL 1 <nil>
[db    ] TRACE 2025/01/07 07:04:55 
		SELECT COALESCE(SUM(charged_kwh * co2_per_kwh) / SUM(charged_kwh), 0)
		FROM sessions
		WHERE finished >= "2024-01-08 07:04:55.573" 
		AND charged_kwh > 0 
		AND co2_per_kwh IS NOT NULL 1 <nil>
[db    ] TRACE 2025/01/07 07:04:55 
		SELECT COALESCE(SUM(charged_kwh * solar_percentage) / SUM(charged_kwh), 0)
		FROM sessions
		WHERE finished >= "2024-12-31 07:04:55.583" 
		AND charged_kwh > 0 
		AND solar_percentage IS NOT NULL 1 <nil>
[db    ] TRACE 2025/01/07 07:04:55 
		SELECT COALESCE(SUM(charged_kwh), 0)
		FROM sessions
		WHERE finished >= "2024-12-31 07:04:55.583" 
		AND charged_kwh > 0 
		AND solar_percentage IS NOT NULL 1 <nil>
[db    ] TRACE 2025/01/07 07:04:55 
		SELECT COALESCE(SUM(charged_kwh * price_per_kwh) / SUM(charged_kwh), 0)
		FROM sessions
		WHERE finished >= "2024-12-31 07:04:55.583" 
		AND charged_kwh > 0 
		AND price_per_kwh IS NOT NULL 1 <nil>
[... 133 lines cut ...]
		WHERE finished >= "1925-02-01 19:07:32.256" 
		AND charged_kwh > 0 
		AND price_per_kwh IS NOT NULL 1 <nil>
[db    ] TRACE 2025/01/07 19:07:32 
		SELECT COALESCE(SUM(charged_kwh * co2_per_kwh) / SUM(charged_kwh), 0)
		FROM sessions
		WHERE finished >= "1925-02-01 19:07:32.256" 
		AND charged_kwh > 0 
		AND co2_per_kwh IS NOT NULL 1 <nil>
[site  ] DEBUG 2025/01/07 19:07:51 ----
[lp-1  ] DEBUG 2025/01/07 19:07:51 charge power: 0W
[lp-1  ] DEBUG 2025/01/07 19:07:51 charge currents: [0 0 0]A
[script] DEBUG 2025/01/07 19:07:51 /fupcService/pvCurrentPower.bat: 0
[site  ] DEBUG 2025/01/07 19:07:51 pv power: 0W
[site  ] DEBUG 2025/01/07 19:07:51 grid power: 2296W
[site  ] DEBUG 2025/01/07 19:07:51 grid currents: [1.35 1.54 7.22]A
[site  ] DEBUG 2025/01/07 19:07:51 site power: 2196W
[lp-1  ] DEBUG 2025/01/07 19:07:51 charge total import: 8600.002kWh
[lp-1  ] DEBUG 2025/01/07 19:07:51 charger status: A
[lp-1  ] DEBUG 2025/01/07 19:07:51 next soc poll remaining time: 59m39s
[site  ] DEBUG 2025/01/07 19:08:11 ---- 
</pre>

=> Das ergibt für mich insofern Sinn, da mit `WHERE finished >= "2024-12-31 19:07:32.247"` der Ladevorgang vom 31.12.2024 (endet 13:59 Uhr) im 2. Logteil _**nicht**_ mehr erfasst wird...
=>wohingegen er vormittags mit `WHERE finished >= "2024-12-31 07:04:55.583"` in der Abfrage noch voll drin liegt.

andig, 2025-01-08:
/cc @naltatis

naltatis, 2025-01-08:
Jo, der Grund ist, dass wir bei der "Dieses Jahr" Betrachtung aktuell nicht sauber vom 1.1.XXXX anfangen, sondern einfach Tage zurückrechnen. Heißt die aktuelle Uhrzeit spielt noch eine Rolle. Das sollte sich relativ einfach lösen lassen.

https://github.com/evcc-io/evcc/blob/master/core/stats.go#L38

closed 2025-01-08 by naltatis