evcc-io/evcc #18018: Can we have the maxPower parameter (Load Management) available trough API?

RikMe, 2025-01-02:
**Is your feature request related to a problem? Please describe.**
NO, Possible enhancement

**Describe the solution you'd like**
Would be great to be able to dynamically set the maxPower parameter for load management.

Potential use case: "Capaciteitstarief" in Belgium.

**Describe alternatives you've considered**
None

**Additional context**
Add any other context or screenshots about the feature request here.

VolkerK62, 2025-01-02:
you can override the configured value with `GetMaxPower`
see https://github.com/evcc-io/evcc/issues/14261#issuecomment-2449611775

andig, 2025-01-02:
Are we missing any documentation here?

closed 2025-01-02 by andig

RikMe, 2025-01-02:
Didn't find it in the documentation yet. Would also be great to have this in the API so I can flow directly into for ex. Home Assistant.

The current solution trough MQTT works fine though!

andig, 2025-01-05:
Documentation should follow with https://github.com/evcc-io/docs/issues/658

pergolafabio, 2025-06-10:
Hey @RikMe , how are you using the capaciteitstarief in evcc?