evcc-io/evcc #18020: Neuer Charger 'AC Thor 9S'

Lepidopteron, 2025-01-02:
Derzeit werden von my-PV die Meter
* AC ELWA 2
* AC ELWA-E
* AC•THOR
* WiFi Meter
unterstützt und der Charger / Heizstab:
* my-PV AC ELWA 2 (https://download.my-pv.com/partner/Dokumentation%20AC-THOR%20oder%20AC%20ELWA%202%20mit%20Growatt%20DE230622.pdf)

my-PV bietet aber via AC Thor auch externe Heizstäbe an, die neu via API gesteuert werden können. Früher hatte AC Thor die Anbindung eines Smartmeters voraus gesetzt, jetzt kann man den Heizstab via dem API selber steuern: https://www.my-pv.com/de/info/my-pv-cloud/
Vorteil: Die Priorisierung des Stromüberschusses kann richtig verteilt werden.

Dokumentation AC Thor: https://download.my-pv.com/acthor/acthor_doc_de_a0021702.pdf
Swagger Doku des API: https://api.my-pv.com/api-docs/

Unser AC Thor 9s läuft mit einem 6 kW Heizstab
Ich hatte unseren  Charger als elwa2 Heizstab konfiguriert, da stimmt aber die Watt-Zahl immer noch und er 'heizt' nur mit 15W

```
meters:
 - name: ACThor
   type: template
   template: thor
   usage: aux
   host: 192.168.4.18 # IP-Adresse oder Hostname

chargers:
 - name: my_ELWA_Heizstab
   type: template
   template: ac-elwa-2
   # Modbus TCP
   modbus: tcpip
   id: 1
   host: 192.168.4.18 # Hostname
   port: 502 # Port

loadpoints:
 - title: Heizstab # display name for UI
   charger: my_ELWA_Heizstab # Wallbox Referenz
   meter: ACThor
   mode: pv # charge mode (off, now, minpv, pv)
   phases: 1
   mincurrent: 0.125
   maxcurrent: 26 # (6 kW bei 230V ?)
   priority: 0
   enable:
    threshold: -300 # einschalten, wenn 1 Minute lang mindestens 300 W Uberschuss vorhanden ist
    delay: 1m
   disable:
    threshold: 300  # maximum import power (W)
    delay: 1m
```

Das API erscheint mir relativ einfach, vielleicht gibt es sonst auch eine andere Lösung?

andig, 2025-01-03:
Ich finde in der Doku leider nichts zu Modbus Registern- die bräuchte es.

Lepidopteron, 2025-01-03:
> Ich finde in der Doku leider nichts zu Modbus Registern- die bräuchte es.

Dann macht es wahrscheinlich eher Sinn einen POST mit 'generiertem Body' abzusetzen mit dem HTTP Plugin?
https://docs.evcc.io/docs/reference/plugins#http-lesenschreiben

Geht das Plugin denn auch für chargers?

Ansonsten geht Ansteuerung via Modbus RTU/TCP schon, aber dann halt mit den jewieligen Protokollen?
Dann müsste der AC Thor die Bezugsleistung wahrscheinlich bereinigt vom EVCC bekommen? (Siehe S. 36 https://download.my-pv.com/acthor/acthor_doc_de_a0020410.pdf). Also in den Steuerungseinstellungen nicht auf den Smartmeter/Wechselrichter eingestellt sein, sondern auf EVCC?

andig, 2025-01-03:
> Dann müsste der AC Thor die Bezugsleistung wahrscheinlich bereinigt vom EVCC bekommen?

Nein. evcc ist master muss den Thor steuern. Dafür brauchts Modbus Register. Bisher fehlen die...

Lepidopteron, 2025-01-03:
Hm, vielleicht übersehen ich etwas, aber in der Dokumentation steht, dass sowohl `Adjustable Modbus TCP` als auch `Adjustable Modbus RTU` als Steueroption hinterlegt sind. Auf Seite 36 findet sich diese Einstellung
[image]

andig, 2025-01-03:
Da ist Thor der master. Das ist nicht was wir brauchen.

Lepidopteron, 2025-01-03:
Aber wenn die AC-THOR Nummer >1 ist, wäre er doch als Slave einstellbar?

andig, 2025-01-03:
Es braucht eine Doku der Modbusregister. Ich mache zu bis die vorliegt.

Lepidopteron, 2025-01-03:
Diese reichen nicht?
https://github.com/DrBing/ioBroker.AC-THOR/blob/master/Holdig_registers.csv

andig, 2025-01-03:
Das hilft. Der einzige Unterschied zur Elwa scheint das Leistungsregister zu sein- 1060 statt 1074?

closed 2025-01-04 by andig

Lepidopteron, 2025-01-08:
Mit der Version `v0.132.1` jetzt verfügbar, vielen Dank.
Mit folgenden Einstellungen kann die Ansteuerung erfolgen:
Obwohl der Heizstab mit 3 Phasen angesteuert wird, ist es egal ob 1 oder 3 Phasen eingestellt sind. Die Regelung übernimmt AC Thor. Wichtig aber: maxcurrent sollte 32A sein, damit man wirklich die Maximalleistung übergibt, die möglich wäre. Auch das regelt AC Thor selbstständig

**1. AC Thor: Steuerungs-Einstellungen**
```
Ansteuerungs-Typ: Modbus TCP
Zeitablauf Ansteuerung: 1m
```

**2. Einstellungen EVCC**
```
meters:
 - name: ACThor
   type: template
   template: thor
   usage: aux
   host: 192.168.178.2 # IP-Adresse oder Hostname
chargers
 - name: heizstab
   type: template
   template: ac-thor
   modbus: tcpip
   id: 1
   host: 192.168.178.2 # IP-Adresse oder Hostname
   port: 502 # Port
loadpoints:
 - title: Heizstab # display name for UI
   charger: heizstab # Charger reference
   meter: ACThor # Used for temperature
   mode: pv # charge mode (off, now, minpv, pv)
   phases: 1
   mincurrent: 0.125
   maxcurrent: 32
   priority: 0
   enable:
    threshold: -200 # einschalten, wenn 1 Minute lang mindestens 200 W Uberschuss vorhanden ist
    delay: 1m
   disable:
    threshold: 200  # maximum import power (W)
    delay: 1m
```
[image]
[image]

andig, 2025-01-08:
> Ein Punkt, der nicht geht: Er liefert auch bei 'Off'/'Solar', wenn keine Sonnenenergie zur Verfügung steht die minimal mögliche Leistung. Das müsste egtl. '0' sein 

Ich verstehe die Frage nicht. Falsche minCurrent am Ladepunkt?

Lepidopteron, 2025-01-08:
> > Ein Punkt, der nicht geht: Er liefert auch bei 'Off'/'Solar', wenn keine Sonnenenergie zur Verfügung steht die minimal mögliche Leistung. Das müsste egtl. '0' sein
> 
> Ich verstehe die Frage nicht. Falsche minCurrent am Ladepunkt?

Also, wenn der Toggle auf 'Off' steht bzw. auf 'Solar' und KEIN Überschuss besteht, müsste bei Power eigentlich '0.0 kW' stehen, er lädt aber mit 28-30Watt permanent.
minCurrent ist ja 0.125 oder größer. Geht es nicht, dass er gar nicht lädt, wenn er 'Off' ist?

andig, 2025-01-08:
Schau bitte ins Logfile. Für "aus" wird Register 1000 auf 0 gesetzt. Der Rest ist vmtl. Eigenverbrauch des Gerätes.

Lepidopteron, 2025-01-08:
> Schau bitte ins Logfile. Für "aus" wird Register 1000 auf 0 gesetzt. Der Rest ist vmtl. Eigenverbrauch des Gerätes.

Habe folgendes in den Logs:
```
[lp-3 ] DEBUG 2025/01/08 23:00:41 pv charge current: 0A = 0A + -3.39A (779W @ 1p)
[lp-3 ] DEBUG 2025/01/08 23:00:41 projected site power 779W >= 200W disable threshold
[lp-3 ] DEBUG 2025/01/08 23:00:41 pv disable timer start: 1m0s
[lp-3 ] DEBUG 2025/01/08 23:00:41 pv disable in 1m0s
[lp-3 ] DEBUG 2025/01/08 23:00:41 max charge current: 0.125A
[lp-3 ] DEBUG 2025/01/08 23:00:50 charge power: 28W
[lp-3 ] DEBUG 2025/01/08 23:01:00 charge power: 29W
[lp-3 ] DEBUG 2025/01/08 23:01:10 charge power: 30W
[lp-3 ] DEBUG 2025/01/08 23:01:11 charger status: B
[lp-3 ] DEBUG 2025/01/08 23:01:11 pv charge current: 0A = 0A + -3.36A (772W @ 1p)
[lp-3 ] DEBUG 2025/01/08 23:01:11 projected site power 772W >= 200W disable threshold
[lp-3 ] DEBUG 2025/01/08 23:01:11 pv disable in 30s
[lp-3 ] DEBUG 2025/01/08 23:01:11 pv disable timer remaining: 30s
[lp-3 ] DEBUG 2025/01/08 23:01:20 charge power: 30W
[lp-3 ] DEBUG 2025/01/08 23:01:30 charge power: 29W
[lp-3 ] DEBUG 2025/01/08 23:01:40 charge power: 29W
[lp-3 ] DEBUG 2025/01/08 23:01:42 charger status: B
[lp-3 ] DEBUG 2025/01/08 23:01:42 pv charge current: 0A = 0A + -3.35A (770W @ 1p)
[lp-3 ] DEBUG 2025/01/08 23:01:42 projected site power 770W >= 200W disable threshold
[lp-3 ] DEBUG 2025/01/08 23:01:42 pv disable in 0s
[lp-3 ] DEBUG 2025/01/08 23:01:42 pv disable timer elapsed
[lp-3 ] DEBUG 2025/01/08 23:01:42 charger disable
[lp-3 ] DEBUG 2025/01/08 23:01:42 wake-up timer: stop
[lp-3 ] DEBUG 2025/01/08 23:01:50 charge power: 29W
```

docolli, 2025-01-10:
Ich habe evcc auf 0.132.1 und jetzt meinen ACThor auch mal drüber eingerichtet.
Leider wird der ACThor nicht deaktiviert, wenn Mode=Off.
Ich habe ein Out of Sync und er geht immer wieder mal auf 1,2kW.
[image]
[image]
[image]

```yaml
[ac-thor] TRACE 2025/01/10 11:27:23 modbus: send 01 21 00 00 00 06 01 03 03 eb 00 01
[ac-thor] TRACE 2025/01/10 11:27:23 modbus: recv 01 21 00 00 00 05 01 03 02 00 09
[lp-3 ] DEBUG 2025/01/10 11:27:23 charger status: B
[ac-thor] TRACE 2025/01/10 11:27:23 modbus: send 01 22 00 00 00 06 01 03 03 e9 00 01
[ac-thor] TRACE 2025/01/10 11:27:23 modbus: recv 01 22 00 00 00 05 01 03 02 01 f8
[ac-thor] TRACE 2025/01/10 11:27:23 modbus: send 01 23 00 00 00 06 01 03 03 e8 00 01
[ac-thor] TRACE 2025/01/10 11:27:23 modbus: recv 01 23 00 00 00 05 01 03 02 04 e2
[lp-3 ] WARN 2025/01/10 11:27:23 charger out of sync: expected disabled, got enabled
[lp-3 ] DEBUG 2025/01/10 11:27:23 charger enable
[ac-thor] TRACE 2025/01/10 11:27:23 modbus: send 01 24 00 00 00 09 01 10 03 e8 00 01 02 00 00
[ac-thor] TRACE 2025/01/10 11:27:23 modbus: recv 01 24 00 00 00 06 01 10 03 e8 00 01
[lp-3 ] DEBUG 2025/01/10 11:27:23 charger disable
[lp-3 ] DEBUG 2025/01/10 11:27:23 wake-up timer: stop
[lp-3 ] DEBUG 2025/01/10 11:27:33 charge power: 5W
[lp-3 ] DEBUG 2025/01/10 11:27:43 charge power: 5W
[ac-thor] TRACE 2025/01/10 11:27:43 modbus: send 01 25 00 00 00 06 01 03 03 e8 00 01
[ac-thor] TRACE 2025/01/10 11:27:43 modbus: recv 01 25 00 00 00 05 01 03 02 00 05
[ac-thor] TRACE 2025/01/10 11:27:43 modbus: send 01 26 00 00 00 09 01 10 03 e8 00 01 02 05 27
[ac-thor] TRACE 2025/01/10 11:27:43 modbus: recv 01 26 00 00 00 06 01 10 03 e8 00 01
[lp-3 ] DEBUG 2025/01/10 11:27:53 charge power: 1249W
[ac-thor] TRACE 2025/01/10 11:27:53 modbus: send 01 27 00 00 00 06 01 03 03 eb 00 01
[ac-thor] TRACE 2025/01/10 11:27:53 modbus: recv 01 27 00 00 00 05 01 03 02 00 09
[lp-3 ] DEBUG 2025/01/10 11:27:53 charger status: B
[ac-thor] TRACE 2025/01/10 11:27:53 modbus: send 01 28 00 00 00 06 01 03 03 e9 00 01
[ac-thor] TRACE 2025/01/10 11:27:53 modbus: recv 01 28 00 00 00 05 01 03 02 01 f7
[ac-thor] TRACE 2025/01/10 11:27:53 modbus: send 01 29 00 00 00 06 01 03 03 e8 00 01
[ac-thor] TRACE 2025/01/10 11:27:53 modbus: recv 01 29 00 00 00 05 01 03 02 04 e1
[lp-3 ] DEBUG 2025/01/10 11:28:03 charge power: 1246W
[lp-3 ] DEBUG 2025/01/10 11:28:13 charge power: 1255W
[ac-thor] TRACE 2025/01/10 11:28:13 modbus: send 01 2a 00 00 00 06 01 03 03 e8 00 01
[ac-thor] TRACE 2025/01/10 11:28:13 modbus: recv 01 2a 00 00 00 05 01 03 02 04 df
[ac-thor] TRACE 2025/01/10 11:28:13 modbus: send 01 2b 00 00 00 09 01 10 03 e8 00 01 02 05 27
[ac-thor] TRACE 2025/01/10 11:28:13 modbus: recv 01 2b 00 00 00 06 01 10 03 e8 00 01
[lp-3 ] DEBUG 2025/01/10 11:28:23 charge power: 1246W
[ac-thor] TRACE 2025/01/10 11:28:23 modbus: send 01 2c 00 00 00 06 01 03 03 eb 00 01
[ac-thor] TRACE 2025/01/10 11:28:23 modbus: recv 01 2c 00 00 00 05 01 03 02 00 09
[lp-3 ] DEBUG 2025/01/10 11:28:23 charger status: B
[ac-thor] TRACE 2025/01/10 11:28:23 modbus: send 01 2d 00 00 00 06 01 03 03 e9 00 01
[ac-thor] TRACE 2025/01/10 11:28:23 modbus: recv 01 2d 00 00 00 05 01 03 02 01 f7
[ac-thor] TRACE 2025/01/10 11:28:23 modbus: send 01 2e 00 00 00 06 01 03 03 e8 00 01
[ac-thor] TRACE 2025/01/10 11:28:23 modbus: recv 01 2e 00 00 00 05 01 03 02 04 de
[lp-3 ] WARN 2025/01/10 11:28:23 charger out of sync: expected disabled, got enabled
[lp-3 ] DEBUG 2025/01/10 11:28:23 charger enable 

```

andig, 2025-01-10:
Hättest Du nochmal mehr Log? Ich frage mich schon, woher das

> charger out of sync: expected disabled, got enabled

kommt?

docolli, 2025-01-10:
Ich melde mich heute Abend nochmals.

Eventuell hilft aber das hier generell weiter: 
https://www.offgridtec.com/media/product_attachements/my-PV_AC-ELWA-2_Documentation-of-controls.pdf

Ich habe übrigens den ACThor 9s. Und ein E3DC System, das den ACThor automatisch gefunden (und konfiguriert ?) hat und ich kann den dort nicht entfernen. Aber der ACThor sollte nur noch vom evcc System Modbus Befehle entgegen nehmen, das ist auf 192.168.1.50.

andig, 2025-01-10:
Es kann nur einer steuern!

docolli, 2025-01-10:
So, jetzt hier mal mehr Log. Wenn du eine spezifische Nachricht suchst, sag Bescheid, dann schau ich nomals ins Log.

Schau ich den ACTHOR 9s per Webinterface an (Update alle 10s), so ändert sich der Status ständig, wenn evcc läuft.
Ist evcc beendet, bleibt der Status stabil auf 4 (keine Verbindung):
[image]

Frage ich die Register 1000 bis 1009 direkt per Modbus ab
[image]
sehe ich, dass 1003 (status) bei mir den Wert 7 zurück gibt, auch wenn im Webinterface im Status "0, standby" steht.

Setze ich eine Leistung (100W) in Register 1000, so ändert sich der Status auf den Wert 9
[image]
Im Webinterface staht dann "1, Heizen".

Das Heizen hört aber nach ~60s auf, der Wert von Register 1000 fällt erst auf 5 und dann nach einigen weiteren Sekunden auf 0 ab. Muss die Heizleistung immer wieder zyklisch in Register 1000 geschrieben werden? 
Ist der Timeout dazu der Wert hier?
[image]
Dann muss die Zeit in der evcc den Wert neu beschreibt und die im ACTHOR eingetragene Zeit zueinander passen, also im ACTHOR sollte ein größerer Wert stehen, als evcc braucht einen neuen Wert zu schreiben.

```yaml
[lp-3 ] DEBUG 2025/01/10 15:08:03 charger enable
[ac-thor] TRACE 2025/01/10 15:08:03 modbus: send 0a 8c 00 00 00 09 01 10 03 e8 00 01 02 00 00
[ac-thor] TRACE 2025/01/10 15:08:03 modbus: recv 0a 8c 00 00 00 06 01 10 03 e8 00 01
[lp-3 ] DEBUG 2025/01/10 15:08:03 charger disable
[lp-3 ] DEBUG 2025/01/10 15:08:03 wake-up timer: stop
[lp-3 ] DEBUG 2025/01/10 15:08:13 charge power: 5W
[ac-thor] TRACE 2025/01/10 15:08:13 modbus: send 0a 8d 00 00 00 06 01 03 03 e8 00 01
[ac-thor] TRACE 2025/01/10 15:08:13 modbus: recv 0a 8d 00 00 00 05 01 03 02 00 05
[ac-thor] TRACE 2025/01/10 15:08:13 modbus: send 0a 8e 00 00 00 09 01 10 03 e8 00 01 02 00 1c
[ac-thor] TRACE 2025/01/10 15:08:13 modbus: recv 0a 8e 00 00 00 06 01 10 03 e8 00 01
[lp-3 ] DEBUG 2025/01/10 15:08:23 charge power: 31W
[lp-3 ] DEBUG 2025/01/10 15:08:33 charge power: 5W
[ac-thor] TRACE 2025/01/10 15:08:33 modbus: send 0a 8f 00 00 00 06 01 03 03 eb 00 01
[ac-thor] TRACE 2025/01/10 15:08:33 modbus: recv 0a 8f 00 00 00 05 01 03 02 00 09
[lp-3 ] DEBUG 2025/01/10 15:08:33 charger status: B
[ac-thor] TRACE 2025/01/10 15:08:33 modbus: send 0a 90 00 00 00 06 01 03 03 e9 00 01
[ac-thor] TRACE 2025/01/10 15:08:33 modbus: recv 0a 90 00 00 00 05 01 03 02 02 0d
[ac-thor] TRACE 2025/01/10 15:08:33 modbus: send 0a 91 00 00 00 06 01 03 03 e8 00 01
[ac-thor] TRACE 2025/01/10 15:08:33 modbus: recv 0a 91 00 00 00 05 01 03 02 00 05
[lp-3 ] DEBUG 2025/01/10 15:08:43 charge power: 5W
[ac-thor] TRACE 2025/01/10 15:08:43 modbus: send 0a 92 00 00 00 06 01 03 03 e8 00 01
[ac-thor] TRACE 2025/01/10 15:08:43 modbus: recv 0a 92 00 00 00 05 01 03 02 00 05
[ac-thor] TRACE 2025/01/10 15:08:43 modbus: send 0a 93 00 00 00 09 01 10 03 e8 00 01 02 00 1c
[ac-thor] TRACE 2025/01/10 15:08:43 modbus: recv 0a 93 00 00 00 06 01 10 03 e8 00 01
[lp-3 ] DEBUG 2025/01/10 15:08:53 charge power: 36W
[lp-3 ] DEBUG 2025/01/10 15:09:03 charge power: 5W
[ac-thor] TRACE 2025/01/10 15:09:03 modbus: send 0a 94 00 00 00 06 01 03 03 eb 00 01
[ac-thor] TRACE 2025/01/10 15:09:03 modbus: recv 0a 94 00 00 00 05 01 03 02 00 09
[lp-3 ] DEBUG 2025/01/10 15:09:03 charger status: B
[ac-thor] TRACE 2025/01/10 15:09:03 modbus: send 0a 95 00 00 00 06 01 03 03 e9 00 01
[ac-thor] TRACE 2025/01/10 15:09:03 modbus: recv 0a 95 00 00 00 05 01 03 02 02 0d
[ac-thor] TRACE 2025/01/10 15:09:03 modbus: send 0a 96 00 00 00 06 01 03 03 e8 00 01
[ac-thor] TRACE 2025/01/10 15:09:03 modbus: recv 0a 96 00 00 00 05 01 03 02 00 05
[ac-thor] TRACE 2025/01/10 15:09:13 modbus: send 0a 97 00 00 00 06 01 03 03 e8 00 01
[ac-thor] TRACE 2025/01/10 15:09:13 modbus: recv 0a 97 00 00 00 05 01 03 02 00 05
[ac-thor] TRACE 2025/01/10 15:09:13 modbus: send 0a 98 00 00 00 09 01 10 03 e8 00 01 02 00 1c
[ac-thor] TRACE 2025/01/10 15:09:13 modbus: recv 0a 98 00 00 00 06 01 10 03 e8 00 01
[lp-3 ] DEBUG 2025/01/10 15:09:14 charge power: 5W
[lp-3 ] DEBUG 2025/01/10 15:09:23 charge power: 36W
[lp-3 ] DEBUG 2025/01/10 15:09:33 charge power: 5W
[ac-thor] TRACE 2025/01/10 15:09:33 modbus: send 0a 99 00 00 00 06 01 03 03 eb 00 01
[ac-thor] TRACE 2025/01/10 15:09:33 modbus: recv 0a 99 00 00 00 05 01 03 02 00 09
[lp-3 ] DEBUG 2025/01/10 15:09:33 charger status: B
[ac-thor] TRACE 2025/01/10 15:09:33 modbus: send 0a 9a 00 00 00 06 01 03 03 e9 00 01
[ac-thor] TRACE 2025/01/10 15:09:33 modbus: recv 0a 9a 00 00 00 05 01 03 02 02 0d
[ac-thor] TRACE 2025/01/10 15:09:33 modbus: send 0a 9b 00 00 00 06 01 03 03 e8 00 01
[ac-thor] TRACE 2025/01/10 15:09:33 modbus: recv 0a 9b 00 00 00 05 01 03 02 00 05
[lp-3 ] WARN 2025/01/10 15:09:33 charger out of sync: expected disabled, got enabled
[lp-3 ] DEBUG 2025/01/10 15:09:33 charger enable
[ac-thor] TRACE 2025/01/10 15:09:33 modbus: send 0a 9c 00 00 00 09 01 10 03 e8 00 01 02 00 00
[... 23 lines cut ...]
[lp-3 ] DEBUG 2025/01/10 15:10:33 charge power: 5W
[ac-thor] TRACE 2025/01/10 15:10:33 modbus: send 0a a4 00 00 00 06 01 03 03 eb 00 01
[ac-thor] TRACE 2025/01/10 15:10:33 modbus: recv 0a a4 00 00 00 05 01 03 02 00 09
[lp-3 ] DEBUG 2025/01/10 15:10:33 charger status: B
[ac-thor] TRACE 2025/01/10 15:10:33 modbus: send 0a a5 00 00 00 06 01 03 03 e9 00 01
[ac-thor] TRACE 2025/01/10 15:10:33 modbus: recv 0a a5 00 00 00 05 01 03 02 02 0d
[ac-thor] TRACE 2025/01/10 15:10:33 modbus: send 0a a6 00 00 00 06 01 03 03 e8 00 01
[ac-thor] TRACE 2025/01/10 15:10:33 modbus: recv 0a a6 00 00 00 05 01 03 02 00 05
[... cut ...]
[lp-3 ] DEBUG 2025/01/10 15:11:03 charger status: B
[ac-thor] TRACE 2025/01/10 15:11:03 modbus: send 0a aa 00 00 00 06 01 03 03 e9 00 01
[ac-thor] TRACE 2025/01/10 15:11:03 modbus: recv 0a aa 00 00 00 05 01 03 02 02 0e
[ac-thor] TRACE 2025/01/10 15:11:03 modbus: send 0a ab 00 00 00 06 01 03 03 e8 00 01
[ac-thor] TRACE 2025/01/10 15:11:03 modbus: recv 0a ab 00 00 00 05 01 03 02 00 05
[lp-3 ] WARN 2025/01/10 15:11:03 charger out of sync: expected disabled, got enabled
[lp-3 ] DEBUG 2025/01/10 15:11:03 charger enable 
```

andig, 2025-01-10:
> sehe ich, dass 1003 (status) bei mir den Wert 7 zurück gibt, auch wenn im Webinterface im Status "0, standby" steht.
> Setze ich eine Leistung (100W) in Register 1000, so ändert sich der Status auf den Wert 9

evcc behandelt Status 2 als "laden". Sonst verbunden. Scheint als würde Thor ein anderes Mapping als die Elwa erfordern? Welches?

> Edit: Ist es möglich, dass der Code in charger/mypv.go den Status des ACTHOR 9s nicht korrekt erkennt? Dort wird auf 2 geprüft (soweit ich den go-code verstehe) hier kommt aber 7 bzw. 9 zurück.

Gut möglich.

> Das Heizen hört aber nach ~60s auf, der Wert von Register 1000 fällt erst auf 5 und dann nach einigen weiteren Sekunden auf 0 ab. Muss die Heizleistung immer wieder zyklisch in Register 1000 geschrieben werden?

Register 1000 wird alle 30s geschrieben solange der Ladepunkt enabled ist.

docolli, 2025-01-10:
Bin dran...

Habe in Register 1000 eine Last von 8000W gesetzt und dann die Register 1000 bis 1079 ausgelesen. Der ACTHOR 9s hat 3 Ausgänge á 3000W. Das von Dir aktuell genutzte Register 1074 gibt nur den Wert für einen dieser Ausgänge zurück!

[image]
[image]

Register 1000 gibt nach dem Schreiben lesend die Gesamtleistung zurück. Solange bis der Wert in "Zeitablauf Ansteuerung" abgelaufen ist, dann fällt die Leistung auf 0W ab. Standardmäßig sind das 10s, dieser Wert muss mindestens auf die 30s gesetzt werden, in der evcc die Werte neu schreibt. Am besten 60s nehmen, dann ist man auf der sicheren Seite.

Die Register 1074/1075/1076 geben die Leistungswerte der **einzelnen** Ausgänge (Phasen) zurück. Also entweder alle 3 auslesen, oder warum nicht zyklisch Register 1000 lesen und die aktuelle Gesamtleistung bekommen? Das sollte, wenn ich die Doku richtig verstehe, mit allen MyPV Komponenten funktionieren, egal ob 1p oder 3p Geräte.

Hast du schon Zeit gehabt dir dieses Dokument anzusehen? Dort ist eine gute Doku der Modbus Register der MyPV Serie.
https://www.offgridtec.com/media/product_attachements/my-PV_AC-ELWA-2_Documentation-of-controls.pdf

andig, 2025-01-11:
Die Elwa Doku nutzt nichts- der funktioniert ja. 

> Und den Wert von Register 1060 kann ich auch nicht richtig mit der von mir gesetzten Leistung korrellieren

Das war das Register das mir genannt wurde. Ansonsten sieht du in mypv.go ja welche Register verwendet werden. Einfach anpassen bis es passt. Ich kann hier mangels Testobjekt nur nach Vorgabe arbeiten.

walburgf, 2025-01-11:
Hier ist die Liste der Modbus Register, die der AC Thor und AC Thor 9s nutzt.
[Documentation-Controls_AC-THOR_AC-THOR 9s_EN241205.pdf](https://github.com/user-attachments/files/18386043/Documentation-Controls_AC-THOR_AC-THOR.9s_EN241205.pdf)

Beim Status haben die Thors folgende Bedeutungen:
        0….. Off
        1-8… device start-up
        9… operation
        >=200 Error states power stage

Weiterhin kann man beim AC Thor 9s 3 Heizstäbe anschliessen mit 3 Temperaturfühlern.
Die Temperaturfühler haben die Modbus Register 1001, 1030 und 1031
Da evcc momentan nur 1 Temperatur (= SOC) handeln kann nutze ich bisher eine manuelle Charger COnfig wo ich aus den 3 Temperaturen eine Median Temperatur ermittle, die ich aus dem Json vom Thor errechne. Das kann man natürlich auch per Modbus auslesen. 
Dazu verwende ich folgende Formel:
   jq: "[.temp1, .temp2, .temp3] | map(. | select (. != null)) | add / length" # parse response json, get the median temperature
Das wäre ein Vorschlag meinerseits... tbd.
Man kann natürlich über evcc auch alle 3 Temperaturen als 3 SOCs darstellen, die jeweils mit einem Limit versehen werden könnten, aber ich denke das ist erstens Aufwand und zweitens bringt das auch nicht wirklich viel.....

andig, 2025-01-11:
Danke! Danach is 1060 die aktuelle Leistung, 1000 scheint weiterhin die VORGABEleistung zu sein.

> Die Temperaturfühler haben die Modbus Register 1001, 1030 und 1031

Kann man ermitteln, welche der 3 jeweils relevant sind? Andererseits: auch der Mittelwert ist ja eher falsch. Theoretisch könnten das sogar 3 einzelne Speicher sein???

Status: bei der Elwa ist 2==Heat. Was bedeutet die 9==Operation beim Thor? Operation = Heizbetrieb?

walburgf, 2025-01-11:
bei den Temp-Fühler sind die relevant, die eine Temperatur anzeigen, d.h. einen Wert > 0 haben.
Dass die Temp.-Fühler in anderen Speichern drin sind, ist unwahrscheinlich. Das sollte man angehen, wenn es wirklich der Fall sein sollte.....
Deshalb habe ich auch die Formel oben so gemacht:
jq: "[.temp1, .temp2, .temp3] | map(. | select (. != null)) | add / length" 
    1. Ermittle die Werte alle 3 Fühler
    2. selektiere alle, die einen Wert != null bzw. >0 haben
    3. addiere die Temperaturen
    4. dividiere die Summe durch die Anzahl der "relevanten" Fühler

9 bedeutet, dass der AC Thor heizt... Da in der Doku 9... steht, würde ich auf 9-199 prüfen, da >=200 ja Error bedeutet

walburgf, 2025-01-11:
beim Lesen der Leistung solltest Du wieder das Register 1000 verwenden. Register 1000 ist ein R/W Register.
Wenn Du z.B: 200W in das Register 1000 schreibst und es danach wieder ausliest, kann es sein, dass 197W zurückgegeben werden. Dies ist abhängig von der internen Regelung des AC Thor.
Das Register 1000 ist von MyPV vorgesehen für das Steuern, sowie das Auslesen der Leistung....

walburgf, 2025-01-11:
ich habe gerade nochmal geprüft:
die aufgenommene Leistung des AC THor 9s ist die Summe der Register 1074 + 1075 + 1076

andig, 2025-01-11:
> Im Register 1060 steht kein "vernünftiger" Wert drin....

Das wäre dann aber nicht laut Doku und auch abweichend zur Elwa wo 1074 richtig zu sein scheint?

walburgf, 2025-01-11:
Du hast Recht.... P(gesamt) = 1074 + 1075 + 1076

docolli, 2025-01-11:
Und was bekommst du, wenn die 1074+1075+1076 rechnest? Vermutlich auch nur 197W. 😉 Das ist die aktuelle Gesamtleistung!
Siehe mein Bild der Modbus Register oben.
Ich würde vorschlagen Register 1000 zu lesen, die Werte der einzelnen 3 Heizstäbe intersssieren hier nicht.

docolli, 2025-01-11:
Hab mit dem Register 1060 etwas gespielt, um dahinter zu kommen, was das bedeutet. Der Wert ist auch im data.json vorhanden und heißt dort `load_nom`. Das ist der Nomialwert der angeschlossen Last. Den ermittelt der AC THOR irgendwie, WENN Strom fließt.

Ist also Register 1000 auf 0, dann ist auch 1060 auf dem Wert 0. Da ändert sich auch nichts, wenn man 100W oder 500W vorgibt, der Wert bleibt auf 0! Erst ab 750W erscheint dort ein Wert und der ist bei mir mit 2962 nahe an den 3000W meines Heizstabes.
Wobei ich eigentlich einen 3x3000W Heizstab habe.... Bei 5000W Vorgabe in Register 1000, stehen bei mir dort 2774.

docolli, 2025-01-11:
Brauchen wir wirklich die Unterscheidung AC ELWA und AC THOR im Code, wenn wir zum Auslesen der Leistung Register 1000 nehmen?

walburgf, 2025-01-11:
meines Erachtens kann man aus dem Register 1000 die Leistung auslesen, da es die Summe der Register 1074 bis 1076 ist.

walburgf, 2025-01-11:
der Unterschied zur Elwa ist aber die SOC Ermittlung beim AC Thor 9s, da dort 3 Temperaturfühler angeschlossen sein können... Siehe mein [Kommentar ](https://github.com/evcc-io/evcc/issues/18020#issuecomment-2585282662) oben

andig, 2025-01-11:
Gerne mal probieren. Danke!

docolli, 2025-01-11:
> der Unterschied zur Elwa ist aber die SOC Ermittlung beim AC Thor 9s, da dort 3 Temperaturfühler angeschlossen sein können... Siehe mein [Kommentar ](https://github.com/evcc-io/evcc/issues/18020#issuecomment-2585282662) oben

Ja, dafür macht das Sinn! Ich habe nur 1 Fühler angeschlossen und der kommt auf 1001.

Den Fix teste ich nachher gerne! Vielen Dank dafür!!!

docolli, 2025-01-11:
Heizstab wird erkannt, auch der Status und steht im Modus PV aktuell korrekt bei 0W (Sonne scheint nicht mehr).
[image]

```yaml
[lp-3 ] INFO 2025/01/11 17:42:43 loadpoint 3:
[lp-3 ] INFO 2025/01/11 17:42:43 mode: pv
[lp-3 ] INFO 2025/01/11 17:42:43 charger: power ✓ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-3 ] INFO 2025/01/11 17:42:43 meters: charge ✓
[lp-3 ] INFO 2025/01/11 17:42:43 charge: power ✓ energy ✗ currents ✗
[lp-3 ] DEBUG 2025/01/11 17:42:43 phase timer inactive
[lp-3 ] DEBUG 2025/01/11 17:42:43 pv timer inactive
[ac-thor] TRACE 2025/01/11 17:42:43 modbus: send 00 01 00 00 00 06 01 03 03 e8 00 01
[ac-thor] TRACE 2025/01/11 17:42:43 modbus: recv 00 01 00 00 00 05 01 03 02 00 00
[lp-3 ] DEBUG 2025/01/11 17:42:43 charge power: 0W
[lp-3 ] DEBUG 2025/01/11 17:42:56 charge power: 0W
[lp-3 ] DEBUG 2025/01/11 17:43:06 charge power: 0W
[ac-thor] TRACE 2025/01/11 17:43:06 modbus: send 00 02 00 00 00 06 01 03 03 eb 00 01
[ac-thor] TRACE 2025/01/11 17:43:06 modbus: recv 00 02 00 00 00 05 01 03 02 00 07
[lp-3 ] DEBUG 2025/01/11 17:43:06 charger status: B
[lp-3 ] INFO 2025/01/11 17:43:06 car connected
[lp-3 ] DEBUG 2025/01/11 17:43:06 pv timer elapse
[lp-3 ] DEBUG 2025/01/11 17:43:06 pv timer inactive
[ac-thor] TRACE 2025/01/11 17:43:06 modbus: send 00 03 00 00 00 06 01 03 03 e9 00 01
[ac-thor] TRACE 2025/01/11 17:43:06 modbus: recv 00 03 00 00 00 05 01 03 02 02 04
[ac-thor] TRACE 2025/01/11 17:43:06 modbus: send 00 04 00 00 00 06 01 03 03 e8 00 01
[ac-thor] TRACE 2025/01/11 17:43:06 modbus: recv 00 04 00 00 00 05 01 03 02 00 00
[lp-3 ] DEBUG 2025/01/11 17:43:06 pv charge current: 0A = 0A + -3.82A (2638W @ 3p)
[lp-3 ] DEBUG 2025/01/11 17:43:06 pv enable timer reset
[lp-3 ] DEBUG 2025/01/11 17:43:06 pv timer inactive
[lp-3 ] DEBUG 2025/01/11 17:43:16 charge power: 0W
[lp-3 ] DEBUG 2025/01/11 17:43:37 charge power: 0W
[lp-3 ] DEBUG 2025/01/11 17:43:47 charge power: 0W
[ac-thor] TRACE 2025/01/11 17:43:47 modbus: send 00 05 00 00 00 06 01 03 03 eb 00 01
[ac-thor] TRACE 2025/01/11 17:43:47 modbus: recv 00 05 00 00 00 05 01 03 02 00 07
[lp-3 ] DEBUG 2025/01/11 17:43:47 charger status: B
[ac-thor] TRACE 2025/01/11 17:43:47 modbus: send 00 06 00 00 00 06 01 03 03 e9 00 01
[ac-thor] TRACE 2025/01/11 17:43:47 modbus: recv 00 06 00 00 00 05 01 03 02 02 04
[ac-thor] TRACE 2025/01/11 17:43:47 modbus: send 00 07 00 00 00 06 01 03 03 e8 00 01
[ac-thor] TRACE 2025/01/11 17:43:47 modbus: recv 00 07 00 00 00 05 01 03 02 00 00
[lp-3 ] DEBUG 2025/01/11 17:43:47 pv charge current: 0A = 0A + -4.11A (2834W @ 3p)
[lp-3 ] DEBUG 2025/01/11 17:43:56 charge power: 0W
[lp-3 ] DEBUG 2025/01/11 17:44:06 charge power: 0W
[lp-3 ] DEBUG 2025/01/11 17:44:16 charge power: 0W
[ac-thor] TRACE 2025/01/11 17:44:16 modbus: send 00 08 00 00 00 06 01 03 03 eb 00 01
[ac-thor] TRACE 2025/01/11 17:44:16 modbus: recv 00 08 00 00 00 05 01 03 02 00 07
[lp-3 ] DEBUG 2025/01/11 17:44:16 charger status: B
[ac-thor] TRACE 2025/01/11 17:44:16 modbus: send 00 09 00 00 00 06 01 03 03 e9 00 01
[ac-thor] TRACE 2025/01/11 17:44:16 modbus: recv 00 09 00 00 00 05 01 03 02 02 04
[ac-thor] TRACE 2025/01/11 17:44:16 modbus: send 00 0a 00 00 00 06 01 03 03 e8 00 01
[ac-thor] TRACE 2025/01/11 17:44:16 modbus: recv 00 0a 00 00 00 05 01 03 02 00 00
[lp-3 ] DEBUG 2025/01/11 17:44:16 pv charge current: 0A = 0A + -4.16A (2869W @ 3p)
[lp-3 ] DEBUG 2025/01/11 17:44:26 charge power: 0W
[lp-3 ] DEBUG 2025/01/11 17:44:36 charge power: 0W
[lp-3 ] DEBUG 2025/01/11 17:44:46 charge power: 0W
[ac-thor] TRACE 2025/01/11 17:44:46 modbus: send 00 0b 00 00 00 06 01 03 03 eb 00 01
[ac-thor] TRACE 2025/01/11 17:44:46 modbus: recv 00 0b 00 00 00 05 01 03 02 00 07
[lp-3 ] DEBUG 2025/01/11 17:44:46 charger status: B
[ac-thor] TRACE 2025/01/11 17:44:46 modbus: send 00 0c 00 00 00 06 01 03 03 e9 00 01
[ac-thor] TRACE 2025/01/11 17:44:46 modbus: recv 00 0c 00 00 00 05 01 03 02 02 04
[ac-thor] TRACE 2025/01/11 17:44:46 modbus: send 00 0d 00 00 00 06 01 03 03 e8 00 01
[ac-thor] TRACE 2025/01/11 17:44:46 modbus: recv 00 0d 00 00 00 05 01 03 02 00 00
[lp-3 ] DEBUG 2025/01/11 17:44:46 pv charge current: 0A = 0A + -4.16A (2867W @ 3p)
[lp-3 ] DEBUG 2025/01/11 17:44:56 charge power: 0W
[lp-3 ] DEBUG 2025/01/11 17:45:06 charge power: 0W 
```

[image]
Leistung min 0.125A. 

Was aber nicht richtig funktioniert ist der Wechsel von Schnell auf Off. Hier läuft es kurze Zeit mit 5W und dann geht die Leistung wieder hoch auf das eingestellte Maximum (=26A).
[... 265 lines cut ...]
[lp-3 ] DEBUG 2025/01/11 17:52:16 wake-up timer: stop
[lp-3 ] DEBUG 2025/01/11 17:52:16 charge power: 5629W
[ac-thor] TRACE 2025/01/11 17:52:16 modbus: send 00 64 00 00 00 06 01 03 03 eb 00 01
[ac-thor] TRACE 2025/01/11 17:52:16 modbus: recv 00 64 00 00 00 05 01 03 02 00 09
[lp-3 ] DEBUG 2025/01/11 17:52:16 charger status: C
[ac-thor] TRACE 2025/01/11 17:52:16 modbus: send 00 65 00 00 00 06 01 03 03 e9 00 01
[ac-thor] TRACE 2025/01/11 17:52:16 modbus: recv 00 65 00 00 00 05 01 03 02 01 f7
[ac-thor] TRACE 2025/01/11 17:52:16 modbus: send 00 66 00 00 00 06 01 03 03 e8 00 01
[ac-thor] TRACE 2025/01/11 17:52:16 modbus: recv 00 66 00 00 00 05 01 03 02 15 fd
[lp-3 ] DEBUG 2025/01/11 17:52:26 charge power: 5W
[ac-thor] TRACE 2025/01/11 17:52:35 modbus: send 00 67 00 00 00 06 01 03 03 e8 00 01
[ac-thor] TRACE 2025/01/11 17:52:35 modbus: recv 00 67 00 00 00 05 01 03 02 00 05
[ac-thor] TRACE 2025/01/11 17:52:35 modbus: send 00 68 00 00 00 09 01 10 03 e8 00 01 02 17 5c
[... cut ...]
[ac-thor] TRACE 2025/01/11 17:52:46 modbus: recv 00 69 00 00 00 05 01 03 02 00 09
[lp-3 ] DEBUG 2025/01/11 17:52:46 charger status: C
[ac-thor] TRACE 2025/01/11 17:52:46 modbus: send 00 6a 00 00 00 06 01 03 03 e9 00 01
[ac-thor] TRACE 2025/01/11 17:52:46 modbus: recv 00 6a 00 00 00 05 01 03 02 01 f8
[ac-thor] TRACE 2025/01/11 17:52:46 modbus: send 00 6b 00 00 00 06 01 03 03 e8 00 01
[ac-thor] TRACE 2025/01/11 17:52:46 modbus: recv 00 6b 00 00 00 05 01 03 02 15 ff
[lp-3 ] DEBUG 2025/01/11 17:52:56 charge power: 5618W 
```

docolli, 2025-01-11:
Noch eine Info zum Verhalten von Register 1003 beim Setzen von "0" in Register 1000, also Abschalten des Heizstabs (erster Befehl im Log).
Hier das Log vom Modbus Master, ich lese die Register 1000 bis 1003 in einem Rutsch.
Man sieht, dass die `03 c3` (=963W) rasch auf `00 05` (=5W) abfallen, aber der Status bleibt lange auf `00 09` (ganz hinten).
Dann geht der Status für ein paar Sekunden auf `00 ca` bis er endlich auf `00 07` abfällt.

```sh
>18:16:31: TX: 01 03 00 00 00 06 ff 06 03 e8 00 00 
>18:16:31: RX: 01 03 00 00 00 06 ff 06 03 e8 00 00 
>18:16:31: Write succeeded: Function code:6
>18:16:31: TX: 01 04 00 00 00 06 ff 03 03 e8 00 04 
>18:16:31: RX: 01 04 00 00 00 0b ff 03 08 03 c3 02 13 02 58 00 09 
>18:16:31: Read succeeded: Function code:3.
>18:16:33: TX: 01 05 00 00 00 06 ff 03 03 e8 00 04 
>18:16:33: RX: 01 05 00 00 00 0b ff 03 08 00 05 02 13 02 58 00 09 
>18:16:33: Read succeeded: Function code:3.
>18:16:34: TX: 01 06 00 00 00 06 ff 03 03 e8 00 04 
>18:16:34: RX: 01 06 00 00 00 0b ff 03 08 00 05 02 13 02 58 00 09 
>18:16:34: Read succeeded: Function code:3.
>18:16:35: TX: 01 07 00 00 00 06 ff 03 03 e8 00 04 
>18:16:35: RX: 01 07 00 00 00 0b ff 03 08 00 05 02 13 02 58 00 09 
>18:16:35: Read succeeded: Function code:3.
>18:16:36: TX: 01 08 00 00 00 06 ff 03 03 e8 00 04 
>18:16:36: RX: 01 08 00 00 00 0b ff 03 08 00 05 02 13 02 58 00 09 
>18:16:36: Read succeeded: Function code:3.
>18:16:37: TX: 01 09 00 00 00 06 ff 03 03 e8 00 04 
>18:16:37: RX: 01 09 00 00 00 0b ff 03 08 00 05 02 13 02 58 00 09 
>18:16:37: Read succeeded: Function code:3.
>18:16:38: TX: 01 0a 00 00 00 06 ff 03 03 e8 00 04 
>18:16:38: RX: 01 0a 00 00 00 0b ff 03 08 00 04 02 13 02 58 00 09 
>18:16:38: Read succeeded: Function code:3.
>18:16:39: TX: 01 0b 00 00 00 06 ff 03 03 e8 00 04 
>18:16:39: RX: 01 0b 00 00 00 0b ff 03 08 00 04 02 13 02 58 00 09 
>18:16:39: Read succeeded: Function code:3.
>18:16:40: TX: 01 0c 00 00 00 06 ff 03 03 e8 00 04 
>18:16:40: RX: 01 0c 00 00 00 0b ff 03 08 00 06 02 13 02 58 00 09 
>18:16:40: Read succeeded: Function code:3.
>18:16:41: TX: 01 0d 00 00 00 06 ff 03 03 e8 00 04 
>18:16:41: RX: 01 0d 00 00 00 0b ff 03 08 00 05 02 13 02 58 00 09 
>18:16:41: Read succeeded: Function code:3.
>18:16:42: TX: 01 0e 00 00 00 06 ff 03 03 e8 00 04 
>18:16:42: RX: 01 0e 00 00 00 0b ff 03 08 00 05 02 13 02 58 00 09 
>18:16:42: Read succeeded: Function code:3.
>18:16:43: TX: 01 0f 00 00 00 06 ff 03 03 e8 00 04 
>18:16:43: RX: 01 0f 00 00 00 0b ff 03 08 00 05 02 13 02 58 00 09 
>18:16:43: Read succeeded: Function code:3.
>18:16:44: TX: 01 10 00 00 00 06 ff 03 03 e8 00 04 
>18:16:44: RX: 01 10 00 00 00 0b ff 03 08 00 05 02 13 02 58 00 09 
>18:16:44: Read succeeded: Function code:3.
>18:16:45: TX: 01 11 00 00 00 06 ff 03 03 e8 00 04 
>18:16:45: RX: 01 11 00 00 00 0b ff 03 08 00 05 02 13 02 58 00 09 
>18:16:45: Read succeeded: Function code:3.
>18:16:46: TX: 01 12 00 00 00 06 ff 03 03 e8 00 04 
>18:16:46: RX: 01 12 00 00 00 0b ff 03 08 00 05 02 13 02 58 00 09 
>18:16:46: Read succeeded: Function code:3.
>18:16:47: TX: 01 13 00 00 00 06 ff 03 03 e8 00 04 
>18:16:47: RX: 01 13 00 00 00 0b ff 03 08 00 05 02 13 02 58 00 09 
>18:16:47: Read succeeded: Function code:3.
>18:16:48: TX: 01 14 00 00 00 06 ff 03 03 e8 00 04 
>18:16:48: RX: 01 14 00 00 00 0b ff 03 08 00 05 02 13 02 58 00 09 
>18:16:48: Read succeeded: Function code:3.
>18:16:49: TX: 01 15 00 00 00 06 ff 03 03 e8 00 04 
>18:16:49: RX: 01 15 00 00 00 0b ff 03 08 00 05 02 13 02 58 00 09 
>18:16:49: Read succeeded: Function code:3.
>18:16:50: TX: 01 16 00 00 00 06 ff 03 03 e8 00 04 
>18:16:50: RX: 01 16 00 00 00 0b ff 03 08 00 04 02 13 02 58 00 09 
>18:16:50: Read succeeded: Function code:3.
>18:16:51: TX: 01 17 00 00 00 06 ff 03 03 e8 00 04 
>18:16:51: RX: 01 17 00 00 00 0b ff 03 08 00 05 02 13 02 58 00 09 
>18:16:51: Read succeeded: Function code:3.
>18:16:52: TX: 01 18 00 00 00 06 ff 03 03 e8 00 04 
[... 38 lines cut ...]
>18:17:05: TX: 01 25 00 00 00 06 ff 03 03 e8 00 04 
>18:17:05: RX: 01 25 00 00 00 0b ff 03 08 00 01 02 13 02 58 00 ca 
>18:17:05: Read succeeded: Function code:3.
>18:17:06: TX: 01 26 00 00 00 06 ff 03 03 e8 00 04 
>18:17:06: RX: 01 26 00 00 00 0b ff 03 08 00 01 02 13 02 58 00 ca 
>18:17:06: Read succeeded: Function code:3.
>18:17:07: TX: 01 27 00 00 00 06 ff 03 03 e8 00 04 
>18:17:07: RX: 01 27 00 00 00 0b ff 03 08 00 00 02 13 02 58 00 ca 
>18:17:07: Read succeeded: Function code:3.
>18:17:08: TX: 01 28 00 00 00 06 ff 03 03 e8 00 04 
>18:17:08: RX: 01 28 00 00 00 0b ff 03 08 00 00 02 13 02 58 00 ca 
>18:17:08: Read succeeded: Function code:3.
>18:17:09: TX: 01 29 00 00 00 06 ff 03 03 e8 00 04 
>18:17:09: RX: 01 29 00 00 00 0b ff 03 08 00 00 02 13 02 58 00 ca 
>18:17:09: Read succeeded: Function code:3.
>18:17:10: TX: 01 2a 00 00 00 06 ff 03 03 e8 00 04 
>18:17:10: RX: 01 2a 00 00 00 0b ff 03 08 00 00 02 13 02 58 00 07 
>18:17:10: Read succeeded: Function code:3.
>18:17:11: TX: 01 2b 00 00 00 06 ff 03 03 e8 00 04 
>18:17:11: RX: 01 2b 00 00 00 0b ff 03 08 00 00 02 13 02 58 00 07 
>18:17:11: Read succeeded: Function code:3.
>18:17:12: TX: 01 2c 00 00 00 06 ff 03 03 e8 00 04 
>18:17:12: RX: 01 2c 00 00 00 0b ff 03 08 00 00 02 13 02 58 00 07 
>18:17:12: Read succeeded: Function code:3.
```

Lepidopteron, 2025-01-11:
> Noch eine Info zum Verhalten von Register 1003 beim Setzen von "0" in Register 1000, also Abschalten des Heizstabs (erster Befehl im Log). evcc läuft hier NICHT, ich habe nur die Kontrolle über den AC THOR per Modbus Master. Hier das Log vom Modbus Master, ich lese die Register 1000 bis 1003 in einem Rutsch. Man sieht, dass die `03 c3` (=963W) rasch auf `00 05` (=5W) abfallen, aber der Status bleibt lange auf `00 09` (ganz hinten). Dann geht der Status für ein paar Sekunden auf `00 ca` bis er endlich auf `00 07` abfällt.
> 
> Hat evcc damit ein Problem, dass nach Abschalten (setzen von Leistung 0) es noch >30s dauert, bis der Status vom AC THOR endlich von 9 auf 7 geht?
> 
> ```shell
> >18:16:31: TX: 01 03 00 00 00 06 ff 06 03 e8 00 00 
> >18:16:31: RX: 01 03 00 00 00 06 ff 06 03 e8 00 00 
> >18:16:31: Write succeeded: Function code:6
> >18:16:31: TX: 01 04 00 00 00 06 ff 03 03 e8 00 04 
> >18:16:31: RX: 01 04 00 00 00 0b ff 03 08 03 c3 02 13 02 58 00 09 
> >18:16:31: Read succeeded: Function code:3.
> >18:16:33: TX: 01 05 00 00 00 06 ff 03 03 e8 00 04 
> >18:16:33: RX: 01 05 00 00 00 0b ff 03 08 00 05 02 13 02 58 00 09 
> >18:16:33: Read succeeded: Function code:3.
> >18:16:34: TX: 01 06 00 00 00 06 ff 03 03 e8 00 04 
> >18:16:34: RX: 01 06 00 00 00 0b ff 03 08 00 05 02 13 02 58 00 09 
> >18:16:34: Read succeeded: Function code:3.
> >18:16:35: TX: 01 07 00 00 00 06 ff 03 03 e8 00 04 
> >18:16:35: RX: 01 07 00 00 00 0b ff 03 08 00 05 02 13 02 58 00 09 
> >18:16:35: Read succeeded: Function code:3.
> >18:16:36: TX: 01 08 00 00 00 06 ff 03 03 e8 00 04 
> >18:16:36: RX: 01 08 00 00 00 0b ff 03 08 00 05 02 13 02 58 00 09 
> >18:16:36: Read succeeded: Function code:3.
> >18:16:37: TX: 01 09 00 00 00 06 ff 03 03 e8 00 04 
> >18:16:37: RX: 01 09 00 00 00 0b ff 03 08 00 05 02 13 02 58 00 09 
> >18:16:37: Read succeeded: Function code:3.
> >18:16:38: TX: 01 0a 00 00 00 06 ff 03 03 e8 00 04 
> >18:16:38: RX: 01 0a 00 00 00 0b ff 03 08 00 04 02 13 02 58 00 09 
> >18:16:38: Read succeeded: Function code:3.
> >18:16:39: TX: 01 0b 00 00 00 06 ff 03 03 e8 00 04 
> >18:16:39: RX: 01 0b 00 00 00 0b ff 03 08 00 04 02 13 02 58 00 09 
> >18:16:39: Read succeeded: Function code:3.
> >18:16:40: TX: 01 0c 00 00 00 06 ff 03 03 e8 00 04 
> >18:16:40: RX: 01 0c 00 00 00 0b ff 03 08 00 06 02 13 02 58 00 09 
> >18:16:40: Read succeeded: Function code:3.
> >18:16:41: TX: 01 0d 00 00 00 06 ff 03 03 e8 00 04 
> >18:16:41: RX: 01 0d 00 00 00 0b ff 03 08 00 05 02 13 02 58 00 09 
> >18:16:41: Read succeeded: Function code:3.
> >18:16:42: TX: 01 0e 00 00 00 06 ff 03 03 e8 00 04 
> >18:16:42: RX: 01 0e 00 00 00 0b ff 03 08 00 05 02 13 02 58 00 09 
> >18:16:42: Read succeeded: Function code:3.
> >18:16:43: TX: 01 0f 00 00 00 06 ff 03 03 e8 00 04 
> >18:16:43: RX: 01 0f 00 00 00 0b ff 03 08 00 05 02 13 02 58 00 09 
> >18:16:43: Read succeeded: Function code:3.
> >18:16:44: TX: 01 10 00 00 00 06 ff 03 03 e8 00 04 
> >18:16:44: RX: 01 10 00 00 00 0b ff 03 08 00 05 02 13 02 58 00 09 
> >18:16:44: Read succeeded: Function code:3.
> >18:16:45: TX: 01 11 00 00 00 06 ff 03 03 e8 00 04 
> >18:16:45: RX: 01 11 00 00 00 0b ff 03 08 00 05 02 13 02 58 00 09 
> >18:16:45: Read succeeded: Function code:3.
> >18:16:46: TX: 01 12 00 00 00 06 ff 03 03 e8 00 04 
> >18:16:46: RX: 01 12 00 00 00 0b ff 03 08 00 05 02 13 02 58 00 09 
> >18:16:46: Read succeeded: Function code:3.
> >18:16:47: TX: 01 13 00 00 00 06 ff 03 03 e8 00 04 
> >18:16:47: RX: 01 13 00 00 00 0b ff 03 08 00 05 02 13 02 58 00 09 
> >18:16:47: Read succeeded: Function code:3.
> >18:16:48: TX: 01 14 00 00 00 06 ff 03 03 e8 00 04 
> >18:16:48: RX: 01 14 00 00 00 0b ff 03 08 00 05 02 13 02 58 00 09 
> >18:16:48: Read succeeded: Function code:3.
> >18:16:49: TX: 01 15 00 00 00 06 ff 03 03 e8 00 04 
> >18:16:49: RX: 01 15 00 00 00 0b ff 03 08 00 05 02 13 02 58 00 09 
> >18:16:49: Read succeeded: Function code:3.
> >18:16:50: TX: 01 16 00 00 00 06 ff 03 03 e8 00 04 
> >18:16:50: RX: 01 16 00 00 00 0b ff 03 08 00 04 02 13 02 58 00 09 
> >18:16:50: Read succeeded: Function code:3.
> >18:16:51: TX: 01 17 00 00 00 06 ff 03 03 e8 00 04 
> >18:16:51: RX: 01 17 00 00 00 0b ff 03 08 00 05 02 13 02 58 00 09 
> >18:16:51: Read succeeded: Function code:3.
> >18:16:52: TX: 01 18 00 00 00 06 ff 03 03 e8 00 04 
> >18:16:52: RX: 01 18 00 00 00 0b ff 03 08 00 06 02 13 02 58 00 09 
[... 39 lines cut ...]
> >18:17:05: Read succeeded: Function code:3.
> >18:17:06: TX: 01 26 00 00 00 06 ff 03 03 e8 00 04 
> >18:17:06: RX: 01 26 00 00 00 0b ff 03 08 00 01 02 13 02 58 00 ca 
> >18:17:06: Read succeeded: Function code:3.
> >18:17:07: TX: 01 27 00 00 00 06 ff 03 03 e8 00 04 
> >18:17:07: RX: 01 27 00 00 00 0b ff 03 08 00 00 02 13 02 58 00 ca 
> >18:17:07: Read succeeded: Function code:3.
> >18:17:08: TX: 01 28 00 00 00 06 ff 03 03 e8 00 04 
> >18:17:08: RX: 01 28 00 00 00 0b ff 03 08 00 00 02 13 02 58 00 ca 
> >18:17:08: Read succeeded: Function code:3.
> >18:17:09: TX: 01 29 00 00 00 06 ff 03 03 e8 00 04 
> >18:17:09: RX: 01 29 00 00 00 0b ff 03 08 00 00 02 13 02 58 00 ca 
> >18:17:09: Read succeeded: Function code:3.
> >18:17:10: TX: 01 2a 00 00 00 06 ff 03 03 e8 00 04 
> >18:17:10: RX: 01 2a 00 00 00 0b ff 03 08 00 00 02 13 02 58 00 07 
> >18:17:10: Read succeeded: Function code:3.
> >18:17:11: TX: 01 2b 00 00 00 06 ff 03 03 e8 00 04 
> >18:17:11: RX: 01 2b 00 00 00 0b ff 03 08 00 00 02 13 02 58 00 07 
> >18:17:11: Read succeeded: Function code:3.
> >18:17:12: TX: 01 2c 00 00 00 06 ff 03 03 e8 00 04 
> >18:17:12: RX: 01 2c 00 00 00 0b ff 03 08 00 00 02 13 02 58 00 07 
> >18:17:12: Read succeeded: Function code:3.
> ```

Offtopic: Aber wie kann ich denn auf den Modbus einen Registerwert setzen und explizit auslesen? :)

walburgf, 2025-01-12:
irgendwie funktioniert das Laden mit dem AC Thor 9s noch nicht korrekt.

Ich habe mal einen Film gedreht, bei dem ich parallel die Anzeige im AC Thor 9s, die evcc Oberfläche und die Inhalte der Register (in Home Assistant) aufgezeichnet habe.

https://github.com/user-attachments/assets/1c11fece-7a89-4b41-aa93-343239747fda

Ich habe in evcc den AC Thor vom Status "Aus" auf "Schnell" geändert. Meine Erwartung war nun, dass das Laden mit voller Leistung durchgeführt wird.
Man sieht auch im MyPV Fenster, dass die volle Leistung am Heizstab ankommt. Der Status (Reg. 1003) geht auch auf 9.
Danach alterniert jedoch die Leistung am AC Thor immer zwischen Maximalleistung und 5W hin und her. Dies heisst für mich, dass die Steuerung durch evcc noch nicht richtig funktioniert.

Nach Beendigung des Videos habe ich in evcc den AC Thor wieder auf "Aus" gestellt. Nun hat das Laden jedoch nicht gestoppt, sondern hat weiterhin zwischen Max-Leistung und 5W alterniert. Ich musste den Heizstab in der MyPV Oberfläche deaktivieren. Erst dann hat evcc erkannt, dass der Heizstab aus ist. Anscheinend sendet evcc keine Leistungsvorgabe 0W an den AC Thor...

andig, 2025-01-12:
> Dies heisst für mich, dass die Steuerung durch evcc noch nicht richtig funktioniert.

Du siehst im trace log, welche Register geschrieben werden. 

Ich denke das Problem hier ist wiefolgt: der Thor (anscheinend entgegen der Elwa) zeigt auf 1000 die echte Leistung an- das inkludiert auch den Standbyverbrauch. Die 1000 nutzen wir aber auch um zu schauen ob der Thor "enabled" ist, dafür wird mit 0 verglichen. Wenn da aber keine 0 kommt obwohl evcc eine rein geschrieben hat wirds schwierig... Das kann man dann im Logfile nachlesen...

Long story short: an welchem Register kann man ablesen, ob der Thor eine Ladefreigabe hat? Bei der Elwa scheint 1000=0 zu kommen und damit ist klar "keine Ladefreigabe". Wo sehen wir das beim Thor?

walburgf, 2025-01-12:
was meinst Du mit Ladefreigabe?
Es soll immer geladen werden, solange "actual Soc" < "Soc Limit", d.h. die Temperatur im Speicher noch nicht die max. Temperatur erreicht hat. Das liest Du ja aus den Registern 1001 (Temp1) und 1002 (max Temp) aus.
im Modus PV soll dann der Überschuss in Register 1000 geschrieben werden, im Modus Schnell die durch maxcurrent definierte Leistung.
Beim AC Thor 9s, der 3 Temperaturfühler haben kann, sollte der "actual Soc" der Mittelwert der angeschlossenen Temperaturfühler sein.
Zusätzlich kann man im Status Register (1003) sehen, dass der AC Thor auf das Laden wartet (1 bis 8) oder heizt (9)

pvssch, 2025-01-12:
Die "Ladefreigabe" müsste beim AC Thor9s das Register 1081 sein

[Documentation-Controls_AC-THOR_AC-THOR 9s_EN231009.pdf](https://github.com/user-attachments/files/18390002/Documentation-Controls_AC-THOR_AC-THOR.9s_EN231009.pdf)

Die echte Leistung kommt von 1000

andig, 2025-01-12:
> müsste

Bitte testen. Ich hab keine Lust mehr, dass noch 40 mal in beliebige Richtungen zu ändern, tut mir leid...

walburgf, 2025-01-12:
hab's gerade getestet.
Device State (1081) = 1 bedeutet, dass der AC Thor 9s angeschaltet ist und Kommandos empfangen kann.
Device State (1081) = 0 bedeutet, dass der AC Thor 9s ausgeschaltet ist und keine Kommandos entgegennimmt.
In der Regel ist ein AC Thor immer eingeschaltet, da er ja darauf "wartet" Leistungsvorgaben zu erhalten, um die Heizstäbe dann zu steuern. Wenn Device state = 0, kann evcc den AC THor ignorieren.
Bei Device state = 1 sollte die Regelung so laufen wie [hier beschrieben](url)

docolli, 2025-01-12:
Bei mir ist das Register 1081 aber auch auf "1", wenn die Leistung im Register 1000 auf "0" ist. Das ist nicht das was wir hier suchen! Es geht um eine Rückmeldung vom AC Thor ob er gerade heizt oder nicht. Also analog zum Status einer Wallbox.

Meiner Meinung nach ist das Register 1003 die bessere Info zum Status. Leider ändert sich der Wert nach Schreiben von 0W ins Register 1000 erst nach >30s. Und auch der Wert von Register 1000 ist nach Schreiben von "0" erstmal eine ganze Weile bei 5W (vermutlich Standbyverbrauch) bis er nach >30s auf "0" abfällt.
evcc scheint aber nur 0W als "Status B" zu erkennen.
Könnte es helfen im Code für den AC Thor das Limit auf z.B. 10W hochzusetzen, also Verbrauch <10W ist Status B?

A= nicht verbunden
B= verbunden, nicht ladend
C= ladend

andig, 2025-01-12:
@docolli es geht um enable/disable, nicht um lädt/lädt nicht. Wenn Temperatur erreicht ist mag das also richtig sein mit 1081.

walburgf, 2025-01-12:
Da Heizstäbe analoge Systeme sind, dauert es immer etwas bis sich Werte stabilisieren....

StatusC: Register 1003 = 9
StatusB: Register 1003 = 1 bis 8

walburgf, 2025-01-12:
meiner Meinung nach ist der AC Thor9s ist immer enabled, wenn das Register 1081 auf 1 steht...

docolli, 2025-01-12:
@andig Was ist denn beim Heizstab der Unterschied zwischen enable/disable und heizt/heizt nicht?

Den AC THOR enablest du, indem du einen Wert >0 ins Register 1000 schreibst, dann heizt er. Register 1003 geht dann auf "9"
Zum disablen schreibst du "0" (Watt) ins Register 1000, dann hört er auf zu heizen, verharrt aber noch 30s auf im Register 1000 auf 5W und erst dann geht Register 1003 auf 7 zurück und Register 1000 auf 0.

andig, 2025-01-12:
Alles schon 1000mal erklärt. Ladefreigabe beim Fahrzeug ist unabhängig davon, ob der Akku voll ist und damit ob wirklich geladen wird. Ist hier auch nicht anders. Es sind einfach unterschiedliche Dinge.

docolli, 2025-01-12:
Mein Sensor hat aktuell ~52°C, ich habe die max. Heiztemp herunter gesetzt auf 50°C. Selbst dann bleibt Register 1081 auf "1".
ABER:
Wenn ich dann z.B. "2500"(W) ins Register 1000 schreibe, und gleich danach wieder auslese, dann bleibt es auf "0". Der Heizstab bleibt also wie erwartet aus (disabled).

walburgf, 2025-01-12:
Register 1081 hängt mit diesem Schalter des AC Thor zusamen:
Er aktiviert/deaktiviert den AC THor generell....

[image]

Ich denke die Steuerung des AC Thor sollte nur über die Temperaturen (1001, 1030, 1031) und das Register 1000 und 1003 gesteuert und überwacht werden

docolli, 2025-01-12:
> Wenns kein passwndes ein/aus Register gibt dann müssen wir uns den enable state halt merken.

Register 1003 geht sofort von 7 auf 9, wenn in Register 1000 ein Wert>0 geschrieben wird.
Aber nach Schreiben von 0 in Register 1000 dauert es bei mir (gibt es da noch was einstellbares im AC THOR Menu?) zwischen 30 und 50s bis der Wert von Register 1003 wieder von 9 auf 7 geht.

Ich habe im folgenden in den Modbus Nachrichten die relevanten Bytes für Leistung (1000) und Status (1003) markiert:
[image]

[image]

Er geht im Register 1003 sogar für 5s auf 202 (=Error power stages), dann auf 7

walburgf, 2025-01-12:
die Frage ist, ob es nicht einfacher ist, enable/disable beim AC Thor nicht zu verwenden und einfach enabled zu setzen, wenn das Register 1081 auf 1 steht. Der AC braucht nicht aktiviert zu werden, sondern kann über Register 1000 einfach gesteuert werden.
Der Status kann aus Register 1003 gelesen werden.
Und die Notwendigkeit einer Zuweisung von Leistung ergibt sich aus den Temperaturen....
... oder sehe ich das zu einfach?

docolli, 2025-01-12:
Was haltet ihr von Register 1077? 
Laut Doku und Fußnote 3 könnte uns dies die benötige Info bieten.

Ich teste mal, @walburgf  kannst du auch mal schauen, ob es bei Dir was sinnvolles ausgibt?

[image]
[image]

andig, 2025-01-12:
> Wenn ich dann z.B. "2500"(W) ins Register 1000 schreibe, und gleich danach wieder auslese, dann bleibt es auf "0". Der Heizstab bleibt also wie erwartet aus (disabled).

Erwartet ist enabled. Das versuche ich ja die ganze Zeit zu sagen. Denk einfach an die Wallbox… 

Aber seis drum: entweder Status merken oder Schwellwert nutzen. Ein Register scheints ja nicht zu geben.

walburgf, 2025-01-12:
@docolli : Register 1077 würde ich nicht nutzen, da es auch Dinge anzeigt, die mit der Steuerung nichts zu tun haben, sondern auch z.B. Netzwerkfehler angezeigt werden....

Ich würde wie oben beschrieben, enabled immer auf 1 setzen, wenn das Register 1081 1 ist.... das aktivie enable/disable nicht verwenden und den Rest über die Temperaturen, und Register 1000 und 1003 regeln/monitoren....

docolli, 2025-01-12:
@andig Nach meinen Tests gerade eben, ist das Register 1077 das was wir suchen!

[image]
[image]
[image]

Das geht von "0" auf "1", wenn eine Heizleistung gesetzt wird und SOFORT wieder auf "0", wenn man die Heizleistung auf "0" setzt.
Ja das Register hat noch mehr Bedeutungen, wir müssen nur die möglichen Registerwerte korrekt auf Status A / B / C mappen.

docolli, 2025-01-12:
0 = Standby / Status B
1 = Heizt / Status C
3 = Heizt nicht, da Sollwert erreicht/überschritten / Status B
4 = Kein Modbus Steuersignal seit dem in der AC THOR Steuerung eingestellten Timeout empfangen / Status A 
       (sollte aber nie vorkommen, solange evcc per Modbus pollt)

Kann mir hier mal jemand kurz helfen was ein Block ist und wie ich den im AC THOR setze, um zu testen, was Register 1077 dann ausgibt?
[image]

[32 more replies]