evcc-io/evcc #18022: Sonnenbatterie Netzladung nur bis 79% möglich

marleyan333, 2025-01-02:
### Describe the bug

Ich lade über evcc zwischen 0-5 Uhr (Tarif octopus intelligent go) meine Sonnenbatterie eco 8.0 mit einer Kapazität von 12 kwh über das Netz auf, parallel lade ich über den Porsche Mobile charger connect meinen Taycan. Letzteres funktioniert problemlos, allerdings lädt die Sonnenbatterie trotz in der evcc.yaml korrekt hinterlegter Kapazität von 12 kwh (versuchsweise habe ich auch schon höhere Kapazitäten versucht) nur bis 79% und stellt dann die Ladung des Akkus ein. 

### Steps to reproduce

1. Netzladen Sonnebatterie aktiv
2. Geplante Ladung Auto im Zeitraum 0-5 uhr
3. Ladung Sonnenbatterie bricht bei 79 % ab
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 2b36a4391d07b25fc98fc4e77dadfc6edd01f8519313cc4d1949ad1aef2d31b6

interval: 30s # control cycle interval

meters:
- type: template
  template: sonnenbatterie 
  usage: grid  
  host: 192.168.178.41  
  port: 8080  
  name: grid1
- type: template
  template: sonnenbatterie 
  usage: pv  
  host: 192.168.178.41  
  port: 8080  
  name: pv2
- type: template
  template: sonnenbatterie 
  usage: battery  
  host: 192.168.178.41  
  port: 8080  
  token: xxxx
  capacity: 50
  chargepower: 3300
  name: battery3

chargers:
- type: template
  template: pmcc 
  ski: xxxx  
  name: wallbox5

vehicles:
- type: template
  template: iso15118 
  title: Taycan  
  capacity: 83.7  
  name: ev4

loadpoints:
- title: Garage
  charger: wallbox5
  vehicle: ev4
  mode: pv

site:
  title: Mein Zuhause
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3

tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.295 # EUR/kWh (default)
    zones:
      - days: Mo-So
        hours: 0-5
        price: 0.1609 # EUR/kWh (werktags 0-5 Uhr)

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.08 # [currency]/kWh

eebus:
  certificate:
      private: |
          -----BEGIN EC PRIVATE KEY-----
          
          -----END EC PRIVATE KEY-----
      public: |
          -----BEGIN CERTIFICATE-----
```

### Log details

```text
Erster Abschnitt, zu Zeitpunkt an dem das Laden der Batterie noch funktioniert:

Jan 01:49:02 raspberrypi evcc[900612]: [lp-1  ] DEBUG 2025/01/01 02:49:02 charge power: 10721W
Jan 01 02:49:02 raspberrypi evcc[900612]: [lp-1  ] DEBUG 2025/01/01 02:49:02 charge currents: [15.9 16 15.5]A
Jan 01 02:49:02 raspberrypi evcc[900612]: [site  ] DEBUG 2025/01/01 02:49:02 pv power: 0W
Jan 01 02:49:02 raspberrypi evcc[900612]: [site  ] DEBUG 2025/01/01 02:49:02 grid power: 18077W
Jan 01 02:49:02 raspberrypi evcc[900612]: [site  ] DEBUG 2025/01/01 02:49:02 battery soc: 79%
Jan 01 02:49:02 raspberrypi evcc[900612]: [site  ] DEBUG 2025/01/01 02:49:02 battery power: -3301W
Jan 01 02:49:02 raspberrypi evcc[900612]: [site  ] DEBUG 2025/01/01 02:49:02 site power: 14876W
Jan 01 02:49:02 raspberrypi evcc[900612]: [lp-1  ] DEBUG 2025/01/01 02:49:02 detected active phases: 3p
Jan 01 02:49:02 raspberrypi evcc[900612]: [lp-1  ] DEBUG 2025/01/01 02:49:02 charger status: C
Jan 01 02:49:02 raspberrypi evcc[900612]: [lp-1  ] DEBUG 2025/01/01 02:49:02 plan: charge 2h16m8s between 2025-01-01 02:49:03 +0100 CET until 2025-01-01 07:00:00 +0100 CET (power: 11040W, avg cost: 0.166)
Jan 01 02:49:02 raspberrypi evcc[900612]: [eebus ] DEBUG 2025/01/01 02:49:02 Recv: d:_i:19667_PorscheEVSE-00025806:Measurement to Measurement notify 6112898 MeasurementListData 02:48:57 raspberrypi evcc[900612]: [eebus ] DEBUG 2025/01/01 02:48:57 ski bb74c3d74f3dcf4e839fda7e0b7b60d95acda3a3 event cem-evcem-DataUpdateEnergyCharged

Zweiter Abschnitt, Laden der Batterie bricht ab:

Jan 01 02:49:37 raspberrypi evcc[900612]: [site  ] ERROR 2025/01/01 02:49:37 pv 1 power: Get "http://192.168.178.41:8080/api/v1/status": dial tcp 192.168.178.41:8080: i/o timeout
Jan 01 02:49:37 raspberrypi evcc[900612]: [site  ] DEBUG 2025/01/01 02:49:37 pv power: 0W
Jan 01 02:49:37 raspberrypi evcc[900612]: [site  ] ERROR 2025/01/01 02:49:37 grid power: Get "http://192.168.178.41:8080/api/v1/status": dial tcp 192.168.178.41:8080: i/o timeout

Jan 01 02:49:37 raspberrypi evcc[900612]: [site  ] ERROR 2025/01/01 02:49:37 pv 1 power: Get "http://192.168.178.41:8080/api/v1/status": dial tcp 192.168.178.41:8080: i/o timeout
Jan 01 02:49:37 raspberrypi evcc[900612]: [site  ] DEBUG 2025/01/01 02:49:37 pv power: 0W
Jan 01 02:49:37 raspberrypi evcc[900612]: [site  ] ERROR 2025/01/01 02:49:37 grid power: Get "http://192.168.178.41:8080/api/v1/status": dial tcp 192.168.178.41:8080: i/o timeout

Jan 01 02:50:43 raspberrypi evcc[900612]: [site  ] ERROR 2025/01/01 02:50:43 pv 1 power: Get "http://192.168.178.41:8080/api/v1/status": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
Jan 01 02:50:43 raspberrypi evcc[900612]: [site  ] DEBUG 2025/01/01 02:50:43 pv power: 0W
Jan 01 02:50:43 raspberrypi evcc[900612]: [site  ] ERROR 2025/01/01 02:50:43 grid power: Get "http://192.168.178.41:8080/api/v1/status": context deadline exceeded (Client.Timeout exceeded while awaiting headers)

an 01 02:51:02 raspberrypi evcc[900612]: [site  ] DEBUG 2025/01/01 02:51:02 ----
Jan 01 02:51:02 raspberrypi evcc[900612]: [lp-1  ] DEBUG 2025/01/01 02:51:02 charge power: 10748W
Jan 01 02:51:02 raspberrypi evcc[900612]: [lp-1  ] DEBUG 2025/01/01 02:51:02 charge currents: [15.9 16 15.5]A
Jan 01 02:51:02 raspberrypi evcc[900612]: [site  ] DEBUG 2025/01/01 02:51:02 grid power: 14888W
Jan 01 02:51:02 raspberrypi evcc[900612]: [site  ] DEBUG 2025/01/01 02:51:02 pv power: 0W
Jan 01 02:51:02 raspberrypi evcc[900612]: [site  ] DEBUG 2025/01/01 02:51:02 battery soc: 79%
Jan 01 02:51:02 raspberrypi evcc[900612]: [site  ] DEBUG 2025/01/01 02:51:02 battery power: -6W
Jan 01 02:51:02 raspberrypi evcc[900612]: [site  ] DEBUG 2025/01/01 02:51:02 site power: 14982W
Jan 01 02:51:02 raspberrypi evcc[900612]: [lp-1  ] DEBUG 2025/01/01 02:51:02 detected active phases: 3p
Jan 01 02:51:02 raspberrypi evcc[900612]: [lp-1  ] DEBUG 2025/01/01 02:51:02 charger status: C
Jan 01 02:51:02 raspberrypi evcc[900612]: [lp-1  ] DEBUG 2025/01/01 02:51:02 plan: charge 2h14m13s between 2025-01-01 02:51:03 +0100 CET until 2025-01-01 07:00:00 +0100 CET (power: 11040W, avg cost: 0.166)
Jan 01 02:51:02 raspberrypi evcc[900612]: [eebus ] DEBUG 2025/01/01 02:51:02 Recv: d:_i:19667_PorscheEVSE-00025806:Measurement to Measurement notify 6113280 MeasurementListData
```

### What type of operating system or environment does evcc run on?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.12

andig, 2025-01-03:
Offensichtlich gibts massive Probleme mit der Batterie zu kommunizieren- warum auch immer. Der Hersteller ist dein Ansprechpartner. evcc kann da gar nichts tun.

closed 2025-01-03 by andig

marleyan333, 2025-01-03:
Als würde den Hersteller interessieren, ob es Probleme mit einer GitHub-Software gibt…. Außerdem geht es ja mit anderer Software problemlos ( sonnenbuddy, früher auch mit clever pv), muss also dann doch an evcc liegen. Aber wenn nicht sein kann, was nicht sein darf, dann schließt man halt einfach das issue. Zumindest ein Hint, woran es liegen könnte, wäre hilfreich gewesen.

marleyan333, 2025-01-03:
@andig , @DerAndereAndi 
Es ist doch nicht zielführend einfach das issue ohne Rücksprache zu schließen, ich will ja nur darauf hinweisen, dass es da ein grundsätzliches Problem in evcc geben muss, denn sonst hätte ich doch die gleichen Probleme bei Verwendung von clever-PV und anderen.  Das ist aber nicht der Fall. Was mir auffällt ist, dass evcc vor Abschluss der vollständigen Ladung den Modus der Batterie von time of use auf self-consumption umstellt. Könnte es daran liegen?  Schaut doch bitte noch mal genauer drüber, wäre schade.

andig, 2025-01-03:
Ich kann deine Anspruchshaltung nicht nachvollziehen. Woran es liegt steht im Logfile: timeout bei der Netzwerkverbindung. Hättest Du zum gleichen Zeitpunkt beliebiges Tool xyz verwendet würde das gleiche Ergebnis entstehen.

> muss also dann doch an evcc liegen

Aus genau diesem Grund nicht nachvollziehbar. Nur weil es deinen Hersteller nicht interessiert heisst das nicht, dass evcc ursächlich ist.

andig, 2025-01-03:
> Könnte es daran liegen?

Ohne Logfile- wie immer- nicht zu beantworten.

marleyan333, 2025-01-03:
Es tritt aber nachweislich bei keinem anderen Tool auf! Habe doch das logfile eingestellt oder was genau braucht ihr da noch, kann gerne mehr liefern.

andig, 2025-01-03:
Es braucht- auch wie immer- ein Logfile das das Problem zeigt, nicht irgendeinen unrelateten Schnipsel. Zur Modusänderung kann ich da nichts sehen.

andig, 2025-01-03:
> Außerdem geht es ja mit anderer Software problemlos ( sonnenbuddy, früher auch mit clever pv), muss also dann doch an evcc liegen. 

Wie redet denn clever mit deinem lokalen WR? Gibts da eine Sst. die besser funktioniert?

marleyan333, 2025-01-03:
Habe versucht, die Stelle rauszusuchen, an der zeitlich das Problem auftreten musste, weil ich sehen konnte, dass da die Ladung abbricht, ich dachte, es macht keinen Sinn euch das ganze logfile einzustellen, war gut gemeint, um euch die Arbeit zu sparen, das ganze file abzusuchen. Und wenn ich wüsste, wo das Problem liegt, bräuchte ich mich ja gar nicht an dich wenden.

marleyan333, 2025-01-03:
Clever PV nutzt die Sonnen Batterie Webhook API. 
[image]

andig, 2025-01-03:
Achso:

```
Jan 01 02:51:02 raspberrypi evcc[900612]: [site  ] DEBUG 2025/01/01 02:51:02 battery soc: 79%
Jan 01 02:51:02 raspberrypi evcc[900612]: [site  ] DEBUG 2025/01/01 02:51:02 battery power: -6W
Jan 01 02:51:02 raspberrypi evcc[900612]: [site  ] DEBUG 2025/01/01 02:51:02 site power: 14982W
Jan 01 02:51:02 raspberrypi evcc[900612]: [lp-1  ] DEBUG 2025/01/01 02:51:02 detected active phases: 3p
Jan 01 02:51:02 raspberrypi evcc[900612]: [lp-1  ] DEBUG 2025/01/01 02:51:02 charger status: C
```

wenn das Auto lädt wird der WR gesperrt (=hold, nicht charge). Das dürfte hier bei den 79% der Fall sein. Ob das so ist ist wiederum mangels ausreichendem Logfile nicht zu erkennen...

marleyan333, 2025-01-03:
> Achso:
>
> ```
> Jan 01 02:51:02 raspberrypi evcc[900612]: [site  ] DEBUG 2025/01/01 02:51:02 battery soc: 79%
> Jan 01 02:51:02 raspberrypi evcc[900612]: [site  ] DEBUG 2025/01/01 02:51:02 battery power: [...]

Soll ich einen längeren Abschnitt des logfile schicken?

DerAndereAndi, 2025-01-03:
@marleyan333 bitte mich hier herauslassen, bin nicht Teil des Projektes und habe daher hiermit nichts zu tun. Vielen Dank.

marleyan333, 2025-01-03:
> @marleyan333 bitte mich hier herauslassen, bin nicht Teil des Projektes und habe daher hiermit nichts zu tun. Vielen Dank.

Sorry, mein Fehler!