evcc-io/evcc #18025: Wrong docu for co2 tariff? Using template as described in docu throws error

dieterlohrstraeter, 2025-01-03:
### Describe the bug

Habe heute evcc neu gestartet und bekam ab dann vermutlich wegen Ausfall des Servers permanente "Bad Gateway" errors für co2 "grünstromindex".
Habe die Doku dazu mit meiner Konfig gecheckt und sah dort eine andere Definition und dachte mir ich probiere es aus.
Leider bekomme ich direkt einen Fehler und evcc bootet mit dieser Konfig nicht.

**Vor Reboot funktionierend:**
```
tariffs:
  co2:
    type: grünstromindex # Beispiel GrünstromIndex (nur Deutschland)
    zip: 59368 # Deine Postleitzahl

  currency: EUR # (default EUR)
...
```

**Beschreibung aus Doku:**
```
Grünstromindex

Regionale Emissionsdaten von https://gruenstromindex.de/. Nur für Deutschland verfügbar.

tariffs:
  co2:
    type: template
    template: grünstromindex
    zip:
```

**Meine Konfig gemäß Doku:**
tariffs:
  co2:
    type: template
    template: grünstromindex # Beispiel GrünstromIndex (nur Deutschland)
    zip: 59368 # Deine Postleitzahl
...
```
liefert Fehler beim Start:
```
Fehler beim Starten. Überprüfe deine Konfiguration und starte neu.
cannot create tariff type 'grünstromindex': decoding failed due to the following error(s): '' has invalid keys: template
```
Habe es wieder zurückgeändert und es läuft wieder.
Ich vermute einen Fehler in der Doku, oder habe ich dabei etwas falsch gemacht?

Passiert auch mit dem neuesten nightly build...

### Steps to reproduce

Ändere type: auf "template" und neuer Parameter template: auf "grünstromindex".

**Beschreibung aus Doku:**
```
Grünstromindex

Regionale Emissionsdaten von https://gruenstromindex.de/. Nur für Deutschland verfügbar.

tariffs:
  co2:
    type: template
    template: grünstromindex
    zip:
```

### Configuration details

```yaml
[main  ] INFO 2025/01/03 11:22:46 evcc 0.131.12
                                               [main  ] INFO 2025/01/03 11:22:46 using config file: /etc/evcc.yaml
                                                                                                                  [db    ] INFO 2025/01/03 11:22:46 using sqlite database: /root/.evcc/evcc.db
                                                                         [bmw   ] DEBUG 2025/01/03 11:22:47 identity.Login - database token found

                            Fehlermeldung:

                                              cannot create tariff type 'grünstromindex': decoding failed due to the following error(s):

    '' has invalid keys: template

Konfiguration (/etc/evcc.yaml):

network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: *****

interval: 10s # control cycle interval

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
sponsortoken: *****
telemetry: true

meters:
- type: template
  template: sonnenbatterie
  usage: grid
  host: SB-xxxx
  port: 8080
  name: grid1
- type: template
  template: sonnenbatterie
  usage: pv
  host: SB-xxxx
  port: 8080
  name: pv2
- type: template
  template: sonnenbatterie
  usage: battery
  host: SB-xxxx
  port: 8080
  name: battery3

chargers:
 - name: sonnencharger
   type: template
   template: etrel
   host: SonnenCharger
   port: 502

#vehicles:
# - type: template
#  template: bmw
#  title: i4-M50
#  user: *****
#  password: *****
#  vin: *****
#  capacity: 84
#  name: ev1
#  cache: 20m
#  icon: car
#  hcaptcha: xxx

loadpoints:
- title: MeineGarage
  charger: sonnencharger
  mode: pv
  enable:
    threshold: 0
    delay: 30s

site:
  title: Mein Zuhause
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
  residualpower: 10

tariffs:
  co2:
    type: template
    template: grünstromindex # Beispiel GrünstromIndex (nur Deutschland)
    zip: 59368 # Deine Postleitzahl

  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.36 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.07 # [currency]/kWh

Version: `0.131.12`
```

### Log details

```text
[main ] FATAL 2025/01/03 11:19:58 cannot create tariff type 'grünstromindex': decoding failed due to the following error(s): '' has invalid keys: template
[main ] FATAL 2025/01/03 11:19:58 will attempt restart in: 15m0s
```

### What type of operating system or environment does evcc run on?

Docker container

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

Version: `0.131.12`

naltatis, 2025-01-03:
Kann ich nicht reproduzieren. Wenn ich mit deiner Config `evcc tariff` aufrufe bekomme ich korrekte Daten:

```
tariffs:
  co2:
    type: template
    template: grünstromindex # Beispiel GrünstromIndex (nur Deutschland)
    zip: 59368 # Deine Postleitzahl

  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.36 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.07 # [currency]/kWh
```

```
$ evcc tariff

feedin:
From                   To                     Price/Cost
2025-01-03 00:00:00    2025-01-03 01:00:00    0.070
2025-01-03 01:00:00    2025-01-03 02:00:00    0.070
2025-01-03 02:00:00    2025-01-03 03:00:00    0.070
...

co2:
From                   To                     Price/Cost
2025-01-03 10:00:00    2025-01-03 11:00:00    167.000
2025-01-03 11:00:00    2025-01-03 12:00:00    146.000
2025-01-03 12:00:00    2025-01-03 13:00:00    129.000
...

grid:
From                   To                     Price/Cost
2025-01-03 00:00:00    2025-01-03 01:00:00    0.360
2025-01-03 01:00:00    2025-01-03 02:00:00    0.360
2025-01-03 02:00:00    2025-01-03 03:00:00    0.360
...
```

Prüf mal bitte ob du irgendwo ein Sonderzeichen/Leerzeichen drin hast wo es nicht hingehört.

Zur Dokumentation: Aktuell funktionieren beide Wege (template und type direkt). Templates sind der zukunftssichere Weg. Die direkte Nutzung als Typ ist aus historischen Gründen noch möglich. Ich schau die Doku mal durch und stelle die alten Schreibeweisen um.

VolkerK62, 2025-01-03:
mach mal ein update auf 0.132.0

andig, 2025-01-03:
> I have verified that the issue is reproducible with the latest nightly build

Das scheint nicht erfolgt zu sein. Closing.

closed 2025-01-03 by andig