evcc-io/evcc #18033: Ampere Storage Pro - Grid values dont make sense

rklueber, 2025-01-03:
### Describe the bug

Installed evcc in Docker. Works like a charm after I understood how the configuration is done. I have Ampere Storage Pro. Adding grid/battery/inverter was easy via IP/modbus.

Anyone else running this hardware without issues? I have the impression the grid meter does not work properly. Saw similar issues which did not lead to a solution for me.

This is evcc
[image]

This is the respective information I read out via HA which make senso to me and are in line with the values I can read on the display of the physical EVU smart meter.

[image]

Any ideas where to look?

### Steps to reproduce

1. Start Docker
2. Values are off

### Configuration details

```yaml
sponsortoken: eyJhbGciOiJIUzI2NiIsInR5cCI6IkpXVCJ9...

network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 9b7394ced66f2e1...

interval: 30s # control cycle interval

meters:
- type: template
  template: saj-h2
  id: 1
  host: 10.0.30.87
  port: 502
  usage: grid
  modbus: tcpip
  name: grid1
- type: template
  template: saj-h2
  id: 1
  host: 10.0.30.87
  port: 502
  usage: pv
  modbus: tcpip
  name: pv2
- type: template
  template: saj-h2
  id: 1
  host: 10.0.30.87
  port: 502
  usage: battery
  modbus: tcpip
  capacity: 12
  defaultmode: 2
  # minsoc: 20
  # maxsoc: 95
  name: battery3

chargers:
- type: template
  template: openwb-2.0
  id: 1
  host: 192.168.193.5
  port: 1502
  modbus: tcpip
  connector: 1
  name: wallbox4
- type: template
  template: openwb-2.0
  id: 1
  host: 192.168.193.6
  port: 1502
  modbus: tcpip
  connector: 2
  name: wallbox5

loadpoints:
- title: Garage
  charger: wallbox4
  mode: minpv
- title: Stellplatz
  charger: wallbox5
  mode: off

site:
  title: Heidbergdamm
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3

tariffs:
  grid:
    type: template
    template: tibber
    token: _N01KQGAJZlsAJPCDOIb...

vehicles:
  - name: tessa
    type: template
    template: tesla
    title: Tessa # Wird in der Benutzeroberfläche angezeigt (optional)
    capacity: 80 # Akkukapazität in kWh (optional)
    accessToken: eyJhbGciOiJSUzI1NiIs...
    refreshToken: EU_8051b3ab64a7ee55...
    vin: LRW...
```

### Log details

```text
[lp-1  ] DEBUG 2025/01/03 15:19:51 charge power: 11161W
[lp-1  ] DEBUG 2025/01/03 15:19:51 charge currents: [16.1 16.2 16.2]A
[lp-2  ] DEBUG 2025/01/03 15:19:51 charge power: 0W
[lp-2  ] DEBUG 2025/01/03 15:19:51 charge currents: [0 0 0]A
[site  ] DEBUG 2025/01/03 15:19:52 grid power: 698W
[site  ] DEBUG 2025/01/03 15:19:56 pv power: 510W
[site  ] DEBUG 2025/01/03 15:19:57 grid currents: [1.34 1.41 1.45]A
[site  ] DEBUG 2025/01/03 15:19:58 battery soc: 11%
[site  ] DEBUG 2025/01/03 15:19:58 battery power: 0W
[site  ] DEBUG 2025/01/03 15:19:58 site power: 798W
[lp-2  ] DEBUG 2025/01/03 15:19:58 charge total import: 0.000kWh
[lp-2  ] DEBUG 2025/01/03 15:19:58 charger status: A
```

### What type of operating system or environment does evcc run on?

Docker container

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.132.0

rklueber, 2025-01-03:
I was able to clone the repro, did the required changes in the template and build/run a docker. I can confirm the suggested change above (0x40A7 to 0x40A0) works.

andig, 2025-01-05:
Laut Template ist 0x40A7 # TotalGridPowerWatt. Das klingt richtig. Was ist 0x40A7? Und funktioniert das dann auch für den SAJ H2?

rklueber, 2025-01-05:
In dem Thread für die HA findet sich (in einem Telegram Chat) ein PDF mit einigen Infos. Aber mehr als die Beschreibung 0x40A0 als "SysTotalLoadWatt" findet sich dort auch nicht. 

Quelle: https://t.me/saj_nooficialoriginal/8487

Ich habe einen H2. Hier funktioniert es mit dem 0x40A0. Hast du einen sinnvollen Wert mit dem Register 0x40A7?

andig, 2025-01-05:
Ich komme mangels Telegram nicht drauf :O

rklueber, 2025-01-05:
[4_5868569929443185242.pdf](https://github.com/user-attachments/files/18310179/4_5868569929443185242.pdf)

andig, 2025-01-05:
[image]

Danke. Klingt für mich, als wäre 0x40A0 die Inverterleistung, 40A7H aber tatsächlich Netz. Bekommst Du beim H2 auf beiden Registern die gleichen Werte?

@premultiply was machen wir damit?

rklueber, 2025-01-05:
Nein auf beiden Regsiern kommen unterschiedliche Werte. Auf 0x40A7 kommen Werte die nicht passen. Siehe initialer Screenshot.

Mit der modification sieht das besser aus bei mir.

[image]

premultiply, 2025-01-05:
Bei 0x40A0 passt keine der Beschreibungen in diversen Dokumenten von SAJ. Da geht es nur um Loads (Hausverbrauch).
Von der Beschreibung her ergibt nur `0x40A7` oder vielleicht noch `0x40AD` größeren Sinn.
Was enthält denn `0x40AD` ?

rklueber, 2025-01-08:
0x40AD liefert 0W bei mir. 
 
[image]

rklueber, 2025-01-08:
Danke für das kritische Nachfragen. Mein System wird gerade durch die Batterie gespeist und in dem Zustand stimmt die Anzeige in EVCC mit dem geänderten Register auf 0x40A0 wieder nicht. Ergo ist das nicht das richtige Register ,-). Euer Verdacht war begründet. 

Welchen Wert erwartet EVCC denn für ein Meter dort?

Kann man das irgendwo nachlesen oder erklären? Dann kann ich versuchen an meinem System den richtigen Wert zu finden.

andig, 2025-01-08:
Battery/PV: Zähler am Gerät bevor es ins Haus geht. Grid: Netzübergabepunkt.

rklueber, 2025-01-09:
Ich hab jetzt per QModMaster die Werte mal ausgelesen. 0x40AD macht Sinn. Aber ... ich hab irgendwie einen Offset on 1 bei den Registern. Seht hier. Eigentlich müsste der Block bei 16533 starten und dann das 25te Register den Wert für Grid haben. 

[image]

Die Register 2/3/4/5 sehen wie die "Direction" Werte aus, die eigentlich ein Register früher starten sollten. Dann kommen die 7*2 ungenutzen Werte und dann get es mit TotalLoadPower weiter ...

````
decode_instructions_part_2 = [
                ("directionPV", None),  
                ("directionBattery", "decode_16bit_int"),  
                ("directionGrid", "decode_16bit_int"),  
                ("directionOutput", None),  
                (None, "skip_bytes", 14),  
                ("TotalLoadPower", "decode_16bit_int"),  
                ("CT_GridPowerWatt", "decode_16bit_int"),  
                ("CT_GridPowerVA", "decode_16bit_int"),  
                ("CT_PVPowerWatt", "decode_16bit_int"),  
                ("CT_PVPowerVA", "decode_16bit_int"),  
                ("pvPower", "decode_16bit_int"),  
                ("batteryPower", "decode_16bit_int"),  
                ("totalgridPower", "decode_16bit_int"),  
                ("totalgridPowerVA", "decode_16bit_int"),  
                ("inverterPower", "decode_16bit_int"),  
                ("TotalInvPowerVA", "decode_16bit_int"),  
                ("BackupTotalLoadPowerWatt", "decode_16bit_uint"),  
                ("BackupTotalLoadPowerVA", "decode_16bit_uint"),  
                ("gridPower", "decode_16bit_int"),  
            ]
````

Lese ich mit QModMaster ab 16534 ... bekomme ich auch den Grid Wert als 25tes Register

[image]

Insofern sieht es für mich so aus, als ob 0x40AD schon irgendwie richtig ist. Ob das Delta von 1 nun von QModMaster kommt, oder die Register wirklich um 1 verschoben sind ... Ich bin mit meinem Latein erst mal am Ende.

Was mache ich flasch?

Hab viel gelernt die letzten Stunden ,-).

Aktuell antwortet mein H2 über modbus auch nicht mehr ... ich hoffe der fängt sich wieder. Oder habt ihr ne Idee?

````
[lp-1  ] DEBUG 2025/01/09 13:33:51 charge power: 0W
[lp-1  ] DEBUG 2025/01/09 13:33:51 charge currents: [0 0 0]A
[lp-2  ] ERROR 2025/01/09 13:34:01 charge power: dial tcp 192.168.193.6:1502: i/o timeout
[lp-2  ] ERROR 2025/01/09 13:34:11 charge currents: dial tcp 192.168.193.6:1502: i/o timeout
[site  ] ERROR 2025/01/09 13:34:12 battery 1 power: read failed: read tcp 172.23.0.2:34346->10.0.30.87:502: read: connection reset by peer
[site  ] ERROR 2025/01/09 13:34:13 battery 1 soc: read failed: read tcp 172.23.0.2:34356->10.0.30.87:502: read: connection reset by peer
[site  ] ERROR 2025/01/09 13:34:14 pv 1 power: read failed: read tcp 172.23.0.2:34358->10.0.30.87:502: read: connection reset by peer
[site  ] ERROR 2025/01/09 13:34:14 grid power: read failed: read tcp 172.23.0.2:34336->10.0.30.87:502: read: connection reset by peer
````

rklueber, 2025-01-09:
LAN vom Ampere Storage Pro kurz gezogen und wieder ran. Geht alles wieder.

Long story short. 0x40AD ist richtig und funktioniert.

PR #18148 (2025-01-09): SAJ H2: fix grid power
Fixes #18033
closed 2025-01-09 by andig