evcc-io/evcc #18035: Mobile UI: direct link to charge point

naltatis, 2025-01-03:
Add an option to directly focus a charge point via URL on smaller screens.

_Originally posted by @naltatis in https://github.com/evcc-io/evcc/discussions/17869#discussioncomment-11656315_

cschlipf, 2025-01-07:
Bin auch sehr daran interessiert um die entsprechenden Widgets meiner Grafana Dashboards (https://github.com/ha-puzzles/evcc-grafana-dashboards/) direkt verlinken zu können. Siehe auch #18093 für meinen Use Case.

premultiply, 2025-01-07:
Out of scope here but maybe add a "kiosk mode" view (option) at the same that limits the view to a single specified loadpoint?

cschlipf, 2025-01-07:
> Out of scope here but maybe add a "kiosk mode" view (option) at the same that limits the view to a single specified loadpoint?

I wouldn't need that right now. But I see the use case. This would enable someone to embed a loadpoint card in other UIs, like Home Assistant dashboards.

smartlader-de, 2025-01-08:
I don't think that's where @premultiply was going @cschlipf .

I imagine a scenario where you could have a:
- RaspiZero powering a Browser in Kiosk Mode (either it's own evcc instance or an external one)
- In a little box with a nice HiDef Touchscreen like the HyperPixel (I have a few one here...for testing...)
https://www.adafruit.com/product/3578

Is that what you envision @premultiply ?

premultiply, 2025-01-08:
Yes I was thinking about multiple use cases in addition to the main feature request here like
- restrict view and operation to the "own" charger for app-style UI usage
- add local touch display mini-computer/tablet to operate only a specific charger

cschlipf, 2025-01-08:
I don't see however how this contradicts what I was saying above ;-)

Maybe with an URL option to hide the power distribution bar on top.

However this is - as said - far beyond this request.

naltatis, 2025-01-08:
Yes, let's move the discussions about the "beyond direct linking" uses cases to https://github.com/evcc-io/evcc/discussions/17869#discussioncomment-11656315 where this first-step-feature originated from.

closed 2025-02-02 by andig