evcc-io/evcc #18036: smart eq login broken

user86000, 2025-01-03:
### Describe the bug

because of change from mercedes in the cloud

the mercedes me integration for homeassistant still works for smart eq

### Steps to reproduce

1. start evcc
2.
3.
...

### Configuration details

```yaml
vehicles:
- name: smart_eq
  type: template
  template: smart 
  user: xxx  
  password: yyy
  vin: W1Azzz  
  capacity: 17 
  phases: 1
```

### Log details

```text
[main  ] ERROR 2025/01/03 17:45:11 creating vehicle smart_eq failed: cannot create vehicle type 'template': cannot create vehicle type 'smart': login failed: resume not found
```

### What type of operating system or environment does evcc run on?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

andig, 2025-01-03:
According to https://github.com/evcc-io/evcc/discussions/17098 SmartEQ is broken.

andig, 2025-01-03:
Funktioniert https://github.com/TA2k/ioBroker.smart-eq noch?

user86000, 2025-01-03:
Das hier funktioniert noch
https://github.com/ReneNulschDE/mbapi2020

andig, 2025-01-04:
@ReneNulschDE is there something we could do for Smart EQ?

ReneNulschDE, 2025-01-04:
Ja, man kann die MB-Komponente nach einer Anpassung nutzen:

1. Der api.vehicles check muß ausgebaut werden (der gibt bei Smarts nichts zurück) --> der darauf folgende api.status call mit der VIN/FIN aus der Config funktioniert

Es gibt aber Meldungen, dass charging_power und ChargingStatus bei einigen Smarts nicht mehr von der API geliefert werden.

user86000, 2025-01-04:
Ja mit der mercedes Konfig in evccc konnte ich die beiden token erfolgreich erzeugen, 
Beim login kam dann kein Fahrzeug gefunden

Beim HA ist es au h die MB Komponente, die noch funktioniert beim smart.

closed 2025-01-04 by andig