evcc-io/evcc #18043: SMA SunnyBoy 3.0 Speedwire abfrage funktioniert nicht.

ForwardThinkerDev, 2025-01-04:
### Describe the bug

Einer meiner 3 SMA Wechselrichter lässt sich nicht mit Speedwire abfragen.
Die Komplette Anlage ist mit Cat7 Kabel per LAN verbunden. Im SMA HomeManager2.0 funktioniert die Abfrage.
Ist die Koexistenz der Systeme ein Problem?

Die 2x STP6.0 funktionieren einwandfrei.

By the way kann die PV Leistung auch vom SMA HomeManager 2.0 Abgerufen werden oder funktioniert dieser in EVCC nur als Netzzähler?
Ich habe auch noch einen neuen SMA eCharger allerdings ist dieser momentan in die SMA Software eingebunden. So weit ich weiß wird dieser noch nicht unterstützt?

### Steps to reproduce

1. SMA SunnyBoy 3.0 via Speedwire einrichten
2. EVCC Startet mit Fehler
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: ***

interval: 30s # control cycle interval

sponsortoken: ***

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

hems:
  type: sma
  allowcontrol: false

meters:
# - type: template
#  template: siemens-pac2200 
#  id: 1  
#  host: 192.168.0.56  
#  port: 502  
#  usage: pv  
#  modbus: tcpip  
#  name: pv1

- name: my_grid
  type: template
  template: sma-home-manager
  usage: grid
  host: 192.168.0.21 # IP-Adresse oder Hostname

- type: template                                                                 
  template: sma-inverter-speedwire                                               
  usage: pv                                                                      
  host: 192.168.0.20                                                             
  password: ***                                                           
  name: pv1   

- type: template                                                                 
  template: sma-inverter-speedwire                                               
  usage: pv                                                                      
  host: 192.168.0.44                                                             
  password: ***                                                            
  name: pv2

- type: template                                                                 
  template: sma-inverter-speedwire                                               
  usage: pv                                                                      
  host: 192.168.0.41                                                             
  password: ***                                                            
  name: pv3   

chargers:
- type: template
  template: abb 
  id: 49  
  host: 192.168.0.49  
  port: 502  
  modbus: tcpip  
  name: wallbox1    

#- type: template                                                                 
#  template: homewizard                                                           
#  host: 192.168.0.95                                                             
#  standbypower: 5                                                                
#  integrateddevice: false                                                        
#  heating: false                                                                 
#  icon: car                                                                      
#  name: wallbox2                                                                                                                          

vehicles:
- type: template
  template: bmw 
  title: BMW i3s  
  icon: car  
  user: ***  
  password: ***
  vin: ***
  capacity: 37.9  
  cache: 15m  
  mode: pv  
  region: EU  
  name: ev1
  hcaptcha: ***
- type: template                                                                 
  template: mercedes                                                             
  title: EQV                                                                     
  icon: van                                                                      
  user: ***                                                
  region: EMEA                                                                   
  accessToken: ***                              
  refreshToken: ***                   
  capacity: 90                                                                   
  phases: 3                                                                      
  mode: pv                                                                                                                                                                                                   
  name: ev2  

loadpoints:
- title: Abb TerraAC
  charger: wallbox1
  vehicle: ev1
  mode: pv
  priority: 1
  enable:
    threshold: 0
    delay: 5m
  disable:
    threshold: 0
    delay: 30m

#- title: Plug HomeWizzard
#  charger: wallbox2
#  vehicle: 
#  mode: off
#  priority: 2
#  enable:
#    threshold: 0
#    delay: 5m
#  disable:
#    threshold: 0
#    delay: 30m

tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.25

  feedin:
    type: fixed
    price: 0.08

site:
  title: Home
  meters:
    grid: my_grid
    pv:
      - pv1
      - pv2
      - pv3
```

### Log details

```text
[main ] FATAL 2025/01/04 14:36:46 cannot create meter 'pv3': cannot create meter type 'template': cannot create meter type 'sma': login failed
[main ] FATAL 2025/01/04 14:36:46 will attempt restart in: 15m0s
```

### What type of operating system or environment does evcc run on?

Docker container

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.132.0

VolkerK62, 2025-01-04:
Hmm ... die fehlermeldung sagt "Login failed".
Typo im Passwort???

ForwardThinkerDev, 2025-01-04:
> Hmm ... die fehlermeldung sagt "Login failed". Typo im Passwort???

Leider nein. Im SMA HM2.0 System haben alle Geräte das selbe Anlagen Passwort wurde auch am WR Webserver getestet.

ForwardThinkerDev, 2025-01-04:
EVCC verwendet das Passwort für das BN Level Benutzer.
SMA Applikationen verwenden das Anlagenpasswort = Installateurpwasswort

closed 2025-01-04 by ForwardThinkerDev

andig, 2025-01-04:
Lesson learned: Fehlermeldung lesen. Und dann nochmal lesen ;)