evcc-io/evcc #18047: Funktion "Verhindere Entladung im Schnell-Modus und bei geplantem Laden" verhindert bei Sonnenbatterie Netzladung

marleyan333, 2025-01-04:
### Describe the bug

Seit Längerem habe ich das Problem, dass ein Netzladen der Batterie immer nur bis ca. 80% möglich war. Wie ich nun durch systematisches Durchgehen der Debug-Logs festgestellt habe, scheint das Problem in meinem Fall durch die im Titel genannte Funktion ausgelöst zu werden. Nach Abstellen selbiger ging es zumindest gestern Nacht dann wieder. Vielleicht hilft diese Info auch anderen und möglicherweise will sich der Admin des Projekts der Sache nun doch annehmen, nachdem das Problem zunächst beim Batteriehersteller vermutet wurde.

### Steps to reproduce

1. Aktivieren der Funktion "Verhindere Entladung im Schnell-Modus und bei geplantem Laden"
2. Wenn die geplante Ladung meines Autos beendet wird erfolgt "set battery mode: normal" (= Eigenverbrauchsoptimierung, kein Netzladen möglich) unter Zeitstempel 01:48:50
3. Deaktivieren der genannten Funktion: battery mode bleibt auf "charge" --> Netzladen funktioniert wieder! unter Zeitstempel 04:27:50

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: xxxx

interval: 30s # control cycle interval

meters:
- type: template
  template: sonnenbatterie 
  usage: grid  
  host: 192.168.178.41  
  port: 8080  
  name: grid1
- type: template
  template: sonnenbatterie 
  usage: pv  
  host: 192.168.178.41  
  port: 8080  
  name: pv2
- type: template
  template: sonnenbatterie 
  usage: battery  
  host: 192.168.178.41  
  port: 8080  
  token: xxxx
  capacity: 50
  chargepower: 3300
  name: battery3

chargers:
- type: template
  template: pmcc 
  ski: xxxx 
  name: wallbox5

vehicles:
- type: template
  template: iso15118 
  title: Taycan  
  capacity: 83.7  
  name: ev4

loadpoints:
- title: Garage
  charger: wallbox5
  vehicle: ev4
  mode: pv

[... 199 lines cut ...]
Jan 04 04:27:49 raspberrypi evcc[1017982]: [lp-1  ] DEBUG 2025/01/04 04:27:49 charge power: 0W
Jan 04 04:27:49 raspberrypi evcc[1017982]: [lp-1  ] DEBUG 2025/01/04 04:27:49 charge currents: [0 0 0]A
Jan 04 04:27:50 raspberrypi evcc[1017982]: [eebus ] DEBUG 2025/01/04 04:27:50 Send: d:_i:19667_PorscheEVSE-00025806 notify 10869 DeviceDiagnosisHeartbeatData
Jan 04 04:27:50 raspberrypi evcc[1017982]: [site  ] DEBUG 2025/01/04 04:27:50 set battery mode: charge
Jan 04 04:27:50 raspberrypi evcc[1017982]: [site  ] DEBUG 2025/01/04 04:27:50 grid power: 45W
Jan 04 04:27:50 raspberrypi evcc[1017982]: [site  ] DEBUG 2025/01/04 04:27:50 pv power: 10W
Jan 04 04:27:50 raspberrypi evcc[1017982]: [eebus ] DEBUG 2025/01/04 04:27:50 Recv: d:_i:19667_PorscheEVSE-00025806:Measurement to Measurement notify 6675999 MeasurementListData
Jan 04 04:27:50 raspberrypi evcc[1017982]: [eebus ] DEBUG 2025/01/04 04:27:50 ski bb74c3d74f3dcf4e839fda7e0b7b60d95acda3a3 event cem-evcem-DataUpdateCurrentPerPhase
Jan 04 04:27:50 raspberrypi evcc[1017982]: [eebus ] DEBUG 2025/01/04 04:27:50 ski bb74c3d74f3dcf4e839fda7e0b7b60d95acda3a3 event cem-evcem-DataUpdatePowerPerPhase
Jan 04 04:27:50 raspberrypi evcc[1017982]: [eebus ] DEBUG 2025/01/04 04:27:50 ski bb74c3d74f3dcf4e839fda7e0b7b60d95acda3a3 event cem-evcem-DataUpdateEnergyCharged
Jan 04 04:27:50 raspberrypi evcc[1017982]: [site  ] DEBUG 2025/01/04 04:27:50 battery soc: 1%
Jan 04 04:27:50 raspberrypi evcc[1017982]: [site  ] DEBUG 2025/01/04 04:27:50 battery power: 240W
Jan 04 04:27:50 raspberrypi evcc[1017982]: [site  ] DEBUG 2025/01/04 04:27:50 site power: 385W
Jan 04 04:27:50 raspberrypi evcc[1017982]: [lp-1  ] DEBUG 2025/01/04 04:27:50 charger status: B
Jan 04 04:27:51 raspberrypi evcc[1017982]: [eebus ] DEBUG 2025/01/04 04:27:51 Recv: d:_i:19667_PorscheEVSE-00025806:Measurement to Measurement notify 6676001 MeasurementListData
Jan 04 04:27:51 raspberrypi evcc[1017982]: [eebus ] DEBUG 2025/01/04 04:27:51 ski bb74c3d74f3dcf4e839fda7e0b7b60d95acda3a3 event cem-evcem-DataUpdateCurrentPerPhase
```

### What type of operating system or environment does evcc run on?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

andig, 2025-01-04:
@rivengh ist das eine bekannte Einschränkung? Dann wäre es toll, wenn das noch im Template erwähnt werden könnte? Vielen Dank!

VolkerK62, 2025-01-04:
Das Setzen des `battery mode` ist doch eigentlich unabhängig von der Hardware, oder?

Ich frage mich wo das herkommt
`[site  ] DEBUG 2025/01/04 01:48:50 set battery mode: normal`

Egal wie ich das bei mir nachstelle, sobald die Batterie geladen wird, ist und bleibt  `battery mode` immer auf `charge`.

marleyan333, 2025-01-04:
> Das Setzen des `battery mode` ist doch eigentlich unabhängig von der Hardware, oder?
> 
> Ich frage mich wo das herkommt `[site ] DEBUG 2025/01/04 01:48:50 set battery mode: normal`
> 
> Egal wie ich das bei mir nachstelle, sobald die Batterie geladen wird, ist und bleibt `battery mode` immer auf `charge`.

So sollte es eigentlich sein. Aber warum geht man überhaupt diesen Weg, dass der Mode jedesmal über die API umgestellt wird? Man könnte die Batterie doch einfach im time of use Mode lassen, damit geht dann eh alles.

andig, 2025-01-04:
Warum willst du das überhaupt evcc machen lassen wenn Deine Batterie alles von selbst kann?

marleyan333, 2025-01-04:
> Warum willst du das überhaupt evcc machen lassen wenn Deine Batterie alles von selbst kann?

Die Steuerung der Fahrzeugladung kann sie nunmal nicht alleine 😉 und die Kombination Sonnenbuddy für Einstellung der Batterieladung einerseits und kostenabhängiger Ladung des Autos über evcc andererseits scheint sich zu beißen. Außerdem ist die Ladeststeuerung über die Porsche App eine totale Katastrophe, da kann ich nicht mal einfach die Ladung anhalten. Letzteres mache ich eben mit evcc. Außerdem ist evcc insgesamt die beste Software, die ich kenne, weil man eben alles unter Dach hat, da gibt es nichts vergleichbares. Also an dieser Stelle auch ein Kompliment an die Macher!

VolkerK62, 2025-01-04:
>Man könnte die Batterie doch einfach im time of use Mode lassen, damit geht dann eh alles.

Das mag für dich passen, aber die Implementierung in evcc ist hardware-unabhängig.
Je nach Bedingung wird der entsprechende Mode (normal, hold, charge) gesetzt und je nach Hardware auf unterschiedliche Art und Weise umgesetzt.

Deshalb wundert es auch, dass der Mode auf `normal`gesetzt wird, wenn der vorher auf `charge` war.
War er denn vorher auf `charge`? Das ist im Log nicht ersichtlich.

VolkerK62, 2025-01-04:
>die Kombination Sonnenbuddy für Einstellung der Batterieladung einerseits und kostenabhängiger Ladung des Autos über evcc andererseits scheint sich zu beißen.

Ah, jetzt wird ein Schuh draus. Die Batterieladung war nicht durch evcc getriggert, sondern extern???

marleyan333, 2025-01-04:
> > die Kombination Sonnenbuddy für Einstellung der Batterieladung einerseits und kostenabhängiger Ladung des Autos über evcc andererseits scheint sich zu beißen.
> 
> Ah, jetzt wird ein Schuh draus. Die Batterieladung war nicht durch evcc getriggert, sondern extern???

Nein, jetzt nicht mehr. Das hatte ich mal versucht, ging aber auch nicht.

marleyan333, 2025-01-04:
> Das Setzen des `battery mode` ist doch eigentlich unabhängig von der Hardware, oder?
> 
> Ich frage mich wo das herkommt `[site ] DEBUG 2025/01/04 01:48:50 set battery mode: normal`
> 
> Egal wie ich das bei mir nachstelle, sobald die Batterie preisgesteuert geladen wird, ist und bleibt `battery mode` immer auf `charge`.

Und du hast auch eine Sonnenbatterie?

VolkerK62, 2025-01-04:
Nein. 
Soweit ich das verstanden habe ist der Mode nicht von der Hardware abhängig.
Sondern nur davon ob das Auto "schnell" oder "preisgesteuert" geladen wird (Mode hold) oder die Batterie preisgesteuert geladen wird (Mode charge).
Wobei "Mode charge" alles andere übersteuert.
Aber das sind nur Beobachtungen aus Diskussionen und Tests mit meiner Dummy-Batterie. Evtl. liege ich da auch falsch.

marleyan333, 2025-01-04:
> Das mag für dich passen, aber die Implementierung in evcc ist hardware-unabhängig. Je nach Bedingung wird der entsprechende Mode (normal, hold, charge) gesetzt und je nach Hardware auf [...]

Also so geht es los:
Jan 04 00:00:19 raspberrypi evcc[1017982]: [site  ] DEBUG 2025/01/04 00:00:19 ----
Jan 04 00:00:19 raspberrypi evcc[1017982]: [lp-1  ] DEBUG 2025/01/04 00:00:19 charge power: 0W
Jan 04 00:00:19 raspberrypi evcc[1017982]: [lp-1  ] DEBUG 2025/01/04 00:00:19 charge currents: [0 0 0]A
Jan 04 00:00:19 raspberrypi evcc[1017982]: [site  ] DEBUG 2025/01/04 00:00:19 grid power: 6749W
Jan 04 00:00:19 raspberrypi evcc[1017982]: [site  ] DEBUG 2025/01/04 00:00:19 pv power: 0W
Jan 04 00:00:19 raspberrypi evcc[1017982]: [site  ] DEBUG 2025/01/04 00:00:19 battery soc: 0%
Jan 04 00:00:19 raspberrypi evcc[1017982]: [site  ] DEBUG 2025/01/04 00:00:19 battery power: -2281W
Jan 04 00:00:19 raspberrypi evcc[1017982]: [site  ] DEBUG 2025/01/04 00:00:19 battery has priority at soc 0% (< 100%)
Jan 04 00:00:19 raspberrypi evcc[1017982]: [site  ] DEBUG 2025/01/04 00:00:19 site power: 6849W
Jan 04 00:00:19 raspberrypi evcc[1017982]: [lp-1  ] DEBUG 2025/01/04 00:00:19 charger status: B
Jan 04 00:00:19 raspberrypi evcc[1017982]: [eebus ] DEBUG 2025/01/04 00:00:19 Send: d:_i:19667_PorscheEVSE-00025806 write 2842 LoadControlLimitListData
Jan 04 00:00:19 raspberrypi evcc[1017982]: [lp-1  ] DEBUG 2025/01/04 00:00:19 max charge current: 16A
Jan 04 00:00:19 raspberrypi evcc[1017982]: [eebus ] DEBUG 2025/01/04 00:00:19 Send: d:_i:19667_PorscheEVSE-00025806 write 2843 LoadControlLimitListData
Jan 04 00:00:19 raspberrypi evcc[1017982]: [lp-1  ] DEBUG 2025/01/04 00:00:19 charger enable
Jan 04 00:00:19 raspberrypi evcc[1017982]: [lp-1  ] DEBUG 2025/01/04 00:00:19 wake-up timer: start
Jan 04 00:00:19 raspberrypi evcc[1017982]: [lp-1  ] DEBUG 2025/01/04 00:00:19 pv timer elapse
Jan 04 00:00:19 raspberrypi evcc[1017982]: [lp-1  ] DEBUG 2025/01/04 00:00:19 pv timer inactive

Dann so:
an 04 00:00:49 raspberrypi evcc[1017982]: [site  ] DEBUG 2025/01/04 00:00:49 ----
Jan 04 00:00:49 raspberrypi evcc[1017982]: [lp-1  ] DEBUG 2025/01/04 00:00:49 charge power: 10610W
Jan 04 00:00:49 raspberrypi evcc[1017982]: [lp-1  ] DEBUG 2025/01/04 00:00:49 charge currents: [15.9 16 15]A
Jan 04 00:00:49 raspberrypi evcc[1017982]: [site  ] DEBUG 2025/01/04 00:00:49 grid power: 17323W
Jan 04 00:00:49 raspberrypi evcc[1017982]: [site  ] DEBUG 2025/01/04 00:00:49 pv power: 15W
Jan 04 00:00:49 raspberrypi evcc[1017982]: [site  ] DEBUG 2025/01/04 00:00:49 battery soc: 0%
Jan 04 00:00:49 raspberrypi evcc[1017982]: [site  ] DEBUG 2025/01/04 00:00:49 battery power: -2285W
Jan 04 00:00:49 raspberrypi evcc[1017982]: [site  ] DEBUG 2025/01/04 00:00:49 battery has priority at soc 0% (< 100%)
Jan 04 00:00:49 raspberrypi evcc[1017982]: [site  ] DEBUG 2025/01/04 00:00:49 site power: 17423W
Jan 04 00:00:49 raspberrypi evcc[1017982]: [lp-1  ] DEBUG 2025/01/04 00:00:49 charger status: C
Jan 04 00:00:49 raspberrypi evcc[1017982]: [lp-1  ] INFO 2025/01/04 00:00:49 start charging ->
Jan 04 00:00:49 raspberrypi evcc[1017982]: [lp-1  ] DEBUG 2025/01/04 00:00:49 wake-up timer: stop

Dann 1 min später geht Batterie auf "hold": -->Ladung Batterie bricht ab, Fahrzeug wird weiter geladen

Jan 04 00:01:19 raspberrypi evcc[1017982]: [site  ] DEBUG 2025/01/04 00:01:19 ----
Jan 04 00:01:19 raspberrypi evcc[1017982]: [lp-1  ] DEBUG 2025/01/04 00:01:19 charge power: 10612W
Jan 04 00:01:19 raspberrypi evcc[1017982]: [lp-1  ] DEBUG 2025/01/04 00:01:19 charge currents: [15.9 16 15.2]A
Jan 04 00:01:20 raspberrypi evcc[1017982]: [eebus ] DEBUG 2025/01/04 00:01:20 Send: d:_i:19667_PorscheEVSE-00025806 notify 2874 DeviceDiagnosisHeartbeatData
Jan 04 00:01:20 raspberrypi evcc[1017982]: [site  ] DEBUG 2025/01/04 00:01:20 set battery mode: hold
Jan 04 00:01:20 raspberrypi evcc[1017982]: [site  ] DEBUG 2025/01/04 00:01:20 grid power: 17279W
Jan 04 00:01:20 raspberrypi evcc[1017982]: [site  ] DEBUG 2025/01/04 00:01:20 pv power: 10W
Jan 04 00:01:20 raspberrypi evcc[1017982]: [site  ] DEBUG 2025/01/04 00:01:20 battery soc: 1%
Jan 04 00:01:20 raspberrypi evcc[1017982]: [site  ] DEBUG 2025/01/04 00:01:20 battery power: -2265W
Jan 04 00:01:20 raspberrypi evcc[1017982]: [site  ] DEBUG 2025/01/04 00:01:20 battery has priority at soc 1% (< 100%)
Jan 04 00:01:20 raspberrypi evcc[1017982]: [site  ] DEBUG 2025/01/04 00:01:20 site power: 17379W
Jan 04 00:01:20 raspberrypi evcc[1017982]: [lp-1  ] DEBUG 2025/01/04 00:01:20 detected active phases: 3p
Jan 04 00:01:20 raspberrypi evcc[1017982]: [lp-1  ] DEBUG 2025/01/04 00:01:20 charger status: C

Um 01:35 passiert etwas unerwartetes: Batterie lädt nun wieder, ohne dass zuvor der Battery mode von hold geändert worden wäre:

Jan 04 01:35:49 raspberrypi evcc[1017982]: [site  ] DEBUG 2025/01/04 01:35:49 ----
Jan 04 01:35:49 raspberrypi evcc[1017982]: [lp-1  ] DEBUG 2025/01/04 01:35:49 charge power: 10691W
Jan 04 01:35:49 raspberrypi evcc[1017982]: [lp-1  ] DEBUG 2025/01/04 01:35:49 charge currents: [15.9 15.9 15.4]A
Jan 04 01:35:49 raspberrypi evcc[1017982]: [site  ] DEBUG 2025/01/04 01:35:49 grid power: 18532W
Jan 04 01:35:49 raspberrypi evcc[1017982]: [site  ] DEBUG 2025/01/04 01:35:49 pv power: 0W
Jan 04 01:35:49 raspberrypi evcc[1017982]: [site  ] DEBUG 2025/01/04 01:35:49 battery soc: 1%
Jan 04 01:35:49 raspberrypi evcc[1017982]: [site  ] DEBUG 2025/01/04 01:35:49 battery power: -3300W
[... cut ...]
Jan 04 04:27:50 raspberrypi evcc[1017982]: [eebus ] DEBUG 2025/01/04 04:27:50 ski bb74c3d74f3dcf4e839fda7e0b7b60d95acda3a3 event cem-evcem-DataUpdateEnergyCharged
Jan 04 04:27:50 raspberrypi evcc[1017982]: [site  ] DEBUG 2025/01/04 04:27:50 battery soc: 1%
Jan 04 04:27:50 raspberrypi evcc[1017982]: [site  ] DEBUG 2025/01/04 04:27:50 battery power: 240W
Jan 04 04:27:50 raspberrypi evcc[1017982]: [site  ] DEBUG 2025/01/04 04:27:50 site power: 385W
Jan 04 04:27:50 raspberrypi evcc[1017982]: [lp-1  ] DEBUG 2025/01/04 04:27:50 charger status: B

VolkerK62, 2025-01-04:
warum lädt um 0:19 Uhr die Batterie?
Gibt es davor ein `set battery mode: charge`?

>Um 04:27:29_
Jan 04 04:27:29 raspberrypi evcc[1017982]: [site ] DEBUG 2025/01/04 04:27:29 set grid charge limit: 0.2
Um 04:27:49: Jetzt wechselt der battery mode auf charge, ich denke das war die Rektion auf mein Deaktivieren des "Entladungsverhinderungsbuttons": --> ab hier lädt es dann regelrecht bis 05:00 = Ende des festgelegten Ladezeitraums

Ich denke eher, der Mode geht auf charge wegen `set grid charge limit: 0.2`

VolkerK62, 2025-01-04:
Das muss fremd-gesteuert sein.
Warum sonst wird bei bei "mode normal"  wieder geladen?
```
Jan 04 01:48:50 raspberrypi evcc[1017982]: [site ] DEBUG 2025/01/04 01:48:50 set battery mode: normal
Jan 04 01:48:50 raspberrypi evcc[1017982]: [site ] DEBUG 2025/01/04 01:48:50 battery power: -3301W
```

marleyan333, 2025-01-04:
> Das muss fremd-gesteuert sein. Warum sonst wird bei bei "mode normal" wieder geladen?
> 
> ```
> Jan 04 01:48:50 raspberrypi evcc[1017982]: [site ] DEBUG 2025/01/04 01:48:50 set battery mode: normal
> Jan 04 01:48:50 raspberrypi evcc[1017982]: [site ] DEBUG 2025/01/04 01:48:50 battery power: -3301W
> ```

Ja, das verstehe ich auch nicht... werde heute mal die app sonenbuddy komplett deinstallieren, um auszuschließen, dass die da reinfunkt. Wobei ich sie eh schon deaktiviert hatte, aber vielleicht reicht das nicht?!

marleyan333, 2025-01-04:
> warum lädt um 0:19 Uhr die Batterie? Gibt es davor ein `set battery mode: charge`?
>
>
> Ich denke eher, der Mode geht auf charge wegen `set grid charge limit: 0.2`

Wie kommt es zu diesem "set grid charge limit"? Und was genau bewirkt das?

marleyan333, 2025-01-04:
> warum lädt um 0:19 Uhr die Batterie? Gibt es davor ein `set battery mode: charge`?
>
>
> Ich denke eher, der Mode geht auf charge wegen `set grid charge limit: 0.2`

Auf deine Frage oben: gibt nur: set battery mode: hold bei 00:01:20

Jan 04 00:01:19 raspberrypi evcc[1017982]: [site  ] DEBUG 2025/01/04 00:01:19 ----
Jan 04 00:01:19 raspberrypi evcc[1017982]: [lp-1  ] DEBUG 2025/01/04 00:01:19 charge power: 10612W
Jan 04 00:01:19 raspberrypi evcc[1017982]: [lp-1  ] DEBUG 2025/01/04 00:01:19 charge currents: [15.9 16 15.2]A
Jan 04 00:01:20 raspberrypi evcc[1017982]: [eebus ] DEBUG 2025/01/04 00:01:20 Send: d:_i:19667_PorscheEVSE-00025806 notify 2874 DeviceDiagnosisHeartbeatData
Jan 04 00:01:20 raspberrypi evcc[1017982]: [site  ] DEBUG 2025/01/04 00:01:20 set battery mode: hold
Jan 04 00:01:20 raspberrypi evcc[1017982]: [site  ] DEBUG 2025/01/04 00:01:20 grid power: 17279W
Jan 04 00:01:20 raspberrypi evcc[1017982]: [site  ] DEBUG 2025/01/04 00:01:20 pv power: 10W
Jan 04 00:01:20 raspberrypi evcc[1017982]: [site  ] DEBUG 2025/01/04 00:01:20 battery soc: 1%
Jan 04 00:01:20 raspberrypi evcc[1017982]: [site  ] DEBUG 2025/01/04 00:01:20 battery power: -2265W
Jan 04 00:01:20 raspberrypi evcc[1017982]: [site  ] DEBUG 2025/01/04 00:01:20 battery has priority at soc 1% (< 100%)
Jan 04 00:01:20 raspberrypi evcc[1017982]: [site  ] DEBUG 2025/01/04 00:01:20 site power: 17379W
Jan 04 00:01:20 raspberrypi evcc[1017982]: [lp-1  ] DEBUG 2025/01/04 00:01:20 detected active phases: 3p
Jan 04 00:01:20 raspberrypi evcc[1017982]: [lp-1  ] DEBUG 2025/01/04 00:01:20 charger status: C

rivengh, 2025-01-04:
> Das muss fremd-gesteuert sein. Warum sonst wird bei bei "mode normal" wieder geladen?
> 
> ```
> Jan 04 01:48:50 raspberrypi evcc[1017982]: [site ] DEBUG 2025/01/04 01:48:50 set battery mode: normal
> Jan 04 01:48:50 raspberrypi evcc[1017982]: [site ] DEBUG 2025/01/04 01:48:50 battery power: -3301W
> ```

Der Ladevorgang könnte durch die Batteriepflege der Sonnenbatterie gestartet worden sein. Wenn die Batterie länger als 30 Tage nicht vollgeladen wurde, wird eine Vollladung initiiert. Das kommt gerade in der dunklen Jahreszeit durchaus mal vor.

marleyan333, 2025-01-04:
> Der Ladevorgang könnte durch die Batteriepflege der Sonnenbatterie gestartet worden sein. Wenn die Batterie länger als 30 Tage nicht vollgeladen wurde, wird eine Vollladung initiiert. Das kommt [...]

Dagegen spricht, dass ich das Problem ja jede Nacht habe...