evcc-io/evcc #18057: Huawei Luna: AC-Laden wird nach Netzladen deaktiviert

jori0001, 2025-01-05:
### Describe the bug

Nachdem die Huawei Luna Batterie aus dem Netz geladen wurde, deaktiviert EVCC das AC-Laden. Das muss bei kaskadierten Anlagen (>1 Wechselrichter) allerdings immer aktiviert sein. Sonst wird die Batterie nur vom primären Wechselrichter geladen, was insbesondere im Winter problematisch ist, da man eh nur schwer die Batterie wieder voll bekommt. 

### Steps to reproduce

Luna aus dem Netz laden

### Configuration details

```yaml
Wird geliefert, falls benötigt
```

### Log details

```text
Wird geliefert, falls benötigt
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

v0.132.0

VolkerK62, 2025-01-05:
Den Parameter `forceaccharging` kennst du?
Doku https://docs.evcc.io/docs/devices/meters#sun2000  -Batterie ->show advanced options

closed 2025-01-05 by andig

jori0001, 2025-01-05:
Schande über mein Haupt. Das habe ich nicht gekannt. Sorry. 🤦🏼‍♂️