evcc-io/evcc #18065: 32A Berechnung bei Fahrzeugen die nur 16A können und 3 bzw. 1 Phasig Unterscheidung

Schimi1983, 2025-01-05:
HI,

mein Ladepunkt (Alfen Single-pro) kann 22kw...

der ID.5 (und auch ein Tesla) bei 3 Phasen "nur" 11kw....

Wenn die Wallbox auf 32A steht (22kw)... berechnet er die ladedauer mit einer ladeleistung von 22KW
[image]

natürlich kann ich das fürs Fahrzeug in der config auf 16A beschränken (dann berechnet er das richtig)...

Problem ist dann aber... wenn er 1-Phasig laden soll (PV-Überschuss z.B.) dann können beide Fahrzeuge (ID.5 und Tesla ModelY) 32A verarbeiten (ca. 7,4kw)...
Limitiere ich aber in der config die Fahrzeuge auf 16A, dann wird bei einphasigem laden auch nur max mit 16A geladen....

Wäre also cool das man für ein und 3 Phasig eine max. Ampere eingeben kann, oder die Wallboxen versuchen (wenn die Wallbox  es her gibt) bei 1-Phasig immer die 32A zu erreichen bevor sie auf 3-Phasig wechselt...?

Ich hoffe es ist einigermaßen verständlich, was ich meine :-)

mucki12, 2025-01-05:
Ja - ist ein Problem. Damit die Berechnung passt bin ich mittlerweile wieder auf 16A zurück. Mit Phasenumschaltung hat man dann zwar eine kleine Lücke beim Wechsel von 1- auf 3-phasig, aber in Deutschland sind 1-phasig ja eh nur 20A erlaubt.

premultiply, 2025-01-05:
Das Problem lässt sich normalerweise gut über ein LM-Circuit entschärfen.
Limit des Circuits der Box auf 11 kW und auf 20A setzen.
Die Box muss natürlich dann dauerhaft auf den höchsten Phasenstrom (z. B. hier 20A) eingestellt werden.
Im dreiphasigen Modus wird die Leistung und somit der Phasenstrom via LM beschränkt.

Bei mir sieht das LM z. B. so aus:
```yaml
circuits:
- name: main
  maxCurrent: 50
  maxPower: 34500
  meter: shm
- name: wallbox1
  maxCurrent: 20
  maxPower: 12000
  meter: charger
  parent: main
```

andig, 2025-01-05:
> berechnet er die ladedauer

Wovon reden wir hier konkret? Mangels Logfile geschlossen.

andig, 2025-01-05:
> Ich hoffe es ist einigermaßen verständlich, was ich meine :-)

Klar ist nur, das jegliche angefragte Information fehlt.

closed 2025-01-05 by andig