evcc-io/evcc #18073: Geladene kWh für Ladevorgang falsch ausgewertet

Schedl-Michael, 2025-01-06:
### Describe the bug

Obwohl Zählerstände zu Beginn und Ende des Ladevorgangs richtig erkannt wurden, hat evcc falsch gerechnet.
Der Wert für die geladene Energiemenge wird uns nicht nachvollziehbaren Gründen mit mehr als 4 Mio kWh angegeben, was natürlich die ganze Auswertung unbrauchbar macht.
[image]
[image]
Kann ich als Anwender die Eintragung des Ladevorgangs in der Datenbank ändern oder bleibt mir als Laie nur die Möglichkeit den betreffenden Ladevorgang zu löschen?
Im Moment habe ich noch die Hoffnung, dass sich mit dem nächsten regulären Ladevorgang mit alles einpendelt und evcc die Fehlauswertung nachträglich berichtigt. (?)
Oder wird die Differenz der Zählerstände am Ende des Ladevorgangs ermittelt und fix in den Datensatz übernommen?

### Steps to reproduce

Da es sich um eine allgemeine Anfrage handelt, habe ich auf diese Angaben verzichtet.

### Configuration details

```yaml
Da es sich um eine allgemeine Anfrage handelt, habe ich auf diese Angaben verzichtet.
Falls doch erforderlich, reiche ich die evcc.yaml noch nach.
```

### Log details

```text
Da es sich um eine allgemeine Anfrage handelt, habe ich auf diese Angaben verzichtet.
```

### What type of operating system or environment does evcc run on?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

v0.132.0

StefanSchoof, 2025-01-06:
Ist ein bisschen komplizierter. Einstieg für den Hintergrund https://github.com/evcc-io/evcc/issues/13726

Schedl-Michael, 2025-01-06:
Ergänzend möchte ich noch hinzufügen, dass bei dem betreffenden Ladevorgang ohne Unterbrechung im Modus "Schnell" geladen wurde. Die PV-Anlage lag unter Schnee, weshald der Sonnenanteil 0,0% betragen hat.

closed 2025-01-06 by premultiply

tm-107, 2025-01-08:
Auch wenn das hier (kommentarlos) geschlossen wurde, glaube ich nicht unbedingt, dass es mit https://github.com/evcc-io/evcc/issues/13726 zu tun hat.
Konfiguration und Logfile wären aber trotzdem hilfreich gewesen ;-)

StefanSchoof, 2025-01-08:
> Konfiguration und Logfile wären aber trotzdem hilfreich gewesen ;-)

Das ist ja auch der Grund. Aus dem Bug Template das man beim Anlegen:

> NOTE Due to amount of issues raised we will close incomplete issues (unclear description what the issue actually is, missing log file, unsupported platform) without further comments.

Schedl-Michael, 2025-01-09:
Die Basisdaten sind ja irgendwo vorhanden: Zählerstand zu Ladebeginn und Zählerstand bei Ladeende.
**Ist es für die Entwickler von evcc denkbar, im UI eine Schaltfläche zu implementieren, die eine neue Auswertung ermöglicht?**
Wenn die Differenz neu gebildet wird, passt für mich alles wieder.
Ansonsten bleibt mir nur, auf 22,3 kWh für einen Ladevorgang in der Statistik zu verzichten, indem ich diese Position lösche. ;-)