evcc-io/evcc #18077: Not possible to have 2 evcc instances with eebus enabled

dog-rgb, 2025-01-06:
### Describe the bug

When I want to connect from another eebus device (in my case a vaillant heating pump), I am not able to select the correct evcc eebus interface as only one "EVCC_HEMS_01" is shown up in the vaillant app. The shown ski belongs to one of the evcc instances, but it is not possible to get the other one when both evcc instances are running.

### Steps to reproduce

1. Having 2 instances of evcc in the same local network, both have eebus configured via "evcc eebus-cert"
2. Open the other eebus device app to setup the connection to evcc
3. There is only one item called "EVCC_HEMS_01" visible

### Configuration details

```yaml
------------------------------------------------------------------- Instance 1:

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc14.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: <>

interval: 30s # control cycle interval

# database configuration for persisting charge sessions and settings
database:
  type: sqlite
  dsn: /evcccfg/evcc.db

# evcc Sponsortoken, dog-rgb bei Github, 15000 Tage gültig (41 Jahre ab 29.06.2024)
sponsortoken: <>

# mqtt message broker
mqtt:
  broker: 192.168.178.21:1883
  topic: evcc-14 # root topic for publishing, set empty to disable publishing
  # clientid: foo
  user: <>
  password: <>

meters:
- type: template
  template: victron-energy 
  usage: grid  
  host: 192.168.178.38  
  port: 502  
  name: grid1
- type: template
  template: victron-energy 
  usage: pv  
  host: 192.168.178.38  
  port: 502  
  name: pv2
- type: template
  template: victron-energy 
  usage: battery  
  host: 192.168.178.38  
  port: 502  
  name: battery3

chargers:
- type: template
  template: openwb 
  host: 192.168.178.32  
  connector: 1  
  phases1p3p: true  
  name: wallbox5
- type: template
  template: go-e-v3 
  host: 192.168.178.33  
  name: wallbox1
#  phases1p3p: true  

vehicles:
- type: template
  template: enyaq 
  title: Enyaq  
  icon: car  
  user: <>
  password: <>  
  phases: 3  
  cache: 15m  
  mode: pv  
  minCurrent: 6  
  maxCurrent: 16  
  timeout: 10s  
  name: ev4

- type: template
  template: vw 
  title: e-Up
  icon: van
  user: <>
  password: <>  
  phases: 2
  mode: pv
  capacity: 32  
  timeout: 10s  
  name: ev5

- type: template
  template: offline 
  title: Gast 1p  
  phases: 1  
  mode: off
  name: ev1

- type: template
  template: offline 
  title: Gast 2p  
  phases: 2  
  mode: off
  name: ev2

- type: template
  template: offline 
  title: Gast 3p  
  phases: 3  
  mode: off
  name: ev3

loadpoints:
- title: Großer Carport
  charger: wallbox5
  mode: off
- title: Kleiner Carport
  charger: wallbox1
  mode: off

site:
  title: Haus 14
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.3051 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.0761 # [currency]/kWh

eebus:
  certificate:
    public: |
      -----BEGIN CERTIFICATE-----
<>      -----END CERTIFICATE-----
      
    private: |
      -----BEGIN EC PRIVATE KEY-----
<>      -----END EC PRIVATE KEY-----

  shipid: EVCC-1400000000000000

------------------------------------------------------------------- Instance 2:

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc14a.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: <>

interval: 10s # control cycle interval

# database configuration for persisting charge sessions and settings
database:
  type: sqlite
  dsn: /evcccfg/evcc.db

# mqtt message broker
mqtt:
  broker: 192.168.178.21:1883
  topic: evcc # root topic for publishing, set empty to disable publishing
  # clientid: foo
  user: <>
  password: <>

meters:
  - name: sungrow-grid
    type: template
    template: sungrow-hybrid
    usage: grid
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.178.89 # Hostname
    port: 502 # Port
    timeout: 10s # optional#- type: template
  - name: sungrow-pv
    type: template
    template: sungrow-hybrid
    usage: pv
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.178.89 # Hostname
    port: 502 # Port
    timeout: 10s # optional#  template: shelly-3em 
  - name: sungrow-bat
    type: template
    template: sungrow-hybrid
    usage: battery
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.178.89 # Hostname
    port: 502 # Port
    timeout: 10s # optional

  - name: shelly4elli
    type: template
    template: shelly-3em
    usage: charge
    host: 192.168.178.88 # IP-Adresse oder Hostname
    
chargers:
- type: template
  template: ellipro 
  ski: <>
  ip: 192.168.178.34  
  name: wallbox4

vehicles:
- type: template
  template: enyaq 
  title: Enyaq  
  icon: car  
  user: <>
  password: <>  
  phases: 3  
  cache: 15m  
  mode: pv  
  minCurrent: 6  
  maxCurrent: 16  
  timeout: 10s  
  name: ev4

- type: template
  template: vw 
  title: e-Up
  icon: van
  user: <>
  password: <>
  phases: 2  
  mode: pv
  capacity: 32  
  timeout: 10s  
  name: ev5

- type: template
  template: offline 
  title: Gast 1p  
  phases: 1  
  name: ev1

- type: template
  template: offline 
  title: Gast 2p  
  phases: 2  
  name: ev2

- type: template
  template: offline 
  title: Gast 3p  
  phases: 3  
  name: ev3

loadpoints:
- title: Carport 14A
  charger: wallbox4
  meter: shelly4elli
  mode: off

site:
  title: Haus 14A
  meters:
    grid: sungrow-grid
    pv:
    - sungrow-pv
    battery:
    - sungrow-bat

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.359 # [currency]/kWh
  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.0735 # [currency]/kWh

eebus:
  certificate:
      private: |
          -----BEGIN EC PRIVATE KEY-----
<>          -----END EC PRIVATE KEY-----
      public: |
          -----BEGIN CERTIFICATE-----
<>          -----END CERTIFICATE-----
```

### Log details

```text
N/A
```

### What type of operating system or environment does evcc run on?

Docker container

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.11

andig, 2025-01-06:
Would this work for you?

dog-rgb, 2025-01-06:
Sounds good in general. But for me it is not clear where to put the name definition for EEBUS_DEVICE_CODE. Into the evcc.yaml or to the environment variables or ...?

andig, 2025-01-06:
Goes into environment.

dog-rgb, 2025-01-06:
Fine for me, thanks.
So I will set it before creating the certificate.