evcc-io/evcc #18082: BMW, PulsarPlus - MinPV mit max. Leistung

wpb52, 2025-01-06:
### Describe the bug

Ich habe 2 Testreihen durchgeführt und einen Fehler und ein Workarround gefunden. Ziel ist es `Min+PV` ein zu stellen.
`Min+PV` nach `Aus `funktioniert nicht, es wird immer mit 16A geladen (siehe unten)

Mit Version A komme ich nie auf 6A. 
Version B funktioniert immer.

### Steps to reproduce

A) `Aus` =0A  -> **`MinPV = 16A`**   -> `Schnell `=16A -> **`MinPV  =16A`**
B) `Aus` =0A ->  `Schnell =16A` -> **`MinPV  = 6A`**

Die Fehlermeldung der Reihe A. 
[image]

In Reihe B gibt es keine Fehlermeldungen.

### Configuration details

```yaml
Konfiguration (/etc/evcc.yaml):

log: debug
levels:
  #cache: trace
  ocpp: trace
  bmw: trace
  lp-1: trace
  loadpoints: trace

interval: 36s #control cycle interval ggf. reduzieren

sponsortoken: *****

site:
  title: Home # display name for UI
  meters:
    grid: my_grid
    pv:   my_pv

loadpoints:
  - title: Garage # display name for UI
    charger: my_charger # charger
    vehicle: my_car # default vehicle

meters:
  - name: my_grid
    type: template
    template: sma-home-manager
    usage: grid
    host: 192.168.188.32 # IP-Adresse oder Hostname
    interface: eth0 # Netzwerkschnittstelle (optional)

  - name: my_pv
    type: template
    template: sma-inverter-modbus
    usage: pv                   # Modbus TCP
    modbus: tcpip
    id: 3
    host: 192.168.188.31        # Hostname
    port: 502                   # Port

chargers:
  - name: my_charger
    type: template
    template: pulsarplus
    remotestart: true           # wieder aktiv seit 2024.10.30 16:35 mfuchs1984
    stationid: Garage           
    metervalues: Energy.Active.Import.Register,Power.Active.Import,SoC,Current.Offered,Power.Offered,Current.Import,Voltage

vehicles:
  - name: my_car
    type: template
    template: bmw
    title: BMWiX
    user: *****
    password: *****
    capacity: 76 # in kWh
    hcaptcha: P1_eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.hadwYXNza2V5xQQMKWVzSMDlQ6f_3yAMXYRPt0Wnb5EYHUgSxDyYION1mASsu-ctHriur-4UwkfCB1sJ3cQmgy04huo9bSb3s4xFL-rODrvZOu5CGFMe6BslP_pqlRWbnNuEX7Yk7PYPabTQevHLsYkTKO74Ih73ItpTNyg2xltdOuqGcRgG3usJQbUTgfUMYFY9T52tRvwcWkUPy92j8gcM4bi_3m37nyyjAwJ-JEpc9JC_rE1XkqNpQkSBUECpyUOJRZ7A-RO8LNY5UsHV9o8Csr3SYcOIyvx5IrDlrZSW1yrl_uc6V3YmzNJoAEyAhI_-SuImMGtHn-PFUSY7weNfsvZgrguM-xiqQhoyRkg60WIyaCQtoYh5BU5vL1LB_Cg-MxDMORlIOaNZvUm_dmLoLtvgUSe6TXHhzQuZPWhhqqlTVMSeREVRDLtfQZSQ4g28_joMGdVem680iRRzbx8VLL6IVvXzEFjqof5SH3zE0QtX7ndXfytOaukpSEeFA0qd-ICl8H9z_5Ews5DiB1X_iDV-O4ZcgR3tTPHPEnhRxHQj9W0gC7hLwRDixbt48v79TH8Fzya1k3easwTUP2QQIKNDvS-BtvEv9L6wCenhpBIApR0-kzA1S6NTiLrrzjkbrcCMTatAtpcqAo-kJ96-Da1NlrBHJOMBdHy_XMxpsmQn0jzVzTJ1r2k9ss8rXKP5cjK9PFY50BVi3ox_OtdB0eL8585y0pg11lSZjWsa12jQG9Cq1-RPb9Z90fzS6JmGDR91fH1cK_aKRTtj4tvUeSAl1AwCMpd2amkieyC9Uv8FVmVa89XH0XXvNiWgjtIQoAkjhGyBO45hvHewFRvJTogfmZn-Sr24uZ8RrxZZzWXpoTnwq6BddbvK1Ot65M7GzPhSuehv3HUbjag7UHICwRyr1JcWfLAcnjdT45xWqm7kGFWF8St0crmgzvEwCQrDwmcPdL0eRFlJm3L36orQkzopVHRICRRa_OvXH5rD4n7heIN_rvfwjhGOKWBYStjE623LuFEZehavOb_Yk1e5doVIK-6j_HpDHowMpemFgBVtDMIPGtQkE3mq5FEVEpW0UyN6AtVUs9sEtwXRh62XnDuGuD5ZDR92SY9z8nigOvTlwUG2l2DivzBKtYq4gDJL2cdiEhp83IyQvc-IKK96uDLZKqp5UoHzw3srCwT7HyxNrlmsFB2tVIZ1HBW9xuKNe48Ao3Zk0JAXR-_2oQSFc62QL1a04kSgfJxUS1meuEbVS6d-VW9qa-58EuNRGnWziBgvfjzOhbC3hLyHvSxi1ueEnFzhbaA9Wu6cROSckul-ENTt-YhEAzq44VN5XX_PmN31pMAohSDXhhZHD8U4D9fx4aPAjHwdxFYaM7_gKWgbzYI1U6NleHDOZ3in2KhzaGFyZF9pZM4UPIQfomtyqDMxZWEyNTVkonBkAA.-pUrBqpx0ivVErQYZMfAwz_vbol2sO5KSjRR42FWtTk

tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.35 # EUR/kWh
  feedin:
    type: fixed
    price: 0.10 # EUR/kWh

Version: `0.132.0`
```

### Log details

```text
Die log-Datei liefere ich bei Bedarf nach.

Da evcc gerade mal läuft habe ich kein Nightly getestet. Sorry für den falschen Haken.
```

### What type of operating system or environment does evcc run on?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.132.0

closed 2025-01-06 by andig

wpb52, 2025-01-06:
Here is the log. Test started 16:05 LondonTime.

[2025.01.06 16h05 evcc.log](https://github.com/user-attachments/files/18322097/2025.01.06.16h05.evcc.log)

andig, 2025-01-06:
> MinPV = 16A

Why do you have a minpv current of 16A?

>  metervalues:

Why are you configuring this?

wpb52, 2025-01-06:
_Why do you have a minpv current of 16A?_  this is my question: Why do I have 16A instead of 6A in Seires A and why does it work in series B korrectly.

_metervalues_  it was recommended to use this option.

Firmware of PulsarPlus is fw-6. I think it doesn't matter.

In my opinion there is a loop with start like A) and the 6A were not set right. In B) it works.
Maybe if the WB has a failure I need a logic way to force the company to exchange it on their cost.

andig, 2025-01-06:
> Why do I have 16A instead of 6A in Seires A and why does it work in series B korrectly.

I guess the WB wrongly reports offered current 16A. The fw5 template fixed this by adding

    metervalues: -Current.Offered,Power.Offered

wpb52, 2025-01-06:
Excuse my short way to **_Kepner Tregoe_** solving a problem. 
The main question there is:   when does it work - and when does it NOT work.
This is shown above in series B - A. All tests are done in 1p.
So it is not depending on PulsarPlus or fw but a logic prob of evcc. 
Do you agree?

I can perform some tests if you define it.

wpb52, 2025-01-06:
The power (A) is measured parallel to the delayed evcc display.

mfuchs1984, 2025-01-06:
The log shows that evcc is sending a limit of 6 Ampere but the wallbox stays at 16, not only Current.Offered, but also Current.Import. Same applies to Power.Offered and Power.Active.Import.

In case I remember correctly, I have not seen a single Pulsar Plus with FW 6.4.X running correctly with evcc here yet. 5.X and 6.1.19 seem to run well but both need the pulsarplus-fw5 template.

Some people tried to contact the vendor to fix it but were not yet successful.

https://github.com/evcc-io/evcc/issues/15490
https://github.com/evcc-io/evcc/discussions/15555#discussioncomment-10403296

I'd recommend to report the bug to the vendor and when possible, to go back to an older firmware.

wpb52, 2025-01-07:
Den Hersteller stellen wir erst mal nach hinten, da fehlen mMn. belastbare, reproduzierbare Fehlerbilder.

Ich werde also im 1. Schritt evcc stoppen, dann das template pulsarplus-fw5 testen...
Im weiteren Schritt ein factory reset der Wallbox machen, mal sehen welche Version bei der Pulsarplus kommt.

mfuchs1984, 2025-01-07:
Wenn die Wallbox ein Set ChargingProfile akzeptiert, muss sie sich daran halten, da gibt's keine Diskussion, das ist belastbar.

mfuchs1984, 2025-01-07:
Factory reset nur machen, wenn du weiterhin Probleme hast, fürs erste reicht ein restart der Wallbox.

mfuchs1984, 2025-01-18:
Wallbox.com hat eine neue Firmwareversion für die Pulsar Plus freigegeben, 6.7.17
https://support.wallbox.com/de/rn/veroeffentlichung-vom-13-januar-2025/

Evtl. hilft die ja.

wpb52, 2025-01-18:
Danke für die Info. Mit 1p läuft es gerade über Stunden ohne 'max. Leistung'. Morgen kann ich 3p testen. Das warte ich noch ab bevor ich update.