evcc-io/evcc #18093: Direkte Verlinkung auf Loadpoint/Sichtbarkeit des ausgewählten Loadpoints

cschlipf, 2025-01-06:
**Is your feature request related to a problem? Please describe.**
Ich habe ein Grafana Dashboard auf dem zwei Gauges für die jeweiligen Loadpoints sind. Nun würde ich gerne die Loadpoints gerne direkt verlinken. Auf dem Desktop kein Problem - hier sieht man immer alle Loadpoints auf einen Blick. Auf dem Smartphone sieht man aber immer den ersten Loadpoint.

Mein Dashboard:
[image]

Hier würde ich gerne auf den Stellplatz klicken um ihn direkt in EVCC anzuzeigen. Das Ergebnis ist aber immer, dass die Garage erscheint und ich immer erst swippen muss:

[image]

**Describe the solution you'd like**

Es wäre super, wenn man per URL parameter den angezeigten loadpoint konfigurieren könnte. Zum Beispiel ein URL parameter `loadpoint=2` um gleich initial den zweiten Loadpoint anzuzeigen:

[image]

closed 2025-01-06 by andig

cschlipf, 2025-01-06:
@andig - gibt es einen solchen Parameter schon? Falls ja, wie lautet der?

VolkerK62, 2025-01-07:
Hört sich an wie https://github.com/evcc-io/evcc/issues/18035

cschlipf, 2025-01-07:
> Hört sich an wie #18035

Dankeschön!