evcc-io/evcc #18100: Meter: Huawei SUN2000 - Second battery is not measured - Only battery_1

heinemannj, 2025-01-07:
### Describe the bug

Connection to inverter via modbus tcpip:
- Connect to inverter's internal WLAN (NOT to SDongle)
- Connect via modbusProxy

Setup:

```
  #
  # Huawei SUN2000-6KTL-M1 (Hybrid inverter, High current)
  # - Huawei DTSU666-H Smart Power Sensor
  # - Huawei LUNA2000-10-S0 Battery 10kWh
  # - Huawei LUNA2000-5-S0 Battery 5kWh
  #
```

**Second battery is not measured - Only battery_1.**

### Steps to reproduce

See  configuration details.

My Huawei Solar PV system is fully integrated within Home Assistant via huawei_solar integration by [wlcrs](https://github.com/wlcrs/huawei_solar).

[image]

HA integration is clearly indicating that evcc is only measuring `battery_1` and not `battery_2`.
Please use within evcc the huawei modbus registers for `batteries` and not only for `battery_1`.

[image]

[image]

[image]

[image]

### Configuration details

```yaml
root@evcc:~# evcc dump --cfg
[main  ] INFO 2025/01/07 09:44:47 evcc 0.132.0 (b728e23c)
[main  ] INFO 2025/01/07 09:44:47 using config file: /etc/evcc.yaml
[db    ] INFO 2025/01/07 09:44:47 using sqlite database: /root/.evcc/evcc.db
[eebus ] INFO 2025/01/07 09:44:47 Local SKI: xxx
[eebus ] DEBUG 2025/01/07 09:44:47 starting websocket server on :4712
[eebus ] ERROR 2025/01/07 09:44:47 websocket server error: listen tcp :4712: bind: address already in use
[eebus ] DEBUG 2025/01/07 09:44:47 mdns: announce
[eebus ] DEBUG 2025/01/07 09:44:47 mdns: using zeroconf
[eebus ] DEBUG 2025/01/07 09:44:47 mdns: new - ski: xxx name: vr921_21231200202609620933046941N4 brand: Vaillant model: VR921 typ: Gateway identifier: 21231200202609620933046941N4 register: false host: vr921-21231200202609620933046941N4.local. port: 12480 addresses: [192.168.178.143]
[circuit-steuVE_circuit] DEBUG 2025/01/07 09:44:49 validation of max power disabled
[circuit-steuVE_circuit] DEBUG 2025/01/07 09:44:49 validation of max phase current disabled
[lp-1  ] DEBUG 2025/01/07 09:44:50 charge total import: 0.000kWh

Konfiguration (/etc/evcc.yaml):

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local# .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: *****

[... 302 lines cut ...]
em_2_c
------
Power:  0W
Energy: 0.0kWh

em_1_c
------
Power:  0W
Energy: 0.0kWh

[eebus ] DEBUG 2025/01/07 09:35:02 mdns: stop announcement
root@evcc:~#
```

### What type of operating system or environment does evcc run on?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.132.0 (b728e23c)

premultiply, 2025-01-07:
> ```
>   - name: battery_1
>     type: template
>     template: huawei-sun2000
>     id: 1
>     host: 192.168.178.149
>     port: 502
>     usage: battery
>     modbus: tcpip
>     timeout: 15s
>     storageunit: 1
>     capacity: 15.0
> ```

There is just one unit configured in your configuration file.

closed 2025-01-07 by premultiply

heinemannj, 2025-01-07:
Huawei
SUN2000MA
V100R001C00
Modbus Interface Definitions
Issue 08
Date 2024-07-04

[Solar_Inverter_Modbus_Interface_Definitions_2024-07-18 (4).pdf](https://github.com/user-attachments/files/18330440/Solar_Inverter_Modbus_Interface_Definitions_2024-07-18.4.pdf)

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

heinemannj, 2025-01-07:
> > ```
> >   - name: battery_1
> >     type: template
> >     template: huawei-sun2000
> >     id: 1
> >     host: 192.168.178.149
> >     port: 502
> >     usage: battery
> >     modbus: tcpip
> >     timeout: 15s
> >     storageunit: 1
> >     capacity: 15.0
> > ```
> 
> There is just one unit configured in your configuration file.

Can you please verify that the following config is compliant:

```
  - name: battery_1
    type: template
    template: huawei-sun2000
    id: 1
    host: 192.168.178.149
    port: 502
    usage: battery
    modbus: tcpip
    timeout: 15s
    storageunit: 1
    capacity: 10.0
  - name: battery_2
    type: template
    template: huawei-sun2000
    id: 1
    host: 192.168.178.149
    port: 502
    usage: battery
    modbus: tcpip
    timeout: 15s
    storageunit: 2
    capacity: 5.0
```