evcc-io/evcc #18101: Weitere Icons für Loadpoints

diehappy69, 2025-01-07:
Hallo Zusammen,

gibt es eine Möglichkeit die Auswahl der Icons zu erhöhen? Ich habe 2 Wallboxen und mehrere Schalt/Mess-Steckdosen und würde gerne die unterschiedlichen Verbraucher visualisieren?

Grüße
Sebastian

Tombra1889, 2025-01-07:
Wie bekomme ich denn überhaupt ein Icon in einen Loadpoint?
[image]
Zumindest beim Alfen-Template wird keins eingeblendet.

diehappy69, 2025-01-07:
Wenn es sich um eine Wallbox handelt (keine Steckdose), dann kannst Du das Icon in der Rubrik Vehicle für jedes Fahrzeug setzen...

VolkerK62, 2025-01-07:
Die Liste in der Doku ist nicht aktuell.
Es gibt schon erheblich mehr https://github.com/evcc-io/evcc/pull/14008

naltatis, 2025-01-07:
@diehappy69 an welche Icons hast du denn so gedacht? Ich würde es weiterhin bei einer kurierten Auswahl lassen, aber wir können die gerne erweitern. Unsere Icons stammen zum Großteil aus Material Symbols:

https://icones.js.org/collection/material-symbols?variant=Outline+Rounded

diehappy69, 2025-01-08:
Hi, 
wictig wären die typischen Großverbraucher (Herd, Trockner, PC, Lüftung etc.). Aber es wäre auch gut unterschiedliche Symbole für die Fahrzeuge zu haben..

Grüße

closed 2025-01-08 by andig