evcc-io/evcc #18103: MinPV not triggering the charge end

sbidy, 2025-01-07:
### Describe the bug

In the `minpv` configuration, the charging session will not stop if the threshold of the `pv charge current` is below the `min. current` (see last log line).
It seems an issue of version 0.132.0 frontend.

### Steps to reproduce

1. Configure 20% minimum charging
2. Configure min current
3. connect the car
4. Charging will start
5. The 20% will reach AND PV current is not over the configured min current

### Configuration details

```yaml
log: debug
levels:
  cache: error

mqtt:
  broker: 1xxxx
  topic: evcc # root topic for publishing, set empty to disable publishing
  # clientid: foo
  user: mqtt

interval: 30s # control cycle interval

tariffs:
  currency: EUR
  grid:
    type: fixed
    price: xx

meters:
- name: pv1
  type: template
  template: hoymiles-opendtu
  usage: pv
  host: xx
- type: template
  template: tasmota-sml
  usage: grid
  host: xx
  name: grid1
- name: charge1
  type: custom
  power:
    source: http
    uri: http://xx/cm?cmnd=Status+8
    jq: if .StatusSNS.ENERGY.Power[0] > 100 then .StatusSNS.ENERGY.Power[0] else 0 end
  energy:
      source: http
      uri: http://xx/cm?cmnd=Status+8
      jq: .StatusSNS.ENERGY.Total
  currents:
    - source: http
      uri: http://xx/cm?cmnd=Status+8
      jq: .StatusSNS.ENERGY.Current[0]
    - source: http
      uri: http://xx/cm?cmnd=Status+8
      jq: .StatusSNS.ENERGY.Current[1]
    - source: http
      uri: http://xx/cm?cmnd=Status+8
      jq: .StatusSNS.ENERGY.Current[2]

chargers:
- type: template
  template: tasmota
  host: xx
  channel: 1
  standbypower: -1
  name: wallbox1

vehicles:
- type: template
  template: skoda
  title: Octavia RS
  capacity: 10
  name: ev1

loadpoints:
- title: Garage
  charger: wallbox1
  vehicle: ev1
  mode: now
  phases: 1
  meter: charge1

site:
  title: Something
  meters:
    grid: grid1
    pv:
    - pv1
```

### Log details

```text
[lp-1 ] DEBUG 2025/01/07 10:17:27 set charge mode: minpv
[lp-1 ] DEBUG 2025/01/07 10:17:27 pv timer reset
[lp-1 ] DEBUG 2025/01/07 10:17:27 pv timer inactive
[lp-1 ] DEBUG 2025/01/07 10:17:27 charge power: 0W
[lp-1 ] DEBUG 2025/01/07 10:17:27 charge currents: [0.091 0.228 0.031]A
[site ] DEBUG 2025/01/07 10:17:27 grid power: 296W
[site ] DEBUG 2025/01/07 10:17:27 pv power: 10W
[site ] DEBUG 2025/01/07 10:17:27 site power: 296W
[lp-1 ] DEBUG 2025/01/07 10:17:28 charge total import: 518.110kWh
[lp-1 ] DEBUG 2025/01/07 10:17:28 charger status: C
[lp-1 ] DEBUG 2025/01/07 10:17:28 soc estimated: 100.00% (vehicle: 100.00%)
[lp-1 ] DEBUG 2025/01/07 10:17:28 vehicle soc: 100%
[lp-1 ] DEBUG 2025/01/07 10:17:28 vehicle range: 47km
[lp-1 ] DEBUG 2025/01/07 10:17:28 pv charge current: min 2A > 0.713A (296W @ 1p, battery: false)
```

### What type of operating system or environment does evcc run on?

Docker container

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.132.0

StefanSchoof, 2025-01-07:
Das ist die Beschreibung von Min+PV
https://docs.evcc.io/docs/features/solar-charging#minpv-modus

> Im Unterschied zum PV-Modus wird die Ladung nicht unterbrochen.

closed 2025-01-07 by premultiply

sbidy, 2025-01-07:
Das macht keinen Sinn.

Also folgendes:
1. Ich definiere 20% min.-Ladung
2. Ich schließe das Fahrzeug an
3. Es wird in jeden Fall auf 20% geladen
5. Wenn 20% erreicht und der PV Strom nicht über dem definierten `Min. Current ` liegt, sollte die Ladung unterbrochen werden
5.1. Wenn 20% erreicht und der PV Strom (bzw. der Überschuss) über dem `Min. Current` liegt, soll die Ladung wieder aufgenommen bzw. weitergeführt werden.

Ich bin mir auch sehr sicher, dass das so auch schon funktioniert hat.
Es macht keinen Sinn nach der Mindestladung einfach weiter zu laden, ohne den PV Strom zu berücksichtigen. In diesem Fall würde ja immer stumpf bei jedem Ladevorgang der unter den (bei mir) 20% liegen sollte, voll geladen werden.

StefanSchoof, 2025-01-07:
Das ist der PV Modus. Der mindest Prozent wird auch im PV Modus berücksichtigt. Das min bei min+PV bezieht sich auf eine ununterbrochene Mindest Ladestrom.