evcc-io/evcc #18104: Anzeige Verbrauch einer Mess/Schalt steckdose von Homematic IP ohne Einbindung als Ladepunkt

diehappy69, 2025-01-07:
Hallo Zusammen,

ich habe mehrere Homematic IP Steckdosen mit Messfunktion im Einsatz und möchte nur die Messwerte unter "Out" darstellen:
Also: Ladepunkt 1, 2, Stromverbrauch Server, Maschmaschine, Trocker....
[image]
 
Aktuell habe ich sie als Meter, Charger und Loadpoint eingetragen. So tauchen sie ja bei den Wallboxen auf, was aber nicht erforderlich bzw. gewünscht ist.
[image]

Wie kann ich das konfigurieren?
Dank Euch
Sebastian

Tombra1889, 2025-01-07:
Kommt hoffentlich bald mit den: https://github.com/evcc-io/evcc/pull/7236

premultiply, 2025-01-07:
Das ist derzeit nicht möglich.
Es gibt bislang nur die Ext-Meters, die aber nicht in der UI angezeigt werden.

andig, 2025-01-07:
...mit https://github.com/evcc-io/evcc/pull/17943 werden sie allerdings erfasst und über Influx, Mqtt etc ausgegeben.

UI ist nicht geplant.

closed 2025-01-07 by andig