evcc-io/evcc #18109: Heidelberg Amperfied Wallbox OCPP: Lademodus wechseln nicht möglich

Saueee, 2025-01-07:
### Describe the bug

Vor zwei Wochen hatte ich bereits ein Ticket eröffnet. Dort hat sich herausgestellt das die Wallbox kein Smart Charging unterstützt: https://github.com/evcc-io/evcc/issues/17836
Dies hat sich in der Zwischenzeit geändert und ein Update hat diese Funktion eingebaut.
Die SmartCharging Profile werden laut Log auch akzeptiert.

Beim Ändern des Lademodus (Aus/PV/PV+Min/Schnell) kommt es dennoch zu zwei Fehlern:
charger logic error: current mismatch (got 0A, expected 6A)
charger out of sync: expected disabled, got enabled

Außerdem kommt es beim Starten des Servers zu einem Fehler:
ocpp message (368316152): OccurrenceConstraintViolation - Field Call.Payload.MeterValue[0].SampledValue[0].Value required but not found for feature MeterValues

Wo könnte das Problem liegen?

### Steps to reproduce

1. Amperfied per OCPP verbinden
2. Auto anschließen und Laden
3. Lademodus wechseln

### Configuration details

```yaml
sponsortoken: XXX
interval: 30s
log: debug
site:
  title: XXX
  meters:
    grid: hager
    pv:
      - pv-anlage

loadpoints:
  - title: Wallbox 1
    charger: wb1
    mode: off
    enable:
      threshold: -4000
      delay: 2m
    disable:
      delay: 20m
      threshold: 1000
  - title: Wallbox 2
    charger: wb2
    mode: off
    enable:
      threshold: -4000
      delay: 2m
    disable:
      delay: 20m
      threshold: 1000
  - title: Wallbox 3
    charger: wb3
    mode: off
    enable:
      threshold: -4000
      delay: 2m
    disable:
      delay: 20m
      threshold: 1000
  - title: Wallbox 4
    charger: wb4
    mode: off
    enable:
      threshold: -4000
      delay: 2m
    disable:
      delay: 20m
[... 137 lines cut ...]
[ocpp ] TRACE 2025/01/07 12:18:59 send wb2: [2,"3094096727","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Current.Offered"}]
[ocpp ] TRACE 2025/01/07 12:18:59 recv wb2: [3,"3094096727", { "status": "Accepted" }]
[ocpp ] TRACE 2025/01/07 12:18:59 send wb2: [2,"806525571","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Power.Offered"}]
[ocpp ] TRACE 2025/01/07 12:18:59 recv wb2: [3,"806525571", { "status": "Accepted" }]
[ocpp ] TRACE 2025/01/07 12:18:59 send wb2: [2,"3656639882","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"SoC"}]
[ocpp ] TRACE 2025/01/07 12:19:00 recv wb2: [3,"3656639882", { "status": "Accepted" }]
[ocpp ] TRACE 2025/01/07 12:19:00 send wb2: [2,"2877776271","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Power.Active.Import,Energy.Active.Import.Register,Current.Import,Voltage,Current.Offered,Power.Offered,SoC"}]
[ocpp ] TRACE 2025/01/07 12:19:00 recv wb2: [3,"2877776271", { "status": "Accepted" }]
[ocpp ] TRACE 2025/01/07 12:19:00 send wb2: [2,"2716089583","TriggerMessage",{"requestedMessage":"MeterValues"}]
[ocpp ] TRACE 2025/01/07 12:19:00 recv wb2: [3,"2716089583", { "status": "Accepted" }]
[ocpp ] TRACE 2025/01/07 12:19:00 recv wb2: [2,"32773065","MeterValues", { "meterValue": [ { "timestamp": "2025-01-07T11:18:59Z", "sampledValue": [ { "context": "Trigger", "measurand": "Power.Active.Import", "value": "" }, { "context": "Trigger", "measurand": "Energy.Active.Import.Register", "value": "" }, { "context": "Trigger", "measurand": "Current.Import", "value": "" }, { "context": "Trigger", "measurand": "Voltage", "value": "" }, { "context": "Trigger", "measurand": "Current.Offered", "value": "" }, { "context": "Trigger", "measurand": "Power.Offered", "value": "" }, { "context": "Trigger", "measurand": "SoC", "value": "" } ] } ], "connectorId": 0, "transactionId": 0 }]
[ocpp ] ERROR 2025/01/07 12:19:00 ocpp message (32773065): OccurrenceConstraintViolation - Field Call.Payload.MeterValue[0].SampledValue[0].Value required but not found for feature MeterValues ([2,"32773065","MeterValues", { "meterValue": [ { "timestamp": "2025-01-07T11:18:59Z", "sampledValue": [ { "context": "Trigger", "measurand": "Power.Active.Import", "value": "" }, { "context": "Trigger", "measurand": "Energy.Active.Import.Register", "value": "" }, { "context": "Trigger", "measurand": "Current.Import", "value": "" }, { "context": "Trigger", "measurand": "Voltage", "value": "" }, { "context": "Trigger", "measurand": "Current.Offered", "value": "" }, { "context": "Trigger", "measurand": "Power.Offered", "value": "" }, { "context": "Trigger", "measurand": "SoC", "value": "" } ] } ], "connectorId": 0, "transactionId": 0 }])
```

### What type of operating system or environment does evcc run on?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.132.0

mfuchs1984, 2025-01-07:
Die Wallbox sendet leere Messwerte, da muss wohl Heidelberg nochmal ran
`[ocpp ] TRACE 2025/01/07 12:19:00 recv wb2: [2,"32773065","MeterValues", { "meterValue": [ { "timestamp": "2025-01-07T11:18:59Z", "sampledValue": [ { "context": "Trigger", "measurand": "Power.Active.Import", "value": "" }, { "context": "Trigger", "measurand": "Energy.Active.Import.Register", "value": "" }, { "context": "Trigger", "measurand": "Current.Import", "value": "" }, { "context": "Trigger", "measurand": "Voltage", "value": "" }, { "context": "Trigger", "measurand": "Current.Offered", "value": "" }, { "context": "Trigger", "measurand": "Power.Offered", "value": "" }, { "context": "Trigger", "measurand": "SoC", "value": "" } ] } ], "connectorId": 0, "transactionId": 0 }]
`

Die anderen Probleme können, aber müssen nicht damit zusammenhängen. Für diese braucht es Logfiles mit mehr Kontext, die kurzen Ausschnitte sind nicht genug. Am besten jeweils ab Start von evcc, Auto anstecken, Problem reproduzieren, ganzes log von Anfang an hochladen.

Saueee, 2025-01-07:
Nur eine Wallbox aktiviert (wb4)

1. Alles neugestartet
2. Auto dran
3. Karte dran
4. Von PV (Default) auf Aus geschaltet

-> Auto lädt weiter mit Full Speed

Kompletter Log:
[evcc-20250107-134900-trace.log](https://github.com/user-attachments/files/18332716/evcc-20250107-134900-trace.log)

closed 2025-01-07 by premultiply

Saueee, 2025-01-07:
@premultiply 
Könntest du mir kurz helfen und sagen welche Einheiten falsch sind bzw. was falsch zurückgegeben wird?
Dann kontaktiere ich den Hersteller.

mfuchs1984, 2025-01-07:
Ohne Auto:
- Wallbox meldet meterValues ohne values 
`[ocpp ] TRACE 2025/01/07 12:19:00 recv wb2: [2,"32773065","MeterValues", { "meterValue": [ { "timestamp": "2025-01-07T11:18:59Z", "sampledValue": [ { "context": "Trigger", "measurand": "Power.Active.Import", "value": "" }, { "context": "Trigger", "measurand": "Energy.Active.Import.Register", "value": "" }, { "context": "Trigger", "measurand": "Current.Import", "value": "" }, { "context": "Trigger", "measurand": "Voltage", "value": "" }, { "context": "Trigger", "measurand": "Current.Offered", "value": "" }, { "context": "Trigger", "measurand": "Power.Offered", "value": "" }, { "context": "Trigger", "measurand": "SoC", "value": "" } ] } ], "connectorId": 0, "transactionId": 0 }] `
Mit Auto:
- Wallbox lädt mit 16 A, meldet aber Current.Offered und Power.Offered mit 0, zudem wird SoC in der Einheit Volt angegeben.
``
Der zentrale Fehler ist aber natürlich, dass sie die SetChargingProfile Nachrichten zwar akzeptiert, aber dennoch einfach mit 16 A weiter lädt.