evcc-io/evcc #18118: UI Bedienungsproblem im Vollbildmodus Safari iPhone 15 pro

gennergy, 2025-01-07:
### Describe the bug

Um EVCC das Look&Feel einer App zu geben, habe ich über den Safaribrowser eine Verknüpfung auf dem iPhone Homescreen angelegt. Die UI übernimmt auch den Vollbildmodus und die Menüs und das Eingabefeld des Browser werden nicht mehr angezeigt.

Jedo lässt sich das Burgermenu lässt im Hochkantmodus nicht öffnen, sowie das Ladeplanungs - Popup nicht schließen. Die Bedienelemente liegen hinter dem iOS Menu. Siehe Screenshot

[image]
[image]

### Steps to reproduce

1. Öffne EVCC UI im Safaribrowser auf iPhone 15 pro (eventuell auch andere iOS Geräte)
2. Drücke auf das Teilen Symbol und wähle "Zum Home-Bildschirm", anschließend "Hinzufügen"
3. Starte UI mit neu erstelltem Symbol auf dem iOS Homescreen 
4. Burger Menu kann nicht bedient werden im Hochkant Modus
5. nach drehen des iPhones um 90° ist das Burgemenu erreichbar.
6. Das gleiche gilt für das Ladeplanungs Popup. Es kann nicht geschlossen werden.
7. Das Einstellungs Popup für günstiges Netzladen lässt sich allerdings schließen.
...

### Configuration details

```yaml
evcc v0.132.0
iphone 15 pro
iOS-Version 18.2
```

### Log details

```text
keine Fehler im Log
```

### What type of operating system or environment does evcc run on?

Docker container

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.132.0

StefanSchoof, 2025-01-07:
Das hängt mit dem iOS Update zusammen. Bei anderen hat geholfen den Link vom Home Screen zu löschen und neu zu erstellen.

#12941

closed 2025-01-07 by andig

gennergy, 2025-01-07:
Konnte es nachvollziehen. Neues Symbol erstellen hat geholfen. Danke!