evcc-io/evcc #18125: cannot create meter type 'custom' nach update auf 0.132.1

softservicegit, 2025-01-08:
### Describe the bug

evcc startet nach dem update nicht mehr. Config ist unverändert.

### Steps to reproduce

1. update auf 0.132.1
2.
3.
...

### Configuration details

```yaml
meters:
  - name: solaredge_grid
    type: template
    template: solaredge-hybrid
    usage: grid
    modbus: tcpip
    id: 1
    host: 192.168.10.6 # Hostname
    port: 526
    timeout: 3s
```

### Log details

```text
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2025/01/08 08:10:28 evcc 0.132.1
[main  ] INFO 2025/01/08 08:10:28 using config file: /config/evcc.yaml
[main  ] FATAL 2025/01/08 08:10:37 cannot create meter 'solaredge_grid': cannot create meter type 'template': cannot create meter type 'custom': power: read tcp 192.168.10.67:47702->192.168.10.6:526: i/o timeout
[main  ] FATAL 2025/01/08 08:10:37 will attempt restart in: 15m0s
```

### What type of operating system or environment does evcc run on?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.132.1

softservicegit, 2025-01-08:
auch nach dem downgrade auf 0.132.0 kommt der selbe Fehler. 
Die letzte Änderung, die in der Richtung passiert ist, ist das update des WR auf die aktuelle Version 4.22.44.

StefanSchoof, 2025-01-08:
`read tcp 192.168.10.67:47702->192.168.10.6:526: i/o timeout`

Da antwortet dein solaredge nicht. Schon versucht den einmal neu zu starten?

premultiply, 2025-01-08:
Da antwortet nur der WR nicht (mehr).
Nix für uns zu tun.

closed 2025-01-08 by premultiply

softservicegit, 2025-01-08:
Ich gebe zu, meine log Auszug war - wie es immer so ist - völliger Blödsinn und am Thema vorbei.

Richtig ist folgendes:
`starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2025/01/08 10:00:36 evcc 0.132.1
[main  ] INFO 2025/01/08 10:00:36 using config file: /config/evcc.yaml
[mqtt  ] INFO 2025/01/08 10:00:36 connecting evcc-1170240778 at tcp://mqtt.sartool.de:1883
[mqtt  ] DEBUG 2025/01/08 10:00:36 tcp://mqtt.sartool.de:1883 connected
[main  ] INFO 2025/01/08 10:00:36 listening at :7070
[main  ] FATAL 2025/01/08 10:00:39 cannot create meter 'solaredge_grid': cannot create meter type 'template': cannot create meter type 'custom': power: sunspec model not found: [{203 0 W}]
[main  ] FATAL 2025/01/08 10:00:39 will attempt restart in: 15m0s`

premultiply, 2025-01-08:
> power: sunspec model not found: [{203 0 W}]

Der WR liefert keine Daten des Grid-Meters mehr im erwarteten SunSpec Modell 203.

Ich würde mal tippen das Gerät braucht einen Neustart, die Konfiguration wurde verändert oder die Firmware ist anderweitig ganz kaputt.

softservicegit, 2025-01-08:
Danke, ich teste. und melde mich