evcc-io/evcc #18142: HTTP provider shared cache

Lenart12, 2025-01-09:
**Is your feature request related to a problem? Please describe.**
Currently, each http provider configuration in evcc independently fetches (and caches) data from the same HTTP endpoint. If multiple states rely on the same response, but different pipeline, still requests would be fired for each provider independently. This leads to unnecessary network request and Increased latency.

**Describe the solution you'd like**
Implement a shared cache for HTTP providers. It would be a boolean option for http provider, something like `sharedcache: true` or modifying cache property to be object instead of time to be like:
```yaml
source: http
cache:
  enabled: true
  shared: true
```
If option for shared cache would be enabled for provider, it would, when returning a cache, return the cached body from a global store of cached requests. If cache is not hit it would follow the normal path of fetching the data, and storing it in the global cache. 

Global store of http cache would be a map of something like `hash(method, uri, auth, insecure, headers)` -> `(timestamp, byte[]) | nil`.

**Describe alternatives you've considered**
Alternative solution is possible by sharing all http request caches without configuration.

It is possible currently to do something like this by implementing this logic in shared javascript, small *pseudo js* example:
```yaml
javascript:
  - vm: shared
    script: |
      let cachedBody = null;
      let cachedTime = null;
      function getState() {
        if (validCache(cachedTime, cachedBody)) return cachedBody;
        cachedBody = doRequest();
        cachedTime = currentTime();
        return cachedBody;
      }

vehicles:
  - name: vehicle_1
    type: custom
    soc:
      source: js
      vm: shared
      script: |
        doPipline(getState(), 'soc')
    range:
      source: js
      vm: shared
      script: |
        doPipline(getState(), 'range')
```

**Additional context**
This feature could be used right away, at least in `templates/definition/vehicle/tesla-ble.yaml`, and would have cached at least 4 requests, but probably also with many others.

```diff
@@ templates/definition/vehicle/tesla-ble.yaml 46:0 @@
    soc:
      source: http
      uri: {{ .url }}:{{ .port }}/api/1/vehicles/{{ .vin }}/vehicle_data?endpoints=charge_state
      jq: .response.response.charge_state.battery_level
      timeout: 30s
+     cache: 30s
+     sharedcache: true
    limitsoc:
      source: http
      uri: {{ .url }}:{{ .port }}/api/1/vehicles/{{ .vin }}/vehicle_data?endpoints=charge_state
      jq: .response.response.charge_state.charge_limit_soc
      timeout: 30s
+     cache: 30s
+     sharedcache: true
    range:
      source: http
      uri: {{ .url }}:{{ .port }}/api/1/vehicles/{{ .vin }}/vehicle_data?endpoints=charge_state
      jq: .response.response.charge_state.battery_range
      scale: 1.60934
      timeout: 30s
+     cache: 30s
+     sharedcache: true
    status:
      source: http
      uri: {{ .url }}:{{ .port }}/api/1/vehicles/{{ .vin }}/vehicle_data?endpoints=charge_state
      method: GET
      jq: (if (.response.response.charge_state.charging_state == "Charging") then "C"
        elif (.response.response.charge_state.charging_state == "Stopped") then "B"
        elif (.response.response.charge_state.charging_state == "NoPower") then "B"
        elif (.response.response.charge_state.charging_state == "Complete") then "B" 
        else "A" end)
      timeout: 30s
+     cache: 30s
+     sharedcache: true
```

andig, 2025-01-09:
I‘m not a big fan of a custom, configurable caching mechanism. Is there an actual need for it? Assuming http cache headers are in place, wouldn‘t it be sufficient to make https://github.com/evcc-io/evcc/blob/master/provider/http.go#L104 shared instead of per provider?

andig, 2025-01-09:
Here‘s a quick attempt, didn‘t test this.

Lenart12, 2025-01-09:
Hey, tested your code and it doesnt work.
Its my first time interacting with go code, much less evcc codebase so thats why I did it this way 😅.

I dont think its a good idea to rely on http servers to support server side caching since for a lot api's it would not make sense (why cache constantly changing state?); except in this specific use case where you can only extract one piece of data per request. Anyways, if your way works eventualy, im all for it, it looks much simpler :^) . I just want some mechanism that would prevent sending multiple requests to TeslaBleHttpProxy (or anything else but thats just my specific use case). Each requests takes quite some time since it is opening a BLE connection for every request.

This is how I test code:
* `test_server.py` [https://hastebin.skyra.pw/terepamoqo.py](url)
* `shared_cache.py` [https://hastebin.skyra.pw/hawuwuqolu.yaml](url)

Lenart12, 2025-01-09:
> I‘m not a big fan of a custom, user-configurable caching mechanism. Is there an actual need for it?

Another thing. Is this not how currently caching works?
```go
// request executes the configured request or returns the cached value
func (p *HTTP) request(url string, body string) ([]byte, error) {
        // Check if it is time to cache with user configured timeout
	if time.Since(p.updated) >= p.cache {
		p.var, p.err = doRequest() //... Shortened function body as it is not relevant
		p.updated = time.Now() // Update last request time
	}
       // Here return either localy cached value (did not take if path) or updated new path (did take if path)
	return p.val, p.err
}
```
This looks like "custom" caching to me. If I understand your PR there would be a second layer of caching somewhere inside `doRequest` path, but right now this is not working.

andig, 2025-01-09:
I agree, there is some sort of custom caching. It is very stupid though since it just "caches", not taking different URLs etc into account.

> Hey, tested your code and it doesnt work.

It should: 

> Http servers still need to send proper etag/cache headers.

Does it? If not- and since you're in control of the server application- add proper headers there?

>  since for a lot api's it would not make sense (why cache constantly changing state?)

Because for many servers, state isn't constantly changing. Instead, it is updated maybe every second or every 5 seconds etc. That's perfectly suitable for HTTP cache headers.

Lenart12, 2025-01-09:
Not everybody is privileged enough to to be able to control server side application and making a proxy for a proxy sounds like a lot of work, because cache control in the context of that specific ble proxy would not really make sense imo. But `TeslaBleHttpProxyCacheControlProxy` does sound fun if somebody wants to make it 😁

Anyways, I gather from your responses, that you do not want this use case to be supported. I will stop puting my efforts into this and just accept that there will be a lot of overhead and run slower. But if thats not the case I'm happy to work with you on an accepted solution.

andig, 2025-01-09:
> I will stop puting my efforts into this and just accept that there will be a lot of overhead and run slower. But

Afaia nothing will run slower in this case. If there‘s any problem lets work with @wimaha to get cache headers implemented.

closed 2025-01-09 by andig

wimaha, 2025-01-09:
If I understand correctly we need to implement the cache headers in `TeslaBleHttpProxy`. I think we can do this. Please open an issue for that at the proxy GitHub.

And: `TeslaBleHttpProxy` does not open a new ble connection for every request. The same ble connection is used for 30 seconds. (That is the maximum Tesla allows.) So every request in this timeframe is “fast”.

andig, 2025-01-10:
@Lenart12 thank you for the issue- in any case we got shared HTTP cache now!