evcc-io/evcc #18145: Support Smart Meter Victron VM-3P75CT

pellenbeck, 2025-01-09:
There is the new smart meter VM-3P75CT from Victron, which measures very quickly and precisely and makes the data available via ModBus Ip.

I would be delighted if this were also supported natively.

https://www.victronenergy.de/meters-and-sensors/energy-meter

VolkerK62, 2025-01-09:
did you try one of the available templates?
https://docs.evcc.io/docs/devices/meters#victron

andig, 2025-01-09:
If not: happy to take a PR

pellenbeck, 2025-01-12:
I have tried the existing Victron templates `cg-em24 `and `cg-emt3xx` without success.

I have tried a connection via Modbus in each case. Error message:
[site ] ERROR 2025/01/12 13:47:34 grid power: read failed: dial tcp xx.xx.xx.xx:502: connect: connection refused

Information on the Modbus ID and register address can be found here:
https://communityarchive.victronenergy.com/questions/258646/modbus-id-and-register-address-of-the-vm-3p75ct.html

andig, 2025-01-12:
> connection refused

There‘s no network connection. Modbus is the least of problems here.

pellenbeck, 2025-01-14:
> > connection refused
> 
> There‘s no network connection. Modbus is the least of problems here.

The evcc instance is not limited in terms of network access. The VictronConnect app on the smartphone (in the same network) can easily access the device via IP and read the live data.

pellenbeck, 2025-01-14:
Just checked the older Carlo Gavazzi Meters: They use Modbus over TCP but the new Victron meter uses UDP!

mdkeil, 2025-01-16:
Do you have a victron multiplus device?, than you can use the default victron/Grid template.

andig, 2025-01-16:
Or just use UDP…

pellenbeck, 2025-01-16:
> Do you have a victron multiplus device?, than you can use the default victron/grid template.

Is under construction - not yet.

> Or just use UDP…

How? `tcpid` is the only supported method in Victron templates `cg-em24` and `cg-emt3xx` - if I am not wrong.

andig, 2025-01-16:
Oh, I see. You can try a `custom` meter then.

andig, 2025-01-20:
...or just add `udp` to the relevant templates.

closed 2025-01-23 by andig

dragonlord83, 2025-04-16:
Hello, I have the same problem.
My meter is connected directly to the rest of my Victron system via the network.
Now I would like to use this meter instead of the previous Tibber Pulse.

But I can't get it to work.

This is what my current settings look like, but they aren't working. How do I change them?

meters:
- name: my-grid
  type: template
  template: victron-energy
  usage: grid
# source: modbus
  id: 1
  uri: 192.XXX.XXX.XX:502
  upd: true

pellenbeck, 2025-04-16:
I now use the "victron-energy" template with this configuration:

```
meters:
- name: victron_grid
  type: template
  template: victron-energy
  usage: grid
  host: xxx.xxx.xxx.xxx # IP-Adresse oder Hostname
  port: 502 # Port (optional)
  meterid: 40 # Grid-Energiezähler VRM instance- kann im VRM Portal oder im RemoteUI ausgelesen werden. (optional)
```

Works without any problem.

AlexHuebi, 2025-06-13:
I just thought I might throw my 2 cents in here.

@andig 
> ...or just add `udp` to the relevant templates.

Will not work. The VM-3P75CT not only uses Modbus/UDP instead of Modbus/TCP but also uses different registers.

In case anyone will ever get to it: 
The only 'official' documentation of the used Registers is in there: 
https://github.com/victronenergy/dbus-modbus-client/blob/master/victron_em.py

And since I'd like to see it implemented, but I neither have the skill to work with the Go-Codebase and Templates nor do I currently possess the time to implement it, here are the relevant registers, just in case it helps when someone has some spare time...

```
Voltage Registers (16 bit Signed INT, Scaling Factor x100): 
L1: 0x3040 
L2: 0x3048
L3: 0x3050

Current Registers (16bit Signed INT, Scaling Factor x100): 
L1: 0x3041
L2: 0x3049
L3: 0x3051

Power Registers (32bit Signed INT): 
L1: 0x3082
L2: 0x3086
L3: 0x308A
```

Btw, as far as I could see, there isn't any official reference in any of the provided Docs referencing Modbus/UDP, so that might also need an update...

Mvanmierlo, 2026-03-04:
I am all down doing the testwork .. this I can do, but without anyone pointing me in the right direction / documentation in regard how to setup a custom device I am clueless as a new user.

But willing to put in the efford if no development is needed. As mentioned I did my fair share to get this up and running in HomeAssistant.