evcc-io/evcc #18154: Tesla: SOC is not displayed while charging

LaTell, 2025-01-09:
### Describe the bug

I updated to version 0.132.1. I have adapted my vehicle configuration to the new "tesla-ble" template because I use wimaha's BLE HTTP Proxy.
Since then, Tesla's SOC has not been displayed.

in the log you can see a parsing error in the vehicle soc.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
vehicles:
  - name: Tessy
    type: template
    template: tesla-ble
    title: Tessy
    capacity: 78
    vin: LRWYGXXXXXXXXXXXXXXXXX
    url: http://192.168.1.xxx # URL des Tesla BLE HTTP Proxy
    port: 8080
    icon: car
    phases: 3
    minCurrent: 5
    maxCurrent: 16
```

### Log details

```text
lp-1  ] DEBUG 2025/01/09 16:40:48 charge power: 4131W
[lp-1  ] DEBUG 2025/01/09 16:40:48 charge currents: [6.1 6.1 6.1]A
[site  ] DEBUG 2025/01/09 16:40:48 grid power: 4541W
[site  ] DEBUG 2025/01/09 16:40:48 grid powers: [1550 1605 1386]W
[site  ] DEBUG 2025/01/09 16:40:48 grid currents: [6.94 7.04 6.14]A
[site  ] DEBUG 2025/01/09 16:40:48 pv 1 power: 0W
[site  ] DEBUG 2025/01/09 16:40:48 site power: 4541W
[lp-1  ] DEBUG 2025/01/09 16:40:48 charge voltages: [224 229 225]V
[lp-1  ] DEBUG 2025/01/09 16:40:48 detected connected phases: 3p
[lp-1  ] DEBUG 2025/01/09 16:40:48 detected active phases: 3p
[lp-1  ] DEBUG 2025/01/09 16:40:48 charger status: C
[lp-1  ] ERROR 2025/01/09 16:40:50 vehicle soc: strconv.ParseFloat: parsing "<nil>": invalid syntax
[lp-1  ] DEBUG 2025/01/09 16:40:50 pv charge current: min 6A > 0A (4541W @ 3p, battery: false)
[site  ] DEBUG 2025/01/09 16:41:18 ----
[lp-1  ] DEBUG 2025/01/09 16:41:18 charge power: 4133W
[lp-1  ] DEBUG 2025/01/09 16:41:18 charge currents: [6.1 6.1 6.1]A
[site  ] DEBUG 2025/01/09 16:41:18 grid power: 4553W
[site  ] DEBUG 2025/01/09 16:41:18 pv 1 power: 0W
[site  ] DEBUG 2025/01/09 16:41:18 grid powers: [1547 1610 1396]W
[site  ] DEBUG 2025/01/09 16:41:18 grid currents: [6.94 7.07 6.15]A
[site  ] DEBUG 2025/01/09 16:41:18 site power: 4553W
[lp-1  ] DEBUG 2025/01/09 16:41:18 charge voltages: [223 229 226]V
[lp-1  ] DEBUG 2025/01/09 16:41:18 detected connected phases: 3p
[lp-1  ] DEBUG 2025/01/09 16:41:18 detected active phases: 3p
[lp-1  ] DEBUG 2025/01/09 16:41:18 charger status: C
[lp-1  ] ERROR 2025/01/09 16:41:22 vehicle soc: strconv.ParseFloat: parsing "<nil>": invalid syntax
[lp-1  ] DEBUG 2025/01/09 16:41:22 pv charge current: min 6A > 0A (4553W @ 3p, battery: false)
[site  ] DEBUG 2025/01/09 16:41:48 ----
[lp-1  ] DEBUG 2025/01/09 16:41:48 charge power: 4113W
[lp-1  ] DEBUG 2025/01/09 16:41:48 charge currents: [6 6.1 6.1]A
[site  ] DEBUG 2025/01/09 16:41:48 grid power: 4538W
[site  ] DEBUG 2025/01/09 16:41:48 grid powers: [1545 1605 1388]W
[site  ] DEBUG 2025/01/09 16:41:48 grid currents: [6.94 7.04 6.12]A
[site  ] DEBUG 2025/01/09 16:41:48 pv 1 power: 0W
[site  ] DEBUG 2025/01/09 16:41:48 site power: 4538W
[lp-1  ] DEBUG 2025/01/09 16:41:48 charge voltages: [223 229 226]V
[lp-1  ] DEBUG 2025/01/09 16:41:48 detected connected phases: 3p
[lp-1  ] DEBUG 2025/01/09 16:41:48 detected active phases: 3p
[lp-1  ] DEBUG 2025/01/09 16:41:48 charger status: C
[lp-1  ] ERROR 2025/01/09 16:41:49 vehicle soc: strconv.ParseFloat: parsing "<nil>": invalid syntax
[lp-1  ] DEBUG 2025/01/09 16:41:49 pv charge current: min 6A > 0A (4538W @ 3p, battery: false)
[site  ] DEBUG 2025/01/09 16:42:18 ----

Trace of the Tesla Car Data:

[Tessy ] TRACE 2025/01/09 16:41:48 GET http://192.168.XXXXXXXXXXX:8080/api/1/vehicles/LRWYXXXXXXXXXXXXX/vehicle_data?endpoints=charge_state
[Tessy ] TRACE 2025/01/09 16:41:49 {"result":true,"reason":"The command was successfully processed.","vin":"LRWYXXXXXXXXXXXXX","command":"vehicle_data","response":{"charge_state":{"timestamp":1736437308,"charging_state":"Charging","charge_limit_soc":82,"charge_limit_soc_std":80,"charge_limit_soc_min":50,"charge_limit_soc_max":100,"battery_heater_on":false,"not_enough_power_to_heat":false,"max_range_charge_counter":0,"fast_charger_present":false,"fast_charger_type":"ACSingleWireCAN","battery_range":158.58333,"est_battery_range":126.21084,"ideal_battery_range":158.58333,"battery_level":51,"usable_battery_level":50,"charge_energy_added":0.04,"charge_miles_added_rated":0,"charge_miles_added_ideal":0,"charger_voltage":229,"charger_pilot_current":16,"charger_actual_current":6,"charger_power":4,"trip_charging":false,"charge_rate":15.074166,"charge_port_door_open":true,"scheduled_charging_mode":"ScheduledChargingModeOff","scheduled_departure_time":0,"scheduled_departure_time_minutes":0,"supercharger_session_trip_planner":false,"scheduled_charging_start_time":0,"scheduled_charging_pending":false,"user_charge_enable_request":false,"charge_enable_request":true,"charger_phases":2,"charge_port_latch":"Engaged","charge_current_request":6,"charge_current_request_max":16,"charge_amps":6,"off_peak_charging_enabled":false,"off_peak_charging_times":"","off_peak_hours_end_time":0,"preconditioning_enabled":false,"preconditioning_times":"all_week","managed_charging_active":false,"managed_charging_user_canceled":false,"managed_charging_start_time":0,"charge_port_cold_weather_mode":false,"charge_port_color":"ChargePortColorFlashingGreen","conn_charge_cable":"IEC","fast_charger_brand":"Tesla","minutes_to_full_charge":530}}}
```

### What type of operating system or environment does evcc run on?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.132.1

andig, 2025-01-09:
> [lp-1  ] ERROR 2025/01/09 16:41:22 vehicle soc: strconv.ParseFloat: parsing "<nil>": invalid syntax

Seems there is no soc. Needs trace log to validate. Closing until available.

closed 2025-01-09 by andig

LaTell, 2025-01-09:
here is the Trace log from my Car

[evcc-20250109-164945-trace.log](https://github.com/user-attachments/files/18367420/evcc-20250109-164945-trace.log)

With my old vehicle configuration it works properly 
vehicles:
  - name: Tessy
    type: custom
    title: Tessy
    icon: car
    capacity: 78
    chargeenable:
      source: http
      uri: "http://192.168.1.12:8080/api/1/vehicles/LRWYGCXXXXXXXXXX/command/{{if .chargeenable}}charge_start{{else}}charge_stop{{end}}"
      method: POST
      body: ""
    maxcurrent: # set charger max current (A)
      source: http
      uri: http://192.168.1.12:8080/api/1/vehicles/LRWYGCXXXXXXXXXX/command/set_charging_amps
      method: POST
      body: '{"charging_amps": "{{.maxcurrent}}"}'
    wakeup: # vehicle wake up command
      source: http
      uri: http://192.168.1.12:8080/api/1/vehicles/LRWYGCXXXXXXXXXX/command/wake_up
      method: POST
      body: ""
    soc:
      source: http
      uri: http://192.168.1.12:8080/api/1/vehicles/LRWYGCXXXXXXXXXX/vehicle_data?endpoints=charge_state
      method: GET
      jq: .response.charge_state.battery_level
      timeout: 30s
    limitsoc:
      source: http
      uri: http://192.168.1.12:8080/api/1/vehicles/LRWYGCXXXXXXXXXX/vehicle_data?endpoints=charge_state
      method: GET
      jq: .response.charge_state.charge_limit_soc
      timeout: 30s
    range:
      source: http
      uri: http://192.168.1.12:8080/api/1/vehicles/LRWYGCXXXXXXXXXX/vehicle_data?endpoints=charge_state
      method: GET
      jq: .response.charge_state.battery_range
      scale: 1.60934
      timeout: 30s
    status:
      source: http
      uri: http://192.168.1.12:8080/api/1/vehicles/LRWYGCXXXXXXXXXX/vehicle_data?endpoints=charge_state
      method: GET
      jq: (if (.response.response.charge_state.charging_state == "Charging") then "C"
        elif (.response.response.charge_state.charging_state == "Stopped") then "B"
        elif (.response.response.charge_state.charging_state == "NoPower") then "B"
        elif (.response.response.charge_state.charging_state == "Complete") then "B"
        else "A" end)
      timeout: 30s

wimaha, 2025-01-09:
Looks like you are not using an old version of the proxy. Please try updating TeslaBleHttpProxy to the latest version.

LaTell, 2025-01-10:
@wimaha @andig 
Yes, it was because I had installed an old version. I thought I had done the update, but apparently it didn't work before. Now it worked the second time.
Thank you very much for your efforts