evcc-io/evcc #18166: Batterieentleerung bei Schnell-Laden trotz Sperre in Funktion Hausbatterie

Johprit, 2025-01-10:
### Describe the bug

Beim Schnellladen wird die Batterie mit maximaler Leistung entleert, obwohl die Sperre im Menu Hausbatterie aktiv war.

### Steps to reproduce

1.  Schnellladen aktiviert   -  Entleerung der Batterie
2. Ladung gestoppt
3. Version aktualisiert
4. Schnellladen aktiviert  -  Entleerung der Batterie
5. PV-min aktiviert  -   Entlleerung der Batterie
[evcc fehler.docx](https://github.com/user-attachments/files/18379408/evcc.fehler.docx)

6.
7.
...

### Configuration details

```yaml
go-e Wallbox
Audi Q4
evcc auf pi
```

### Log details

```text
keine
```

### What type of operating system or environment does evcc run on?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.132  sowie 0.132.1

andig, 2025-01-10:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

closed 2025-01-10 by andig

Johprit, 2025-01-22:
Hier die logs

[evcc-20250122-164722-trace.log](https://github.com/user-attachments/files/18508444/evcc-20250122-164722-trace.log)
[evcc-20250122-164644-debug.log](https://github.com/user-attachments/files/18508445/evcc-20250122-164644-debug.log)